guino/BazzDoorbell #1: magic err on uart

DiederikRhee, 2020-12-01:
When I connect to the device using a USB ttl interface I get this output:

hisi-sdhci: 0
PPS:Mar 18 2020 19:07:17 meari_c5    0
magic err

I cannot login...

guino, 2020-12-01:
Do you see any countdown immediately at power on ? mine says something like ... "1" then "0" then it boots up -- you have to press a key on the terminal to interrupt the countdown and it should ask for a password (which I don't have).
If you don't have a countdown then your device may be more locked down then others. 
If you have a countdown and press any key and it doesn't prompt for a password then you may have an issue/bad connection with your TX (PC side, RX camera side) which may be preventing the board from detecting the key press to prompt for a password.
If you press the 'reset' button when you power up it should try to read the ppsMmcTool.txt on the mmc card, see https://github.com/DanTLehman/orion_sc008ha and https://github.com/DanTLehman/orion_sc008ha/issues/1

DiederikRhee, 2020-12-01:
Ah yes it does countdown. I got it working now, but no password

closed 2020-12-02 by DiederikRhee