guino/BazzDoorbell #2: No programmer, No UART, No problem!

guino, 2020-12-05:
Based on the interest in this project I went ahead and spent some more time (a lot actually) in ghidra looking at the boot loader code and managed to get the ppsMmcTool.txt file format figured out and found a way to use it to modify the boot settings in order to run a script in the SD card during boot process of the camera.
I assume this can be used for any device using this PPStrong boot loader (Tuya cameras/doorbells like Geeni, Merkury, Bazz, Meari, etc).

WARNING: The process below will **require** a SD card to always be present during power-on or the device WILL NOT BOOT. The process is always reversible but keep that in mind.

I'm going to lay down the steps here for a 'manual' process but may try to automate it later, so without delays:

**1**-Verify your camera/doorbell is compatible using its local IP address (i.e. 192.168.x.x, etc) -- this is NOT the public IP displayed in the TUYA app (and likely other apps), if you're reading this there's a good chance you know how to find the IP from your router. Open a web browser and load this address: http://admin:056565099@192.168.x.x/proc/cmdline the result (kernel command line) should be something like this:
```
mem=36 console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd
```
Copy/save that response (kernel command line) as we'll need it next (and also in case you want to restore original settings) -- If you get no response, or some very different response, please stop now -- chances are you're using the wrong IP or this won't work for your device. Feel free to post your kernel command line if you have questions.

**2**-Create a file called 'env' with the following contents:
```
bootargs=mem=36 console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=0 ppsWatchInitEnd ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,T
```
VERY IMPORTANT DETAILS ABOUT THIS FILE:
-The part after bootargs= should be the same as you got on step 1 (from your device) with the **ppsAppParts=0** (instead of ppsAppParts=5).
-The file has **one single line** "bootargs=..." and a new line (enter) at the end. It is a long line (on purpose) so it may show up as multiple lines in a browser so please account for it. A sample env file is attached for your conveninence.
-The single line **MUST** have the same exact size as I posted above (924 characters + new line). Your kernel command line may be different which may require you to adjust (remove or add) to the ThankYou... text so that you match the size of the line.
-There's a 0x00 (Zero character) at the end of the file -- if this is removed by your text editor (ie notepad, etc) you'll need to be sure it is there (or the process will not work). You may need to use a hex editor to change the last character to a "00" making sure there's at least a new line (0A) at the end of the line before the "00". Again the sample env file is a good reference.

**3**-Copy these 3 files to the root of a fat32 formatted SD card (do not place them in any 'folders'): env, ppsMmcTool.txt and initrun.sh -- **MAKE SURE** there's no 'upgrade.bin' file in the SD card or this could cause problems. Be sure to properly 'eject' (or unmount) the SD card before removing it from the computer.

**4**-Power off your device and insert the SD card with the 3 files in the SD card slot.

**5**-Press-and-hold the reset button, then power on the device (i.e. power wires/USB cable) and **continue holding the reset button for 5 seconds after power on** then let the device boot. It will take longer than usual (precisely 10 seconds longer) for it to fully boot up as that's part of the initial boot script.

**6**-Repeat step 1, this time yout kernel command line should look like this:
```
mem=36 console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=0 ppsWatchInitEnd ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,T mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppPart
```
(Notice how the 'ppsAppPart' is cut off at the end -- that's the intention and as long as it doesn't say ppsAppParts=n it should work)

**7**-Now browse to this address: curl http://admin:056565099@192.168.x.x/proc/self/root/tmp/hack -- it should say "done" which is the indication everything is working as designed.

**8**-You can now delete ppsMmcTool.txt and env files from the SD card but there's no harm leaving them there. You **MUST** always have initrun.sh in the SD card during boot or the device WILL NOT BOOT.

**9**-Customize initrun.sh to your needs, you can get a copy of your ppsapp file by removing the # from the indicated line, or copy the files I posted in the project to enable telnet, httpd and/or set the user/passord for telnet login. 

**10**-For RTSP: If your camera/doorbell is 1080P chances are my ppsapp-rtsp will work for you (use my custom.sh along with initrun.sh commands provided). If your camera is 720P chances are this [ppsapp-rtsp](https://github.com/Jordan-Jarvis/Merkury-Smart_cam-720p-work/files/5633531/ppsapp-rtsp.zip) will work for you. 

##### TROUBLESHOOTING / RESTORE

If my ppsapp doesn't work for you feel free to post your ppsapp (and provide your camera model + firmware version so I can take a look).

If you wish to 'restore' the operation of your camera with SD card (remove the 'hack'):

**1**-Edit/create the env file to be this:
```
bootargs=mem=36 console=ttyAMA0,115200n8
```
NOTE: You should match the mem and console parameters as they were originally (step 1 of install process) and the env file **must** also have the new line (0x0A) and 0x00 (Zero character) at the end.

**2**-Copy ppsMmcTool.txt and env to the root of SD card (initrun.sh is NOT needed).

**3**-Follow steps 4, 5 and 6 of the install process, on step 6 the kernel command line should look like it was originally (before any changes).

If for some reason you can't get the the initrun.sh script to run please post a copy of your env file (zip format so I can verify it) and your kernel command line (before and after install attempt) so I can take a look.

If we gather enough ppsapp versions/models I may try to automate the patch process to enable RTSP.

It would be simple to make an application that reads the kernel command line from a provided IP and prepares the SD card files automatically (as well as a 'restore' file) -- If there's enough demand/interest I may work on that.

If you'd like to buy me a beer/coffee in appreciation of the effort I put in to make the above possible, feel free to:
http://paypal.me/wbbo
cash app: $wbbo

Enjoy!

[ppshack.zip](https://github.com/guino/BazzDoorbell/files/5646566/ppshack.zip)

solidssss, 2020-12-05:
Thank you so much for sharing this!
At step 6 I'm getting the same text as in step 1. Not sure what I'm doing wrong. 
This is what I'm getting from step 1:
`mem=64M console=ttySAK0,115200n8 mtdparts=spi0.0:256k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,2496k(sys),4608k(app),640k(cfg) ppsAppParts=5 ip=192.168.1.xx:::255.255.255.0 eth=xx:xx:xx:xx:xx:xx`

Could you please take a look at my env file? [Here it is](https://drive.google.com/file/d/1teLgFVSeaKVZnGMhFfcEks81P1MtvDRF/view?usp=sharing)
ppsMmcTool.txt and initrun.sh is also on thee root of the sd card. I noticed that a folder called SDT with files is also created when I boot the camera.

rkkoszewski, 2020-12-05:
@solidssss That basically means that it did not work for you.

I was just trying it out on my Nedis camera (Also Tuya) with the same result. The ppsMmcTool.txt way has eider been patched out or the different cameras are using a different format.

I'll try to hook up my reader to the UART port to see if it shows some kind of error.

Just for notes, my cmdline is similar to yours but still different:
`mem=23M ethphy=3#3  console=ttyAMA0,115200 loglevel=0 ppsdebug=off mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg),2240k(sys),5m(app),448k(cfg) ppsAppParts=5 ip=192.168.1.10:::255.255.255.0 eth=xx:xx:xx:xx:xx:xx`

LouDnl, 2020-12-05:
Great effort 👌🏻👌🏻👍🏻👍🏻

jandy123, 2020-12-05:
My cmd line is same as @guino, just instead of mem=36 I have mem=37M. Please notice the additional "M". Could this be your problem @rkkoszewski, @solidssss  ? I will also try, but have to find a uSD card.

Btw., which fw do you guys have on your cams ?

adwiraguna, 2020-12-05:
can you please hack my ppsapp?
my camera is Bullet 4S, firmware version 2.9.7
thanks.

[ppsapp](https://drive.google.com/file/d/1pBzJwDj8Pz7V3hIMzhd3HVe2yVniLHhA/view?usp=sharing)

jandy123, 2020-12-05:
I can confirm, it works here too ! Great ! 
Thanks @guino.

solidssss, 2020-12-05:
> My cmd line is same as @guino, just instead of mem=36 I have mem=37M. Please notice the additional "M". Could this be your problem @rkkoszewski, @solidssss ? I will also try, but have to find a uSD card.
> 
> Btw., which fw do you guys have on your cams ?

I tried without the "M", but it still isn't working. 
Here is some info of the device:
`{"devname":"Smart Home Camera","model":"Speed 4C","serialno":"xxxxxxxxx","softwareversion":"2.7.4","hardwareversion":"S4C_A1_V10_1245","firmwareversion":"ppstrong-a2-tuya2_teco-2.7.4.20191111"`

LouDnl, 2020-12-05:
@guino would it be possible for you to send me the dump of your flash chip? It seems that my readout is corrupted in the part before u-boot which contains the magic part. This is due to my clip not being stable enough. I've removed the chip from the board and get stable readouts now.

guino, 2020-12-05:
> Thank you so much for sharing this!
> At step 6 I'm getting the same text as in step 1. Not sure what I'm doing wrong.
> This is what I'm getting from step 1:
> `mem=64M console=ttySAK0,115200n8 mtdparts=spi0.0:256k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,2496k(sys),4608k(app),640k(cfg) ppsAppParts=5 ip=192.168.1.xx:::255.255.255.0 eth=xx:xx:xx:xx:xx:xx`
> 
> Could you please take a look at my env file? [Here it is](https://drive.google.com/file/d/1teLgFVSeaKVZnGMhFfcEks81P1MtvDRF/view?usp=sharing)
> ppsMmcTool.txt and initrun.sh is also on the root of the sd card. I noticed that a folder called SDT with files is also created when I boot the camera.
> I got a Nedis camera (WIFICI20CGY)

I sent a request to access the file in google, I'll wait for access or you can just attach the env file here. But I'll tell you right now that if the line didn't change at all, it can only be:
-You didn't hold reset while powering on the device (it can be hard)
OR
-The env file was not correct
OR
-Your firmware may be too different for it to work -- please provide the information from http://admin:056565099@IP/devices/deviceinfo when you can.

guino, 2020-12-05:
> ppsAppParts=5

I would expect any device that has that ppsAppParts=5 to work, but I only have looked at 2 firmware files so I have no way of knowing. If you have a UART/TTL 3.3V and can get to the serial terminal it may be helpful if you send me the terminal output while you're powering it on with the reset button pressed (with the 3 files in place. Chances are it will show whatever it is not liking. You can also pose the http://admin:056565099@IP/devices/deviceinfo so I have an idea.

guino, 2020-12-05:
> My cmd line is same as @guino, just instead of mem=36 I have mem=37M. Please notice the additional "M". Could this be your problem @rkkoszewski, @solidssss ? I will also try, but have to find a uSD card.
> 
> Btw., which fw do you guys have on your cams ?

Mine is 2.9.2 if I didn't post it before.

guino, 2020-12-05:
> can you please hack my ppsapp?
> my camera is Bullet 4S, firmware version 2.9.7
> thanks.
> 
> [ppsapp](https://drive.google.com/file/d/1pBzJwDj8Pz7V3hIMzhd3HVe2yVniLHhA/view?usp=sharing)

I will take a look today.

guino, 2020-12-05:
> I can confirm, it works here too ! Great !
> Thanks @guino.
> 
> Edit: Btw. I have fw version 2.9.6. Still my ppsapp is different in size than the one uploaded by @guino. Time to fire up ghidra...

It takes me 5 mins if you zip/post it here...

guino, 2020-12-05:
> I tried without the "M", but it still isn't working.
> Here is some info of the device:
> `{"devname":"Smart Home Camera","model":"Speed [...]

This looks like older firmware which should be simpler to get into, I'm going to send you a few links to see if you can provide more information. It is possible your booloader may not even have a password in which case the UART route could be an easy option.

solidssss, 2020-12-05:
@guino
I'm sure that I pressed and held the reset button during power on, I can feel it clicking.
You should be able to access the env file from google drive now.
Here's some info from my device:
`{"devname":"Smart Home Camera","model":"Speed 4C","serialno":"xxxxxxxxx","softwareversion":"2.7.4","hardwareversion":"S4C_A1_V10_1245","firmwareversion":"ppstrong-a2-tuya2_teco-2.7.4.20191111"`

guino, 2020-12-05:
> @guino would it be possible for you to send me the dump of your flash chip? It seems that my readout is corrupted in the part before u-boot which contains the magic part. This is due to my clip not being stable enough. I've removed the chip from the board and get stable readouts now.

I will zero my cfg partition and post it since the fw.bin has serial/key info which could cause problems if passed around. Or you can provide me a way to send it privately (email, url, etc).

LouDnl, 2020-12-05:
> I will zero my cfg partition and post it since the fw.bin has serial/key info which could cause problems if passed around. Or you can provide me a way to send it privately (email, url, etc).

Do you have discord? LouD#4930

guino, 2020-12-05:
> can you please hack my ppsapp?
> my camera is Bullet 4S, firmware version 2.9.7
> thanks.
> 
> [ppsapp](https://drive.google.com/file/d/1pBzJwDj8Pz7V3hIMzhd3HVe2yVniLHhA/view?usp=sharing)

Here's the patched ppsapp to enable RTSP:
[ppsapp-rtsp-2.9.7.zip](https://github.com/guino/BazzDoorbell/files/5647724/ppsapp-rtsp-2.9.7.zip)

guino, 2020-12-05:
> Do you have discord? LouD#4930

@LouDnl Sorry don't have discord, I barely have skype/whatsapp. Here's my bin file with the enc and cfg partitions zeroed out:
[public-bazz-2.9.6.zip](https://github.com/guino/BazzDoorbell/files/5647747/public-bazz-2.9.6.zip)

If your enc and cfg partitions are the same size (or smaller than mine) I would expect you could just copy them into this bin file and flash it to your device. All enc partitions I've seen are 64k and if your cfg partition is bigger there's a good chance most of it is empty which would allow you to just copy the used part into the bin file. Something to try at least.

LouDnl, 2020-12-05:
> @LouDnl Sorry don't have discord, I barely have skype/whatsapp. Here's my bin file with the enc and cfg partitions zeroed out: [...]

Thanks, both my partitions seem in tact. Will try 👍🏻👍🏻

guino, 2020-12-05:
@solidssss your env file appears to be fine. Can you open this http://admin:056565099@IP/proc/self/root/etc/init.d/S90PPStrong and post the response here -- it seems like the bootloader is different (older) and that may be causing the difference. Only way for me to find out would be if you connected onto the UART and provided the output while doing step 5 of the install. If you have a way to plug into the UART email me and we can try a few things.

LouDnl, 2020-12-05:
Well I think I indeed completely bricked my camera 💸 the led doesn't even come on anymore. Everything gets power but thats about it. This happened before I took off the chip from the pcb when I turned it on but forgot to take off the pcb clip 🤦🏻‍♂️ some power must have gone somewhere it shouldnt have.

Gonna check my uart wiring tomorrow because they have power at the pcb. I hope thats it but I dont think so as the led wont even light up.

But as this is really searching for a needle in a haystack I might aswell buy a new one.

guino, 2020-12-05:
@LouDnl sorry to hear that - I am assuming you saw my warning about making the 3.3v mod to the programmer. On the UART front I always connect GND, TX and RX and power the camera by normal means (usb or doorbell wiring).
I also have noticed that when reading/writing the flash (with pin 6 disconnected) the light on the device would faintly turn on, but I limited connection of the clip to when I needed it.

LouDnl, 2020-12-05:
> @LouDnl sorry to hear that - I am assuming you saw my warning about making the 3.3v mod to the programmer. On the UART front I always connect GND, TX and RX (leave power disconnected) and power the camera by normal means (usb or doorbell wiring) - again UART ttl level is 3.3v
> I also have noticed that when reading/writing the flash (with pin 6 disconnected) the light on the device would faintly turn on, but I limited connection of the clip to when I needed it.

Yea my uart vcc isnt connected either, I meant the rx/tx have power from the chip like they should but something else isnt working. I didn't mod my ch341a board, havent had any problems in the past but I should have been more carefull. All other times I took the clip straight off, if it even stayed on as they usually dont. But this time it did and I forgot about it when I turned it on. So something backfired and I dont know what.

adwiraguna, 2020-12-06:
> > can you please hack my ppsapp?
> > my camera is Bullet 4S, firmware version 2.9.7
> > thanks.
> > [ppsapp](https://drive.google.com/file/d/1pBzJwDj8Pz7V3hIMzhd3HVe2yVniLHhA/view?usp=sharing)
> 
> @adwiraguna Here's the patched ppsapp to enable RTSP:
> [ppsapp-rtsp-2.9.7.zip](https://github.com/guino/BazzDoorbell/files/5647724/ppsapp-rtsp-2.9.7.zip)

thank you very much @guino, but different from your original ppsapp-rtsp it doesn't open rtsp stream on port 8554.

guino, 2020-12-06:
> thank you very much @guino, but different from your original ppsapp-rtsp it doesn't open rtsp stream on port 8554.

I will run it on mine and see what the log shows - if you enabled Telnet on yours you should be able to kill the running ppsapp and run this one from the terminal and see it it starts rtsp on a different port or if it shows an error.

adwiraguna, 2020-12-06:
> I will run it on mine and see what the log shows - if you enabled Telnet on yours you should be able to kill the running ppsapp and run this one from the terminal and see it it starts rtsp on a [...]

I'm having trouble login to telnet, because I don't know the correct hash to use in passwd file

guino, 2020-12-06:
> 
> I'm having trouble login to telnet, because I don't know the correct hash to use in passwd file

Do a passwordless telnet , just run telnetd like this: 

/mnt/mmc01/busybox tenetd -l /bin/sh

lschafroth, 2020-12-06:
Here is my ppsapp.  This is from a Mercury MI-CW017-101W 1080P camera from Walmart.
Does this need patched to get RTSP to work? Not sure of the next steps.
[ppsapp.zip](https://github.com/guino/BazzDoorbell/files/5648266/ppsapp.zip)

adwiraguna, 2020-12-06:
> I will run it on mine and see what the log shows - if you enabled Telnet on yours you should be able to kill the running ppsapp and run this one from the terminal and see it it starts rtsp on a [...]

@guino the program log doesn't contain any "rtsp" or "echoshow_init"

guino, 2020-12-06:
> @guino the program log doesn't contain any "rtsp" or "echoshow_init"

Can you provide a copy of the terminal output? You can blank out any keys and such, it looks like on your version there was nothing calling the function that checks if it should run the echoshow code. Did you try my ppsapp by any chance?

I tried your version on my doorbell and it froze up.

guino, 2020-12-06:
> Here is my ppsapp. This is from a Mercury MI-CW017-101W 1080P camera from Walmart. Firmware: 2.9.6
> Does this need patched to get RTSP to work? Not sure of the next steps.
> [ppsapp.zip](https://github.com/guino/BazzDoorbell/files/5648266/ppsapp.zip)
> 
> Here is my cmdline:
> mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd
> 
> I dont see a custom.sh and I dont know how to flash your ppsapp-rtsp file.

Did you follow the steps in the first post above to install the hack? If you did and everything worked, just copy my ppsapp-rtsp as “ppsapp” to the SD card, copy my custom.sh file and uncomment the bottom section of initrun.sh that runs custom.sh. Those files that are not in the first post above are here: https://github.com/guino/BazzDoorbell

If my ppsapp (same version as yours) doesn’t work I can look/patch yours, just let me know.

lschafroth, 2020-12-06:
There isn't a custom.sh file. That's what i don't know how to do without it.

adwiraguna, 2020-12-06:
> Can you provide a copy of the terminal output? You can blank out any keys and such, it looks like on your version there was nothing calling the function that checks if it should run the echoshow [...]

@guino [here](https://pastebin.com/jFGHXRvZ) the output 

yes, I tried your version and it open rtsp port but on my smart life app the camera shown as Doorbell not as outdoor camera, and after a few minutes looks like ppsapp is crashed

guino, 2020-12-06:
> There isn't a custom.sh file. That's what i don't know how to do without it.

It is there with all mmc files:
https://github.com/guino/BazzDoorbell/blob/master/mmc/custom.sh

solidssss, 2020-12-06:
> @solidssss your env file appears to be fine. Can you open this http://admin:056565099@IP/proc/self/root/etc/init.d/S90PPStrong and post the response here -- it seems like the bootloader is different [...]

@guino I have a raspberry pi 4 that I use to flash devices with, but I'm not sure where to connect wires to the camera.
Here's the output from S90PPStrong:
`#!/bin/sh

export PATH=/usr/bin:/sbin/:/usr/sbin:/bin

RED="[1;31m"
NORMAL="[0;39m"

echo "${GREEN} 2015 PPStrong Tech Cop.Ltd.${NORMAL}"

mkdir -p /opt/pps
MTDNUM=`cat /proc/cmdline | sed 's/.*ppsAppParts=\([0-9]\).*/\1/'`

# debug
MTDNUM=5

case $MTDNUM in
         5)
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
			break
			;;
		 7)
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
		 	break
		 	;;
		 0)     
			sleep 10
			mount -t vfat /dev/mmcblk0p1 /opt/pps
			break
			;;
		 *)
			MTDNUM=5
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
		 	;;
esac

echo "/opt/pps/" > /tmp/PPStrong.runpath
[ -e /opt/pps/initrun.sh ] && cp /opt/pps/initrun.sh /tmp/PPStart && chmod +x /tmp/PPStart && /tmp/PPStart`

guino, 2020-12-06:
Ok here's something I just found out: I ran the ppsapp 2.9.7 (newer than mine 2.9.6) on my device and it changed something in the SPI flash to where it no longer would boot!
Based on the above I will **strongly recommend NOT running any ppsapp newer than your original version** or it might make changes to your SPI flash that are incompatible with your original version. I can only expect running an older version of ppsapp (i.e. my 2.9.6) on a board with newer firmware (2.9.7+) is ok because it won't think there's anything needed to 'upgrade'. But I found out the hard way and had to use the programmer to restore my flash.

guino, 2020-12-06:
> @guino [here](https://pastebin.com/jFGHXRvZ) the output
> 
> yes, I tried your version and it open rtsp port but on my smart life app the camera shown as Doorbell not as outdoor camera, and after a few minutes looks like ppsapp is crashed

I am going to look at your ppsapp version again and try to 'inject' a call to the RTSP start function -- that will be new territory but I think it can be done -- just need to find some code that doesn't run to overwrite it (your log will be helpful for that).

guino, 2020-12-06:
@solidssss  your S90PPStrong is what we want it to be so the only reason for it to fail is that the commands in ppsMmcTool.txt failed. The only way we can move forward with your board would be for you to get your UART work OR you could potentially update to a newer firmware (hopefully 2.9.6) so the bootloader works like mine (so the patch works). Ideally it would be best to get a copy of your flash first (which would require a programmer) in case you wanted to go back to it, but it's your call.

guino, 2020-12-06:
> Here is my ppsapp. This is from a Mercury MI-CW017-101W 1080P camera from Walmart. Firmware: 2.9.6
> Does this need patched to get RTSP to work? Not sure of the next steps. [...]

I can look/patch your ppsapp -- I will try to do it soon. 
I assume you got it by uncommenting the command in initrun.sh after you installed the hack ?
Your command line is exactly as mine so there's a very good chance you have similar version of firmware as I do and I would expect everything to work once I patch the ppsapp for rtsp.

solidssss, 2020-12-06:
> @solidssss your S90PPStrong is what we want it to be so the only reason for it to fail is that the commands in ppsMmcTool.txt failed. The only way we can move forward with your board would be for [...]

In the tuya app, it says it has the latest firmware (2.7.4). So I guess the only way to update firmware is with UART? 
I don''t know where to connect the wires on the board to read/write the flash. Maybe you or someone else know?
Here's a image of the board (Nedis WIFICI20CGY): 
[image]
The wires in the picture goes to diffrent motors for pan/tilting and speaker. 
On the backside of the board its just some resistors, capacitors and sd-card reader. No pads.

LouDnl, 2020-12-06:
> > @solidssss your S90PPStrong is what we want it to be so the only reason for it to fail is that the commands in ppsMmcTool.txt failed. The only way we can move forward with your board would be for you to get your UART work OR you could potentially update to a newer firmware (hopefully 2.9.6) so the bootloader works like mine (so the patch works). Ideally it would be best to get a copy of your flash first (which would require a programmer) in case you wanted to go back to it, but it's your call.
> 
> @guino
> In the tuya app, it says it has the latest firmware (2.7.4). So I guess the only way to update firmware is with UART? 
> I don''t know where to connect the wires on the board to read/write the flash. Maybe you or someone else know?
> Here's a image of the board (Nedis WIFICI20CGY): 
> [image]
> The wires in the picture goes to diffrent motors for pan/tilting and speaker. 
> On the backside of the board its just some resistors, capacitors and sd-card reader. No pads.

Can you show the other side of the board? The uart pins are on the top side of the chip, so looking at this picture they are most likely on the other side.

solidssss, 2020-12-06:
@LouDnl
Oh, I missed 4 pads on the other side. Could it be them? Where do I connect tx/rx and gnd?
[image]

LouDnl, 2020-12-06:
> @LouDnl
> Oh, I missed 4 pads on the other side. Could it be them? Where do I connect tx/rx and gnd?
> [image]
> 
> 
> 

Well they could be, but looking at where they are going, they end up on the bottom side of the chip. Pinout shows uart at the top. Maybe they go through an inside layer. Just test them and youll know

solidssss, 2020-12-06:
@LouDnl
Just measured those 4 pads on the bottom, its 1 gnd + 3 3.3v. On the other side all of the pads are either gnd or 3.3/5v, except for the 4 pads near the motor cables, they are probably just motor pads. I don't know where to connect rx/tx.

LouDnl, 2020-12-06:
> @LouDnl
> Just measured those 4 pads on the bottom, its 1 gnd + 3 3.3v. On the other side all of the pads are either gnd or 3.3/5v, except for the 4 pads near the motor cables, they are probably just for the motor. I don't know where to connect rx/tx.
> I may be wrong but, tx/rx cannot be on 3.3/5v right?

Well that could be your uart.
https://learn.sparkfun.com/tutorials/logic-levels

jandy123, 2020-12-06:
@guino: I've patched my ppsapp to enable rtsp, but unfortunately it seems to fail with something like:

```
IPC DEFS < ENABLE_ECHO_SHOW:1 ENABLE_CHROMECAST:1 ENABLE_CLOUD_STORAGE:1 >'
[2020-12-6 13:52:19.958] tuya_ipc_echoshow.c tuya_ipc_echoshow_init(821) begin___, video channel: 0
[13:52:19.957 DEBUG tuya_pps_media.c:992] Init EchoShow[01-01 01:00:10-- TUYA Err][tuya_ipc_echoshow.c:861] create log seq <http> fail..0
[2020-12-6 13:52:19.958] tuya_ipc_echoshow.c tuya_ipc_echoshow_init(864) end___, video step(6000), audio step(2)

```
There is no mentioning of rtsp in the log. Could you please share the relevant part of your patched ppsapp ? 

Thanks in advance !

solidssss, 2020-12-06:
> Well that could be your uart.
> https://learn.sparkfun.com/tutorials/logic-levels

Nope, sadly it was controlls for one of the engines. I'm stuck now

jandy123, 2020-12-06:
@guino: Please forget my question... I was patching the wrong address. After correction, it simply works !

Some more details about my setup: 
- video doorbell bought from Action NL
- firmware version 2.9.6
- size of ppsapp executable: 4090336 bytes
- versions:
 tuya_p2p_sdk_v3.1.12
 TUYA IPC SDK V:4.7.10

jilleb, 2020-12-06:
@guino, maybe it's useful to create a ppsapp version table here.
for instance:

| firmware version | hardware version |  original ppsapp MD5 | device  | patched ppsapp | 
| ------------- |:-------------:| -----:| -----:| -----:|
| ppstrong-c51-tuya2_lcs-2.9.6.20200628 | BE8S_H1_V10_433 | 50AD9C96C65C0E446D8B3D5C8C828957 | Action LSC Doorbell | 

etc cetera. It will probably save you from a lot of ghidra work for each and every specific ppsapp 😄

russinnes, 2020-12-06:
Busy last 24h here. Just catching up now. 

Everything mostly works fine on the Energizer Connect doorbell (bell5s clone). The ppsapp / ppsapp-rtsp in the repo caused a kernel panic I'm assuming, device frozen. The referenced ppsapp with rtsp for 720p cams would crash, but continually get reloaded via the ./config.sh loop, so the device doesnt freeze up at least. In the latter scenario, the device had to be re-paired with the app, like it lost its config contents. This worked a few times, now does not. This has however created an interesting problem, the device (which was registered in tuya/energizer connect (same service)), now shows as registered to another user and needs to be unbound from that user before I can pair it again to my app(s). I havn't had a chance to dig in to it, perhaps ppsapp updates the tuya .db located on the device. I'll dig in to it and report what I find.

LouDnl, 2020-12-06:
Well for now this adventure looks over for me. Everything gets power but the led on the board no longer turns on (not broken).
There do not seem to be any short circuits as even the rx and tx of the uart get the right ttl levels but no information comes through.
For reference here are some pictures of my camera's board. Bought at Action NL and branded as LSC Smart Connect indoor IP-camera.
[image][image]

jilleb, 2020-12-06:
[ppsapp.zip](https://github.com/guino/BazzDoorbell/files/5649232/ppsapp.zip)
| firmware version | hardware version |  original ppsapp MD5 | device  |
| ------------- |:-------------:| -----:| -----:|
| ppstrong-c51-tuya2_lcs-2.9.6.20200628 | BE8S_H1_V10_433 | 50AD9C96C65C0E446D8B3D5C8C828957 | Action LSC Doorbell |

guino, 2020-12-06:
> I didn't see this ppsapp version posted yet. Diving into Ghidra now to see if I can patch it myself smile crossed_fingers
> 
> [ppsapp.zip](https://github.com/guino/BazzDoorbell/files/5649232/ppsapp.zip)
> firmware version 	hardware version 	original ppsapp MD5 	device
> ppstrong-c51-tuya2_lcs-2.9.6.20200628 	BE8S_H1_V10_433 	50AD9C96C65C0E446D8B3D5C8C828957 	Action LSC Doorbell

Love the idea, obviously we'll need cooperation from everyone to post their patched file along with version. I would say it's better to post it on a separate issue (that only contains the list) so people don't have to keep digging thru this discussion to find anything useful.

jilleb, 2020-12-06:
I totally agree.

jandy123, 2020-12-06:
+1. Also some instructions how to get the required data from table.

jandy123, 2020-12-06:
@jilleb How do I get the required table data ?

guino, 2020-12-06:
I created a new issue for anyone to post: issue #3 -- @jandy123 the way to get the information is on the first post.

jandy123, 2020-12-06:
@jilleb Well, yours is done already ;). Have a look here https://github.com/guino/BazzDoorbell/issues/3

jandy123, 2020-12-06:
@guino: Now that we have telnet, ftp, rtsp, how about the next steps ?
- protect from fw updates. Already when I start tuyasmart it asks me about fw. update. @guino: Would a fw update by safe ? 
- gain access to plain mqtt. Tuya use encrypted mqtt communication, which makes difficult to use e.g. localtuya with hassio.

[753 more replies]

closed 2021-04-14 by guino