guino/BazzDoorbell #3: List of RTSP Enabled ppsapp files

guino, 2020-12-06:
I'll start with my own device
| firmware version | hardware version |  original ppsapp MD5 | device  |
| ------------- |:-------------:| -----:| -----:|
| ppstrong-c51-tuya2_bazz-2.9.6.20200422 | BE8S_H1_V10_915 | 8c03189bcfa0c691bc2f409c15a329da | Bazz Doorbell | 

https://github.com/guino/BazzDoorbell/blob/master/ppsapp-rtsp?raw=true

jandy123, 2020-12-06:
> firmware version 	hardware version 	original ppsapp MD5 	device
>  ppstrong-c51-tuya2_lcs-2.9.6.20200628	BE8S_H1_V10_433 	50ad9c96c65c0e446d8b3d5c8c828957 	Bell 8S
> 
> https://github.com/guino/BazzDoorbell/blob/master/ppsapp-rtsp?raw=true

guino, 2020-12-06:
| firmware version | hardware version |  original ppsapp MD5 | device  |
| ------------- |:-------------:| -----:| -----:|
| ppstrong-c2-weeye-1.0.0.20160101? | unknown | 671a7f2b18a289880b1bd3b4f2cafdcf | Merkury-Smart_cam-720p | 

https://github.com/Jordan-Jarvis/Merkury-Smart_cam-720p-work/files/5633531/ppsapp-rtsp.zip

guino, 2020-12-06:
| firmware version | hardware version |  original ppsapp MD5 | device  |
| ------------- |:-------------:| -----:| -----:|
| ppstrong-c51-tuya2_geeni-2.9.6 | cw017-101W-1080P | ec64bd06e807db52f50c3009827ab676 | merkury-cw017-101W-1080P | 

[merkury-cw017-101W-1080P-2.9.6-ppsapp-rtsp.zip](https://github.com/guino/BazzDoorbell/files/5649407/merkury-cw017-101W-1080P-2.9.6-ppsapp-rtsp.zip)

russinnes, 2020-12-07:
Firmware Version | Hardware Version | Original ppsapp MD5 | device
----------------- | ------------------|-----------------------|---------
ppstrong-c51-tuya2_xtremeecho-2.9.7.20200730|BE5S_H1_V10_915|A56A4D5DE71F262D66E9DA54A64BE971|Smart Home Camera Bell 5S / **Energizer Connect Doorbell**

[ppsapp-rtsp-bell5s.zip](https://github.com/guino/BazzDoorbell/files/5650267/ppsapp-rtsp-bell5s.zip)

adwiraguna, 2020-12-07:
| Firmware Version | Hardware Version | Original ppsapp MD5 | device |
|--------------------|---------------------|-------------------------|--------|
| ppstrong-c51-tuya2_bardi-2.9.7.20200907 | B4S_V10_H1_2063 | f4874ce6f40ff1a32e2bc1232e157105 | Bardi Outdoor Camera |
[ppsapp-rtsp-2.9.7.zip](https://github.com/guino/BazzDoorbell/files/5650388/ppsapp-rtsp-2.9.7.zip)

guino, 2020-12-09:
For more patch files please check https://github.com/guino/ppsapp-rtsp/issues/1 -- we should only be posting patches from now on (we should not be making ppsapp files available for download -- only patches)

closed 2020-12-09 by guino