guino/BazzDoorbell #4: Off-cloud research

jilleb, 2020-12-08:
@guino, I hope you don't mind me creating a seperate issue, but I think it helps to keep the other topics on-topic and easier to read.

So let's discuss here what's what when it comes to off-cloud/offline usage of the doorbell. Currently it uses the Tuya cloud, which means the following, according to @m11tch research

> when internet connectivity is blocked:
> 
> * you can not view recordings from tuya app, so seems to be streaming directly from SD upon request.
> * RTSP is not available. (connection refused) (will check logs later)
> * snap.cgi and mjpeg.cgi work fine
> * button press does make the "ding dong" sound, but no mqtt message is sent (will check logs later)
> * Camera still switches to night mode if dark

m11tch, 2020-12-08:
@jilleb I was thinking the same as we were taking over the other thread quite a bit :D 
Doing some more research later, will update this post with new findings.

jilleb, 2020-12-08:
dmesg output when doorbell is pressed while it's not connected to the cloud:

```
initS (906): drop_caches: 3
door bell  press down
 Strnio_ioctl
AcShdn,open pwr
CMT211XA_Send size: 68
 ###cfg gpio OK name:spi1_sdo
 cfg gpio ok:spi1_sdo
irq-----
send data:68
initS (906): drop_caches: 3
door bell  press up
send data:68
send data:68
send data:68
send data:68
CMT211XA package Send finished!
initS (906): drop_caches: 3
 Strnio_ioctl
AcShdn,close pwr
```

jilleb, 2020-12-08:
> I don't see motion detected in dmesg though, perhaps motion detection is happening in tuya cloud or it is part of ppsapp which is not fully started yet?

Good question.. I'm looking for it in Ghidra. Sounds like something that would possibly happen in the cloud. I'll get back on this.

guino, 2020-12-08:
I have been downloading/viewing my .data files for the longest time - they play just fine in VLC when you set the demux to h264 demuxer (in advanced settings). It is kind of a pain having to switch between automatic and h264 but it works - just no audio.
I personally like having the tuya app, so I plan to keep it. Just wanted to be able to use RTSP, mjpeg and be able to backup the SD card files as they’re created - and now I can do it all.
Also I don’t record everything, only motion events and my camera doesn’t run hot at all. 
@jandy123 if your camera is running hot, check if you don’t have any left over streamer-arm or other processes running in the background from testing, etc. The mjpeg support I put together is just a jerry rig, so the ideal solution is that an application should be made for that purpose (and to make it more efficient).

guino, 2020-12-08:
The fact that I don’t record everything may also impact if the camera runs hot or not (haven’t tried recording everything) plus I have a 9v transformer at the house doorbell.

m11tch, 2020-12-08:
> > I don't see motion detected in dmesg though, perhaps motion detection is happening in tuya cloud or it is part of ppsapp which is not fully started yet?
> 
> Good question.. I'm looking for it in Ghidra. Sounds like something that would possibly happen in the cloud. I'll get back on this.

maybe getting ppsap to run propperly is enough?, i'm no expert in Ghidra 😞 maybe it is enough if we can disable the mqtt connection in ppsap as it is just constantly trying to reconnect, each time with a longer sleep time.. and not doing much else...

`[01-01 01:00:35-- TUYA Debug][mqtt_client.c:1230] mqtt fail_cnt:6 and sleeptime:4756 ms`

jandy123, 2020-12-08:
@guino: 
- Welcome back ! Guess, different timezones, jobs, etc.

- I have no other apps running, streaming etc. And this is without enabling continuous recording. Just record on event setting in the app. It's also interesting that for you, apparently, 9V AC  is enough. I couldn't get it running with anything (apart from USB) but with the supplied 12V DC adapter.
- If I run it from USB,  then it's getting a bit warm on the opposite side of the lens. If I run it from the 12V adapter, it gets quite hot around where the wires are connected. Again, I suspect some LDO and/or diode bridge ? I.e., dropping from 12V AC to 3.3V has to happen somewhere...

jandy123, 2020-12-08:
For me rtsp streaming still works after blocking outside connection.

jilleb, 2020-12-08:
I'll check. Rtsp does work after blocking web access. But upon reboot it's dead.

I just restarted ppsapp from the console by accident.. got  a lot of interesting output I will post later when I'm back at the pc.

guino, 2020-12-08:
I would expect recording continuously would make it hotter than recording events only - have not paid attention to be honest. The other doorbell I had (EZVIZ DB1) was hot as hell compared to this one. It may require more current to run at lower voltage (power=voltage * current) so maybe a bigger low voltage transformer would work, just not sure if it’s worth messing with it when it is already working.

If you guys are going to be doing stuff offline using the logs  I think we should patch the debug log function to always print the output (there’s some sort of output level setting which I haven’t tried to find yet).

Regarding mqtt are those messages encrypted or require secret keys? Cause if not you could just write a new address in ppsapp and use your own server? Just a thought. It should be possible to ‘patch’ ppsapp into thinking the connection is working but that’s a bunch more work specially to do in multiple ppsapp versions.

In my case I have a guest network that only has internet access where I put these untrusted devices , then I limit what I access with iptables/ebtables, so I don’t mind that they’re online.

m11tch, 2020-12-08:
@guino connections to the tuya mqtt server are over TLS and require username/password. for internal usage this is ofc not necessary.. I will try redirecting the connection to a internal mqtt server that supports TLS to see if that is enough..

jandy123, 2020-12-08:
@guino 
>  It may require more current to run at lower voltage 

Nope, this is not the cause for heating, since it's definitely cooler at 5V from USB. It's just some linear regulator which transforms the 12V > 3.3V that dissipates heat, I guess. But then, I didn't even open the device yet...

>  If you guys are going to be doing stuff offline using the logs I think we should patch the debug log function to always print the output (there’s some sort of output level setting which I haven’t tried to find yet).

As I said, I get same log output from ppsapp, so I can still see the forwarded mqtt events.

jandy123, 2020-12-08:
Also, in  my case, the RTSP server seems to be created on the fly, when a client request (from LAN) is made. This works (as it should !!)  without the cloud struff. I can post logs if needed.

jandy123, 2020-12-08:
@m11tch 

> I will try redirecting the connection to a internal mqtt server that supports TLS to see if that is enough..

This won;t work. I've already looked at the communication in wireshark...

guino, 2020-12-08:
It seems they have made changes from 2.9.6 to 2.9.7 in a number of aspects (ie blocking HD RTSP, changes to log functions, some logging and probably more) -- those are the things I noticed when patching some of the files.

m11tch, 2020-12-08:
> @m11tch
> 
> > I will try redirecting the connection to a internal mqtt server that supports TLS to see if that is enough..
> 
> This won't work. I've already looked at the communication in wireshark...

I don't need the actual data, was just hoping that it would be enough to let ppsap to start up further.. 

@jandy123 RTSP does not seem to happen for me, what kind of request are you sending?

jandy123, 2020-12-08:
@guino 
> I  belive both me and jandy are running 2.9.6 though..

Well, yes, but then we have different ppsapp...

jandy123, 2020-12-08:
As noted in the other thread by @adwiraguna,  CPU usage in rtsp streaming mode gets to 100%. IS this only with blocked tuya servers ???

jandy123, 2020-12-08:
@m11tch 

> what kind of request are you sending? 

What exactly you mean ? I just say `vlc rtsp://192.168.2.179:8554`

m11tch, 2020-12-08:
> As noted in the other thread by @adwiraguna, CPU usage in rtsp streaming mode gets to 100%. IS this only with blocked tuya servers ???

Perhaps this is why they decided to block HD rtsp streams in 2.9.7? 

@jandy123 I will try to open rtsp stream again later today when connection to tuya is blocked.. will come back with the results..

jandy123, 2020-12-08:
Well, CPU usage when streaming over RTSP is not that bad, actually.

Assuming blocked tuya servers, idling is round 3-5% (reported by top). When HD rtsp streaming to one client, it goes up to about 17-20%. The process itself ppsapp does take about 100% CPU, though.

jilleb, 2020-12-08:
> Also, in my case, the RTSP server seems to be created on the fly, when a client request (from LAN) is made. This works (as it should !!) without the cloud struff. I can post logs if needed.

We have the same doorbell/firmware/ppsapp, but it isn't working 

debug log from ppsapp:
``` 
info.p2p_factory_type: 14
---- uuid_licence_switch 0
info.p2p_factory_type: 14
name: Smart Home Camera
factory: PPSTRONG
device_type: 191
model: Bell 8S
hardware_version: BE8S_H1_V10_433
software_version: 2.9.6
firmware_version: ppstrong-c51-tuya2_lcs-2.9.6.20200628
sn: xxxxxxx
p2p_uuid:
identity_card: xxxxxxxxxx
licence_id: xxxxxxxxxxxx
licence_key: xxxxxxxxxxxxxxx
this board do not support temp/ humidity
legace mtd get info mtd num  7
mtd name bld num 0 eb size 0x10000 data sie 0x30000
mtd name env num 1 eb size 0x10000 data sie 0x10000
mtd name enc num 2 eb size 0x10000 data sie 0x10000
mtd name sysflg num 3 eb size 0x10000 data sie 0x10000
mtd name sys num 4 eb size 0x10000 data sie 0x310000
mtd name app num 5 eb size 0x10000 data sie 0x440000
mtd name cfg num 6 eb size 0x10000 data sie 0x50000
[08:35:02.197 INFO pps_watchdog.c:292] Watchdong enabled....
start rea random
build data:Mar 31 2020,pps_media_version:20200325--4.3.9------>sys_cfg_size:4128
###version:1.0.2,set sensor Type gc2063mipi
[Func]:cmos_set_wdr_mode [Line]:666 [Info]:linear mode
read i2c error ! :ffffffff  -1
read i2c error ! :ffffffff  -1
read i2c error ! :ffffffff  -1
read i2c error ! :ffffffff  -1
[Func]:DrcCheckCmosParam [Line]:2195 [Info]:Invalid u8Asymmetry!
pipe:0,I2c:0___init____addr:110
=========================================================
=== hisiev300 gc2063mipi sensor 1080P30fps(mipi port) init success!=====
=========================================================
vi init ok----
set ok-----------
2 close aec-----
------------set default aec para
id::0
id::1
id::2
id::3
id::4
id::5
id::6
id::7
id::8
id::9

---->sensorName:gc2063mipi,modeName:Bell^8S,lensInfo:f2.4A

pthread exit:4,yuvTask
BCD init ok---------------

pthread exit:9,defaultDsp
ioctl: Bad file descriptor
open out
shartp  set sta:0
-----------------------------------seq:43--->
input vol=[90], output vol=[50], aec_enable=[1]
[... 193 lines cut ...]
[01-01 01:00:05-- TUYA Debug][tuya_iot_wifi_api.c:340] wifi netstat changed to:5  -->>
[08:35:12.903 DEBUG main_app.c:96] ipc status changed:5[08:35:12.903 DEBUG main_app.c:109] mqtt is offline
[01-01 01:00:05-- TUYA Debug][tuya_iot_wifi_api.c:343] wifi netstat changed to:5  <<--
[01-01 01:00:05-- TUYA Debug][mqc_app.c:911] get_gw_nw_status:0
[01-01 01:00:05-- TUYA Err][mqtt_client.c:1036] mqtt task is not permit to execute. short sleep..
[08:35:13.770 INFO third_party_base_fun.c:291] open file failed
open: No such file or directory
[01-01 01:00:06-- TUYA Debug][app_agent.c:870] Setup TCP Server On Port:6668
store dns info to position 7 for m2.tuyaeu.com 0x12b81f5a
[01-01 01:00:06-- TUYA Debug][mqtt_client.c:1048] select mqtt host:m2.tuyaeu.com
[01-01 01:00:06-- TUYA Notice][mqtt_client.c:1061] mqtt get serve ip success
[01-01 01:00:06-- TUYA Debug][mqtt_client.c:734] serverIP 314056538 port:8883
[01-01 01:00:10-- TUYA Debug][tuya_iot_wifi_api.c:340] wifi netstat changed to:6  -->>
[08:35:17.870 DEBUG main_app.c:96] ipc status changed:6[08:35:17.870 DEBUG main_app.c:109] mqtt is offline
[01-01 01:00:10-- TUYA Debug][tuya_iot_wifi_api.c:343] wifi netstat changed to:6  <<--
[08:35:17.951 DEBUG tuya_network_manager.c:535] WIFI[0] Get IP:192.168.1.97
[01-01 01:00:10-- TUYA Debug][app_agent.c:692] Create UDP Socket To Broadcast
[08:35:22.000 DEBUG pps_net_manager_client.c:189] NETWORK KEEPALIVE
[08:35:22.962 DEBUG tuya_network_manager.c:535] WIFI[0] Get IP:192.168.1.97
[08:35:27.972 DEBUG tuya_network_manager.c:535] WIFI[0] Get IP:192.168.1.97
[08:35:32.964 DEBUG tuya_network_manager.c:535] WIFI[0] Get IP:192.168.1.97
[08:35:37.952 DEBUG tuya_network_manager.c:535] WIFI[0] Get IP:192.168.1.97
[08:35:42.966 DEBUG tuya_network_manager.c:535] WIFI[0] Get IP:192.168.1.97
[08:35:43.000 DEBUG pps_net_manager_client.c:189] NETWORK KEEPALIVE
```

jandy123, 2020-12-08:
@jilleb I now see what you mean. If start ppsapp *without* tuya cloud connection, indeed, rtsp streaming does not work. Seems that ppsapp gets stuck on connecting to tuya hosts:

`
[01-01 01:01:57-- TUYA Err][mqtt_client.c:1051] resolve mqtt host Fail:m2.tuyaeu.com erro:-100
[18:12:27.766 DEBUG tuya_network_manager.c:535] WIFI[0] Get IP:192.168.2.179
[18:12:28.009 DEBUG pps_net_manager_client.c:189] NETWORK KEEPALIVE
[18:12:32.773 DEBUG tuya_network_manager.c:535] WIFI[0] Get IP:192.168.2.179
[18:12:37.782 DEBUG tuya_network_manager.c:535] WIFI[0] Get IP:192.168.2.179
gethostbyname error 0

`

m11tch, 2020-12-08:
> > Also, in my case, the RTSP server seems to be created on the fly, when a client request (from LAN) is made. This works (as it should !!) without the cloud struff. I can post logs if needed.
> 
> We have the same doorbell/firmware/ppsapp, but it isn't working
> 
> debug log from ppsapp:
> 
> [debug.txt](https://github.com/guino/BazzDoorbell/files/5660683/debug.txt)
> 
> The relevant part seems to be:
> IPC DEFS < ENABLE_ECHO_SHOW:1 ENABLE_CHROMECAST:1 ENABLE_CLOUD_STORAGE:1 >'
> 
> if we can influence these definitions.. would be cool

I noticed some config files in /home/cfg, maybe we can do something with those to manipulate? although I don't see anything usefull at first glance..

jandy123, 2020-12-08:
@jilleb 

I never get to 

> IPC DEFS < ENABLE_ECHO_SHOW:1 ENABLE_CHROMECAST:1 ENABLE_CLOUD_STORAGE:1 >

In my case, ppsapp gets stuck into

`[01-01 01:01:57-- TUYA Err][mqtt_client.c:1051] resolve mqtt host Fail:m2.tuyaeu.com erro:-100
[18:12:27.766 DEBUG tuya_network_manager.c:535] WIFI[0] Get IP:192.168.2.179
[18:12:28.009 DEBUG pps_net_manager_client.c:189] NETWORK KEEPALIVE
[18:12:32.773 DEBUG tuya_network_manager.c:535] WIFI[0] Get IP:192.168.2.179
[18:12:37.782 DEBUG tuya_network_manager.c:535] WIFI[0] Get IP:192.168.2.179
gethostbyname error 0`

m11tch, 2020-12-08:
> @jilleb
>
> I never get to
>
>
> In my case, ppsapp gets stuck into
>
> `[01-01 01:01:57-- TUYA Err][mqtt_client.c:1051] resolve mqtt host Fail:m2.tuyaeu.com erro:-100 [18:12:27.766 DEBUG [...]

I think connection was not blocked during the startup of this debug log, can you confirm @jilleb ?

jandy123, 2020-12-08:
@m11tch 
> I noticed some config files in /home/cfg

I did see the stuff in /home/cfg. Tried to modify ota:0 in /home/cfg/capa.info in a hope to quiet the tuya app upgrade banner, but it didn't work....

jilleb, 2020-12-08:
The connectiong was blocked all the time.

I started ppsapp from sd-card in the console with the --help commandline, hoping it would generate some useful output.But....... after closing it, the doorbell no longer boots 😆

jandy123, 2020-12-08:
Connection was blocked all the time here too. 

@jilleb Hmm, no boot at all ???

jilleb, 2020-12-08:
> @jilleb Hmm, no boot at all ???

Nope.. but I'm confident it can be fixed 😄

m11tch, 2020-12-08:
I was thinking of only allowing the mqtt connection, to see if this changes anything for ppsapp startup anyone tried this?

jandy123, 2020-12-08:
@m11tch: The is a certificate file in home/ca.crt, but I doubt it will work.

jandy123, 2020-12-08:
It's also worrying that tuya can upgrade the device firmware at any moment and without any warning. 

As a 1st protection, I've just changed the string `upgrade.bin` to `nograde.bin` in `ppsapp`.

m11tch, 2020-12-08:
@jandy123 the CA cert seems to be related to MQTT, so that is also a dead end.. 

I think modification of ppsapp is needed, either to modify the urls to connect to or to bypass the connection stages so that it atleast launches.. 

unfortunately I lack the knowledge to do that :D, I will try some more stuff tomorrow.. hopefully some of the other guys here have some bright ideas :D

guino, 2020-12-09:
to unblock the normal features while offline I think all you may need to do is disable this loop in the main sub:
[image]

jilleb, 2020-12-09:
I'll dive into that.

I also noticed that it doesn't syn the OSD date/time. Would be best if this was an optional thing. 😄 

[image]

m11tch, 2020-12-09:
yeah the ntp servers it uses are also in the cloud,, I think I saw them in on of the config files though? So perhaps that is an easy fix to point it to a internal NTP server...

> I'll dive into that.
> 
> I also noticed that it doesn't sync the OSD date/time when not connected to the cloud. Would be best if this was an optional thing. 😄
> 
> [image]

jandy123, 2020-12-09:
@jilleb Please keep us posted. I could also have a look if needed.

@guino Thanks again !

m11tch, 2020-12-09:
this works to get notifications from button press: 

maybe not the best way, but it works :joy:
```
while true; do
        if dmesg -c | grep -q "door bell  press down"
        then
                /mnt/mmc01/mosquitto_pub -h 192.168.2.2 -m "pushed" -t home/doorbell/button
        fi
done
```

jilleb, 2020-12-09:
After diving into Ghidra, I found some things regarding "flight mode"
It's set in tuya_config.json. Setting it doesn't seem to do anything when it's not connected to the cloud.

jandy123, 2020-12-09:
@jilleb 

> @jandy123 @m11tch: I have an IR cam here, I can take some pics to see what's heating up

I would be very curious to know.

jandy123, 2020-12-09:
I've followed @guino's suggestion and we are one step closer. If time is correctly set before starting rtspapp, then we can have off-line rtsp streaming working. 

What bothers me still is that ppsapp enters some loop still trying to reach Tuya's NTP server... Ideas ?

jandy123, 2020-12-09:
Ok, I think it's done, ppsapp seems to behave normally ! 

Can any of you test, i.e., do you have the Action NL doorbell ?

m11tch, 2020-12-09:
@jandy123 I have the Action NL Doorbell.. gimme the file and i'll test it :) 

in regards to the ntp, did you try changing ntp adresses in dev_settings.json?

jandy123, 2020-12-09:
@m11tch 

> in regards to the ntp, did you try changing ntp adresses in dev_settings.json?

Nope, not yet. Please test; I have a call...

I attached the patched app.
[ppsapp-rtsp2.zip](https://github.com/guino/BazzDoorbell/files/5666222/ppsapp-rtsp2.zip)

jilleb, 2020-12-09:
Weird, I don't have dev_settings.json

jilleb, 2020-12-09:
@jandy123 RTSP feed now works when i'm offline!

jandy123, 2020-12-09:
@jilleb Did you have to set time/date before starting ppsapp ?

jandy123, 2020-12-09:
Well, here it works. Just modify custom.sh to something like:

`  /mnt/mmc01/busybox ntpd -d -q -n -p 192.53.103.108 #ptbtime1.ptb.de
# debug info from ppsapp-rtsp forwarded to output.log
  /mnt/mmc01/ppsapp-rtsp 2>&1 | /mnt/mmc01/log_parser.sh /mnt/mmc01/output.log &
`

jilleb, 2020-12-09:
- setting time works before starting ppsapp 👍 
- motion detection.. haven't tested yet, but I use Surveillance Station for this.
- I did notice that RTSP feed performance is poor shortly after reboot.

jandy123, 2020-12-09:
@jilleb Could you please fire up your IR camera ? I'm still bothered by the fact that it gets quite warm when running on 12V.

Also, do you get the "offer" to update  fw. inthe Tuya app ? I'd like to bypass this bloody nag and to make sure that we won't get surprise updates from Tuya. This is next on my list...

Now, is the latest patched  ppsapp working normally too ? I.e., if wan access to Tuya servers is granted  ;) ?

m11tch, 2020-12-09:
@jandy123 Sorry, work things came up, so did not get to test it earlier.  used your  nptpd suggestion in custom.sh and that works fine 👍 

Results: 
- motion detection works
- button press works 
- RTSP works. seems to be even faster then before (more FPS) 
- ofcourse it still tries to connect to tuya cloud endlessly. 

I also get the offer about fw upgrade each time I launch the app, but at the stage we are now with the modified ppsapp I don't see a reason to allow my doorbell to connect to tuya cloud ever again ;)

jandy123, 2020-12-09:
@m11tch 
> Sorry, work things came up

Well, it's about same here...

>  also get the offer about fw upgrade each time I launch the app, but at the stage we are now with the modified ppsapp I don't see a reason to allow my doorbell to connect to tuya cloud ever again ;)

Yep, sure, still, I'm not yet sure if I want it on/off-line yet.

m11tch, 2020-12-09:
@jandy123 I do have an issue when viewing the RTSP stream via mplayer, after about 15 minutes the doorbell reboots. do you guys  experience similar behavior?

guino, 2020-12-09:
I am glad you guys got a handle on the time thing - I was indeed going to suggest to just set the time from the start script instead of letting ppsapp do it. 
My suggestion with the mqtt stuff was to modify the loop so it no longer waits instead of modifying where it connects to but whatever works for you.
At some point you may need to decide between online/offline but you should have the ability to run multiple ppsapp versions from the SD card based on whatever you see fit. Ir maybe when you’re home you want it offline, maybe you’re traveling and want it online so you can use the app - you could make something in the start script to start the online or offline version based on something else you define (ping to tuya server, calendar information from somewhere or just a query to a local server, etc)

jandy123, 2020-12-09:
@m11tch Well, didn't really test this. But, please do so. As I said, rtsp HD is too much for my OPI + hasssio to handle... I'll stick to the mjpeg.

@guino: 

> My suggestion with the mqtt stuff was to modify the loop so it no longer waits instead....

Well,, that's what I did. Just have a look :). 

==== 

One more question: Can we bypass killing of the original ppsapp ? I'm a bit worried, if we  kill it in whatever state, it may erase everything, put fire to our house,  etc ;).

jilleb, 2020-12-09:
https://imgur.com/gallery/ZJWYDVp
Heat signature pictures here @jandy123

jandy123, 2020-12-09:
@jileb: Thanks a lot ! Since I didn't even have to open my camera (thanks to @guino !!!), I'll have to investigate a bit which chip is what. But, maybe you guys know that already ? When on 12V, it gets hot exactly under the camera and middle, where the wires are connected. This seems to agree with your photos. There is a big with "chip" with few legs getting hot on the side, which probably is some linear regulator ??? For the rest, I couldn't spot anything unusual, right ???

Anyways, what do you guys think ?

jandy123, 2020-12-09:
@jilleb Please test also rtsp streaming while connected to tuya servers. It could be that there is still some hidden timeout... which would be wired, since that would mean on a summer day and with no internet, under normal circumstances, the thing would get fried from so many reboots... Btw., how do we reboot the thing gracefully ???

m11tch, 2020-12-09:
@jandy123 The crashes also occur when it is connected to cloud.. I found this out the hard way when I tested this before blocking its internet connection. (the device rebooted when I just fell asleep and the bootup melody woke me up again 😢 )

[347 more replies]