guino/BazzDoorbell #5: Upgrade with patch

Ierlandfan, 2020-12-10:
According to this http://<ip>/flash/upgrade/release_package we can upgrade by file. 
You can upload a firmware file. Can we create such a file?

guino, 2020-12-10:
From the few bin files we have seen it looks like the file must be signed for the upgrade to accept it, so it is not something I would try without a programmer to backup/restore the flash and I doubt the signing certificate is on the flash for us to create a valid upgrade file. There’s a better chance of getting the upgrade to work with ppsMmcTool.txt which may not require a signed firmware file.

guino, 2020-12-19:
So digging a little more on this I found that the upgrade file doesn’t seem to be signed but it does have headers/compression etc which would have to work out, but the main problem is that I would likely need a more current firmware (2.9.6/2.97) update file to investigate this further.

Ierlandfan, 2020-12-20:
There is also the possibility to upgrade psapp only. http:///flash/upgrade/psapp. Maybe we can use that. (exact link to be found in Ghidra)

guino, 2020-12-20:
@Ierlandfan based on what I have seen ppsapp is stored and compressed ram file system which (cramfs) which doesn't allow any type of changes, so the only way to update ppsapp in the flash is to rebuild the entire cramfs partition (like in did initially with a programmer). I know for a fact there's a way to update the entire cramfs partition (and only that) from an upgrade file but unless I have the upgrade file for one of the current firmware versions (2.9.6 or 2.9.7) it would take too long to figure out the format specially considering there's already a SD card method available.
In any case it should be possible to 'upgrade' with a patch -- just requires someone spending enough time looking at code to figure out the format (and possibly using a programmer to backup/try different things).