guino/BazzDoorbell #6: Chime research

jilleb, 2020-12-11:


m11tch, 2020-12-11:
Not compatible:
[image]

jilleb, 2020-12-11:
I contacted Marmitek about their "connect kit" to be able to work with the old, wired chime. Unfortunately, they don't sell it seperately... But I'm having my hopes up that the Ring Doorbell Power Kit is actually a similar kind of thing, which enables the doorbell to be retrofitted on existing wire/transformer/chime hardware.

jandy123, 2020-12-11:
@jilleb: 

Regarding the ring doorbell power kit, have a look here:  https://www.instructables.com/UK-Ring-Video-Doorbell-Pro-Working-With-Mechanical/ 

Apparently this is how it works:
"
One important note is that you'll need to tell your Ring Doorbell that it is connected to a mechanical chime. This is found in the device settings in the Ring app. When you do this, it basically tells the doorbell to pull a big lump of current (about 1 Amp), and then release the current and this is what energises the coil in the chime making the hammer move and strike the metal bars (the ding, and then the dong).
"

So, you'd really need to test this before going this path. I believe I read somewhere that the LSC app has such an option ??? 

Finally, the "ring doorbell power kit" is just a resistor 25-30 ohm that limits current through the doorbell. Then, given explanation aboe, it all makes sense...

m11tch, 2020-12-11:
> @jilleb:
>
> Regarding the ring doorbell power kit, have a look here: https://www.instructables.com/UK-Ring-Video-Doorbell-Pro-Working-With-Mechanical/
>
> Apparently this is how it works:
> "
> One important [...]

Would this be what the 'external chime' setting is for in dev_settings.json?

jilleb, 2020-12-11:
Probably. It's worth a try.

m11tch, 2020-12-12:
I did some quick measurements on the AC side, I don't really see an increase in current draw when the button is pressed.
I'll try some measurements on DC side later

jandy123, 2020-12-12:
@m11tch If you try, please put some series resistor to limit current, just in case...

m11tch, 2020-12-12:
it seems to draw 0.16A at normal operation, if i press the button, it goes up to about 0.32A for about 1 second and then returns to 0.16A again..  (during my test it was hooked up to the included 12V 1A Adapter) 

I only have a cheap multimeter, so I don't know how accurate the measurements are :D

jandy123, 2020-12-12:
@m11tch Maybe the 12V 1A adapter is the limiting factor ? Question is, would this trigger an analog bell assuming there is a 25-30 ohm resistor in parallel with the bell coils ?

m11tch, 2020-12-12:
@jandy123 I'm not sure, another difference is that normall "bell transformers"(is that the correct english? :joy:) are 6,8,12,24V AC instead of DC like the adapter, does our doorbell even work on 12V AC? 

I currently have a ancient trafo that doesn't go higher then 8V AC and I want to get rid of that thing.. So I'm not sure if I can propperly test and i'm also hesistant to plug the bell into AC considering it comes with a DC adapter..

jandy123, 2020-12-12:
@m11tch  

> does our doorbell even work on 12V AC?

This is a very good question ;). 

> I currently have a ancient trafo that doesn't go higher then 8V AC and I want to get rid of that thing.. So I'm not sure if I can properly test and i'm also hesitant to plug the bell into AC considering it comes with a DC adapter..

I'm in the same situation. The original Bazz doorbell is rated12-24V AC, but who knows ? I cannot easily test, since I do not really have access to it.

jandy123, 2020-12-12:
> Edit2: but from what I read, the marmitek doesn't come with a DC adapter at all.. would that mean the device can run on AC too? confused

For sure it works on AC. It's in the instructions manual.

jilleb, 2020-12-12:
Allright, lets do some additional research then. The Marmitek BuzzLo has the following in the manual:

This product requires an 8 ~ 16 VAC 50 Hz power 
source and a stable internet connection

So 8V AC should work...

jandy123, 2020-12-12:
@jilleb 8V AC would be even better w.r.t. heat generation.

jandy123, 2020-12-12:
Let's see, does the device even work with < 12V DC ?  This should be a 1st safer test.

jilleb, 2020-12-12:
I could test it, although I'm not sure if my old ass transformer would work:

[image]

m11tch, 2020-12-12:
@jilleb my transformer is bakelite, yours is very new compared to mine :D

m11tch, 2020-12-12:
> 😂😂😂
> 
> I found a topic regarding a similar setup, I'll read through it during my coffee break:
> (It's dutch, although I have a suspicion that some of us are Dutch 😂)
> https://gathering.tweakers.net/forum/list_messages/1925242

the fact that we all have the action NL bell should tell you enough :D 
the topic you mentioned is about a compeltely different bell though

jandy123, 2020-12-12:
Look here https://gathering.tweakers.net/forum/list_messages/2031224/1 

Search posts by Allewijn. He claims that the device  works on  an AC 8v Hager beltrafo.

m11tch, 2020-12-12:
so @jilleb are you volunteering? :joy:

jandy123, 2020-12-12:
I'm quite sure device supports 12V AC. I've just reversed the 12VDC adapter wiring and device works.

jandy123, 2020-12-12:
@m11tch My old analog bell looks just like the one you linked to. For sure, if the device works with analog bell alone, same trick won't work withe an electronic one.

m11tch, 2020-12-12:
> @m11tch My old analog bell looks just like the one you linked to. For sure, if the device works with analog bell alone, same trick won't work withe an electronic one.

Well I believe a  analog ding dong chime works differently then a traditional bell :p

jilleb, 2020-12-12:
> so @jilleb are you volunteering? 😂

My daytime job is in the field of testing... So let's do it 😂
And if it fails: I've suffered worse bricks in my life, and each of them is a cool birthdayparty story 😂😂

jandy123, 2020-12-12:
I've tried on my trafo 8V AC. It tries to boot but when about to produce the sounds it resets. I suppose it's just not enough juice...

Will redo the wiring with the 12V DC adapter.

jandy123, 2020-12-12:
Right,so it's all installed here using the 12V DC adapter, although, I'm 100% sure 12V AC would be as good; 

=====

So, all in happy with the result, wife too ;), etc. Now, just time will tell how long it's going to last...

Anyways, thank you guys @guino , @m11tch and @jilleb;  @guino it's just amazing how much time you put in all this ! It's been a great journey here !!!

Well, I'll still keep an eye on this thread and see how it all goes. If you need my input, I'll be around....

m11tch, 2020-12-12:
> Right,so it's all installed here using the 12V DC adapter, although, I'm 100% sure 12V AC would be as good;
>
> =====
>
> So, all in happy with the result, wife too ;), etc. Now, just time will tell how [...]

Yeah I also installed it today with the included AC adapter. @jandy123 thanks for your awesome work patching for offline mode too! 👍

jilleb, 2020-12-12:
Hi guys, indeed, great work!!

I just installed the doorbell on my 8VAC transformer, and i bypassed my good old faithful but crappy chime hardware. It's all working, I'm using Homey with spoken notifications to Google Home ("the bell is ringing" 😂😂) and the Marmitek chime, seen here from the inside:

[image]

I'll make a new issue regarding chime tweaking on the Doorbell side. It's pretty easy, replacing the wav files with a specific wav formatted file.

Hopefully the Marmitek chime has UART access as well, and from the looks of it, it does....fingers crossed for new tones there 😁

m11tch, 2020-12-12:
> Hi guys, indeed, great work!!
>
> I just installed the doorbell on my 8VAC transformer, and i bypassed my good old faithful but crappy chime hardware. It's all working, I'm using Homey with spoken [...]

did you end up buying the Marmitek chime? :D

jilleb, 2020-12-12:
Yes! And it came in the mail today, and it worked withing 1 second 😁

jandy123, 2020-12-12:
Guys, is there a way for a clean reboot ? Still struggling with this, now that the thing is mounted ;).

m11tch, 2020-12-12:
@jandy123 I think I came accross a URL to reboot it in ghidra.. let me check

jandy123, 2020-12-12:
@m11tch Hmm, interesting. How did you come across this one ? Does it unmount sdcard, etc ?

lesleyvanrijn, 2020-12-12:
@jandy123 There is a whole list available in a different repository. [https://github.com/AMoo-Miki/homebridge-tuya-lan/issues/4#issuecomment-732422778](AMoo-Miki/homebridge-tuya-lan#4)

lesleyvanrijn, 2020-12-12:
> I just installed the doorbell on my 8VAC transformer, and i bypassed my good old faithful but crappy chime hardware. It's all working, I'm using Homey with spoken notifications to Google Home ("the bell is ringing" 😂😂) and the Marmitek chime, seen here from the inside:

@jilleb What is triggering the Homey event? Have you added the MQTT client?

guino, 2020-12-12:
I also have 8VAC transformer and it is wired within the house (existing chime) like this: 

[image]

The bell doesn’t ring well like it used to but you can definitely tell the button was pushed. I thought about changing the chime but I kind of like it being muffled specially having the app notification, so it doesn’t wake me up if I am sleeping.

jandy123, 2020-12-12:
@jilleb What's the "main RF chip" in the marmitek chime ? Could you take a picture on the other board side ?

m11tch, 2020-12-12:
> @m11tch Hmm, interesting. How did you come across this one ? Does it unmount sdcard, etc ?

No clue what it actually does, I only spotted it 😂

jilleb, 2020-12-12:
> @jilleb What's the "main RF chip" in the marmitek chime ? Could you take a picture on the other board side ?
The backside:
[image]

MokanHan, 2020-12-13:
> @m11tch I've managed to get the 2020 action chime working with the rf bridge.
> 
> Edit: Assuming Tasmota + Portisch firmware on the rf bridge, just use a rule simiar to this:
> 
> `Rule2 on RfReceived#Data=303630 do backlog RfRaw 1; RfRaw AA B0 29 03 08 0640 0208 1CDE 281819090908190819090908190819081818190819081908181909081909081819 55; RfRaw 0 endon `
> 
> Replace `303630` to whatever your doorbell spits out.

How did You get the AA B0 29 03 08 0640 0208 1CDE 281819090908190819090908190819081818190819081908181909081909081819 55 key. I am trying to togle RfKey1 on rule match but cannot seem to get it working

jandy123, 2020-12-13:
@MokanHan 
> I am trying to togle RfKey1 

This won't work. Seems that the chime sends non-standard packets (as in not  EV1527 compatible).  Thus, you'll have to learn codes from the button using the 0xB1 command, see https://github.com/Portisch/RF-Bridge-EFM8BB1/wiki/Commands

This can then be converted into a 0xB0 command, see "B1 to B0 Conversion Tools" here: https://tasmota.github.io/docs/devices/Sonoff-RF-Bridge-433/ This is how I've got the looong 0xB0 code ;).

MokanHan, 2020-12-13:
Well got it working anyway with
<code>
Rule1 on RfReceived#Data=303631 do backlog rfkey1 1; RfRaw 0 endon
</code>

MokanHan, 2020-12-13:
> @MokanHan
>
>
> This won't work. Seems that the chime sends non-standard packets (as in not EV1527 compatible). Thus, you'll have to learn codes from the button using the 0xB1 command, see [...]

Well got it working anyway with
Rule1 on RfReceived#Data=303631 do backlog rfkey1 1; RfRaw 0 endon

jandy123, 2020-12-13:
@MokanHan That will work _if_ the alarm accepts standard EV1527 24-bit packets. If it doesn't, as for the Action 2020 in my original post, that won't work.

MokanHan, 2020-12-13:
And yes! also got my 50 cent chime working LOL.

<code>Rule2 on RfReceived#Data=303631 do backlog RfRaw 1; RfRaw AA B0 25 05 08 00DC 037A 015E 00A0 21F2 4818181819081929281929292929292929292B192928181818 55; RfRaw 0 endon</code>

jandy123, 2020-12-13:
@MokanHan Well, now you know exactly what I mean :) !

MokanHan, 2020-12-13:
> @MokanHan Well, now you know exactly what I mean :) !

Thanks for the right direction. I forgot to press the 50 cent Bell button all the time :)

MokanHan, 2020-12-13:
A simpele sidenote: do not cut lines. Use your already acquired hardware to flash porting on tasmotized rfbridge

lesleyvanrijn, 2020-12-17:
Just bought a Nedis Chime, but since it is on max volume and a horrible melody (it works though), i'll return that one. 
Then I realised that i bought some microcontroller components a few years ago that could be helpful. So i went ahead and looked for the box with components and started playing around yesterday. 

Grabbed an [ESP-01S Relay with a compatible ESP8266 Microcontroller](https://nl.aliexpress.com/item/4001108959192.html?spm=a2g0o.productlist.0.0.5d04c991pezodl&algo_pvid=4f21b08b-82f8-4906-a05f-3de23ae5eeb8&algo_expid=4f21b08b-82f8-4906-a05f-3de23ae5eeb8-0&btsid=0b0a050b16082022685255295e0845&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603), flashed it with ESPHome, rewired my old trafo in order to provide continuous power to both the doorbell and the old chime and hooked up the ESP-01S relay. 

The ESP8266 is subscribed to the MQTT messages send by the doorbell, and when payload "pushed" is submitted, it will flip the switch for 2 seconds. Works like a charm!

[image]
[image]

jkottier, 2021-01-05:
Since I've not been able to capture the button press with my Sonoff Bridge (original firmware version 3.5.0) I'm using a different approach using an old remote doorbell with chime that also works with 433.

I've captured the old doorbell button from the old doorbell with my Sonoff RF Bridge.

1. My LSC Doorbell sends a MQTT message to my Home Assistant MQTT Broker
2. In Home Assistant I'm using an automation that send a remote command to my Sonoff RF Bridge on the MQTT trigger

Here's my million dollar question ;-) 

What am I missing here why my Sonoff RF Bridge isn't able to capture the button press of my LSC Doorbell?

Firmware version: 2.9.7
Sonoff RF Bridge firmware version: 3.5.0

I've even tried adding `"external_charm":       "1"` to /home/cfg/tuya_config.json but so far no success

jandy123, 2021-01-05:
@jkottier Above I assumed that the RF bridge is flashed using Tasmota + Portisch firmware, see https://tasmota.github.io/docs/devices/Sonoff-RF-Bridge-433/

As far as I can tell you use original Sonoff firmware, right ?

jkottier, 2021-01-06:
> @jkottier Above I assumed that the RF bridge is flashed using Tasmota + Portisch firmware, see https://tasmota.github.io/docs/devices/Sonoff-RF-Bridge-433/
> 
> As far as I can tell you use original Sonoff firmware, right ?

Correct!

RaveGun, 2021-01-07:
I bought this one here:
[Elepowstar IP55 Wireless Doorbell](https://www.amazon.de/dp/B07B8Y6ZLH/ref=cm_sw_r_tw_dp_OM29FbK0HMG24?_encoding=UTF8&psc=1)
[image]

But is NOT compatible :(

What was the one that was compatible? This one from here? [ Marmitek ](https://github.com/guino/BazzDoorbell/issues/6#issuecomment-743323928)

m11tch, 2021-01-07:
> I bought this one here:
> [Elepowstar IP55 Wireless Doorbell](https://www.amazon.de/dp/B07B8Y6ZLH/ref=cm_sw_r_tw_dp_OM29FbK0HMG24?_encoding=UTF8&psc=1)
> [image]
>
> But is NOT compatible :(
>
> What was the [...]

The marmitek should work out of the box yeah.

jkottier, 2021-01-17:
> The doorbell sends a 433mhz RF signal on button press, this is intended to set off a chime.
> 
> Perhaps we can keep a list of tested chimes
> 
> Below is a list of tested so far.
> **Compatible**:
> 
> * marmitek BELL ME (https://www.marmitek.com/nl/smart-me/smart-security/deurbel-met-camera/bell-me-wht-draadloze-deurbel/)
> 

This one is also compatible with my LSC Action Smart Connect doorbell:
https://nl.aliexpress.com/item/1005001566936991.html?spm=a2g0s.9042311.0.0.1dae4c4dXmtw1P

Pairing is really easy:
1. There are 3 buttons on the side of the chime. One to adjust the volume of the chime, and two arrowbuttons to switch the doorbell ringtone. Hold the first button for about 3 seconds. When you release the button the blue light of chime should be on.
2. Now press the doorbell within 5 seconds. If the pairing was succesfull, the chime will blink.

roykaandorp, 2021-03-03:
I use this original chime of Nedis: WIFICDP10GY. Tried before to receive the signal with rflink (esp version which has raw support) and with rtl_433, it seemed that de nedis (bell 5s) transmitted on 868mhz. 
The WIFICDP10GY is not so loud so I built in a d1 mini and wired the speaker output to analog in so it sends a message to domoticz and plays a sound with almost no delay. (This was before this hack was available).
You could also connect it with Alexa and let this trigger something

swanitzek, 2021-05-25:
How did you pair the Doorbell to a wireless chime? There is no option in the app.

royka1, 2021-05-26:
@swanitzek As the manual says:
- Go to the settings menu of the video doorbell
- Select doorbell settings
- Place the gong really close to the doorbell
- Give the button on the bottom of the gong a short press: the light will start flashing.
- Press the "add" button in the app (remove the tone/volume menu first if it is already there)
- The menu will appear.
- Select another tone and ring the doorbell to test

Ierlandfan, 2021-06-21:
`This can then be converted into a 0xB0 command, see "B1 to B0 Conversion Tools" here: https://tasmota.github.io/docs/devices/Sonoff-RF-Bridge-433/ This is how I've got the looong 0xB0 code ;).`

Pressing the videodoorbell:
(rfraw 177)
11:15:11.811 MQT: tele/Sonoff_RF_C234CA/RESULT = {"Time":"2021-06-21T11:15:11","RfRaw":{"Data":"AAA055"}}
11:15:16.220 MQT: tele/Sonoff_RF_C234CA/RESULT = {"Time":"2021-06-21T11:15:16","RfRaw":{"Data":"AA B1 03 0154 0514 050A 281818 55"}}

Rule2 on RfReceived#Data= AAA055?