guino/BazzDoorbell #7: Customizing doorbell sounds

jilleb, 2020-12-14:
Not a bug or issue, but I didn't want to keep anyone from this info, would be cool to add it to a wiki with additional doorbell tweaks or something like that.

It's fairly simple to change your doorbell sounds.
I added the following line to custom.sh, just before telnetd is started and ppsapp is restarted.

```cp /mnt/mmc01/smb_pause.wav /home/sound/dingdong.wav``` 

This will copy the file smb_pause.wav to the file that's played when someone pressed the doorbell. 
But you can't take just any WAV-file, because it will sound like aliens have invaded your house.

Looking at the original doorbell WAV file, it's PCMs16 LE, mono, 8000Hz, 16 bits per sample.
[image]

If you don't have the tools to save or convert your WAV-file, you can use this website that does it for you:
https://www.3cx.com/docs/converting-wav-file/

Enjoy!

lesleyvanrijn, 2020-12-14:
Nice, was looking into this yesterday. Especially to mute the boot sound (had a few reboots two days ago in the middle of the night, and i could hear the boot sound two floors up. But testing the sounds only warning.wav and dingdong.wav would work on my Macbook.

jilleb, 2020-12-14:
Brilliant!

BytheWWW, 2020-12-17:
> Nice, was looking into this yesterday. Especially to mute the boot sound (had a few reboots two days ago in the middle of the night, and i could hear the boot sound two floors up. But testing the sounds only warning.wav and dingdong.wav would work on my Macbook.
> 
> Edit: A little joke for the dutchies over here. Will use this as my doorbell for a while, haha!
> [Ik moest kloppen want de bel doet het niet](https://github.com/guino/BazzDoorbell/files/5688780/Ik.moest.kloppen.want.de.bel.doet.het.niet.wav.zip)

Mwoaaaa Geertje

JasperMC, 2020-12-17:
Awesome idea. Does anyone know how to get a sound file to play on the bell anyway? I can't seem to find any kind of audio device to output to. Was hoping to do some funny things with the Portal Turret sounds (e.g. "Hello friend" on motion detection and "Hey, are you still there? after motion detection ends. If you ring the bell and no one's home, I'd like it to say "Sorry, we're closed"

gielie, 2021-02-08:
Brilliant, I used this one.
https://www.pacdv.com/sounds/voices/open-the-goddamn-door.wav

Nigel1992, 2021-07-29:
@jilleb I added this to the wiki.
Great job!