guino/BazzDoorbell #8: Integrating Onvif (ie Home Assistant)

BytheWWW, 2020-12-15:


BytheWWW, 2020-12-15:
@lesleyvanrijn Did your home assistant find the doorbell automatically?

lesleyvanrijn, 2020-12-15:
@BytheWWW Yep, with this settings:

Name: Whatever you fancy
Host: IP of the device
Port: 8000
User: admin
Passw: admin

After that it will add the camera to your home assistant instance. After that you'll have to update the connection type to TCP.

BytheWWW, 2020-12-15:
@lesleyvanrijn Was trying to connect through port 8554 🤪

It's visible in HA now. Now the main reason for me to integrate it into HA was to be able to hear a chime through Google home when someone was ringing the doorbell

lesleyvanrijn, 2020-12-15:
@BytheWWW That is possible! 

1: Add the  [MQTT middleware](https://github.com/guino/BazzDoorbell/issues/4#issuecomment-740644879) to the doorbell (at the end of that reply is a zip and instructions for adding the middleware). 
2: Create a MQTT server in Home Assistant, follow steps 1-6 of [this tutorial](https://medium.com/@fedorets.alex/hass-io-zigbee2mqtt-io-the-simplest-way-b7375c9fba22)
3: Update configuration.yaml and add:
```
mqtt:
  broker: 127.0.0.1
```
4: Create a new automation:
````
**Trigger:**
platform: mqtt
    topic: home/doorbell/button
    payload: pushed

**Actions:**
  - service: mqtt.publish
    data:
      topic: home/doorbell/button
      payload: idle
  
  - service: camera.play_stream
    data:
      entity_id: camera.{id} << ADD ID
      media_player: media_player.{id} << ADD ID
  
````

BytheWWW, 2020-12-15:
@lesleyvanrijn Awesome! Will have a look later, but thanks for the tip

BytheWWW, 2020-12-15:
@lesleyvanrijn Where can I find the entity and media player ID?

guino, 2020-12-15:
@BytheWWW glad your efforts to get the camera feeds were not a waste!

BytheWWW, 2020-12-15:
@guino Thanks to both you and @lesleyvanrijn 👍🏻

BytheWWW, 2020-12-16:
@lesleyvanrijn Stil having trouble getting the mqtt to work. Is there any way to see if the mosquitto_pub is being loaded?

BytheWWW, 2020-12-17:
Finally got it working! I needed to put my home assistant username and password in the log_parser.sh, otherwise it wouldn't work. Just need to stream the video to my phone now.

TopdRob, 2020-12-18:
Is it possible to use mqtt with the original pspapp or does that only work with a patched?
And how would the custom.sh look if it's possible to use the original pspapp.

The example of @m11tch uses a patched pspapp
```
# debug info from ppsapp  forwarded to output.log
  /mnt/mmc01/ppsapp  2>&1 | /mnt/mmc01/log_parser.sh /mnt/mmc01/output.log &
# no debug info  
#  /mnt/mmc01/ppsapp 2>&1 | /mnt/mmc01/log_parser.sh /dev/null &
```

guino, 2020-12-18:
@TopdRob I don't use HA/mqtt but I am pretty sure this will work with original ppsapp (no patch) because none of the patches we made messes with the output of ppsapp. I would expect it to work, but it's possible different versions of ppsapp would have different outputs.

In custom.sh you just have to change the line from ```/mnt/mmc01/ppsapp &``` (there's only one line exactly like that almost at the bottom of the file), to be this instead:
```
/mnt/mmc01/ppsapp 2>&1 | /mnt/mmc01/log_parser.sh /dev/null &
```

BytheWWW, 2020-12-18:
@TopdRob It works with the original ppsapp. I've got it working on my Google home speaker with HA an mqtt. (there is a small lag but that's too be expected)

MokanHan, 2020-12-24:
Any idea how I can make HA cast to Google smart display on button press?

MokanHan, 2020-12-27:
> Any idea how I can make HA cast to Google smart display on button press? Onvif is working in HA.
> 
> My automations.yaml is as follows:
> 
> `
> 
> * alias: deurbell
>   trigger:
>   
>   * platform: mqtt
>     topic: home/doorbell/button
>     payload: pushed
>   
>   Action:
>   - service: mqtt.publish
>   data:
>   topic: home/doorbell/button
>   payload: idle
>   ```
>   - service: cast.lovelace_view
>     data:
>        dashboard_path: lovelace
>        entity_id: camera.doorbell_mainstream
>        media_player: media_player.living_room
>   ```
> 
> `
> Yaml loads fine but nothing happens.

It is working!

https://youtu.be/-TfjgEMkXek

Tyrian81, 2021-01-02:
> @BytheWWW That is possible!
>
> 1: Add the [MQTT middleware](https://github.com/guino/BazzDoorbell/issues/4#issuecomment-740644879) to the doorbell (at the end of that reply is a zip and instructions [...]

Have it working but without the 
payload: idle

What is that meant for ?

BytheWWW, 2021-01-02:
@Tyrian81 i think the payload idle is to 'reset' the pushed state, but I might be wrong. Anyway, can't hurt to add it.

RaveGun, 2021-01-03:
Hi,

I had it working with RTSP, MJPEG and now with ONVIF (I recofigured the WIFI to have the camera on the same VLAN).

Unfortunately it does not matter in which mode is the camera configured, once I start to stream the content to my Lenovo Smart Display. There is about 5 seconds ore more lag to realty.
Maybe it is the Smart Display that is to weak or it is the HA on NUC that is lagging.

I tried without success to start casting directly the camera MJPEG URL that I can watch in VLC but it did not worked. I am not sure did not worked.

Thank you anyway for all the development so far. It is really a great $30 camera, now.

jkottier, 2021-01-05:
Next to the delay of the stream itself you can gain a small win by making the mqtt call a bit earlier.
There is another logline that is logged just before the Tuya call mentioning the doorbell button is pressed.

Adjust in log_parser.sh:
`elif contains "$BUF" "##doorbell_push 3"; then`
to:
`elif contains "$BUF" "button ev: doorbell"; then`

RaveGun, 2021-01-07:
Thank you for the hint. I will try it out.

Suspect-byte, 2021-01-10:
@lesleyvanrijn Hi,
How did you turn on onvif on your camera? Because I get constantly an error on hassio named: `Unknown error occurred
`

lesleyvanrijn, 2021-01-10:
@Suspect-byte start a telnet session, modify home/cfg/tuya_config.json and restart ppsapp.

Suspect-byte, 2021-01-11:
@lesleyvanrijn i am really new to telnet. So I got as far as the terminal with putty and cd to home/cfg/ but how can i edit the tuya_config.json file? I found on google that you can use `vi`, but when I use it i get this error `vi: applet not found`. Do you have another idea? Or a guide that can tell me how?

guino, 2021-01-11:
@Suspect-byte the easiest way for people unfamiliar with vi is this: https://github.com/guino/BazzDoorbell/issues/15#issuecomment-753655742 that said, only 2.9.7 firmware has onvif support. From what I hear you should be able to configure the camera with the RTSP stream (not onvif).

Tyrian81, 2021-02-19:
When i point my NVR (shinobi) to the RTSP stream my doorbell spontaniously reboots over time , any workaround ?

guino, 2021-02-19:
@Tyrian81 there's a known issue with cameras rebooting after 13 minutes of viewing the RTSP on **UDP** (which is default in most clients). If you use **TCP** this does not happen -- I have left it on overnight several times for testing. The question is how do you configure your NVR to use TCP instead of UDP (if that's possible). If your NVR doesn't support TCP your only option would be to setup a proxy on a computer using ffmpeg so it connects to the camera on TCP and serves the stream in UDP to your NVR.

guino, 2021-02-19:
ONVIF/RTSP are virtually the same as long as you can specify TCP it should be fine. RTSP is more widely supported and gives you a little more control I think. MJPEG is fine if you don't mind the lower resolution and no audio.

guino, 2021-02-19:
looks like shinobi does support RTSP over TCP: https://shinobi.video/docs/settings

Tyrian81, 2021-02-19:
@guino Thank you very much set it to use TCP transport !! Yeah continous recording !!

Ps. anyway to get 2 way comms working ?

guino, 2021-02-19:
@Tyrian81 you can use the app for that. Excluding the app how did you want to send the audio to it ? You can currently play wav files on the device (even uploaded files) using play.cgi

roykaandorp, 2021-02-25:
I've not tried it myself yet, but maybe this could work with a nvr if you have a firmware that has no onvif built in.
https://github.com/kristian/rpos-gateway

guino, 2021-02-25:
@roykaandorp Other than requiring a separate computer/board to run the onvif service for the camera I would expect it to work.

guino, 2021-05-25:
@zuzuman the onvif video feed should have sound if that’s what you’re asking. There’s no sound **input** support (to play on the device from onvif). For playing sounds on the device the only thing we have is play.cgi (see #10)

zuzuman, 2021-05-25:
> @zuzuman the onvif video feed should have sound if that’s what you’re asking. There’s no sound **input** support (to play on the device from onvif). For playing sounds on the device the only thing we have is play.cgi (see #10)

I have integrated my doorbell with onvif, but sound is missing (live preview and records) - can see video only

guino, 2021-05-25:
@zuzuman I would suggest:
1-check that there’s sound using the standard phone app (tuya/smartlife/etc). This is to verify your hardware is working correctly.
2-check if the sound works with RTSP (no onvif). This is to verify the RTSP video feed works correctly (because it is also used by onvif). You should definitely try different software/apps such as VLC, mplayer, etc
3-test the onvif with different software/phone apps.

Depending the results the problem can be better defined, but there’s a good chance it could just be the app you’re using and the poor onvif support.

zuzuman, 2021-05-25:
> @zuzuman I would suggest:
> 1-check that there’s sound using the standard phone app (tuya/smartlife/etc). This is to verify your hardware is working correctly.
> 2-check if the sound works with RTSP (no [...]

1) Tuya app - sound present 
2) VLC - sound present 
3) Video recorded with ONVIF integration on Home Assistant. - sound missing

AndreiGr, 2022-12-01:
> Finally got it working! I needed to put my home assistant username and password in the log_parser.sh, otherwise it wouldn't work. ~Just need to stream the video to my phone now~ Not gonna go through all that trouble at this moment, I'm just going to use the tuya app for that. That way I have two way speech also. The main thing was to get the doorbell to sound through my google home speaker, and now it does! 30 euros well spend.
> 
> Learned a lot in a short time. Thanks again to all who helped me, especially @guino and @lesleyvanrijn! You guys rock!

Where exactly did you put the username and password, in the logparser.sh file?
Could you help me with an example?
I think this is why mqtt does not work for me as well.

Also, I assume that for the IP you put the HA ip, right?

guino, 2022-12-06:
@AndreiGr did you take a look here: https://github.com/guino/BazzDoorbell/wiki/%5BHow-to%5D-Integrate-with-Home-Assistant,-HomeBridge,-Domoticz,-etc the user/password information for logparser would be in the mosquitto_pub or mqtt_pub line in logparser.

AndreiGr, 2022-12-06:
Yes, I checked it out, and I managed to figure out where to put the username and password.

Unfortunately, the MQTT messages still aren't working for me. I can't see anything when I listed to those topics in HomeAssistant,

guino, 2022-12-07:
@AndreiGr you should test the mqtt messages manually from command line either in telnet from the device OR using some other machine and mosquitto_pub -- once that is working from command line then you can adjust the logparser to use the same script. I don't use Home Assistant so I don't know what all is required to be configured for MQTT to work with it (I use domoticz and home bridge personally and mqtt works fine there).

AndreiGr, 2022-12-08:
I'm not exactly sure how I could test mqtt from command line. Is there a way to see also the response of the command when using telnet?
The configuration for MQTT in HA is pretty basic, and it definitely works, because I have other devices that communicate with HA through MQTT so the issue is not on the HA side, but on the device side, I think.

guino, 2022-12-08:
@AndreiGr basically you telnet into the device (may need to install telnet in windows): `telnet ip` user `root` and password is whatever you defined in step 9 when rooting: https://github.com/guino/BazzDoorbell/issues/2).

Then assuming you have mosquitto_pub in the SD card it should just typing the command:
```
/mnt/mmc01/mosquitto_pub -h 192.168.x.x -m "detected" -t home/doorbell
```
Basically the same stuff you were editing in log_parser but you execute it by hand instead of waiting for it to detect motion.

Feel free to post a copy of your SD card files (without SDT folder) for review if you like.

AndreiGr, 2023-01-24:
I picked the debug back up today, and after logging into telnet, I kept getting an 'Invalid port given' error, from mosquitto_pub.

That's when I realized that I was specifying the MQTT password using the -p flag, instead of -P. So that was the issue all along.

Thank you for the help with telnet, I don't think I would've figured it out if I wouldn't have seen that error.

KapitanKapitan, 2023-02-19:
Guys any idea what is login and password for onvif (4.0.7v)? Admin:admin not working. I get worng password error on my NVR. ONVIF is turn on in Smart Life app.

guino, 2023-02-20:
@KapitanKapitan  I replied to your question here: https://github.com/guino/BazzDoorbell/issues/95#issuecomment-1436327814