guino/BazzDoorbell #10: Playing audio through the bell speaker

JasperMC, 2020-12-17:
Hello everyone,

I thought it would be interesting to start a topic on how to get audio to play through the bell's speaker. The reason for this is that it could perhaps help us with somehow recreating the talkback feature among other audio related experiments.

Adding some links for inspiration here:
- https://github.com/topics/hi3516ev200 (the SoC in the camera)
- https://github.com/hi35xx/live-streamer/commit/016889b755a8b452b119a173c0d1583f954b5b81

guino, 2020-12-17:
The main thing here is that I think the ppsapp opens the audio device and keeps it open so I don’t know if an outside application would be able to open it again (in parallel).

The quickest rig I can think of is to rename/link an existing audio file and cause something in the ppsapp to trigger/inject a call to play the file. Otherwise you would have to patch ppsapp to stop using the audio device which means not being able to use some features from the phone app.

jilleb, 2020-12-18:
I got the device to change from night vision to regular rgb by calling some URL, I'll look if there's any audio trigger.

guino, 2020-12-19:
From what I can see in the code we should be able to make ppsapp play any wave file we want with a simple dd (write) command but like jpeg functions this would vary based on firmware version. I'm going to give it a quick try with mine and if it works I would need to know what version of firmware you're running (please full version from /devices/deviceinfo) so I can make you something to try.

guino, 2020-12-19:
I was able to make a command to play a wav file on demand, but like I said it requires knowing thr address for the playback request in ppsapp, so that’s firmware specific like the jpeg features. If you tell me which firmware you have I can look it up in ghidra and provide you with a command so you can try/use it yourself.

JasperMC, 2020-12-19:
I can provide the exact device info later but it’s basically the LSC Smart Connect bell at 2.9.7 firmware (using the ppsapp you posted)

JasperMC, 2020-12-19:
Exact device info as promised:

```
{"devname":"Smart Home Camera","model":"Bell 8S","serialno":"061065043","softwareversion":"2.9.7","hardwareversion":"BE8S_H1_V10_433","firmwareversion":"ppstrong-c51-tuya2_lcs-2.9.7.20201020","authkey":"rsjPTgwAhHV2MLcT863SxTvKRvmTW5xL","deviceid":"pp01f385739c31d88c01","identity":"MR2007180100101742","pid":"aaa","WiFi MAC":"[blanked out]"}
```

guino, 2020-12-19:
@JasperMC  This should be done now. Download/save the [play.cgi](https://github.com/guino/BazzDoorbell/blob/master/mmc/cgi-bin/play.cgi) to your SD Card and adjust the request address for your firmware which I published [here](https://github.com/guino/ppsapp-rtsp/issues/1#issuecomment-740937085). 

You can play wav files from telnet/internal script by using:
```
/mnt/mmc01/cgi-bin/play.cgi /home/sound/dingdong.wav
```
OR by URL (assuming you have httpd installed/configured) on the format:
http://user:pass@ip/cgi-bin/play.cgi?/home/sound/dingdong.wav

I highly recommend you check and use the exact .wav file format (bitrate, channels etc) of the existing files as I have no idea if different formats would work or cause any problem (I only tried the existing ones).

WARNING: A path+filename longer than 127 characters can potentially crash your ppsapp, so keep that in mind and/or adjust play.cgi to check the length of the patn+filename if you have any concerns.

NOTICE: This is not a 'patch' it should work regardless of your ppsapp being stock or patched for offline or anything else, so it should also work with whatever we come up for LED (if we get that working).

JasperMC, 2020-12-19:
To quote my env file: "Thank you Guino, Thank you Guino, Thank you Guino" ;)

guino, 2020-12-19:
@JasperMC did you already try/confirm it?

JasperMC, 2020-12-19:
Trying it right now, with the default sounds and a custom one :)

JasperMC, 2020-12-19:
It seems it's still not working unfortunately. I changed the REQ address in the script but can't play the default sounds or custom sounds. I converted my custom sounds to be in the same format as the original, but no luck. I'll give it another try to rule out that it's not my shell scripting haha.

JasperMC, 2020-12-19:
It works! It was a quick scripting mistake.

guino, 2020-12-19:
Thanks for confirming and reporting back!

JasperMC, 2020-12-19:
It had to be done..

JasperMC, 2020-12-19:
Sorry, I had to do it.

https://youtu.be/H2rBc6mi7_4

guino, 2020-12-20:
I added the play.cgi address to the firmware [ppstrong-c51-tuya2_lcs-2.9.6.20200628](https://github.com/guino/ppsapp-rtsp/issues/1#issuecomment-740477496) and created steps to find it on other firmware versions in https://github.com/guino/ppsapp-rtsp

@JasperMC I assume you have not had a chance to try the LED stuff yet?

JasperMC, 2020-12-20:
Not yet! I hope to do so tomorrow. From what I can see the original ppsapp sets the lights to red and then blue when ready before it is killed and the custom one is launched. I’ll throw some LED commands in my custom.sh soon so I can report back :)

JasperMC, 2020-12-20:
Something I did notice today is that the ppsapp might close the audio connection somewhere after a timeout or something. I left my bell running all night with that portal sound triggering on motion but it eventually didnt play anymore

guino, 2020-12-20:
Does the play.cgi script show any error?

JasperMC, 2020-12-23:
Haven't looked at that yet. Does it output that to the console or some log file? I didn't have a telnet connection open the whole time so I might have to try it again.

guino, 2020-12-23:
@JasperMC it should print out any error running it on Telnet or curl or using a web browser.

JasperMC, 2021-01-03:
Update on this; My SD card seems to have been formatted by the bell itself. I have no idea what happened, but will try to recover it. Will try again testing long term. It's possible that my SD card got formatted while running, and that the play.cgi code is fine.

guino, 2021-01-03:
@JasperMC I have said this many times: the ppsapp deletes files from the card when space is low. You should always backup your files and keep a reasonable amount of free space to prevent that from happening -- I use a cleanup.cgi script which I run daily (it is in the github project page). Alternatively you can disable recording (I just use record on motion-only which gives me several months of files in less than 16GB).

JasperMC, 2021-01-03:
Hi @guino,

I'm aware of the deletion and had also set it to motion recording only. I guess it somehow filled up anyway.

I'll change the settings when I can get to them; the bell is not mounted yet anyway, so I don't even need recording.

karlzre, 2021-05-17:
Hi @JasperMC 

Could you share the file format needed for the wav ?
8k/16bit/Mono ?
thx

Nigel1992, 2021-07-26:
@JasperMC How did you manage to play a sound using motion detection ?
log parser? If so, please share :)

Nigel1992, 2021-07-26:
> Hi @JasperMC
> 
> Could you share the file format needed for the wav ?
> 8k/16bit/Mono ?
> thx

PCMs16 LE, mono, 8000Hz, 16 bits per sample.
As mentioned in #7

guino, 2021-07-26:
See https://github.com/guino/BazzDoorbell/issues/4#issuecomment-751512241 for how to call the log parser 

See https://github.com/guino/BazzDoorbell/issues/4#issuecomment-754848862 for the log parser script 

like I said on the other thread: either modify the log parser script to run the play.cgi script or use it to notify a server and make the server call the play.cgi url when the notification arrives. On the he thread of the URLs above there is more information so you may want to read thru it.

I don’t have anything ready for this, but really it’s 1 line change in the log parser to call the play.cgi script instead of making the notification to the server (or together)

JasperMC, 2021-07-27:
Hi @Nigel1992 

As @guino suggested, I added a line to the log parser file to trigger on motion or on button press.

Later on, I also did it through Home Assistant automations. If I'm not home, it would then play a message asking to leave the package at the door.

Nigel1992, 2021-07-27:
@guino @JasperMC 
I'm a noob at Linux.
I cant get it to work :(

Here's my custom.sh
`if [ ! -e /tmp/customrun ]; then
 echo custom > /tmp/customrun
 cp /mnt/mmc01/passwd /etc/passwd
cp /mnt/mmc01/chime.wav /home/sound/dingdong.wav
 /mnt/mmc01/busybox telnetd
 /mnt/mmc01/busybox httpd -c /mnt/mmc01/httpd.conf -h /mnt/mmc01 -p 8080
 if [ -e /mnt/mmc01/ppsapp ]; then
  PPSID=$(ps | grep -v grep | grep ppsapp | awk '{print $1}')
  kill $PPSID
  #/mnt/mmc01/set record_enable 0
  #/mnt/mmc01/set enable_event_record 1
  #/mnt/mmc01/set onvif_enable 1
 /mnt/mmc01/ppsapp
/mnt/mmc01/ppsapp 2>&1 | /mnt/mmc01/log_parser.sh
 fi
fi
if [ ! -e /tmp/cleanup`date +%Y%m%d` ]; then
 rm -rf /tmp/cleanup*
 touch /tmp/cleanup`date +%Y%m%d`
 /mnt/mmc01/cgi-bin/cleanup.cgi > /tmp/cleanup.log
fi
echo "$(date) - | - checking for log_parser.sh in /mnt/mmc01" >> /mnt/mmc01/hack
[1]+  Stopped                    vi custom.sh
`

Here's my log_parser.sh
`
        return 0    # $substring is in $string
    else
        return 1    # $substring is not in $string
    fi
}

main() {
    IFS='$\n'
    while true; do
        read -r BUF;
        if [ $? -ne 0 ]; then
            sleep 1;
            continue
        fi
        if contains "$BUF" "--motion detection alarm --"; then
           /mnt/mmc01/cgi-bin/play.cgi /home/sound/dingdong.wav
        elif contains "$BUF" "##doorbell_push 3"; then
            /mnt/mmc01/busybox wget -O- http://10.10.10.87:8080/doorbell?Doorbel
        fi
    done
}

main
`

guino, 2021-07-27:
@Nigel1992 in order to use the log_parser you must have ppsapp in /mnt/mmc01/ (root of SD card)-- even if you didn't patch it. This is because the only way to capture/parse the ppsapp output (to detect/trigger motion alert) is to kill and run ppsapp with the right parameters and this only happens when you place a ppsapp in /mnt/mmc01/ (the root of the SD card).

You should have a copy of your ppsapp under home/app/ppsapp on the SD card, so you just need to copy that to the root of SD card and reboot after (to make it effective) as other than that your changes should be all you need.

Nigel1992, 2021-07-27:
@guino Thanks, its working now!
Is there a possibility that we "users" are able to contribute to your Wiki ?
I would love to post some tutorials on the things I learned!

guino, 2021-07-27:
@Nigel1992 If you create an issue with whatever information you think it's helpful I can pin it to the main issues so it's easy to find.

jilleb, 2025-03-27:
@guino , sorry for digging up such an old topic...
I've got the doorbell running smoothly for years now 😄 
I am thinking of adding a way to enable the intercom feature, instead of playing my own pre-recorded replies to the doorbell.

I have the doorbell running as an Onvif device in Surveillance Station, and they offer support for the intercom as well, as long as the device reports this feature in the onvif xml... I think it can replace the PTZ section that's currently there, but then I have no clue where the incoming audio stream to ppsapp would be... Did you experiment with this?