guino/BazzDoorbell #9: Controlling the LED

JasperMC, 2020-12-17:
After liberating my bell I am now curious if there's a way we can change the color of the LED and its brightness. This could be fun to customize, or even react to the sound you're playing.

I'm hoping someone can help me research how to do so.

A first hint might be the "pwm" device in /dev. Perhaps the LED is controlled through PWM.

guino, 2020-12-17:
The little I have seen on this suggests the LEDs are gpio controlled (no level/intensity control) and there is a blue and a red one (no RGB). There’s a command you can run to toggle them but ppsapp seems to set them right back to what they are supposed to be (blue/red). I do not yet know if there’s anything hardware side that would prevent both LEDs from being on at the same time.
Basically for intensity/color (other than just blue/red/off) you’re looking at hardware changes likely. 
For control of the on/off of the blue/red LEDs it may be possible to patch ppsapp (to disable its LED control) and allow an outside app/script to manage it.

JasperMC, 2020-12-17:
Thanks Guino! 

Could you share what command you used to change the LED color? I'm mostly just interested in somehow making it sync with audio if I can. Would be a cool feature.

guino, 2020-12-17:
```
/bin # ppsconfig 
Usage: getdevinfo [opt][=value]
        -u                      get p2puuid
        -p                      get p2ptype
        -n                      get product number
        -s                      get serial number
        -m                      get device model
        -t                      get device type
        -h                      get hardware info
        -i                      get identity card info
        -l                      get licence id
        -k                      get secret key
        --redled=[on|off]       turn red led on or off
        --blueled=[on|off]      turn blue led on or off
```
But like I said: when you use this the ppsapp usually restores the led color right back.

guino, 2020-12-19:
It is possible to patch ppsapp to disable its control of the leds but if you do that you would have to control it yourself for everything (ppsapp would no longer manage it). If you want to try that let me know and I can probably make a patch for it.

JasperMC, 2020-12-19:
That would be wonderful actually. Would it be possible to make a combined one also with that audio command you were talking about in the other issue?

If there’s anything I can help with, please let me know as well. I really appreciate it man.

guino, 2020-12-19:
This is UNTESTED and only for version ppstrong-c51-tuya2_lcs-2.9.7.20201020 -- try this 
[ppsapp-noled.zip](https://github.com/guino/BazzDoorbell/files/5720033/ppsapp-noled.zip) to disable ppsapp from controlling the LED. Again you will have to control the LED by yourself for everything (network status, etc) -- I am taking a guess it will boot up into a 'red' state and stay that way until you manually set it to something else using ppsconfig. I am **hoping** this won't have any other side effects like disabling the nigh-vision lights. Please test and report back when you have a chance.

Ierlandfan, 2020-12-23:
Is this also possible for the 2.9.6 version? I just want the (red/blue) leds to stay off when booted up. That red is on while booting is no big deal.

guino, 2020-12-23:
@Ierlandfan a specific patch for 2.9.6 would have to be made but I don’t even have confirmation that this is working on 2.9.7 so right now this is just an idea.

JasperMC, 2020-12-23:
Will test this very soon guys. Sorry for the delay, got caught up with the holiday prepping :)

guino, 2021-02-05:
@JasperMC thanks for confirming this can be done. I'll try to put up instructions on patching to use this feature.

guino, 2021-02-07:
These instructions are for those who wish to manually control the BLUE/RED LED without having ppsapp change it. 

Please notice: if you just want to turn OFF the LED you can simply edit tuya_config.json to set "light_onoff" to  0 (no need to patch anything).

I am not even going to place this in the main repo instructions because I think the interest on this feature is very limited.

Here's how to patch ppsapp for disabling the LED control (so it can be controlled by command line/script using ppsconfig):

1. Follow instrunctions from https://github.com/guino/ppsapp-rtsp until you no longer have a progress bar at the bottom and wrote down the loadable segment value.
2. Click on the middle window (assembly listing) and press CTRL+E (search)
3. type in **led_ctrl** and select "all fields" at the bottom and click 'Next'
4. It should find a function that looks like this ('SetThreadName' may just bun FUN_0XXXXX):
[image]
5. Click the function name on the right side at the top and it should highlight something in the middle window and it will display some XREF values like this:
[image]
6. Double Click on the first XREF value (in this example 00049444)
7. You'll get to a function that looks like below (notice I renamed the function to 'StartThread' but the function name shown does NOT matter):
[image]
8. Click the function name (StartThread in my case or FUN_0XXXX or whatever) so it highlights the code in the middle window:
[image]
9. You want to change the **bl** for a **nop** so the function is never called, so right click the **bl** select 'Patch Instruction' (click OK if warning shows), then delete the contents in the red boxes and type in **nop** on the first box selecting 00 f0 20 e3 on the list) and press enter to confirm
10. You should see the middle window something like the below:
[image]
11. Open the hex editor and go to the address shown in the middle window (- loadable segment), which is 0x49450-0x10000 = 0x39450 then change the bytes 47 E3 00 EB to 00 F0 20 E3 and save the changes.
12. Place the edited ppsapp in the root of SD card and reboot -- you should be able to manually control the LED after the device boots and ppsapp should not longer change it.