guino/BazzDoorbell #13: My S90PPStrong has MTDNUM=5, can I bypass that?

guino, 2020-12-22:
#### Description of issue

We saw a few cases (namely @solidssss @suspect-byte) where the ppshack process worked on steps 1 to 6 but failed on step 7. This is because there was a line the developers forgot to comment out in the firmware and prevented the hack from working. You can confirm this by using this URL http://admin:056565099@ip/proc/self/root/etc/init.d/S90PPStrong and checking if this line is present:
```
MTDNUM=5
```
(Please note if the line shows ```#MTDNUM=5``` with **#** then the original ppshack should work for you so you can just follow steps from #2 instead -- once this processed is better vetted it will likely become the preferred method over the old one as it should still allow the device to boot without a SD card installed).

#### Process

**1**-Verify your camera/doorbell is compatible using its local IP address (i.e. 192.168.x.x, etc) -- this is NOT the public IP displayed in the TUYA app (and likely other apps), if you're reading this there's a good chance you know how to find the IP from your router. Open a web browser and load this address: http://admin:056565099@192.168.x.x/proc/cmdline the result (kernel command line) should be something like this:
```
mem=mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd
```
Copy/save that response (kernel command line) as we'll need it next (and also in case you want to restore original settings) -- If you get no response, or some very different response, please stop now -- chances are you're using the wrong IP or this won't work for your device. Feel free to post your kernel command line if you have questions.

**2**-Prepare a file called 'env' with the following contents (sample env file included in attached zip file):
```
bootargs=mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd - ip=\\${T//_/\\$\\'"\\\\x20"\\'}:::::";T=\\"sleep_5;mkdir_-p_/mnt/mmc01;mount_-t_vfat_/dev/mmcblk0p1_/mnt/mmc01;/mnt/mmc01/initrun.sh&\\";eval"
```
VERY IMPORTANT DETAILS ABOUT THIS FILE:
-The part after bootargs= should be the same as you got on step 1 (from your device) with **NO CHANGES**
-The file has **one single line** "bootargs=..." and a new line (enter) at the end. A sample env file is attached for your conveninence.
-The line size is not important, just make sure the **- ip=...** section is added after the original kernel command line.
-There's a 0x00 (Zero character) at the end of the file -- if this is removed by your text editor (ie notepad, etc) you'll need to be sure it is there (or the process will not work). You may need to use a hex editor to change the last character to a "00" making sure there's at least a new line (0A) at the end of the line before the "00". Again the sample env file is a good reference.

**3**-Copy these 3 files (attached) to the root of a fat32 formatted SD card (do not place them in any 'folders'): env, ppsMmcTool.txt and initrun.sh -- **MAKE SURE** there's no 'upgrade.bin' file in the SD card or this could cause problems. Be sure to properly 'eject' (or unmount) the SD card before removing it from the computer.

**4**-Power off your device and insert the SD card with the 3 files in the SD card slot.

**5**-Press-and-hold the reset button, then power on the device (i.e. power wires/USB cable) and **continue holding the reset button for 5 seconds after power on** then let the device boot. It will take longer than usual (precisely 5 seconds longer) for it to fully boot up as that's part of the initial boot script.

**6**-Repeat step 1, this time your kernel command line should look like this:
```
mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd - ip=${T//_/$'\\x20'}:::::;T=\"sleep_5;mkdir_-p_/mnt/mmc01;mount_-t_vfat_/dev/mmcblk0p1_/mnt/mmc01;/mnt/mmc01/initrun.sh&\";eval mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd
```
(Notice how the **- ip=...** part was added at the end)

**7**-Now browse to this address: http://admin:056565099@192.168.x.x/proc/self/root/mnt/mmc01/hack -- it should say 'done' which is the indication everything is working as designed.

Follow the rest of the steps (starting from step 8) from #2 -- please note the default initrun.sh will already copy the original ppsapp to the SD card (under /home/app/ppsapp).

[ppshack2.zip](https://github.com/guino/BazzDoorbell/files/5731529/ppshack2.zip)

solidssss, 2020-12-22:
I still get the same output on both step 1 & 6. 

I don't think anything is wrong with my env, but I've uploaded it here for you to check it.
[env.zip](https://github.com/guino/BazzDoorbell/files/5732028/env.zip)

guino, 2020-12-22:
@solidssss it seems you and @thomasloven have the same device/firmware (I thought you got steps 1-6 to work). If you can follow the steps in #11 (using the address he posted on that issue) and post your flash.bin file I can try to take a look but there’s no way I can test anything without the device. Otherwise we can only wait to see if @thomasloven makes any progress and posts it (unless you have UART access to play with).

thomasloven, 2020-12-22:
I've been playing around a bit after your helpful hints, and I'm afraid I have some bad news (for me and @solidssss at least).
I am now 100% sure I'm setting `bootargs` correctly, and also 100% sure it is entirely ignored in the end - which I understand is a behavior that can be compiled into the linux kernel.

I managed to extract the flash and a piece of the kernel, but the flash seems to be encrypted - the entropy is 6 bits per byte, `binwalk` finds nothing at all and `file` says it's a PGP key security ring.

...and at this point we're already way above my skill level...

solidssss, 2020-12-22:
@guino  Here is the flash.bin:
```
Scan Time:     2020-12-22 22:21:50
Target File:   /home/pi/flash.bin
MD5 Checksum:  403f31bae8885c8ab490f70a79d0a522
Signatures:    386

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
7963382       0x7982F6        Unix path: /dev/mmcblk0p1_/mnt/mmc01;/mnt/mmc01/initrun.sh&\\";eval"
7995904       0x7A0200        Executable script, shebang: "/bin/sh"
7996115       0x7A02D3        Unix path: /opt/pps/app.tar.gz -C /mnt/mmc01/
7996449       0x7A0421        Unix path: /opt/pps/initrun.sh ] && cp /opt/pps/initrun.sh /tmp/PPStartOrig && chmod +x /tmp/PPStartOrig && /tmp/PPStartOrig &
8061440       0x7B0200        Zip archive data, at least v2.0 to extract, compressed size: 235, uncompressed size: 305, name: env
8061793       0x7B0361        End of Zip archive, footer length: 22

Scan Time:     2020-12-22 22:22:04
Target File:   /home/pi/_flash.bin-2.extracted/env
MD5 Checksum:  94ce6ab00a0207658fd7eccbea532eec
Signatures:    386

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
246           0xF6            Unix path: /dev/mmcblk0p1_/mnt/mmc01;/mnt/mmc01/initrun.sh&\\";eval"

-----------------------------------------------------------------------
```

I tried  81000000, 42000000 and 81808000 with same result, except for diffrent md5 checksums.
It doesn't really look much like yours though.

thomasloven, 2020-12-22:
That looks like what may have been on your SD card before reformating or something...

I did not get the flash extraction to work with the `ppsMmcTool.txt` included in #11. Maybe there are some syntax differences between u-boot versions.
I soldered into a UART I found and ran the same commands with some tweaks.

I have tried to translate them back to "`ppsMmcTool.txt`-format" below. Maybe you could try that if you're brave.

`style=upgrade,,writeAddr=0,,password=nothing,,writeLen=0,,fileName=0;sf probe; sf read 81808000 0 8000; mmc write 81808000 1 8000,,`

guino, 2020-12-23:
@solidssss I agree with @thomasloven it looks like the process to read your camera's flash.bin failed -- I don't believe there's any harm in trying the ppsMmcTool.txt he posted above using the instructions from #11 but it doesn't seem like there's a lot to get from it. I would be willing to take a look at the bin file anyway.

solidssss, 2020-12-23:
@thomasloven I added `style=upgrade,,writeAddr=0,,password=nothing,,writeLen=0,,fileName=0;sf probe;sf read 81808000 0 8000; mmc write 81808000 1 8000,,` to ppsMmcTool.txt and run binwalk again, the result was still the same.

@guino  I've emailed my .bin file to you if you want to take a look at it.

thomasloven, 2020-12-23:
FWIW here's the flash I got.
[flash.bin.zip](https://github.com/guino/BazzDoorbell/files/5734310/flash.bin.zip)

guino, 2020-12-23:
@thomasloven the flash.bin file you posted is only 32Kb in size (0x8000 bytes instead of 0x8000 sectors) -- it should have been at least 8Mb (or 16Mb) if you used #11 -- maybe you can try it again ? it did look like it had information from the flash but it was just a small fragment of it.

guino, 2020-12-23:
@thomasloven I think your flash file should look like this one: https://github.com/Jordan-Jarvis/Merkury-Smart_cam-720p-work/blob/master/Mercury-Smart_cam-720p-dump.bin which is version 2.7.3 -- in fact the file you provided looks similar to it but I only got 32Kb of it, so if we figure out the right uboot commands and use the right dd commands (as I posted in #11) I would hope to get a flash file similar to the linked bin file.

thomasloven, 2020-12-23:
Did I mention being in above my head?

You're absolutely right. Looks like I'm back in the game.
[flash.bin.zip](https://github.com/guino/BazzDoorbell/files/5737187/flash.bin.zip)

guino, 2020-12-23:
@thomasloven good job - this flash looks fine and should be customizeable. That being said: based on your findings that the boot loader seems to ignore the bootargs setting the only way to customize it would be to use #12 which is risky specially considering that the boot loader commands appear to be different from the version I have. My recommendation would be to get a $15 programmer, but if you'd like to try your hand at #12 I can help with the changes/reviewing etc -- just let me know if you feel brave enough to try it. Now that you have a copy of the flash you should be able to use a programmer to restore it in case things go sideways (but my recommendation still would be to use a programmer to read/write the flash directly).

guino, 2020-12-24:
@solidssss @thomasloven can you guys try this url: http://admin:056565099@IP/sys/telnetd/switch -- I believe this may enable telnet on your older firmware versions and if so you should be able to telnet into the device if we figure out the password. Someone suggested it should be root:12345678 but I am not sure about that.

solidssss, 2020-12-24:
@guino I've tried a bunch of passwords from [here](https://gist.githubusercontent.com/gabonator/74cdd6ab4f733ff047356198c781f27d/raw/c750ca79adb3d41992de6dbc55c3054699970126/password.txt), it's some commun telenet tuya camera passwords but none is working. I'll try with bigger password list.

guino, 2020-12-24:
I have also not found that telnet password -- having that would be a less risky way to get into the device than modifying the flash (which seems the only option right now).

Regarding onvif: did you check for settings under http://admin:056565099@ip/proc/self/root/home/cfg/tuya_config.json ? that may have the onvif password if it's different than the default one.

dixnor, 2020-12-24:
If it can be of any  help, the root password hash is easily accessible via the /proc/self/root/etc/passwd uri
Don't know if an md5 cracker can get through and decrypt this password
This is my output using this uri: 

`root:$1$12345678$CTq8UQyYrE.vbbG7E8Mtj1:0:0::/root:/bin/sh`

guino, 2020-12-24:
@dixnor yeah we have had the hash for awhile but still no clue of the password, that’s why my custom.sh just copies a new passwd file to replace the unknown hash. In any case it doesn’t hurt to post the hash again so others can search.

solidssss, 2020-12-25:
@guino nope, /proc/self/root/home/cfg/tuya_config.json did not contain any password, just some status like "sleep_mode":	0.

thomasloven, 2021-01-01:
There's an `/etc/-passwd` too, which has the RSA-encrypted password `helpme` for root, but that does not work for logging in via telnet.

I have been digging through `ppsapp` with a disassembler for a while and believe I may have the format of `upgrade.bin` almost figured out. That could be a way to replace parts of flash by just placing a file on the sd card. Though I'd probably want a rooted device to try it on so I could get the debug output... may need to invest in a flash programmer.

lcouch2018, 2021-01-01:
@thomasloven  you going to build of a cheap camera HA card ? Btw love you card mods man and appreciate you HA contributions.

guino, 2021-01-03:
@solidssss @thomasloven @dixnor I created a new repo with instructions [HERE](https://github.com/guino/Merkury720) for 2.7.x firmware.

thomasloven, 2021-01-03:
I found it already, and got it to work. Thanks for your awesome work!

guino, 2021-01-30:
@Lukey1979 you didn’t post feedback for your results but you had firmware 2.9.0 (and I assume you used the instructions on this issue to get ppsapp), did you get RTSP working? If not you may want to look at this repo: https://github.com/DanTLehman/orion_sc008ha

Lukey1979, 2021-01-30:
Hello @guino , I apologise for not getting back to you. Unfortunately the hack didn't work and I could not access RTSP nor could I access the still images. In fact, with the SD card inserted, the camera would reboot every couple of minutes. I will have a look at that other repo to see if that helps run any way.

guino, 2021-01-30:
@Lukey1979 this sounds exactly like what he experienced with 2.9.0 - so you should be able to make it work with his changes.

Tyrian81, 2021-10-30:
Hello Guino ,

after rooting my tuya doorbell succesfully , Thank you VERY much !!!
i thought to try my indoor tuya cam , but a few bumps in the road ...

4.0.6 firmware

S90PPstrong doesnt have the hashtag #MTDNUM=5 
```
/proc/cmdline/
setenv bootargs mem=64M console=ttySAK0,115200n8 loglevel=10   mtdparts=spi0.0:256k(bld),64k(env),64k(enc),64k(sysflg),3m(sys),4032k(app),640k(cfg) ppsAppParts=5 ip=192.168.1.99:::255.255.255.0 eth=00:55:7b:b5:7d:f7
```
(weird IP range in this one)

also differs from yours ?
any chance this can be rooted ?

guino, 2021-10-31:
@Tyrian81 there are plenty of people who were able to hack 4.0.6 firmware and patch ppsapp -- for that firmware you need to use the instructions from this link: https://github.com/guino/Merkury1080P#customization