guino/BazzDoorbell #14: Is there a discord channel for this?

lcouch2018, 2020-12-31:
Since this is a highly active and we have several ppsapp patches for different firmwares I was wondering if anyone started a discord dedicated to the ppstrong. If no one has I would be glad to. I feel if we pooled together and worked together we could total make the firmware our own with custom functions and no calling out to tuya.

guino, 2020-12-31:
I don't have discord and I don't think anyone has started anything on it, so you're welcome to do it if you like. I don't know if you have seen #4 where they totally disabled the internet for the device and just use ntpd, log_parser, rtsp, cgi-bin scripts and mqtt to completely use the device off-cloud for local LAN access only. I can see the reasoning for wanting to do it but I personally like/use the tuya app myself (just wanted more control over it and know that I could generate custom events without paying for their services, even though I don't do anything with it myself). Any contribution at this point is welcome.

lcouch2018, 2020-12-31:
Yes I saw it and im looking at using the native mqtt code in ppsapp. Also looking at making my own tuya cloud spin off so on checkin will still be control by me. The issue I see with still using the tuya cloud is they see we modified the firmware and will eventually patch it. Knowing how they even do on tvs they can probably push a upgrade without them being enabled or with out you knowing. We are getting a feature they didnt want us to have. Merkcury gave me a " We are currently working on adding that feature."  I just see a lot of stuff we can do with this firmware and since you can make a dev account at tuya and roll your own mobile app/cloud I saw why use theres? I will make a discord tomorrow so we can track everyones development.  I was personally thinking about making the hack process auto the patch part not so sure about but everything else seems simple.