guino/BazzDoorbell #15: update to 2.9.7 possible? (LSC camera, mini11s)

debouttek, 2021-01-03:
Hi
I have patched my LSC indoor camera successful and it was on firmware 2.9.6.
I now restored it to the factory settings and it is stock now. Al works like out of the box.

I would like to update to 2.9.7 on the app from LSC or the Smart life app (both work) and it says that there is no update available.
It it possible to force update it with a BIN file or so?

{"devname":"Smart Home Camera","model":"Mini 11S","serialno":"0606xxxxx","softwareversion":"2.9.6","hardwareversion":"M11S_H1_V10_F23","firmwareversion":"ppstrong-c51-tuya2_lcs-2.9.6.20200628","authkey":"n4cwwrkQ5MrDl8ArB4owlwkIHah7fID0","deviceid":"pp01c4763f98f9710abd","identity":"MR2005211800904487","pid":"aaa","WiFi MAC":"xx:xx:xx:xx:xx:xx"}

RaveGun, 2021-01-03:
Hi,
I do not have the same camera but I have done the "patch" after upgraded to 2.9.7 and actually there is no ppsapp patching needed. You only need to have the correct addresses for mjpeg/image/video .cgis.

Because I use the LSC App, there were a lot of option not present at all in the tuya_config.json.
Or its was just maybe that the configuration got written by the 2.9.6 FW Version, I do not know.

Once I wrote the missing options, like:
```
        "onvif_enable": 1,
        "onvif_pwd":    "admin",
        "external_charm":       1
```
It worked without a problem.
I also managed to use the MQTT integration but for this I had to copy the ppsapp like in the initrun.sh script.

debouttek, 2021-01-03:
@RaveGun 
I cannot update the original firmware to 2.9.7 with the smart life or LSC app. 
Or do you mean to leave the camera on 2.9.6 and change the files?

guino, 2021-01-03:
@debouttek I will warn against running different versions of ppsapp than what the firmware came with - it may brick your device. There’s no ‘real’ advantage of running 2.9.7 vs 2.9.6 the way I see it, but if you want to upgrade I would highly recommend finding an app that has the update available. I did exactly that (ran 2.9.7 ppsapp on 2.9.6 fw) and it bricked my device and I had to use a programmer to restore it. The availability of a firmware may be largely dependent on the hardware version, in your case mini 11s vs bell 8s. The only way I would try a forced update is if I had a programmer, backup of current firmware and was willing to get dirty with soldering work to restore it in case it bricks the device.

debouttek, 2021-01-03:
@guino 

I only would like to update for the onvif. So I wait for it till the smart life app updates my device.

guino, 2021-01-03:
Onvif is only a publishing layer for the rtsp stream. There are some 2.9.6 versions that have onvif support (you can try enabling it in the json config file) it should be safe to do it. You could also try something like this but I expect it will highly depend on the nvr in question: https://community.home-assistant.io/t/connecting-any-rtsp-camera-to-onvif-nvr/115334

debouttek, 2021-01-03:
@guino 
quick guide? Where to find the Json file.

guino, 2021-01-03:
the json file is in /home/cfg/tuya_config.json -- if you're not familiar with linux the easiest thing to do is:
* login by telnet (assuming you configured it)
* execute **cp /home/cfg/tuya_config.json /mnt/mmc01**
* remove card, place it on computer, edit file, save and eject card properly
* insert card on device, wait a few seconds (i.e. 10) so the card is mounted
* execute: **cp /mnt/mmc01/tuya_config.json /home/cfg**

debouttek, 2021-01-06:
> the json file is in /home/cfg/tuya_config.json -- if you're not familiar with linux the easiest thing to do is:
>
> * login by telnet (assuming you configured it)
> * execute **cp [...]

Hi, I don't use telnet to hack the camera. I always removed the card from the camera to the pc.