guino/BazzDoorbell #16: Send message to Google Home speakers when someone is press te bell

m-Aiden, 2021-01-08:
Maybe someone has already found it. 
Is there a way to send a message to Google Home Speakers from the bell via bash or something?

jilleb, 2021-01-09:
I've used Homey to get this working.
You can either call a webhook that triggers a Google TTS event on Homey or link a flow to the 433Mhz signal.

m-Aiden, 2021-01-10:
> I've used Homey to get this working.
> You can either call a webhook that triggers a Google TTS event on Homey or link a flow to the 433Mhz signal.

Found that busybox has no curl or wget? Is there a way to send an http request?

guino, 2021-01-10:
@m-Aiden see:
https://github.com/guino/BazzDoorbell/issues/2#issuecomment-757497746

m-Aiden, 2021-04-13:
Already fix this.. Doorbell send a mqtt message to Home Assistant and they play a sound on Google Home.

closed 2021-04-13 by m-Aiden