guino/BazzDoorbell #17: Hack is unresponsive

kanawati975, 2021-01-09:
Hi there!
Thank you very much for your efforts.
I have "LSC Smart Camera" which is connected via Smart Life app, and I wanted to integrate it to Homeassistant, or at least make use of the inegrated motion sensor.
I followed the instructions, and reached to opening http://admin:056565099@192.168.178.145/proc/self/root/tmp/hack but this URL is not responding.
Here's more info, hoping that you can tell where I'm doing wrong:
**http://192.168.178.145/proc/cmdline:** mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=0 ppsWatchInitEnd - ip=${T//_/$'\\x20'}:::::;T=\"sleep_5;mkdir_-p_/mnt/mmc01;mount_-t_vfat_/dev/mmcblk0p1_/mnt/mmc01;/mnt/mmc01/initrun.sh&\";eval mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd
(_No matter I repeat, ppsAppParts remains 5_)
**http://192.168.178.145/devices/deviceinfo:** {"devname":"Smart Home Camera","model":"Mini 11S","serialno":"060644862","softwareversion":"2.9.6","hardwareversion":"M11S_H1_V10_F23","firmwareversion":"ppstrong-c51-tuya2_lcs-2.9.6.20200628","authkey":"lNBdXky5Yy7hUnFm2JqUaYT48KQGhNP4","deviceid":"pp01d273f1f2ad13f1a9","identity":"MR2007010101400116","pid":"aaa","WiFi MAC":"7c:a7:b0:f6:6b:aa"}
**http://192.168.178.145/proc/self/root/etc/init.d/S90PPStrong:** #MTDNUM=5

Looking forward to your recommendations, and thanks again.

guino, 2021-01-09:
You used #13 which is fine and it installed correctly based on your cmdline. The url you’re using for checking the hack is incorrect for #13 — please use the url in step 7 of #13 which is http://admin:056565099@192.168.x.x/proc/self/root/mnt/mmc01/hack (not the same as in #2). Then follow steps 8 and forward from #2.

Did you check your SD card to see if /home/app/ppsapp was created? If it’s there the hack is working, but if not you may be dealing with SD card format issues as some people have experienced.

kanawati975, 2021-01-09:
Hi guino,
Thank you for the quick response.
The URL http://192.168.178.145/proc/self/root/mnt/mmc01/hack is also not responsive (site can't be reached).
/home/app/ppsapp was created, but after your comment I have removed all files from the SD Card, and repeated the whole Process from step 3, and the URL http://192.168.178.145/proc/self/root/tmp/hack reported "done", while the URL http://192.168.178.145/proc/self/root/mnt/mmc01/hack is also not responsive (unable to handle this request).
What should I do now? Should I return the contents of the SD card and try? Please advise.

guino, 2021-01-09:
If you did the process and got done then the hack should be working and you should have /home/app/ppsapp file and you can look for a patch for it OR post it with your device information so it can be patched

kanawati975, 2021-01-09:
Okay I have completed step 9 and seems everything is in Order now... Here is the ppsapp file.
[ppsapp.zip](https://github.com/guino/BazzDoorbell/files/5791645/ppsapp.zip)

Just want to point out that busybox Filesize is smaller than what you mentioned in your Notes.

I really appreciate your efforts

guino, 2021-01-09:
@kanawati975 I will take a look at your ppsapp soon. In the meantime double check you downloaded the busybox separately from the link on step 9 - it should have the exact size posted or it will likely not work.

kanawati975, 2021-01-09:
@guino Many, Many Thanks...
About Busybox: I just got the correct file and all seems well now.
I appreciate your help and efforts... take your time :)

guino, 2021-01-09:
@kanawati975 your ppsapp has already been patched previously and the patch is here: https://github.com/guino/ppsapp-rtsp/issues/1#issuecomment-740477496

Please use https://www.marcrobledo.com/RomPatcher.js/ to patch your ppsapp with the above patch file then place the patched ppsapp in the root of the SD card with the name **ppsapp**. If you want to use snap/mjpeg.cgi you need to adjust the address in the cgi files with the provided one. And remember to keep some free space in the SD card to prevent the scripts from being deleted (I recommend backing up your files). Lastly you have to reboot the device with the SD card in place after saving/making changes to the files.

kanawati975, 2021-01-09:
@guino  Please excuse my ignorance...
So, I have patched the ppsapp and got a ppsapp.txt file, then I replaced the ppsapp file in /home/app with the new file, and removed the extension. But I didn't get why you said to put it in the root of the SD card. Or do you mean that I leave the unpatched file as it is, and put the patched file in the root of the SD card?
I think there is some free space... The card is 8GB

guino, 2021-01-09:
@kanawati975 I am not sure where you got a ppsapp.txt file as the rom patcher should provide a ppsapp-rtsp file which is the exact same size as the original ppsapp file — you then need to copy that ppsapp-rtsp file (downloaded from rom patcher) to the **root** of the SD card and name it just **ppsapp** and reboot. There’s no need to make any changes to the ppsapp under /home/app/ppsapp (just leave that alone).

kanawati975, 2021-01-09:
@guino in https://www.marcrobledo.com/RomPatcher.js/ I just used creator mode and repeated the same Process. The site gave me a ppsapp-rtsp.ips file.
Okay now I got it.
Thanks

guino, 2021-01-09:
Yeah , creator mode is to build a patch - you use the normal (not creator mode) to patch the ppsapp. I suppose I can make some step by step instructions to usr that page.

kanawati975, 2021-01-09:
The output of the normal mode (not creator mode) is .txt file. In the creator mode I got the .ips file
The contents of the SD card looks like this now.

[image]

guino, 2021-01-09:
Follow this step by step:
* remove the ppsapp file from root of the SD card (leave everything else alone)
* go to https://www.marcrobledo.com/RomPatcher.js/
* Click ‘choose file’ in front of ‘ROM file’ and select the original ppsapp file under /home/app/ppsapp on the SD card (assuming you didn’t do anything to it since the hack was installed)
* Click ‘choose file’ in front of ‘Patch file’ and select the zip file you downloaded from https://github.com/guino/ppsapp-rtsp/issues/1#issuecomment-740477496
* click ‘apply patch’ and save/download the file to the root of the SD card (the default file name will likely be ppsapp-rtsp)
* Rename the saved/downloaded file on the SD card to ppsapp — verify the size is exactly the same as the size of /home/app/ppsapp in the SD card which should be a few megabytes.
* Properly eject the SD card from windows using the tray icons
* Power off device, Insert SD card and power on device.
* Wait for it to boot - you likely will hear the startup sound 2 times. After the 2nd startup time you can try to view thr rtsp feed under rtsp://ip:8554

kanawati975, 2021-01-09:
WOW That was very detailed instructions... Thank you very much.
**Again:** The output of the normal mode (not creator mode) is .txt file. 
In creator mode there is options to change the output file's extension. 
Would it be okay if the output is .txt file, then I remove the extension?
**And again:** Please excuse my ignorance.

guino, 2021-01-09:
**DO NOT USE CREATOR MODE** you are not creating a patch, you are using the patch. Following the steps I provided exactly (it does not say to click on creator mode)

kanawati975, 2021-01-09:
@guino Thank you very much for your help and your patience. I really appreciate it

powerfarmer98, 2021-01-10:
I'm having the same issue with the Rompatcher. I think what @kanawati975 means to tell is that the tool, provided with the correct ppsapp, the downloaded zip and NOT in creator mode, does not create a "ppsapp-rtsp.xxx" but a "ppsapp.txt"
Is this output correct? And do we rename it to "ppsapp" (doing this causes my camera, ending in md5: 8957, to bootloop)
[image]

guino, 2021-01-10:
@powerfarmer98 can you try a different browser ? I just tried it here with the same ppsapp and same ppsapp-rtsp.zip and it asked me to save a ppsapp file when I clicked 'apply patch' -- no TXT extension and the file downloaded should have the exact same size as the original ppsapp file. If for some reason the file is of a different size it should not be used and likely there's something wrong with the browser during the patching process.

guino, 2021-01-10:
@powerfarmer98 @kanawati975 this is what you should see (I am using firefox): 
[image]

powerfarmer98, 2021-01-10:
Firefox indeed gave me an extentionless ppsapp, Both Chrome and edge however gave a ppsapp.txt

guino, 2021-01-10:
Ok, I tried chrome and saw that it gave a .txt file but the size and data in the file seems correct. I am wondering if there's an issue with the patch itself -- the patch posted seems to have different changes than on the patch I made. I am going to quickly review it.

Meanwhile: cam you try just copying the /home/app/ppsapp to the root of the SD card to verify the device boots normally (no bootloop) ?

powerfarmer98, 2021-01-10:
Unfortunately this also gives me a bootloop, so something else must be wrong i guess. Without a ppsapp in the root it boots fine however.

guino, 2021-01-10:
Ok, that is helpfull -- @Suspect-byte is having the same issue apparently. I am reviewing the scripts to see what may be the problem.

guino, 2021-01-10:
@powerfarmer98 @Suspect-byte I'll need someone to help me find out what's going on, so I need the following:
1. copy the **original** ppsapp to the root of the SD card (I understand this causes the bootloop too)
2. Modify custom.sh so it looks like this:
```
#!/bin/sh
if [ ! -e /tmp/customrun ]; then
 echo custom > /tmp/customrun
 cp /mnt/mmc01/passwd /etc/passwd
 /mnt/mmc01/busybox telnetd -l /bin/sh
 /mnt/mmc01/busybox httpd -c /mnt/mmc01/httpd.conf -h /mnt/mmc01 -p 8080
 if [ -e /mnt/mmc01/ppsapp ]; then
  PPSID=$(ps | grep -v grep | grep ppsapp | awk '{print $1}')
  kill $PPSID
  echo $PPSID >> /mnt/mmc01/ppsapp.log
  /mnt/mmc01/ppsapp &>> /mnt/mmc01/ppsapp.log &
 fi
fi
```
(The only changes are in the ppsapp.log lines -- 4th and 3rd last lines)

3. Modify initrun.sh so it looks like this:
```
#!/bin/sh

# Copy app partition if not already done
if [ ! -e /mnt/mmc01/home ]; then 
 MTDNUM=`cat /proc/cmdline | sed 's/.*ppsAppParts=\([0-9]\).*/\1/'`
 mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
 tar xzf /opt/pps/app.tar.gz -C /mnt/mmc01/
fi

# Now flag the hack is done
echo done > /mnt/mmc01/hack

# Try to run custom.sh
while true; do
 if [ -e /mnt/mmc01/custom.sh ]; then
  /mnt/mmc01/custom.sh
 fi
 sleep 10
done
```
(The only change is in the position of the 'sleep 10' line at the bottom section)
4. Make sure the 'home' directory is not deleted from the SD card
5. Boot the device -- wait to see if it bootloops or not.
6. Remove SD card then zip/post the ppsapp.log file that should be created on the SD card.

On my device (2.9.6) the current scripts were working correctly so I am not sure why it is not working for you guys -- I'll need the log to find out.

guino, 2021-01-10:
I also have a possible theory here -- if the device is much slower than mine it is possible that the built-in ppsapp is being started AFTER the ppsapp from the SD card. To test this we need to modify initrun.sh to be this:
```
#!/bin/sh

# Copy app partition if not already done
if [ ! -e /mnt/mmc01/home ]; then 
 MTDNUM=`cat /proc/cmdline | sed 's/.*ppsAppParts=\([0-9]\).*/\1/'`
 mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
 tar xzf /opt/pps/app.tar.gz -C /mnt/mmc01/
fi

# Now flag the hack is done
echo done > /mnt/mmc01/hack

# Try to run custom.sh
while true; do
 sleep 30 
 if [ -e /mnt/mmc01/custom.sh ]; then
  /mnt/mmc01/custom.sh
 fi
done
```
(The above is the same as the initrun.sh in ppshack2.zip with the sleep increased to 30 where it was 10)
If someone can test that it could be helpful as well -- maybe it won't bootloop with that.

kanawati975, 2021-01-10:
@powerfarmer98 Thank you. Yes the problem is in google Chrome. Firefox outputted the file without extension.

@guino I just copied /home/app/ppsapp to / and did not get bootloop. Does this means that --at somepoint-- ppsapp is currupted?

guino, 2021-01-10:
@kanawati975 if you copied /home/app/ppsapp to / and it did not boot loop then it probably ran it correctly -- if you have a chance: please modify inirun.sh to have sleep 30 (instead of sleep 10) and see if it bootloops. If it doesn't boot loop and you hear the startup sound 2 times then go ahead and placed the patched ppsapp in the root of SD card with the name **ppsapp** and reboot -- it should enable RTSP then.

kanawati975, 2021-01-10:
@guino Okay I edited inirun.sh and uncommented the last section and made sleep 30.
The Camera booted twice, and blue led is on (usually meand it's connected to my wifi)

guino, 2021-01-10:
@kanawati975 thats good! Go ahead and put in the patched ppsapp and try rtsp!

kanawati975, 2021-01-10:
@guino just to be clear: 
Now I use the website to patch the ppsapp file that is on the root folder now, so replace it with the original that been copied from /home/app?

guino, 2021-01-10:
You just have to follow these steps exactly now:
https://github.com/guino/BazzDoorbell/issues/17#issuecomment-757366970

kanawati975, 2021-01-10:
@guino Well basically this is what I did (since /ppsapp is a copy of /home/app/ppsapp) 
so I repalced it with the patched file. No issues or bootloops, but unfortunately no RTSP stream is shown on homeassistant

guino, 2021-01-10:
@kanawati975 did you try vlc or mplayer? I don’t think 2.9.6 has onvif if that’s what you’re trying to use. There’s an issue with onvif instructions but I think that only works on 2.9.7

kanawati975, 2021-01-10:
@guino Just did and YES! I got RTSP stream on VLC.

guino, 2021-01-10:
@kanawati975 awesome! Thanksgiving for the feedback. You may want to look at ways to use rtsp (not onvif) on homeassist. It may also be possible to “conver” rtsp to onvif by installing a piece of software as onvif is just a publishing layer that works on top of rtsp.

kanawati975, 2021-01-10:
Onvif didn't work so I tried to integrate it as generic in configuration.yaml
- platform: generic
    name: Indoor camera
    username: admin
    password: 056565099
    still_image_url: http://192.168.178.145:8554/jpg/image.jpg
    stream_source: rtsp://192.168.178.145:8554

No, Thank you @guino for your patience

guino, 2021-01-10:
@powerfarmer98 and @Suspect-byte you should be able to use the original custom.sh and initrun.sh from ppshack2.zip and just adjust the sleep 10 to be sleep 30 to prevent the bootloop issue (and use the rtsp patched ppsapp).

powerfarmer98, 2021-01-10:
@guino Changing sleep to 30 got it to work! rstp also tested and working. Thank for the effort!

guino, 2021-01-10:
I closed the issue as it was determined the issue was due to the sleep value in the initrun.sh from #13 - changing it from 10 to 30 fixes the issue and I have updated the initrun.sh script in #13 to have 30 by default so this should no longer happen.

closed 2021-01-10 by guino