guino/BazzDoorbell #18: Empty SDcard after reboot with red led

debouttek, 2021-01-14:
I have strange behavior. Everything works snap/mjpeg/rtsp and when I pull out the power cable from the camera and lets it reboot the red led stays red. My SDcard is then empty.

I need to put back env/initrun.sh/ppsmmctool.txt and the the camera powers on and red led becomes blue and it works. Then I need to do all steps with MMC files and ppsapp again and when I pull out the power cable I need to do all over again...

What can be done about it?

LSC camera indoor

{"devname":"Smart Home Camera","model":"Mini 11S","serialno":"060653965","softwareversion":"2.9.6","hardwareversion":"M11S_H1_V10_F23","firmwareversion":"ppstrong-c51-tuya2_lcs-2.9.6.20200628","authkey":"xxxx","deviceid":"xxxx","identity":"xxxx","pid":"aaa","WiFi MAC":"xx:xx:xx:xx"}

guino, 2021-01-14:
@debouttek 
This seems like the normal behavior if you have low space on the SD card... what is the SD card size you're using and how much is free? I don't know exactly how much is considered 'low' for the ppsapp but I would make sure you have at least 1GB free space to prevent it from deleting your files from the SD card. I have not had any issues with it deleting my files while keeping around 4GB free space (that's about 90 days worth of motion-only recordings on my front door camera).

debouttek, 2021-01-14:
OK. I used a 2GB card that was laying around. I also switched off the scared registration switch on the LSC app.
I see I only have about 0,36GB free

closed 2021-01-14 by debouttek