guino/BazzDoorbell #19: Hisilicon SDK

4c0n, 2021-02-01:
After a lot of digging, I seem to have found the SDK that was used to make the original firmware:

http://hasecurity.com/HL_support_sdks.aspx

For some reason I can't find it on their own website, so I guess they're being sneaky about this ;) However this contains the source of most of the kernel modules, the fs images, uclibc, etc..
Haven't set up a toolchain with it yet, but it also contains instructions on how to that.

My guess is that ppsapp uses a tuya sdk interface like:
"https://github.com/tuya/tuya-iotos-embeded-multimedia-demo/blob/master/demo_for_ipc/demo_src/tuya_ipc_doorbell_demo.c"

that interacts with the peripherals using the drivers. 

Have you seen anything like that in ghidra?

There's a few missing links left, but could this be helpful with interpretation of the output?

guino, 2021-02-01:
I had posted the links to that SDK before: https://github.com/guino/BazzDoorbell/issues/2#issuecomment-744672236

I did get a more updated version of it from @LouDnl since then but have not had the time to do anything else with it -- it's about 2GB. Send me an email if you want details but the question I have is what specific features you're trying to add ? Some of the sample code resembles stuff I've seen in ghidra but it seems they have a different version of their own 'ppsapp' somewhat similar to what's in the demos.

4c0n, 2021-02-01:
@guino pretty much interested in attempting to build my own firmware image and cutting out tuya completely by writing a new app.
Basically liberating the hardware to do whatever with kind of like openipc.org or maybe actually use that.
Learning what makes these things tick is very entertaining :) After all the things I've seen so far, security wise this device is a total joke and I'm not too sure I'd want to use this thing as an actual doorbell without some sort of construction to physically keep people out at the very least.
Hopefully the code quality of the Tuya code is better than the Hisilicon code quality which looked poor in some places, but I doubt it since it seems that developers kind of put it together using a low coding platform they provide although I'm not 100% sure it's how this product was crafted.

guino, 2021-02-01:
@4c0n if you have the time to do that, I'm sure you can build a your own kernel and use the hisilicon drivers (and/or sources) to do everything you want or just run the openipc hisilicon stuff. Personally the only thing I would like to have is NFS support and I'm not sure even openipc firmware supports that and requires flashing everything new (which I can do) but again is a lot of work -- it is cheaper (time-wise) to just buy something that supports the openipc firmware directly (or something that already supports NFS in my case). The one and only reason I ever bought this doorbell is for the hardware integration with the existing wiring of the house (transformer/bell). For $20 less I could have gotten another 4K POE camera with everything out the box (including NFS), but it wouldn't have a button, 2 way talk or existing wiring support.

4c0n, 2021-02-01:
@guino it's always cheaper to buy some (usually) Chinese product instead. But for me the point is to challenge myself and learn something new. It might be cheaper than a dev board though, but I can imagine even those coming quiet cheap.
I don't know a lot about NFS it's internals, but from my experience it can be a heavy operation and is not very fast... No clue if the device would be able to stay on top of everything and provide NFS access at the same time. Should I reach the point of being able to easily implement it, I'll be sure to let you know.

guino, 2021-02-01:
@4c0n I have similar hi-silicon cameras with even less powerful processors and older kernel that support NFS just fine -- it doesn't take a lot but they just didn't compile support for it in the kernel. NFS support would allow to just use a network mapping as storage instead of a SD card that can be ripped out and stolen along with the device. The closest thing I did was to put a script that syncs the video recordings to my network drive every minute.
If you want to play with it as a challenge, I definitely encourage it. I never tried but if the proper support is added to the kernel it should be possible to use the existing bootloader/firmware to load a kernel from the SD card and mount/use the OS entirely from the SD card which would be a way to keep the firmware nearly intact (so it can be factory reset) and run everything new (kernel+os) from the SD.

4c0n, 2021-02-01:
@guino Right I was thinking you wanted to make the device run the nfs server for some reason, but yeah as a client it shouldn't be an issue. 😃 It would be fun and educational to mess around with regardless!

4c0n, 2021-02-01:
Extracted some interesting sample code from the SDK already: https://github.com/4c0n/uvc_app :smiley:

guino, 2021-02-03:
The SDK can be found here too (old and new):
https://drive.google.com/drive/folders/1LhHEmQdA5V96_o4Vc9YjKDU7vlDMToEp?usp=sharing

4c0n, 2021-02-03:
Cool! I found out that the difference is not so much old or new, but one is the Linux based SDK and the other (new) one is the LiteOS version, which seems to be the one that was used as the base for our device.
Both are incomplete it seems, there's some info in Chinese and Korean about the contents on some sites I found, but fortunately also some translated stuff.
Also found out that there are actually official downloads on the Huawei website, but to be able to download any of it you need to upgrade account privileges by providing a contract or some other kind of proof.

4c0n, 2021-02-03:
https://www.programmersought.com/article/53856306255/
https://www.programmersought.com/article/20946900812/
https://www.programmersought.com/article/69196304451/

It's kind of a strange website, not sure how the content is produced, it seems machine aggregated/translated...

guino, 2021-02-03:
That seems like it has good pointers in setting things up - better than nothing.

4c0n, 2021-02-04:
Found some more SDKs here: https://dl.openipc.org/SDK/HiSilicon/Hi3516Ev200_16Ev300_18Ev300

guino, 2021-02-04:
@4c0n I had posted that address too but you have to “pay” to actually download the files there.

closed 2021-07-22 by 4c0n