guino/BazzDoorbell #20: 2.10.0 - Bell5s, not patched

russinnes, 2021-02-01:
I've gotten down a new rabbit hole here.
Attached (FYI, the firmware update did nothing to prevent it from booting from SD as it was configured prior for 2.9.7). 
It is still loading the 2.9.7 as per the script, after killing the new updated ppsapp. FWIW, the (german?) firmware update text suggested something about ONVIF. 

Attached
[ppsapp-2.10.0.zip](https://github.com/guino/BazzDoorbell/files/5907519/ppsapp-2.10.0.zip)

guino, 2021-02-02:
I can have a look and make a patch today but it sounds like if you remove the patched ppsapp from the SD card and reboot it should have onvif native (without patching) that you may want to try out.

russinnes, 2021-02-02:
The addresses from the jpeg buffer would be handy - the onvif built in is not working, but I havn't dug too far into it. Everything on the bell itself works, except closing of the contacts (I imagine a GPIO to a tranistor or similar) is not working. I had to set it up as a standard camera so it may very well be an entry in the JSON. I'll report back findings, mainly for others out there.

guino, 2021-02-02:
@russinnes I am working on this now, can you post your updated /devices/deviceinfo ?

guino, 2021-02-02:
jpeg address: 0x005b97e8

russinnes, 2021-02-02:
> @russinnes I am working on this now, can you post your updated /devices/deviceinfo ?

Doesnt work on this firmware -- digging thru it right now.

guino, 2021-02-02:
It should still work from what I see in the file but you may need the user/password: http://admin:056565099@IP/devices/deviceinfo

russinnes, 2021-02-02:
> http://admin:056565099@IP/devices/deviceinfo

Nope, doesnt. Nothing running on ::80

guino, 2021-02-02:
@russinnes that sucks, aren't you glad you hacked it while it was on 2.9.7 ?
I believe this is your firmware string (which is what I wanted to get: ppstrong-c51-tuya2_bardi-2.10.0.20201222 do you remember the exact firmware you used before ? I just wanted to post it in https://github.com/guino/ppsapp-rtsp/issues/1 with as much detail as possible (at this point only need the hardware version).

russinnes, 2021-02-02:
> @russinnes that sucks, aren't you glad you hacked it while it was on 2.9.7 ?
> I believe this is your firmware string (which is what I wanted to get): ppstrong-c51-tuya2_bardi-2.10.0.20201222 do you remember the exact firmware you used before ? I just wanted to post it in [guino/ppsapp-rtsp#1](https://github.com/guino/ppsapp-rtsp/issues/1) with as much detail as possible (at this point only need the hardware version).

Yes, entirely. I'm digging around in a shell I'll let you know what I find. Here is the firmware that was on it at update time.  Sorry, I've flashed a variety on it, hard to keep verisons straight. 
[ppsapp.zip](https://github.com/guino/BazzDoorbell/files/5908579/ppsapp.zip)

guino, 2021-02-02:
@russinnes I didn't make the patch for 2.9.7 for that version this is what confused me. I have posted a patch for this 2.10.0 ppsapp here: https://github.com/guino/ppsapp-rtsp/issues/1#issuecomment-771350179

PLEASE NOTICE: I got reports of problems trying to enable the 'old' RTSP on ONVIF enabled versions so I have been adjusting the URL to rtsp://IP:8555 (notice the port is NOT 8554). This should allow the new onvif feature to operate normally, if configured, at whatever port it is configured which I think it's 8554 by default.

russinnes, 2021-02-02:
This is all hillarious, its running on a energizer doorbell cam Bell5s. I'll run the patch right now and let you know.

russinnes, 2021-02-02:
Works like a charm.

guino, 2021-02-02:
@russinnes thanks for the feedback! I am expecting that despite 2.10.0 not having the port 80 responses we should still be able to apply the hack (probably #13 since it doesn't require knowing the cmdline), but obviously you don't have to do it since it was already installed.

guino, 2021-02-03:
@russinnes I can see on ghidra where they blocked off port 80 unless the device hasn’t been serialized yet (if serial is set to 056565099), so I am expecting that all newer firmware will come like that. I guess I will have to wait till the newer firmware is available on more devices so I can get one to play with it. Without the cmdline we would have to use #11 and **guess** the cmdline in order to apply the hack. The other option is #12 which is riskier. There’s a slight chance of making a version of the hack without the cmdline but I will need a device to test with.

guino, 2021-02-09:
@russinnes by the way If you have a change, I would like you to try the info from https://github.com/guino/Merkury720/issues/6#issuecomment-775656600 on 2.9.10 without the hack to see if it works (you can just rename the patched ppsapp from the SD card to something else, reboot to verify port 80 is closed, create the file and verify if port 80 opens).. If you can confirm it I can add it to the steps so port 80 can be used again.

russinnes, 2021-02-09:
@guino yes will do - I assume you're referring to 2.10.0 that I posted a few nights ago. Just confirming the string needs to end with 0x0D 0x0A?

guino, 2021-02-09:
Yes, but I would advise just right clicking the link and “save as...” to avoid formatting issues. The. Just edit the wifi network name (ssid) and password.

russinnes, 2021-02-09:
Yes, works like a charm. Sorry, that 2.10.0 module is on a doorbell (PITA to get SD) so had to do it manually in vi over telnet.

guino, 2021-02-09:
@russinnes thanks for confirming and letting me know. I will update the instructions so others can use it to open port 80.