guino/BazzDoorbell #21: CW017 research

guino, 2021-02-07:
Here's a couple of images. Looks similar to the board in #11 from @swisslegacy

[image]

[image]

_Originally posted by @parkerlreed in https://github.com/guino/BazzDoorbell/issues/2#issuecomment-774789777_

parkerlreed, 2021-02-07:
:+1: Thanks. Good idea keeping tracking ina  separate thread (that's a pain in the ass to expand all the comments in the main)

guino, 2021-02-07:
@parkerlreed yes :) similar but different hardware. The uart pads will likely be on the other side by removing the 4 silver colored screws

parkerlreed, 2021-02-07:
Trying the 2.7.x files from #11 here's what I observe

- Light blinks red and blue after holding reset for 5 seconds
- I release, it sits on red for a second, and then goes blue
- Stays solid blue (and never gets to the "Hey Im booted" beep)
- Wait a couple minutes and check SD card
- Looks to be the original MBR
[image]

2.9.x on my first try resulted in a corrupted MBR which made me think it worked but was the same result (and subsequent tries the MBR stays in tact)

- Red and blue after 5 seconds
- This time stays red for about 20 seconds
- Proceeds to boot normally
[image]

parkerlreed, 2021-02-08:
Inside shot

[image]

guino, 2021-02-08:
There’s usually data left in the sectors of the card from before doing #11 so if you want to see if the process did anything you would have to zero out the sectors before using #11:

dd if=/dev/zero of=/dev/mmcblkX bs=512 count=32768

guino, 2021-02-08:
Serial pins should be the 4 pads on the bottom left of the picture you posted.

guino, 2021-02-08:
You can use the screw hole as ground and test which is ground from the pads, then test which is 3.3v using pin 8 of the flash chip. Or you can just solder wires and swap them until you find what is rx, tx and ground (no need to connect 3.3v). Just make sure you don’t pull hard on wiring after soldering or they will come right out of the board.

guino, 2021-02-08:
This should be pin 8 of the flash (3.3v) :

[image]

parkerlreed, 2021-02-08:
I tried zero'ing out beforehand and no difference. (Stayed blank instead of having any data)

Does the device not care that I blew out the MBR?

guino, 2021-02-08:
The MBR is sector 0 (zero) the command I gave was supposed to start zeroing from sector 1 - but I realize I forgot to ass the parameter seek=1, it should have been:

dd if=/dev/zero of=/dev/mmcblkX bs=512 count=32768 seek=1

that was my bad - you would probably have to redo #11 to build a new MBR now.

parkerlreed, 2021-02-08:
As for the serial, I have a couple Pi Zeros I can try and get working (They do have 3.3v UART right?) 

Thanks for the tips.

guino, 2021-02-08:
Yeah you could wire the serial from a pi zero directly.

guino, 2021-02-08:
@parkerlreed I can’t read the numbers on the chips but based on the previous version of the board the flash may actually be the one on the other side of the board instead of the one I posted above (pin 8 3.3v marked):
[image]

Like I said before you don’t need to connect the 3.3v I only marked it as a way to try and determine which pads to use (so you don’t need to solder a wire to the 3.3v pad).

parkerlreed, 2021-02-10:
```
U-Boot 2013.10.0-AK_V3.0.07 (Sep 19 2020 - 14:46:53)

DRAM:  64 MiB
8 MiB
ANYKA SDHC/MMC4.0: 0
PPS:Sep 19 2020 14:46:58   anyka_c3:  0 
                                        magic err
magic err
command:setenv bootargs mem=64M console=ttySAK0,115200n8 loglevel=10   mtdparts=spi0.0:256k(bld),64k(env),64k(enc),64k(sysflg),3m(sys),4032k(app),640k(cfg) ppsAppParts=5 ip=192.168.1.99:::255.255.255.0 eth=00:55:7b:b5:7d:f7
## Booting kernel from Legacy Image at 81c08000 ...
   Image Name:   Linux-3.4.35
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2789360 Bytes = 2.7 MiB
   Load Address: 81c08000
   Entry Point:  81c08040
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Anyka Linux Kernel Version: 1.1.12
```

parkerlreed, 2021-02-10:
Also for reference here's the listing for this camera https://www.walmart.com/ip/Merkury-Innovations-Smart-Wifi-Camera-1080p-2-Way-Communication/679786717

parkerlreed, 2021-02-10:
Middle two. TX on top in picture. [image]

guino, 2021-02-10:
Send me an email and I can send you something to help out

parkerlreed, 2021-02-10:
Sent. Thanks.

guino, 2021-02-10:
Based on the bootargs shown the method used for Merkury 720p (my other repo) should work if you just adjust the load address (looks like 81c08000?)

parkerlreed, 2021-02-10:
Thanks. Will give it a shot

parkerlreed, 2021-02-10:
So do I do the port 80 enable first and then the rest?

guino, 2021-02-10:
It should just need the commands in the env file (typed in thr uboot since you can), ideally we figure out the address so that hack would work on it.

If your port 80 is disabled be default you can use the ppsFactoryTool.txt to enable it but that’s not important for the hack to work.

parkerlreed, 2021-02-10:
I'm sure I'm probably just being thick here. I am typing the correctly?

[image]

guino, 2021-02-10:
Remove the hack= part and only do what is after the = sign .. the hack= part is to set the environment from file which is not needed in uboot.

guino, 2021-02-10:
Actually nevermind that, we need to figure out where the files are loaded, can you execute **printenv** and post the response ?

parkerlreed, 2021-02-10:
Ok so something like this?

```
setenv bootargs ${bootargs} '- ip=30;/mnt/mmc01/initrun.sh)&:::::;date>/tmp/hack;(sleep
ipaddr=0
bootm 0x81c08000
```

ipaddr is supposed to be it's own command or part of the first one?

guino, 2021-02-10:
Without being familiar with uboot it will be easier to just modify the hack for your camera — with the printenv output I can probably do it.

the commands from the env file would have to be like this I think:

setenv hack ‘setenv bootargs ${bootargs} \'- ip=30;/mnt/mmc01/initrun.sh)&:::::;date>/tmp/hack;(sleep’

then another command:

setenv ipaddr ‘0;run hack;bootm 0x81c08000;’

It would be nice if you execute and post the output for **printenv** and for **help** so I know what is available

parkerlreed, 2021-02-10:
```
pps # printenv
              baudrate=115200
bootargs=mem=64 console=ttySAK0,115200n8 loglevel=10
bootcmd=loadk;bootm 0x81C08000
bootdelay=3
console=ttySAK0,115200n8 loglevel=10 
ethaddr=00:55:7b:b5:7d:f7
ipaddr=192.168.1.99
memsize=64M
netmask=255.255.255.0
serverip=192.168.1.1
setcmd=setenv bootargs console=${console}  mem=${memsize}
sf_hz=20000000
stderr=serial
stdin=serial
stdout=serial
ver=U-Boot 2013.10.0-AK_V3.0.07 (Sep 19 2020 - 14:46:53)

Environment size: 455/4088 bytes
pps # help
          ?       - alias for 'help'
auth    - encrypt product authfile
base    - print or set address offset
bootm   - boot application image from memory
clearCfg- clearCfg
cmp     - memory compare
cp      - memory copy
crc32   - checksum calculation
devmem  - read or write register, now just read
downimage- downimage   - download and write All-Image to FLASH device,partiton table from ENV partition.
downjffs2fs- load usr.jffs2 tftp
downkernel- load uImage tftp
downrootfs- load root.sqsh4 tftp
downsquashfs- load usr.sqsh4 tftp
downuboot- load uboot tftp
env     - environment handling commands
envreset- reset env para
erase   - erase FLASH memory
fatinfo - print information about filesystem
fatload - load binary file from a dos filesystem
fatls   - list files in a directory (default /)
flinfo  - print FLASH memory information
format  - erase all flash except bootloader part
go      - start application at address 'addr'
help    - print command description/usage
loadk   - load kernel to DRAM  
loop    - infinite loop on address range
md      - memory display
mm      - memory modify (auto-incrementing address)
mmc     - MMC sub system
mmcboot - from mmc start
mmcinfo - display MMC info
mw      - memory write (fill)
nm      - memory modify (constant address)
parts   - read out partitions table info.
parts_adjust- 
adjust parts info. Each part's size,offset etc.

pcbMenu - select Pcb
printenv- print environment variables
protect - enable or disable FLASH write protection
readcfg - read config from config.
reset   - Perform RESET of the CPU
run     - run commands in an environment variable
saveenv - save environment variables to persistent storage
setenv  - set environment variables
setloadaddr- set loadaddr to  config infor .
sf      - spi flash sub-system:
tfdownjffs2fs- load usr.jffs2TF
tfdownkernel- load uImageTF
tfdownrootfs- load root.sqsh4TF
tfdownsquashfs- load usr.sqsh4TF
tfdownuboot- load u-boot.binTF
tfupdateimage- tfupdateimage   - download and write All-Image to FLASH device
uartdown- down style 0:tftp 1:uart 2:mmc
upa     - update app image appfile
upb     - update bootloader bldfile
update  - update upgrade.bin
updatecfg- update config from config infor table.
upe     - update auth image authfile
upf     - update firmware, format and update (factory use upffile)
upk     - update uImage kernelfile
upr     - update backup.img
version - print monitor, compiler and linker version
```

parkerlreed, 2021-02-10:
After setting thos two

```
bootargs=mem=64 console=ttySAK0,115200n8 loglevel=10 - ip=30;/mnt/mmc01/initrun.sh)&:::::;date>/tmp/hack;(sleep
ipaddr=0;run hack;bootm 0x81c08000;
```

Should be good to power cycle?

guino, 2021-02-10:
There’s a very good chance that if you just adjust the address in ppsMmcTool.txt and env files from the merkury 720p project to 81c08000 (instead of 81808000) you should be able to install the hack using the steps in that repo - that is what I would try first.

guino, 2021-02-10:
If you executed the commands you don’t eant to power cycle you want to execute **loadk**

guino, 2021-02-10:
I doubt the commands as I posted will work - hard to tell without the device to do printenv after every change. But I am very hopeful that using the existing hack with the correct address will work.

parkerlreed, 2021-02-10:
I reset the env back to original and am trying that now.

parkerlreed, 2021-02-10:
```
U-Boot 2013.10.0-AK_V3.0.07 (Sep 19 2020 - 14:46:53)

DRAM:  64 MiB
8 MiB
ANYKA SDHC/MMC4.0: 0
PPS:Sep 19 2020 14:46:58   anyka_c3:  2 

U-Boot 2013.10.0-AK_V3.0.07 (Sep 19 2020 - 14:46:53)

DRAM:  64 MiB
8 MiB
ANYKA SDHC/MMC4.0: 0
PPS:Sep 19 2020 14:46:58   anyka_c3:  0 
                                        button
cmd:fatload mmc 0 0x81c08000 ppsmmctool.txt
send cmd 8 error, status = 2004
block rw command 8 is failed!
mmc_send_if_cond, Err cnt:0
cdh:test_part_dos DOS_PART_MAGIC_OFFSET ok!
reading ppsmmctool.txt
102 bytes read in 1 ms (99.6 KiB/s)
cmdBuf:fatload mmc 0 0x81c08000 env;env import 81808000;saveenv
reading env
131 bytes read in 0 ms
## Warning: defaulting to text format
## Warning: Input data exceeds 1048576 bytes - truncated
## Info: input data size = 1048578 = 0x100002

Saving Environment to SPI Flash...
Env export buffer too small: 4088, but need 1049053
ERROR: Cannot export environment: errno = 12

at env_sf.c:443/saveenv()
size:131
error: Pack header size error!
error: upgrade.bin unpack error!
reading env
131 bytes read in 1 ms (127.9 KiB/s)
## Warning: defaulting to text format
## Warning: Input data exceeds 1048576 bytes - truncated
## Info: input data size = 1048578 = 0x100002

Saving Environment to SPI Flash...
Env export buffer too small: 4088, but need 1049053
ERROR: Cannot export environment: errno = 12

at env_sf.c:443/saveenv()
readLen:131
read err
resetting ...
heartbeat = 1
m

U-Boot 2013.10.0-AK_V3.0.07 (Sep 19 2020 - 14:46:53)

DRAM:  64 MiB
8 MiB
ANYKA SDHC/MMC4.0: 0
PPS:Sep 19 2020 14:46:58   anyka_c3:  0 
                                        magic err
magic err
command:setenv bootargs mem=64M console=ttySAK0,115200n8 loglevel=10   mtdparts=spi0.0:256k(bld),64k(env),64k(enc),64k(sysflg),3m(sys),4032k(app),640k(cfg) ppsAppParts=5 ip=192.168.1.99:::255.255.255.0 eth=00:55:7b:b5:7d:f7
## Booting kernel from Legacy Image at 81c08000 ...
   Image Name:   Linux-3.4.35
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2789360 Bytes = 2.7 MiB
   Load Address: 81c08000
   Entry Point:  81c08040
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Anyka Linux Kernel Version: 1.1.12
```

That's what I get trying the purely SD method from Merkiry720

guino, 2021-02-10:
This almost looks good but it seems you either created the env file from scratch (you should have downloaded and edited the one I made) OR you deleted the 0x00 character after the new line in the env file. Please make sure the env file has a new line and a character 0x00 at the end try again.

parkerlreed, 2021-02-10:
Yeah I just edited the file in place.

Maybe nano ate something?

guino, 2021-02-10:
It is possible, you may want to use a hex editor to make sure there’s a 00 at the end of the file.

parkerlreed, 2021-02-10:
Ahh that's probably why. nano tends to throw on a /n. Ill make the edit to the base address in the hex editor to make sure.

guino, 2021-02-10:
Give me a sec and I can post a good env file

parkerlreed, 2021-02-10:
The file straight from your repo ends in OA 

[image]

guino, 2021-02-10:
just unzip that over the SD card and overwrite it
[env.zip](https://github.com/guino/BazzDoorbell/files/5955741/env.zip)

guino, 2021-02-10:
The file looks ok, but wrong address on the last image you posted.

parkerlreed, 2021-02-10:
Ok about to boot this card. :crossed_fingers:

guino, 2021-02-10:
post the log when it's done

parkerlreed, 2021-02-10:
Same thing it looks like :/ Made sure to Overwrite the env that's on the card,

```
Type [C-a] [C-h] to see available commands
Terminal ready

U-Boot 2013.10.0-AK_V3.0.07 (Sep 19 2020 - 14:46:53)

DRAM:  64 MiB
8 MiB
ANYKA SDHC/MMC4.0: 0
PPS:Sep 19 2020 14:46:58   anyka_c3:  0 
                                        button
cmd:fatload mmc 0 0x81c08000 ppsmmctool.txt
send cmd 8 error, status = 2004
block rw command 8 is failed!
mmc_send_if_cond, Err cnt:0
cdh:test_part_dos DOS_PART_MAGIC_OFFSET ok!
reading ppsmmctool.txt
102 bytes read in 0 ms
cmdBuf:fatload mmc 0 0x81c08000 env;env import 81808000;saveenv
reading env
131 bytes read in 1 ms (127.9 KiB/s)
## Warning: defaulting to text format
## Warning: Input data exceeds 1048576 bytes - truncated
## Info: input data size = 1048578 = 0x100002
ERROR: Environment import failed: errno = 22

at cmd_nvedit.c:1076/do_env_import()

Saving Environment to SPI Flash...
Env export buffer too small: 4088, but need 520537
ERROR: Cannot export environment: errno = 12

at env_sf.c:443/saveenv()
size:131
error: Pack header size error!
error: upgrade.bin unpack error!
reading env
131 bytes read in 1 ms (127.9 KiB/s)
## Warning: defaulting to text format
## Warning: Input data exceeds 1048576 bytes - truncated
## Info: input data size = 1048578 = 0x100002
ERROR: Environment import failed: errno = 22

at cmd_nvedit.c:1076/do_env_import()

Saving Environment to SPI Flash...
Env export buffer too small: 4088, but need 520537
ERROR: Cannot export environment: errno = 12

at env_sf.c:443/saveenv()
readLen:131
read err
resetting ...
heartbeat = 1
m

U-Boot 2013.10.0-AK_V3.0.07 (Sep 19 2020 - 14:46:53)

DRAM:  64 MiB
8 MiB
ANYKA SDHC/MMC4.0: 0
PPS:Sep 19 2020 14:46:58   anyka_c3:  0 
                                        magic err
magic err
command:setenv bootargs mem=64M console=ttySAK0,115200n8 loglevel=10   mtdparts=spi0.0:256k(bld),64k(env),64k(enc),64k(sysflg),3m(sys),4032k(app),640k(cfg) ppsAppParts=5 ip=192.168.1.99:::255.255.255.0 eth=00:55:7b:b5:7d:f7
## Booting kernel from Legacy Image at 81c08000 ...
   Image Name:   Linux-3.4.35
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2789360 Bytes = 2.7 MiB
   Load Address: 81c08000
   Entry Point:  81c08040
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Anyka Linux Kernel Version: 1.1.12
```

guino, 2021-02-10:
ok, in the ppsMmcTool.txt file you needed to edit all addresses that were 81808000 to 81C08000 -- there's at least one address wrong in that file, please adjust and try again.

parkerlreed, 2021-02-10:
OH I thought env was the only edit *facepalm*

Checking it. Thanks.

parkerlreed, 2021-02-10:
Looks to be good. Waiting for it to boot. It still doing the blinky blue thing. I assume it's booted at that point?

```
Type [C-a] [C-h] to see available commands
Terminal ready

U-Boot 2013.10.0-AK_V3.0.07 (Sep 19 2020 - 14:46:53)

DRAM:  64 MiB
8 MiB
ANYKA SDHC/MMC4.0: 0
PPS:Sep 19 2020 14:46:58   anyka_c3:  0 
                                        button
cmd:fatload mmc 0 0x81c08000 ppsmmctool.txt
send cmd 8 error, status = 2004
block rw command 8 is failed!
mmc_send_if_cond, Err cnt:0
cdh:test_part_dos DOS_PART_MAGIC_OFFSET ok!
reading ppsmmctool.txt
102 bytes read in 1 ms (99.6 KiB/s)
cmdBuf:fatload mmc 0 0x81c08000 env;env import 81C08000;saveenv
reading env
131 bytes read in 0 ms
## Warning: defaulting to text format
## Info: input data size = 130 = 0x82

Saving Environment to SPI Flash...
Writing to SPI flash...Env save done OK
size:131
error: Pack header size error!
error: upgrade.bin unpack error!
reading env
131 bytes read in 0 ms
## Warning: defaulting to text format
## Info: input data size = 130 = 0x82

Saving Environment to SPI Flash...
Writing to SPI flash...Env save done OK
readLen:131
read err
resetting ...
heartbeat = 1
m

U-Boot 2013.10.0-AK_V3.0.07 (Sep 19 2020 - 14:46:53)

DRAM:  64 MiB
8 MiB
ANYKA SDHC/MMC4.0: 0
PPS:Sep 19 2020 14:46:58   anyka_c3:  0 
                                        magic err
magic err
## Booting kernel from Legacy Image at 81c08000 ...
   Image Name:   Linux-3.4.35
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    2789360 Bytes = 2.7 MiB
   Load Address: 81c08000
   Entry Point:  81c08040
   Verifying Checksum ... OK
   XIP Kernel Image ... OK

Starting kernel ...

Uncompressing Linux... done, booting the kernel.
Anyka Linux Kernel Version: 1.1.12
```

parkerlreed, 2021-02-10:
Oh or not

error: Pack header size error!
error: upgrade.bin unpack error!

parkerlreed, 2021-02-10:
Oh I have extra folders!

parkerlreed, 2021-02-10:
[image]

guino, 2021-02-10:
so the hack worked!

parkerlreed, 2021-02-10:
I assume I just need to set my WiFi and stuff now in the root of the SD card files?

guino, 2021-02-10:
post your ppsapp file and I can take a look at it

parkerlreed, 2021-02-10:
[ppsapp.zip](https://github.com/guino/BazzDoorbell/files/5955773/ppsapp.zip)

guino, 2021-02-10:
the hack is applied and working (good) please post a zip with your env+ppsMmcTool.txt file (so I can post it later for others too).

parkerlreed, 2021-02-10:
[A8D6-2942.zip](https://github.com/guino/BazzDoorbell/files/5955775/A8D6-2942.zip)

guino, 2021-02-10:
Thanks! I just wanted to make sure I wouldn't screw something up by editing the files manually myself (and having no way to test it). I am looking at ppsapp in ghidra right now, give me a few mins and I'll let you know what I find.

parkerlreed, 2021-02-10:
To set the WiFi network do I just use that first ppsFactory that is mentioned? Or is there another thing I can use since the hack is applied?

guino, 2021-02-10:
you can try the ppsFactoryTool.txt but at this point I have no Idea if it will work cause this code seems different than all others I have seen so far. There's no harm in trying, download the file to the SD , define ssid and passowrd and insert it (you don't even have to reboot but can if you want to), then see if port 80 is open

[71 more replies]