guino/BazzDoorbell #22: Avidsen Homecam 360

myrulezzz, 2021-02-17:
Hi. I recently bought this camera https://www.avidsen.com/?view=product&lang=en_US&product_id=630. Which is basically a smartlife(tuya) camera running firmware 2.9.1 Port 80 is open but when i try to http i get prompt to login with username and password which i was not able to find. Is there any other way that i might be able to flash it or do i need the kernel from your step 1?

guino, 2021-02-17:
@myrulezzz try user admin and password 056565099 -- such as this link (replace IP with the local network IP of the camera): http://admin:056565099@IP/devices/deviceinfo -- if it shows the firmware information then #13 probably will work for you (based on the version).

guino, 2021-02-17:
@myrulezzz Additionally if the user **admin** and password **056565099** doesn't work (keeps asking for the prompt), please post a screenshot of the popup, but you may be able to use #11 to read the flash of the camera and then (if you post that) I may be able to determine the user/password so you can then use #13 on it. Without the kernel command line there's no easy way to be sure the process will work without causing problems for your device (specially being a different hardware like yours).

myrulezzz, 2021-02-17:
Hi.
I was able to get this with #13:

#!/bin/sh

export PATH=/usr/bin:/sbin/:/usr/sbin:/bin

RED=" [1;31m"
NORMAL=" [0;39m"

echo "${GREEN} 2015 PPStrong Tech Cop.Ltd.${NORMAL}"

mkdir -p /opt/pps
MTDNUM=`cat /proc/cmdline | sed 's/.*ppsAppParts=\([0-9]\).*/\1/'`

# debug
MTDNUM=5

case $MTDNUM in
         5)
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
			break
			;;
		 7)
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
		 	break
		 	;;
		 0)
			sleep 10
			mount -t vfat /dev/mmcblk0p1 /opt/pps
			break
			;;
		 *)
			MTDNUM=5
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
		 	;;
esac

echo "/opt/pps/" > /tmp/PPStrong.runpath
[ -e /opt/pps/initrun.sh ] && cp /opt/pps/initrun.sh /tmp/PPStart &&
chmod +x /tmp/PPStart && /tmp/PPStart

Does it make any sense?

guino, 2021-02-17:
@myrulezzz that response looks right -- #13 is the method that will work on your device (#11 likely can be used to backup your firmware if you wish to do so). I would expect you should also be able to get http://admin:056565099@IP/proc/cmdline for it too (replacing the IP as you did to get the response you posted).

myrulezzz, 2021-02-17:
I was able to do everything but when i
http://admin:056565099@192.168.x.x/proc/self/root/mnt/mmc01/hack i don’t
get any response at all.

guino, 2021-02-17:
@myrulezzz you may have to wait 1-2 mins after boot and retry http://admin:056565099@192.168.x.x/proc/self/root/mnt/mmc01/hack -- in any case, if everything else worked, you should remove the card and see if you got a /home/app/ppsapp in the SD card -- if so it means the hack worked and camera is rooted, just need to continue with the steps.

myrulezzz, 2021-02-17:
Wanger i was not able to get the folder in the sd card. Here is my device
info file. From what i understood i was able to proceed with step after 1
and go to step 6 because i used your file. But now i am removing the sd
card and i am trying to run the kernel url to get my kernel but i don't get
something new.
Here is my device info results if they are helping:
{"devname":"Smart Home Camera","model":"Speed
4S","serialno":"059011004","softwareversion":"2.9.1","hardwareversion":"S4S_H1_V10_F23","firmwareversion":"ppstrong-c51-tuya2_shf-2.9.1.20190907","authkey":"iZTU1cXIO6s15Gb0us5q1qR9xQgrYIe6","deviceid":"pp014a036fb2cc9ca6e1","identity":"MR1908211100917570","pid":"aaa","WiFi
MAC":"c4:3a:35:0c:9c:8a"}

myrulezzz, 2021-02-17:
And something else my camera has the debug entry uncomment

guino, 2021-02-17:
@myrulezzz I knew your camera was going to have the MTDNUM=5 uncommented because of the version (2.9.1). I would lime you to do this:

1- post the results of http://admin:056565099@192.168.x.x/proc/cmdline (using the IP of the camera)

2- make sure the initrun.sh file is in the SD card (if not unzip it from #13) . Then insert the SD card while the camera is powered off, then power the camera on normally. Wait at least 2 minutes after the boot sound, then power it off.

3- check (on the computer) if your sd card has a folder/directory called **home** - if so check if you have a ppsapp file under home/app/ppsapp - if so, zip it and post it here so it can be patched.

Based on the response from 1 and if you had anything on step 3 I may ask you to do one thing or another, but I can guide you thru it.

From the device information you provided you should be able to enable rtsp (even if I have to help a little bit).

myrulezzz, 2021-02-17:
This is my information from step 1:

mem=37M console=ttyAMA0,115200n8
mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4224k(app),448k(cfg)
ppsAppParts=5 ppsWatchInitEnd

myrulezzz, 2021-02-17:
I have copied initrun.sh on the sd card followed your instructions but i am
not able to get folder home on it.

guino, 2021-02-17:
@myrulezzz ok, so the hack is not installed (steps 2 and 3 are irrelevant now).

Please do this: 
1-Format your SD card -- make sure it is formatted as FAT32 -- if you don't see an option for FAT32 you may want to check this page: https://www.howtogeek.com/316977/how-to-format-usb-drives-larger-than-32gb-with-fat32-on-windows/

2-Unzip this [myrulezzz.zip](https://github.com/guino/BazzDoorbell/files/5999207/myrulezzz.zip) into the SD card -- it should only have 3 files (env, initrun.sh and ppsMmcTool.txt)

3-Eject the SD card (properly) using the windows system tray icons -- to prevent corrupting data on it

4-Power off camera, insert SD card, HOLD the reset button and power it on **while holding** the reset button for 5 seconds. After 5 seconds, let go of the reset button, wait for the camera to make the boot sound, then wait 2 minutes and check http://admin:056565099@192.168.x.x/proc/cmdline -- post the response.

myrulezzz, 2021-02-17:
Wagner,
I have followed your steps except that the camera does not have any button
so i switched off the camera inserted the sd card with the files after
formatting the sd card to fat 32, i know that the micro sd is ok because
the camera is creating the folder structure to save videos and pictures.

myrulezzz, 2021-02-17:
The result is the same:

mem=37M console=ttyAMA0,115200n8
mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4224k(app),448k(cfg)
ppsAppParts=5 ppsWatchInitEnd

guino, 2021-02-17:
The camera absolutely has a reset button - it may be called setup and may even be just a tiny hole that you have to push into with a paper clip. Worst case scenario you may have to open it to locate the button but it has to be there.

If you don’t push the button while powering on then all effort is useless.

I did not see a manual for that camera or I would have looked it up.

myrulezzz, 2021-02-17:
There is a paper clip i will try to find the reset button

guino, 2021-02-18:
I am expecting it to be something like this:
[image]

myrulezzz, 2021-02-18:
I am pressing the reset button for 5 seconds and power on the camera the
same time. but the url is still the same. Is that possible?

guino, 2021-02-18:
It is possible but unlikely - do you feel a button being pressed when pushing into it at all? I have seen only one other user (different device) which could not get the hack installed - it could be the SD card format or a problem with the actual hardware. One user reported that it only worked after partitioning and formatting the sd card in linux.

myrulezzz, 2021-02-18:
Yes I feel the button. I don't think it is hardware because as from serial
number it seems that this camera was produced in 2019. I have tried the
whole process from mac os. I will try tomorrow from linux and also. try a
different sd card ad i will let you know. Thanks

guino, 2021-02-18:
@myrulezzz this being a different hardware it may have a different load address which is the case for older hardware on 2.7.x and also #21. The problem is figuring out the address which right now requires opening and connecting to the serial port.

myrulezzz, 2021-03-01:
Hi Wagner i was trying redo the the steps provided for the camera but still
not able to get the url. what do you mean opening and connecting to serial
port?

guino, 2021-03-01:
@myrulezzz It means litteraly opening the device (physically) and connecting (possibly soldering) a TTL-Serial adapter to the Serial port pins on the device. This is the kind of thing you should only do if you have some experience with soldering/electronics (and you would need to have/buy a serial-ttl adapter for 3.3v).

myrulezzz, 2021-03-01:
I have the serial ttl adapter but the board where is located? Inside the
len cover?

guino, 2021-03-01:
@myrulezzz I don/t have a device like yours so the only way to figure out where is the uart/board is for **someone** to open the device (wherever it may open), disassemble it and take pictures of the boards and post them so I can tell you. You always have the option to wait for someone else (with more experience) to get a similar camera so it can be taken appart, but if you're willing (and confident) to do the manual work I can help you like I helped in #21 with a different camera.

myrulezzz, 2021-03-01:
Hi Wagner i was able to get this url after retrying the files:

mem=37M console=ttyAMA0,115200n8
mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4224k(app),448k(cfg)
ppsAppParts=5 ppsWatchInitEnd -
ip=${T//_/$'\\x20'}:::::;T=\"sleep_5;mkdir_-p_/mnt/mmc01;mount_-t_vfat_/dev/mmcblk0p1_/mnt/mmc01;/mnt/mmc01/initrun.sh&\";eval
mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4224k(app),448k(cfg)
ppsAppParts=5 ppsWatchInitEnd

it looks better now?

guino, 2021-03-01:
@myrulezzz That looks correct -- you should check if you have ppsapp under /home/app/ppsapp in the SD card and if so we should be able to patch it.

myrulezzz, 2021-03-01:
> Here is the file
>
> On Mon, Mar 1, 2021 at 5:56 PM Andreas Stylianou <stylianouandr@gmail.com>

here is the file

guino, 2021-03-01:
@myrulezzz you have to go into github and post a ZIP of the file -- just replying an email with the file attached does not attach it to github and you'be been replying into the thread's email. You can alternatively email me the directly (my email is on my github profile).

myrulezzz, 2021-03-01:
[home.zip](https://github.com/guino/BazzDoorbell/files/6062952/home.zip)
Is that ok?

guino, 2021-03-01:
@myrulezzz that looks good, I will post the patch soon.

guino, 2021-03-01:
@myrulezzz Here's the patch: https://github.com/guino/ppsapp-rtsp/issues/1#issuecomment-788099471 Please note this is the first 2.9.1 firmware I have seen and people with 2.9.0 have had to do extra work to get the RTSP functional. I suggest you try snap/mjpeg first and once you verify that works, then try patching ppsapp -- if for some reason you can't view the RTSP video please try viewing the video in the APP (tuya/smartlife/etc) -- if it doesn't work there either then you may have to the extra steps for 2.9.0 posted here: https://github.com/DanTLehman/orion_sc008ha#how-did-i-get-it-working

myrulezzz, 2021-03-01:
Ok i have done the patching as per the guide how to patch rtsp.added the file in the root folder of the sd card. I havent heard the beap twice when switching on the camera. What do you mean to try snap/mpeg?
From smartlife app i am able to view the camera

guino, 2021-03-01:
@myrulezzz

(From the original instructions)
"For mjpeg/snap.cgi support you have to adjust the address for JPEG address in snap.cgi and mjpeg.cgi as posted with your ppsapp patch (or found with ghidra) -- this should work even if you don't patch ppsapp (on step below). The URL should be like this: http://user:password@IP:8080/cgi-bin/snap.cgi (with user:password from httpd.conf)" 
(mjpeg would be:  http://user:password@IP:8080/cgi-bin/mjpeg.cgi )

After booting up you should wait 1-2 minutes and try RTSP on VLC with `rtsp://IP:8554` to see if it works, if it does you should be good to go. If it doesn't: can you show me the results from http://user:password@IP:8080/ (where user:password is whatever is in httpd.conf -- which is user:password by default).

myrulezzz, 2021-03-01:
so you mean to download the mmc folder and adjust the files with username password?

guino, 2021-03-01:
The instructions have steps to place the mmc files in the SD card, so if you followed the instructions you should have cgi-bin with the .cgi files in it as well as httpd.conf, busybox, custom.sh and initrun.sh in the SD card. For mjpeg and snap you need to set the address (posted with the patch) in snap.cgi and mjpeg.cgi.

at a bare minimum you need to have initrun.sh custom.sh and the patched ppsapp in the sd card so you can try RTSP. 

If you have doubt on the files you should just zip everything from the SD card (except for SDT and home folder) and post it here so I can review.

myrulezzz, 2021-03-01:
[sd card.zip](https://github.com/guino/BazzDoorbell/files/6063475/sd.card.zip)
Hi Wagner can you please check the attach zip file?  This is everything on my sd card.

guino, 2021-03-01:
Your ppsapp/initrun.sh and custom.sh look ok, but your busybox file is wrong, please do this:
1. Download busybox correctly as stated in Step 9 of #2 -- this is the link: https://github.com/guino/BazzDoorbell/blob/master/mmc/busybox?raw=true
2. Open cgi-bin/snap.cgi in a text editor and set the address line to: `ADDRESS=48d5e4`
3. Open cgi-bin/mjpeg.cgi in a text editor and set the address line to: `ADDRESS=48d5e4`
4. Open initrun.sh in a text editor and change the line that says `sleep 10` to be `sleep 60`
5. properly eject the card from computer, insert it into device while powered off and let it boot up -- it SHOULD do the startup sound 2 times (you may have to wait a couple of minutes for it to happen) -- please wait at least 5 minutes after power on to see if it does it or not.
7. Try to open on a web browser: http://admin:mypassword@IP:8080/cgi-bin/snap.cgi and http://admin:mypassword@IP:8080/cgi-bin/mjpeg.cgi -- just replace the right IP for the camera.
8. Try in VLC or any phone RTSP app to open RTSP with: rtsp://IP:8554 (just replace the right IP for the camera).

guino, 2021-03-01:
@myrulezzz I just noticed your initrun.sh is wrong too -- you need to download the one from here: https://github.com/guino/BazzDoorbell/files/5792854/ppshack2.zip

myrulezzz, 2021-03-01:
>    6. Try to open on a web browser:
>    7. Try in VLC or any phone RTSP app to open RTSP with: rtsp://IP:8554

Hi Wagner, this worked.Thanks for the support.
Are there any customisations that can be done to the camera?

guino, 2021-03-01:
@myrulezzz RTSP, snap and mjoeg are customizations. You should have telnet enabled too but your passwd file has a space that should not be in there so likely your telnet login won't work until you fix that. You can set the address in play.cgi to play wav files on it if you like. If you want to trigger notifications for motion and button push there's a script you can get from the off-cloud issue (which you likely have to customize). I am not aware of any other customization at this time but you can edit some settings in /home/cfg/tuya_config.json (using telnet) to modify settings that your app may not have as options like type of recording (off/motion/always) and little things like turn off the LED light.

myrulezzz, 2021-03-01:
Ok and this should work with ip cameras software?
Do you have anything in mind?

guino, 2021-03-01:
I use this app myself (on IOS): http://ipcamsoft.com/ but I also use it in homebridge and domoticz. There are lots of things that support RTSP.

myrulezzz, 2021-03-01:
Which plugin are you using in homebridge?

guino, 2021-03-01:
[image]

myrulezzz, 2021-03-02:
Hi Wagner,
Is there any option to utilise streaming resolution, i think after the hack
the resolution is not good at all.

guino, 2021-03-02:
The resolution for snap/mjpeg is lower than RTSP as that is intended for preview and notifications. The RTSP resolution should be whatever is your camera’s resolution. Obviously viewing the stream on a computer with a big screen will look a lot worse than viewing it in a phone with a small screen. In VLC (and some apps) you can check the resolution of the RTSP stream by clicking on the Tools menu and selecting Media/Codec information - it should show your cameras resolution. You don’t have to use snap/mjpeg if you don’t like the resolution - there’s no way to change the resolution on that feature.