guino/BazzDoorbell #23: Integration with Domoticz

gielie, 2021-03-02:
I don’t know if I can ask the following question here?

I have the LCS version and it works perfect with all the tools placed here. 
One thing I’d like to add is an integration with Domoticz. 
If someone push the button or if there is motion detected I’d like to get an signal in my Domoticz, is that even possible?

I created several sounds I can activate via Domoticz and I’d like to activate them by motion or by pushing the button.

guino, 2021-03-02:
I have domoticz and have my cameras configured in it (as cameras). You can get motion/button notifications sent to anything you want if you use the log_parser posted in the off-cloud issue #4 . I have that configured but only to trigger events in homebridge right now -- I don't see why it couldn't be used to trigger events in domoticz.

Here's my log_parser (mostly a copy of what's in the off-cloud issue):
```
#!/bin/sh

# contains(string, substring)
#
# Returns 0 if the specified string contains the specified substring,
# otherwise returns 1.
contains() {
    string="$1"
    substring="$2"
    if test "${string#*$substring}" != "$string"
    then
        return 0    # $substring is in $string
    else
        return 1    # $substring is not in $string
    fi
}

main() {
    IFS='$\n'
    while true; do
        read -r BUF;
        if [ $? -ne 0 ]; then
            sleep 1;
            continue
        fi
        if contains "$BUF" "--motion detection alarm --"; then
            /mnt/mmc01/busybox wget -O- http://10.10.10.87:8080/motion?Doorbell
        elif contains "$BUF" "##doorbell_push 3"; then
            /mnt/mmc01/busybox wget -O- http://10.10.10.87:8080/doorbell?Doorbell
        fi
    done
}

main
```

If you can activate one of your sounds with a http request it should be simple to just modify it above. 
You also have to adjust custom.sh so the log_parser is used, here's my custom.sh:

```
#!/bin/sh
if [ ! -e /tmp/customrun ]; then
 echo custom > /tmp/customrun
 cp /mnt/mmc01/passwd /etc/passwd
 /mnt/mmc01/busybox telnetd
 /mnt/mmc01/busybox httpd -c /mnt/mmc01/httpd.conf -h /mnt/mmc01 -p 8080
 if [ -e /mnt/mmc01/ppsapp ]; then
  PPSID=$(ps | grep -v grep | grep ppsapp | awk '{print $1}')
  kill $PPSID
  /mnt/mmc01/ppsapp 2>&1 | /mnt/mmc01/log_parser.sh &
 fi
fi
```

gielie, 2021-03-03:
i think i have the wrong busybox version.
[code]
/ # busybox
BusyBox v1.26.2 (2019-11-03 17:33:40 PST) multi-call binary.
BusyBox is copyrighted by many authors between 1998-2015.
Licensed under GPLv2. See source distribution for detailed
copyright notices.
[/code]

where can i find the one wit wget in it? I used the from point 9

guino, 2021-03-03:
@gielie if you just execute `busybox` you will execute the internal/flash busybox. Assuming you downloaded the busybox with the link from step 9 , you have to execute busybox with the path to it in the SD card like shown in the scripts above: `/mnt/mmc01/busybox wget http://IP/path/`

gielie, 2021-03-03:
@Guino
I see, I tried your code to start with in my homebridge but I can't get it to work right away. Do I need something else configured in the Homebridge Camera FFmpeg?

Do i understand correctly that the ip starting with 10. is your ip of the homebridge for the http based automation?

could you give me a starting point if I want to use a code like this for Domoticz
http://IP:port/json.htm?type=command&param=switchlight&idx=156&switchcmd=On

guino, 2021-03-03:
@gielie you should be able to just change the wget lines in my script to this: (and you can try the command in telnet to verify it works):
```
/mnt/mmc01/busybox wget -O- "http://IP:port/json.htm?type=command&param=switchlight&idx=156&switchcmd=On"
```
Please notice the quotes must be used since your URL has **&** characters in it (otherwise no need for quotes like on my script).

There's nothing special about the homebridge plugin settings but here's a screenshot of my settings: https://github.com/guino/BazzDoorbell/issues/22#issuecomment-788253245 -- on my script 10.10.10.87 is the IP of my homebridge and 8080 is the port I configured in the plugin (as shown in the screenshot).

The only other important thing to notice is that you must have the **ppsapp** file in the SD card, so if you're running 2.9.7 without a patched ppsapp you'll need to copy ppsapp to the root of the SD card (even if you don't patch it) -- this is so the custom script can detect it, kill the ppsapp and run a new instance of it with the output redirected into log_parser.

gielie, 2021-03-06:
@guino this works like a charm. Now i can have a switch in Domoticz for motion detection and doorbell. So what I did is; when there is nobody home and the doorbell rings a command is executed with a frase "there is nobody home" and one kinda freaky one, between 2000 and 2200 when motion is activated a frase "I can see you" very nice.

i was wondering if it is a good idea to create a wiki, im looking for something I can't find anymore.
I saw a list of http commands to change settings but I can't find it anymore. If there were a wiki it would be easier to find something, I could take care of the Domoticz integration if you'd like.

guino, 2021-03-06:
@gielie I would advise against letting strangers knowing that nobody is home (even letting them know you have cameras) but that's just me.

I am all for creating wiki, sites, etc with helpful content in regards to this and anything else. I just don't have the time to take on any more projects (even just maintaining information somewhere) but I would be willing to contribute to it when I can.