guino/BazzDoorbell #24: Chacon FI02

SamThing, 2021-03-15:
Hi. I recently bought this camera https://chacon.com/pt/camaras-ip/1238-camara-ip-wi-fi-interior-mini-1080p-5411478345268.html. Which is basically a smartlife(tuya) camera running firmware 2.9.5 Port 80 is open but when i try to http i get prompt to login with username and password which i was not able to find. I was able to gain access with #13
Now I have telnet and the hack file is done.

But I'm still getting prompted to login with username and password. (already tried all possible combinations, the serial number and etc)
Also, snap and mjpeg don't work (tiny blank square image)

Some informations about the cam (I was able to get them running /tmp/PPStart ):
name: Smart Home Camera
factory: PPSTRONG
device_type: 74
model: Mini 8S
hardware_version: M8S_H1_V10_F23
software_version: 2.9.5
firmware_version: ppstrong-c51-s_chacon-2.9.5.20200205
sn: 059554188

I got UART to work (readonly it seems) and the hardware aspects are very similar to this one https://github.com/guino/BazzDoorbell/issues/2#issuecomment-774789777

Great work you've done!
Thanks for your time and effort!

SamThing, 2021-03-15:
Some things I noted:

* Even without the SD card the camera boots and works fine
* The default hack proccess didn't work for me, UART gave me magic err but got it working with #13 even with ppsAppParts=5
```
cat /proc/cmdline

mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd - ip=${T//_/$'\\x20'}:::::;T=\"sleep_5;mkdir_-p_/mnt/mmc01;mount_-t_vfat_/dev/mmcblk0p1_/mnt/mmc01;/mnt/mmc01/initrun.sh&\";eval mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4224k(app),448k(cfg) ppsAppParts=5 ppsWatchInitEnd
```

SamThing, 2021-03-15:
```
binwalk flash.bin

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
18304         0x4780          gzip compressed data, has original file name: "u-boot.bin", from Unix, last modified: 2019-07-22 08:33:20
393216        0x60000         uImage header, header size: 64 bytes, header CRC: 0xB176B747, created: 2019-07-22 07:52:05, image size: 3205538 bytes, Data Address: 0x40008000, Entry Point: 0x40008000, data CRC: 0x8846780B, OS: Linux, CPU: ARM, image type: OS Kernel Image, compression type: none, image name: "Linux-4.9.37"
393280        0x60040         Linux kernel ARM boot executable zImage (little-endian)
395744        0x609E0         device tree image (dtb)
409272        0x63EB8         device tree image (dtb)
415228        0x655FC         device tree image (dtb)
419444        0x66674         gzip compressed data, maximum compression, from Unix, last modified: 1970-01-01 00:00:00 (null date)
3584680       0x36B2A8        device tree image (dtb)
3604480       0x370000        CramFS filesystem, little endian, size: 4136960, version 2, sorted_dirs, CRC 0x54D40DD5, edition 1, 1010 blocks, 3 files
7929856       0x790000        JFFS2 filesystem, little endian
8192248       0x7D00F8        Zlib compressed data, compressed
8192496       0x7D01F0        Zlib compressed data, compressed
8194704       0x7D0A90        Zlib compressed data, compressed
8194952       0x7D0B88        Zlib compressed data, compressed
8198124       0x7D17EC        Zlib compressed data, compressed
8198372       0x7D18E4        Zlib compressed data, compressed
8200640       0x7D21C0        Zlib compressed data, compressed
8200888       0x7D22B8        Zlib compressed data, compressed
8203096       0x7D2B58        Zlib compressed data, compressed
8203344       0x7D2C50        Zlib compressed data, compressed
8205552       0x7D34F0        Zlib compressed data, compressed
8205800       0x7D35E8        Zlib compressed data, compressed
8208652       0x7D410C        Zlib compressed data, compressed
8208900       0x7D4204        Zlib compressed data, compressed
8210932       0x7D49F4        Zlib compressed data, compressed
8211180       0x7D4AEC        Zlib compressed data, compressed
8213388       0x7D538C        Zlib compressed data, compressed
8213636       0x7D5484        Zlib compressed data, compressed
8217552       0x7D63D0        Zlib compressed data, compressed
8217800       0x7D64C8        Zlib compressed data, compressed
8221032       0x7D7168        Zlib compressed data, compressed
8221280       0x7D7260        Zlib compressed data, compressed
8225488       0x7D82D0        Zlib compressed data, compressed
8225736       0x7D83C8        Zlib compressed data, compressed
8226992       0x7D88B0        Zlib compressed data, compressed
8227240       0x7D89A8        Zlib compressed data, compressed
8229276       0x7D919C        Zlib compressed data, compressed
8229524       0x7D9294        Zlib compressed data, compressed
8231444       0x7D9A14        Zlib compressed data, compressed
8231692       0x7D9B0C        Zlib compressed data, compressed
8233612       0x7DA28C        Zlib compressed data, compressed
8233860       0x7DA384        Zlib compressed data, compressed
8235780       0x7DAB04        Zlib compressed data, compressed
8236028       0x7DABFC        Zlib compressed data, compressed
8237284       0x7DB0E4        Zlib compressed data, compressed
8237532       0x7DB1DC        Zlib compressed data, compressed
8239568       0x7DB9D0        Zlib compressed data, compressed
8239816       0x7DBAC8        Zlib compressed data, compressed
8242176       0x7DC400        Zlib compressed data, compressed
8242424       0x7DC4F8        Zlib compressed data, compressed
8244340       0x7DCC74        Zlib compressed data, compressed
8244588       0x7DCD6C        Zlib compressed data, compressed
8246504       0x7DD4E8        Zlib compressed data, compressed
8246752       0x7DD5E0        Zlib compressed data, compressed
8248004       0x7DDAC4        Zlib compressed data, compressed
8248252       0x7DDBBC        Zlib compressed data, compressed
8250284       0x7DE3AC        Zlib compressed data, compressed
8250532       0x7DE4A4        Zlib compressed data, compressed
8252448       0x7DEC20        Zlib compressed data, compressed
8252696       0x7DED18        Zlib compressed data, compressed
8254612       0x7DF494        Zlib compressed data, compressed
8254860       0x7DF58C        Zlib compressed data, compressed
8257096       0x7DFE48        Zlib compressed data, compressed
8257344       0x7DFF40        Zlib compressed data, compressed
8257732       0x7E00C4        Zlib compressed data, compressed
[... 137 lines cut ...]
8363128       0x7F9C78        Zlib compressed data, compressed
8363376       0x7F9D70        Zlib compressed data, compressed
8363624       0x7F9E68        JFFS2 filesystem, little endian
8365028       0x7FA3E4        Zlib compressed data, compressed
8365280       0x7FA4E0        Zlib compressed data, compressed
8366240       0x7FA8A0        Zlib compressed data, compressed
8366492       0x7FA99C        Zlib compressed data, compressed
8367452       0x7FAD5C        Zlib compressed data, compressed
8367704       0x7FAE58        Zlib compressed data, compressed
8368660       0x7FB214        Zlib compressed data, compressed
8368908       0x7FB30C        Zlib compressed data, compressed
8369868       0x7FB6CC        Zlib compressed data, compressed
[... cut ...]
8372428       0x7FC0CC        Zlib compressed data, compressed
8372988       0x7FC2FC        JFFS2 filesystem, little endian
8373640       0x7FC588        Zlib compressed data, compressed
8374200       0x7FC7B8        JFFS2 filesystem, little endian
8374852       0x7FCA44        Zlib compressed data, compressed
8375412       0x7FCC74        JFFS2 filesystem, little endian
8376064       0x7FCF00        Zlib compressed data, compressed
8376624       0x7FD130        JFFS2 filesystem, little endian
```

guino, 2021-03-15:
@SamThing since you used #13 the camera can boot without a SD card (just keep in mind that when booting without SD card it will run standard/stock firmware).

For snap/mjpeg to work you need to adjust the jpeg address in the snap.cgi and mjpeg.cgi files - this must be done based on your ppsapp file (which you did not post). Instructions on how to do that are posted in https://github.com/guino/ppsapp-rtsp which also contains instructions on patching ppsapp to enable RTSP.

If you got Telnet access you should have a copy of your ppsapp under /home/app/ppsapp of the SD card. You can either patch it (and find jpeg address) with the instructions on the link above OR post a zip of your ppsapp so I can make a patch for it (and post it).

What I am not exactly clear is what are your trying to get with “http access” - some cameras don’t have that enabled at all, the ones that have that enabled don’t provide you anything other than some information. And in many cameras you cam enable http access with ppsFactoryTool.txt but there’s not “web view” of the camera if that’s what you want. The only thing close to a “web view” is the snap.cgi and mjpeg.cgi links which are added by the hack on port 8080 (not port 80 of the firmware). In any case all tuy cameras that I have seen use the same user/password for http links: admin:056565099 for instance http://admin:056565099@IP/proc/cmdline — if it doesn’t work for you I suggest trying a different web browser.

In any case you will gain nothing from the http links that you don’t already have with Telnet so you really just need to get your ppsapp patched for RTSP and set the jpeg address in snap/mjpeg and you will have RTSP, snapshot and mjpeg features enabled on your camera.

SamThing, 2021-03-15:
Hi @guino , thanks for you quick reply

I'll try to get the address using the instructions you provided. I didn't see that main repo page, only [this one issue](https://github.com/guino/ppsapp-rtsp/issues/1)

Regarding the http access, I mean exactly the built in web server you mentioned (is enabled by default).
The 8080 port hack is working like a charm.

The endpoint http://admin:056565099@ip/proc/cmdline (or anyother on the port 80) doesn't work, it keeps asking me user/pass. Same thing on differents browsers.
Not sure if it's a tuya camera tho. The firmware_version string contains this "chacon" word.
Already tried using the serial number as pass with no luck.

I just need the RSTP stream to be accessible from my PI or something on the network, so patching should do the trick right?

SamThing, 2021-03-15:
[ppsapp.txt](https://github.com/guino/BazzDoorbell/files/6141519/ppsapp.txt) 
Here's my ppsapp file (without the txt extension).
I'll try to patch it later

guino, 2021-03-15:
@SamThing I should be able to patch it and get the jpeg address in a little bit assuming the file is good (haven’t checked yet).

SamThing, 2021-03-15:
Np, but I want to learn something of the proccess too.
I'm a enthusiast of embedded systems and stuff.

Seems I need help anyway since I couldn't find the *ipc_ring* occurrence on my file 😄 
I was not able to follow the LAB_0xxxxx label as it doesn't exists

[image]

[image]

Sorry for the screenshot quality, I'm running ghidra on headless remote machine (ssh -X)

guino, 2021-03-15:
Since you're trying to learn: the LAB_0xxxx is a reference for the main function on newer ppsapp files -- in your older ppsapp the symbol information wasn't removed so it has the 'original' name which is 'main' -- this actually makes it easier to find the main sub since you just have to double click the 'main' word.

SamThing, 2021-03-15:
Cool!

The main function seems a little different that you showed.
Seems like we can call the file passing arguments to disable the watchdog for example.

```
undefined4 main(int param_1,int param_2)

{
  __pid_t _Var1;
  int iVar2;
  int iVar3;
  int iVar4;
  char *__s1;
  code *pcStack200;
  code *pcStack196;
  undefined4 local_c0;
  undefined4 local_b8;
  char acStack172 [128];
  int local_2c;
  
  local_2c = __stack_chk_guard;
  _Var1 = getpid();
  sprintf(acStack172,"/proc/%d/oom_score_adj",_Var1);
  iVar2 = getrlimit64(RLIMIT_STACK,(rlimit64 *)&local_c0);
  if (iVar2 != 0) {
    puts("getrlimit failed!");
LAB_0004e864:
    if (local_2c == __stack_chk_guard) {
      return 0xffffffff;
    }
                    /* WARNING: Subroutine does not return */
    __stack_chk_fail();
  }
  printf("ulimit stack soft: %d hard: %d\n",local_c0,local_b8);
  setenv("WATCHDOG","1",0);
  puts("\n");
  puts("=============================================");
  puts("=============================================");
  puts("\n");
  pcStack200 = pps_malloc2;
  pcStack196 = pps_free;
  pps_cJSON_InitHooks(&pcStack200);
  puts("aaaaaaaaaaaaaaa slb init bbbbbbbbbbbbbbbbbbbbbb");
  puts("aaaaaaaaaaaaaaa slb init bbbbbbbbbbbbbbbbbbbbbb");
  puts("aaaaaaaaaaaaaaa slb init bbbbbbbbbbbbbbbbbbbbbb");
  puts("aaaaaaaaaaaaaaa slb init bbbbbbbbbbbbbbbbbbbbbb");
  pps_mem_global_init(&DAT_002c0000);
  if (1 < param_1) {
    iVar2 = 1;
LAB_0004e7bc:
    do {
      iVar3 = strcmp(*(char **)(param_2 + 4),"-h");
      if (iVar3 == 0) {
        puts(
            " <cmd> -option=value ...\n   -h               help information\n   -d              daemon mode\n   -vcodec=         set video codec: only [h264] support\n   -fps=           set video fps: [1~30]\n   -bps=            set video bps: [>1024]\n   -gop=           set video gop: [>0]\n   -vbr             set video bps ctrl type: variable\n   -cbr            set video bps ctrl type: constant\n   -acodec=         set audio codec: [aacor mpg2], default=mpg2\n   -bitrate=        set audio bitrate:8000/16000/32000/64000/128000/192000\n   -sample=         set audio sample rate:8000/16000/32000/z44100/48000\n   -bitwidth=       set audio bitwidth:  8/16/32,default:16bit\n   -channels=       set audio channels:  1/2, default: 1\n   -print_fps      print audio/video fps"
            );
        goto LAB_0004e864;
      }
      __s1 = *(char **)(param_2 + iVar2 * 4);
      iVar4 = strcmp(__s1,"-d");
      iVar3 = iVar2 * 4;
      if (iVar4 != 0) {
        iVar4 = strcmp(__s1,"--watchdog=no");
        if (iVar4 == 0) {
          iVar2 = iVar2 + 1;
          unsetenv("WATCHDOG");
          if (param_1 <= iVar2) break;
          goto LAB_0004e7bc;
        }
        iVar4 = strcmp(__s1,"--watchdog=yes");
        if (iVar4 == 0) {
          setenv("WATCHDOG","1",0);
        }
        else {
          iVar4 = strcmp(__s1,"-sensor");
          if (iVar4 == 0) {
            iVar2 = iVar2 + 1;
            if (param_1 <= iVar2) break;
            setenv("SENSOR_TYPE",*(char **)(param_2 + iVar3 + 4),0);
          }
        }
      }
      iVar2 = iVar2 + 1;
    } while (iVar2 < param_1);
  }
  register_signal_handlers();
  FUN_0004eae0(&DAT_0029445c,ppsdev_init_plugins_core);
  FUN_0004eae0(&DAT_00294464,ppsdev_init_plugins_user);
  FUN_0004eae0("after",ppsdev_init_plugins_after);
  do {
    sleep(10000);
  } while( true );
}
```

guino, 2021-03-15:
@SamThing This really isn't a 'tuya' device as you said. It is a 'ppstrong' firmware but has no tuya functions. It will take some time to review the code before I know what is available. The rtsp functions we've used so far are from tuya so they're not in this ppsapp. There are however some references to rtsp/onvif in the code so I need to see if we can enable it somehow. The instructions I put together won't be as helpful but the JPEG functions seem to be the same, you can try to use snap/mjpeg by setting this address in snap.cgi/mjpeg.cgi : 035cdc4

guino, 2021-03-15:
@SamThing play.cgi address should be: 035d7e0

SamThing, 2021-03-15:
Awesome!

You rock man!
snap/mjpeg works like a charm, I'll try play.cgi now

guino, 2021-03-15:
@SamThing can you do (linux command) **nmap 192.168.x.x** from your linux machine using the UP of your device to see which ports are open ? it looks like it does have a server on port 8554 -- it may not be started depending on the settings (which we can fix).

It would be VERY helpful in investigating the code if you could do this (using telnet on the device):
```
PPSID=$(ps | grep -v grep | grep ppsapp | awk '{print $1}');   kill $PPSID
/mnt/mmc01/home/app/ppsapp > /mnt/mmc01/ppsapp.log
```
Then open whatever app it came with and start a video feed, then stop it, execute (on telnet) `sync` and take the SD card out and post your log file so it can help me follow the code quicker. You can email it to me if you prefer (my email is on my github profile).

guino, 2021-03-15:
@SamThing can you also check if there's anything useful in /home/cfg/dev_settings.json of the device ? I don't think it has the password for anything but it's worth checking whatever else is in that directory in the device.

SamThing, 2021-03-15:
@guino 

Good news, the play.cgi works like a charm too

I noted that the RTSP port only opens with the app.
Before trying to hack the camera itself, I started with some wireshark stuff.
The RTSP traffic is UDP and the port only opens after the app sent any unknown command to the chacon servers.
Other thing: if you try to get anything when the port 8554 is open, it causes the camera to crash.
Even running nmap does that.

Before the starting the video feed:
```
nmap 192.168.1.226 -p1-8554
Starting Nmap 7.80 ( https://nmap.org ) at 2021-03-15 15:34 WET
Nmap scan report for 192.168.1.226
Host is up (0.0096s latency).
Not shown: 8549 closed ports
PORT     STATE SERVICE
23/tcp   open  telnet
53/tcp   open  domain
80/tcp   open  http
8080/tcp open  http-proxy

Nmap done: 1 IP address (1 host up) scanned in 2.64 seconds
```

After starting the video feed:
```
nmap 192.168.1.226 -p1-8554
Starting Nmap 7.80 ( https://nmap.org ) at 2021-03-15 15:34 WET
Nmap scan report for 192.168.1.226
Host is up (0.0096s latency).
Not shown: 8549 closed ports
PORT     STATE SERVICE
23/tcp   open  telnet
53/tcp   open  domain
80/tcp   open  http
8080/tcp open  http-proxy
8554/tcp open  rtsp-alt

Nmap done: 1 IP address (1 host up) scanned in 2.64 seconds
```

After running this nmap, the camera restarts.
I'll upload the log in a minute

guino, 2021-03-15:
@SamThing  it looks like the http server on this device allows some post commands to come in to do different things (possibly open RTSP too?). I can probably make a patch that would take 'any' user/password for the http requests but right now I haven't found the user/password combination. I just know for a fact it isn't admin:admin (which you already knew).

SamThing, 2021-03-15:
[ppsapp.log](https://github.com/guino/BazzDoorbell/files/6142619/ppsapp.log)

Yes, makes sense.
Not sure if the log file will be usefull, I couldn't start the video feed. The ppsapp starts fine, but the android app hangs forever.

Redirecting the output causes anything that prevents the conection to the camera.

guino, 2021-03-15:
@SamThing the fact that redirecting output prevents connection with the camera makes me think this ppsapp is unable to re-open the hardware handles once it is killed (thus it probably restarts after awhile). You may need to apply the process described here instead: https://github.com/DanTLehman/orion_sc008ha#total-success -- you can adjust the S90ppstrong-290 file so the line `/mnt/mmc01/ppsapp-rtsp &` is changed to `/mnt/mmc01/home/app/ppsapp > /mnt/mmc01/ppsapp.log &` so we have the output while the device works correctly. Then once we have a patch you can change it back to just `/mnt/mmc01/ppsapp-rtsp &`.

1-Did you check the files under /home/cfg to see if there was anything with perhaps a reference to user/password information ?
2-While on telnet, can you post a response to the 'mount' command ?

Right now I can see in the code+log where it calls the function that is supposed to start the rtsp server but I have no way of knowing if that's being started. Sometimes nmap doesn't show the port open even if it is, so maybe you want to try (from your linux machine) `telnet IP 8554` to see if it connects at all -- there's a chance it might (even without patching). Then it would just be a matter of finding the password OR disabling the authentication so any user/password works.

guino, 2021-03-15:
@SamThing additional to the above you may want to try http://IP/search (without user/password) to see if it works -- seems like it may provide an 'ONVIF' response on port 80.

SamThing, 2021-03-15:
Cool, the search endpoint works:
```
// 20210315163737
// http://192.168.1.226/search

{
  "deviceName": "059554188",
  "serialno": "059554188",
  "sn": "ppscf12989d3f4854da0",
  "licenseUsed": 1,
  "licenseId": "ppscf12989d3f4854da0",
  "p2p_uuid": "v3-0595541880000111A",
  "factory_code": 0,
  "factory_code_str": "",
  "model": "Mini 8S",
  "tp": "000000000",
  "ip": "192.168.1.226",
  "mask": "255.255.255.0",
  "gw": "192.168.1.1",
  "mac": "ac:64:cf:ea:1d:3e",
  "interface": "wlan0",
  "version": "2.9.5",
  "server_type": "https://apis-eu-frankfurt.meari.com.cn",
  "capability": "{\"ver\":18,\"cat\":\"camera\",\"caps\":{\"vtk\":3,\"fcr\":0,\"dcb\":1,\"md\":1,\"ptz\":0,\"tmpr\":0,\"hmd\":0,\"pir\":0,\"cst\":0,\"geo\":1,\"nst\":1,\"evs\":1,\"vst\":0,\"led\":1,\"cse\":0,\"dnm\":1,\"cs2\":0,\"bps\":226,\"wfs\":0,\"svc\":0,\"flt\":0,\"rng\":0,\"pwm\":0,\"sd\":1,\"ota\":1,\"hms\":0,\"shd\":0,\"flp\":1,\"slp\":1,\"vec\":0,\"bcd\":0,\"ptr\":0,\"pdt\":0,\"rel\":0,\"ovc\":0}}",
  "produceAuth": 1
}
```

The mount command:
```
mount
rootfs on / type rootfs (rw,size=15864k,nr_inodes=3966)
proc on /proc type proc (rw,relatime)
sysfs on /sys type sysfs (rw,relatime)
tmpfs on /dev type tmpfs (rw,relatime)
devpts on /dev/pts type devpts (rw,relatime,mode=600,ptmxmode=000)
/dev/mtdblock6 on /home/cfg type jffs2 (rw,relatime)
/dev/mmc01 on /mnt/mmc01 type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
```

I couldn't find anything usefull on the cfg folder:
```
ls -la /home/cfg/
-rw-r--r--    1 root     root            14 Mar 16 00:37 region
-rw-r--r--    1 root     root            12 Mar 16 00:37 TZ
-rw-r--r--    1 root     root            15 Jan  1  2016 device.token
-rw-r--r--    1 root     root           422 Mar 16 00:37 sys_time
-rw-r--r--    1 root     root           315 Mar 16 00:37 capa.info
-rw-r--r--    1 root     root            12 Mar 16 00:37 alarm.cfg
-rw-r--r--    1 root     root           370 Mar 16 00:37 dev_settings.json
-rw-r--r--    1 root     root            77 Jan  1  2016 test_info_save.conf
lrwxrwxrwx    1 root     root             9 Mar 15 17:53 cfg -> /home/cfg
-rw-r--r--    1 root     root           260 Mar 16 00:37 wifi.info
-rw-r--r--    1 root     root            74 Jan  1  2016 pps_appserver.json
```

```
cat dev_settings.json
{
        "ntp_pool":     ["time1.aliyun.com", "time2.aliyun.com", "time3.aliyun.com", "time4.aliyun.com", "time5.aliyun.com", "time6.aliyun.com", "time7.aliyun.com"],
        "sleep":        "off",
        "sleep_time":   [],
        "home_geographic":      {
},
        "home_geo_fencing_state":       "0",
        "timezone":     "UTC8:00",
        "motion_detect":        {
                "enable":       0,
                "alarmtype":    1,
                "sensitivity":  4
        },
        "video_mirror": 0
```

```

cat test_info_save.conf
{
        "semi_state":   1,
        "aging_state":  0,
        "madeup_state": 1,
        "auth_state":   1
}
```

guino, 2021-03-15:
@SamThing  cool, did you check if there was anything useful in pps_appserver.json ?

It looks like your device has rw root partition which means we could just modify the /etc/init.d/S90PPstrong file so it mounts the SD card and copies a modified ppsapp to /opt/pps/home/app/ppsapp and redirects the ppsapp line to /tmp/ppsapp.log (so we could get the log later) and reboot to see if the changes are presistent or get reset after reboot. For testing I guess you can just modify anything in that file, reboot and see if the file still has the changes after reboot.

If the changes are persistent (I have cameras that are like that) then we can work on the other changes -- but the changes from the link I posted (Dan's github) should work just the same (with the benefit of being persistent after firmware upgrades).

Let me know

SamThing, 2021-03-15:
Let me try that

Btw, the 8554 seems open (but hidden indeed):

```
telnet 192.168.1.226 8554
Trying 192.168.1.226...
Connected to 192.168.1.226.
Escape character is '^]'.
Connection closed by foreign host.
```

And then the camera restarts

SamThing, 2021-03-15:
```
cat pps_appserver.json
{
        "endpoint":     "https://apis-eu-frankfurt.meari.com.cn",
        "gwCode":       "EU"
}
```

guino, 2021-03-15:
@SamThing with port 8554 open I can only assume the RTSP server is starting up, it's just that it will likely need a user/password that we currently don't have. I can make it so it accepts anything like admin:admin but that still requires us being able to start the ppsapp correctly without having to kill it first -- this means either using Dan's method OR being able to modify that S90PPStronf startup script. Let me know what you find out.

PS: Too bad there was nothing in /home/cfg that could hint to the password.

SamThing, 2021-03-15:
The alterations are not persistent between reboots.

But the Dan's method worked!

Now I have the log file with the steam feed.
Seems like there's some mqqt involved.

[ppsapp.log](https://github.com/guino/BazzDoorbell/files/6143761/ppsapp.log)

SamThing, 2021-03-15:
The password is in the logfile: admin/c4c5bdec9d412936e2b37f89939ab407

guino, 2021-03-15:
@SamThing I would try that with VLC rtsp://admin:c4c5bdec9d412936e2b37f89939ab407@IP:8554/ and also with Streaming/Channels/101 to see if it works at all

SamThing, 2021-03-15:
No luck.

With or without the video stream feed on the camera keeps crashing

guino, 2021-03-15:
@SamThing can you try password just: c4c5bdec9d412936 -- it seems like it has some references to just that in the log.

Since you got Dan's method going, I can make you a patched ppsapp that should accept any password if you like to try.

SamThing, 2021-03-15:
Same result.

That would be great!

guino, 2021-03-15:
[ppsapp-rtsp.zip](https://github.com/guino/BazzDoorbell/files/6143912/ppsapp-rtsp.zip)

Place that one on the SD card and make sure the S90PPStrong-290 file on the SD card points to the correct filename -- it should (hopefully) accept user admin with any password. I am hoping this will allow the RTSP feed to be viewed on one of rtsp://IP:8554/ or with Streaming/Channels/101

SamThing, 2021-03-15:
No luck

The patched version runs as the unpatched.
Requires the same password.
The android app still working and no RTSP stream (even with the right password)

```
ps |grep pp
  924 root      1104 S    ./ppsdsry
  952 root      123m S    /mnt/mmc01/ppsapp-rtsp
```

guino, 2021-03-15:
@SamThing  orry, I forgot to mention they have a specific prevention against using `admin` as the **password** so if you tried admin:admin by any chance try admin:password instead.

You obviously have ppsapp-rtsp running (from the ps output, so I'd like you to try this:

http://admin:password@IP/proc/cmdline to see if it works

And the other thing I was going to ask is if you're trying VLC with TCP or UDP for the RTSP stream. By default I think VLC uses RTSP over UDP, but you can go in the settings ans switch to TCP (Click 'Tools > Preferences > Input / Codecs and select 'RTP over RTSP (TCP)' at the bottom, them click 'Save') you should be trying both to see.

SamThing, 2021-03-15:
hi, only the right password works for the http endpoints, already tried other combinations.

The TCP stream fails as it does with UDP

😢

guino, 2021-03-15:
so you're saying http://admin:RIGHTPASSWORD@IP/proc/cmdline works if you use the password from the log file, but doesn't work with RTSP ?

SamThing, 2021-03-15:
Yes, that's right.

So the patched file works just the same as the unpatched one.
On the http endpoints only the right password works and regarding the RTSP stream, no luck so far.

Looking to the logs, seems that first there's a mqtt connection to the server from both the android client and the camera.
And then, begins an p2p conection between the android app and the camera.
Still trying to get the details from the log.

guino, 2021-03-15:
Did you happen to try the http endpounts like /proc/cmdline with the correct password before using the patched ppsapp file ? I am double checking the changes I made.

guino, 2021-03-15:
@SamThing I think I found a case where my change was not catching, making one more change so you can try..

SamThing, 2021-03-15:
Cool!

Yes, tried several passwords on the http endpoints, only the "original" one works

guino, 2021-03-15:
@SamThing 
ok, try this one -- please try /proc/cmdline first to see if it takes any password (other than admin):
[ppsapp-rtsp.zip](https://github.com/guino/BazzDoorbell/files/6144185/ppsapp-rtsp.zip)

guino, 2021-03-15:
btw I didn't see much more info on the log you posted last time compared to the first one (just fyi) -- I was expecting to see some more stuff related to the RTSP stream.

SamThing, 2021-03-15:
Yeah, me too. I can try get more logs.

No luck with this patch too. Same behaviour, only the right password works on http endpoints. 
RTSP still causes the camera to reboot (both UDP/TCP)

Let me know 
Should I get more logs?

Just out of curiosity: does any other camera you know use mqtt?

guino, 2021-03-15:
tuya cameras use mqtt too but to tuya servers.

More logs may be helpful -- can you triple check that the S90PPStrong-290 on the root of the SD card is running the patched ppsapp-rtsp on the SD card (which by default is on the root of the SD card too) ? I know we modified it initially and just wanted to double check -- there's just no way for the "pps_user_verify_user()" function to be returning anything other than a good response on the last patch. So I am going to check the code for any other 'similar' functions but that one should be working, so it is weird to see no change in behavior.

SamThing, 2021-03-15:
I modified the file to call the custom.sh on the root of the SD card:

```
cat /etc/init.d/S90PPStrong
#!/bin/sh

export PATH=/usr/bin:/sbin/:/usr/sbin:/bin

mkdir -p /opt/pps
mount -t cramfs /dev/mtdblock5 /opt/pps
tar xzf /opt/pps/app.tar.gz -C /
rm -f /home/init.d/S60ppsapp
umount /opt/pps

(
/home/init.d/initS
cd /home/app/
./network &
sleep 1

mkdir -p /mnt/mmc01
mount -t vfat /dev/mmcblk0p1 /mnt/mmc01
sleep 1

/mnt/mmc01/custom.sh &
#/mnt/mmc01/home/app/ppsapp &
#/mnt/mmc01/home/app/ppsapp > /mnt/mmc01/ppsapp.log &

sleep 15
rm /home/app -rf
sleep 20
until [ 1 -gt 2 ]
do
 echo 3 > /proc/sys/vm/drop_caches
 sleep 2
done
) &
```

And on the custom.sh I added the patched file:
```
cat custom.sh
#!/bin/sh
if [ ! -e /tmp/customrun ]; then
 echo custom > /tmp/customrun
 cp /mnt/mmc01/passwd /etc/passwd

 /mnt/mmc01/set record_enable 0
 /mnt/mmc01/set enable_event_record 0
 /mnt/mmc01/set onvif_enable 1

 if [ -e /mnt/mmc01/ppsapp ]; then
  #PPSID=$(ps | grep -v grep | grep ppsapp | awk '{print $1}')
  #kill $PPSID

  #/mnt/mmc01/ppsapp &
  /mnt/mmc01/ppsapp-rtsp &
  #/mnt/mmc01/ppsapp > /mnt/mmc01/ppsapp.log &
 else
  /mnt/mmc01/home/app/ppsapp
  #/mnt/mmc01/home/app/ppsapp > /mnt/mmc01/ppsapp.log &
 fi

 /mnt/mmc01/busybox telnetd -l /bin/sh
 /mnt/mmc01/busybox httpd -c /mnt/mmc01/httpd.conf -h /mnt/mmc01 -p 8080

fi
if [ ! -e /tmp/cleanup`date +%Y%m%d` ]; then
 rm -rf /tmp/cleanup*
 touch /tmp/cleanup`date +%Y%m%d`
 /mnt/mmc01/cgi-bin/cleanup.cgi > /tmp/cleanup.log
fi
```

As you can see, the ps show the patched file:
```
ps |grep pp
  924 root      1104 S    ./ppsdsry
  952 root      123m S    /mnt/mmc01/ppsapp-rtsp
 1439 root      1476 S    grep pp
```

And regarding the version that's running, here you can see the creation date (corresponding to the second one):
```
ls -la ppsapp-rtsp
-rwxr-xr-x    1 root     root       3318036 Mar 15 20:08 ppsapp-rtsp
```

My local time here is 20:32

So no doubt that's the last patched file that's running 😄

guino, 2021-03-15:
sounds good -- I think I need a more comprehensive log so I don't go blind into this thing

guino, 2021-03-15:
@SamThing it would be helpful if the log had some attempts to use RTSP and /proc/cmdline with (correct/wrong password too) as well and the RTSP using the app.

SamThing, 2021-03-15:
Timeline:
* Started rtsp on the android app as soon as the camera shows.
* Stop streaming about 20:39
* Tried several times /proc/cmdline endpoint with user admin and random passwords (password, blank text, serial number etc)
* The last one was using the right password.

[ppsapp.log](https://github.com/guino/BazzDoorbell/files/6144319/ppsapp.log)

SamThing, 2021-03-15:
I can't try using external rtsp as the camera crashes and the log is erased.
I can try appending the log file instead.

Let me try that.
Just need to resolve some persona things first.
Brb ~30m

guino, 2021-03-15:
I think the function I patched is meant only for 'basic' authentication but apparently the requests are going out with 'Digest' so the changes I made are basically ignored. I don't know if there's a way to force basic authentication in the browser/VLC (or whatever RTSP client) but I'm trying to see if I can bypass the password on Digest.

SamThing, 2021-03-15:
Works with basic auth!

And seems you're right:
```
curl -XGET 192.168.1.226/devices/network \
-H "User-Agent: Awesome HTTP Client" --digest --user admin:banana

401
```

```
curl -XGET 192.168.1.226/devices/network \
-H "User-Agent: Awesome HTTP Client" --user admin:banana

200 - ok
```

Not sure if there's a way to try it on rtsp, I'll investigate

guino, 2021-03-15:
I also see some references in the code to /etc/passwd which may indicate it's actually using that user/password list to allow/deny service to the device. I would check/set the contents cf /etc/passwd and try a user/password combination from there to see if it works (i.e. your user+password from telnet).

I also see some indications that it may have a RTSP server on port 554 (separate from 8554) so you may want to try to check if 554 is open with telnet IP 554 and if it's open try rtsp on that as well.

Finally I made another change to try and bypass password check on Digest (if the above doesn't work) -- I doubt it will allow 'everything' to work but I expect at least some stuff to work with Digest:
[ppsapp-test.zip](https://github.com/guino/BazzDoorbell/files/6144824/ppsapp-test.zip)

SamThing, 2021-03-16:
Hi

the 554 port is closed for TCP and UDP (even using the video feed on the android app)
The patch works!
Now I can use any user/pass on both basic auth and digest.
But still no luck on rtsp

guino, 2021-03-16:
@SamThing When you have a chance: can you try to make another log with the **original** ppsapp just trying to access rtsp on the camera from VLC to see if we can get any useful information? I understand the camera may crash but maybe tou can execute`/mnt/mmc01/busybox sync` right after trying to start rtsp so it records data to the SD card before it crashes/reboots - this could be very useful in defining where the change needs to be done.

I assume you tried using a user/password combination that works with telnet (from /etc/passwd) with no success. I also wanted to double check as there have been reports of VLC for mac having issues with RTSP from these cameras - you may want to try with something else like ffplay/ffmpeg or some phone app too.

The only other thing I think you could try is some onvif tool to see what it shows - this device may have basic onvif on port 80 (which should provide information on rtsp feeds available).

SamThing, 2021-03-16:
Hi

fyi: I just updated the busybox with a newer version that has vi, wget and curl to make the things easier

This log file contains the log when trying to get the rtsp feed using vlc on a linux machine using the original ppsapp:
```
vlc --rtsp-tcp rtsp://admin:banana@192.168.1.226:8554

VLC media player 3.0.9.2 Vetinari (revision 3.0.9.2-0-gd4c1aefe4d)
[0000556212055be0] main libvlc: Running vlc with the default interface. Use 'cvlc' to use vlc without interface.
libGL error: No matching fbConfigs or visuals found
libGL error: failed to load driver: swrast
Created new TCP socket 19 for connection
Created new TCP socket 19 for connection
[00007fc230001170] live555 demux error: Failed to connect with rtsp://192.168.1.226:8554
[00007fc230004ad0] main stream error: connection failed: Connection refused
[00007fc230004ad0] satip stream error: Failed to connect to RTSP server 192.168.1.226:8554
[00007fc230004ad0] main stream error: connection failed: Connection refused
[00007fc230004ad0] access_realrtsp stream error: cannot connect to 192.168.1.226:8554
QObject::~QObject: Timers cannot be stopped from another thread
```

Also, tried to use some android apps that get the onvif configuration automatically, but no luck here, since all of them try to reach this invalid */onvif/device_service* endpoint
[ppsapp.log](https://github.com/guino/BazzDoorbell/files/6149156/ppsapp.log)

SamThing, 2021-03-16:
As you can see, there's nothing on the log, the app just stoped.
Please note that the file was appended, so start reading backwards 😄

guino, 2021-03-16:
@SamThing I'm probably going to have to try making a few different changes for you to try when you have a chance.

guino, 2021-03-17:
@SamThing Sorry I had a long day at work. Here are 2 ppsapp mods for you to try:

[ppsapp-test1.zip](https://github.com/guino/BazzDoorbell/files/6152866/ppsapp-test1.zip) This one should try to open RTSP on port 855**5** (not 8554) and should accept any password for RTSP.

[ppsapp-test2.zip](https://github.com/guino/BazzDoorbell/files/6152870/ppsapp-test2.zip) This one is the same as above but using port 8554. I have a feeling that the port 8554 is being for some specific function for the app support and preventing the 'normal' (non-app) RTSP from being available on port 8554 so that's I made the change on the port on the first test above.

Please run the 1st one and do nmap to see if port 8555 opens -- then try to view the video with the app (to see if it still works or not), then try VLC (tcp/udp) using rtsp://admin:password@IP:855**5**/Streaming/Channels/101 to see if you get anything.

Regardless of it working or not, you can try the 2nd one using port 8554 (including app/VLC) using rtsp://admin:password@IP:855**4**/Streaming/Channels/101 -- I am expecting that any password other than 'admin' should work but let me know.

SamThing, 2021-03-17:
Hi, good morning
And thanks for keeping working on it!

Testing the first file shows that the port 8555 is indeed open, but using it causes the camera to reboot, same as the default behaviour of 8554. Just to nmap that port causes that. RTSP don't work either.

The second file behaves the same but on port 8554. 😢

guino, 2021-03-17:
@SamThing does the app work (to view video) with the test1 file? I assume it doesn’t based on your response but want to be sure.

SamThing, 2021-03-17:
no, don't work at all and causes the camera to reboot as it does on the original ppsapp file on port 8554 😢

[20 more replies]