guino/BazzDoorbell #25: Incomplete snap image

Tobydeteckel, 2021-03-18:
Hi hoping for some help,

I get incomplete images from the snap.cgi

LSC Doorbell 2.9.7 non-patched onvif enabled.
[image]

Problem on all 3 bells, But there is a difference in the amount of incompletion.

Greats

guino, 2021-03-18:
This is likely due to the complexity of your image (I can only guess it's all the deails on the brick wall). The default jpeg image buffer is 57kb and in your case you should probably increase it and it should take care of the issue. I would suggest playing with the value a little bit so you don't make it 'too big' which could lead to memory issues and slowness.

This is the line in snap.cgi (Similar in mjpeg.cgi):
`/mnt/mmc01/jpeg-arm /proc/$PPSID/mem $ADDRESS 57336 2> /dev/null`

Just change it to (this is a 50% increase):
`/mnt/mmc01/jpeg-arm /proc/$PPSID/mem $ADDRESS 86000 2> /dev/null`

If you later have issues with the image being cut off when there is movement or changes to the image you can increase it to 114672 but I would try avoid increasing it a lot more than that.

Tobydeteckel, 2021-03-19:
Solved!

Thanks you!

closed 2021-03-19 by Tobydeteckel