guino/BazzDoorbell #26: Lower volume of the doorbell when someone press

m-Aiden, 2021-04-13:
Hope someone can help me here.

Where can I adjust the volume of the bell when someone rings the doorbell .. It is now so loud that they can hear it on the other side of town = D

Had already looked in the Tuya config via telnet, but I can't find it there. He does not want to adjust it via the app.

guino, 2021-04-13:
There’s no such option anywhere by software or hardware as far as I can tell.

The only thing you can do is modify the WAV file so it has a lower volume (amplitude). Once you have a modified wav file you can add a line to custom.sh to copy it over the existing one so it is used instead of the default one (right after the ‘cp ... /etc/passwd line’ ):
`cp /mnt/mmc01/dingdong.wav /home/sound/`

jilleb, 2021-08-30:
I did find some endpoints in the ppsapp that refer to input and output volume.:

/media/audio/output/volume/

Will generate the following json:
`{
	"volume":	35
}`

/media/audio/input/volume/
Will generate the following json:

`{
	"volume":	90
}`