guino/BazzDoorbell #29: Enable motion without app notification

karlzre, 2021-05-22:
Hi, 

Thx for this hack, i have now a super connected doorbell :-)

Is it a way to enable the motion detection without getting the notification on the tuya app ?
I Want to use the motion sensor just to play a sound when motion is detected.
otherwise i receive a lot of notifications on my phone.

Thx

guino, 2021-05-22:
@karlzre you can enable motion detection and either disable notifications OR set a do-not-disturb schedule in the app (for the whole day) to filter out notifications specifically from the doorbell if you don’t want to be notified on the phone. I am not even sure you need to enable motion detection alerts on the app to have the log_parser working.

stack0423, 2021-05-24:
I can confirm that I have received a notification in domoticz about movement outside the tuya app.