guino/BazzDoorbell #30: Door bell to call trough sip

karlzre, 2021-05-22:
Hi, 

do you think it should be possible with buzybox to make a sip call to a pbx or sip device ?
So we can use home pbx to receive the call, we don't need the app anymore.

Thx

guino, 2021-05-22:
@karlzre with busybox I am sure there’s nothing for any kind of audio support. Even if it had what you are asking would require a bunch of coding and a high level of integration with ppsapp - to the point it could be simpler just to make your own application based on hisilicon sdk. This is definitely not something I would work on but your device is rooted and you can play with it all you want (and dig through ppsapp with ghidra) so if you are a coder and you are knowledgeable about sip (and have enough time) you could probably pull it off.

fennec622, 2022-03-02:
@guino 
If two way audio can support that would be amazing

Do you thinks this api can help you

https://developer.tuya.com/en/docs/app-development/avfunction?id=Ka6nuvucjujar

guino, 2022-03-03:
@fennec622 most of the tuya API is based on android/ios libraries (closed source) which won't work on a PC running mac/windows/linux (will only work on a phone), additionally they're device-manufacturer dependent so the fact that they have an API for the feature doesn't mean it will work with all devices as it depends on the manufacturer to enable the feature (which in turn may require extra fees). If I remember correctly they even have something in the API to get RTSP feed from the devices but I have never seen anything that can successfully use it since most manufactures won't enable the feature (maybe won't pay extra for it?).