guino/BazzDoorbell #31: HTTP Error 500

MichelGlazenborg, 2021-05-29:
Was following the guide in #2 and at step 7 i'm receiving a HTTP Error 500. It does appear that the cmdline has changed correctly, but the error persists.

cmdline:
`mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino, mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts`

Software version is: 2.9.6

guino, 2021-05-30:
@MichelGlazenborg Your boot line appears to be correct, you should check if your SD card has the home/app/ppsapp file , if so: you should just proceed with thr rest of the steps even if step 7 is not working.

MichelGlazenborg, 2021-05-30:
@guino It appears that file hasn't been created, but there are 2 other files
```
SDT\060574266\record\2021\05\30\09\094234.data
```
and
```
SDT\060574266\record\2021\05\30\20210530.index
```

MichelGlazenborg, 2021-05-30:
Issue fixed using the ppshack2 from #13 ([http://admin:056565099@ip/proc/self/root/etc/init.d/S90PPStrong](http://admin:056565099@ip/proc/self/root/etc/init.d/S90PPStrong) has #MTDNUM=5 but this hack seems to work without issues)

closed 2021-05-30 by guino