guino/BazzDoorbell #32: Stream Corrupt

stack0423, 2021-05-31:
I have 2 doorbell 1 works fine other gives a corrupt stream,
[picture.zip](https://github.com/guino/BazzDoorbell/files/6568231/picture.zip)
 get a white blank under the video (see attachment), because of this it is not loaded in motioneye.
sometimes it works fine again after reboot but still comes back again.
this is the url of stream:http://192.168.178.22:8080/cgi-bin/mjpeg.cgi

guino, 2021-05-31:
@stack0423 this is the same issue reported/fixed here: https://github.com/guino/BazzDoorbell/issues/25 Please read/adjust in the same way and it should work.

stack0423, 2021-05-31:
ok thanks now the blue led of the pressure bell is off, does that have something to do with it?

stack0423, 2021-06-13:
The stream now works fine without disruption, only the blue LED no longer works. Can I adjust something to get this working again?

guino, 2021-06-13:
The LED state is not modified by the hack/patch unless you modify the settings in tuya_config.json or add a ‘set’ command in custom.sh to do it. Some phone apps and/or firmware versions allow you to modify the LED settings so you should probably check that first (you could have changed it by accident). I would also verify that when you power cycle the camera you can see the blue LED on (even if momentarily) to verify the LED still works hardware wise. If it is working and just doesn’t stay on after it boots then you could modify tuya_config.json to keep it on.