guino/BazzDoorbell #33: Motion video telegram

stack0423, 2021-06-01:
Hi, is it possible to send a motion video to telegram?

guino, 2021-06-01:
@stack0423 Possible: yes. Has it been done? not that I know of. How hard ? well, If you're talking about sending motion ALERT (text message) to telegram I suppose you can just trigger the notification from the bell to a server/machine somewhere and from that send a message o telegram. 

If you're talking about sending the actual VIDEO it's a whole lot more work you'd have to make a fancy script to wait for a motion notification, find the video file associated with it in the SD card and WAIT for it to be done recording THEN send the video to a server/machine somewhere to then CONVERT the video to a 'standard' format and then send the converted video into telegram (and honestly I don't even know if telegram supports video -- never used it). Additionally you will NOT get audio on the video because only proprietary tools are able to convert these videos with audio.

It's definitely something I am not planning to work on -- I already get too many 'motion notification' events from the Tuya app (which include a JPEG) so adding more notifications is not in my to-do list. That said, someone with enough knowledge (scripting/services/ffmpeg/telegram API) should be able to do it.

jkottier, 2021-07-08:
You can achieve this using a (virtual) Tuya switch, Smart Life "Smart" automation and Automation in Home Assistant.

With the Smart automation in the Smart Life / LSC or Tuya app (they are almost the same) you can turn on the (virtual) Tuya Switch (I also turn it off after 5 seconds)
Within Home Assistant you can create an automation that is triggered when the virtual Tuya Switch is turned on.
Next the automation can send a snapshot, create a recording and send it to a Telegram group.

Sending a snapshot is almost instant. Eventhough I'm using the Tuya network for catching the doorbell events it seems way faster than the log_parser or dmsg I've used before.

Sending a recording is rather slow (too slow in my opinion). This is because HA needs to store the recording first before you can send it via Telegram. But if that isn't an issue, this could be your solution.

richaardvark, 2021-09-07:
Another option to achieve what (I think) you are trying to achieve - if you don't mind using an Android-based server approach (and happen to have an old Android phone or tablet/a raspberry pi/extra PC or already running server that you can configure to run an Android virtual machine from) you can use the TinyCAM PRO app as well to have Telegram broadcast/send out motion videos and alerts/do a billion other neat things.  I haven't tried setting up the Telegram feature myself yet but it seems fairly easy. Here's a link to download the latest version of the TinyCAM PRO app that has had any advertisements/trackers/user analytics reporting and  other limitations completely removed.:[Download TinyCAM Pro APK](https://forum.mobilism.org/viewtopic.php?f=429&t=4403058&hilit=ip)