guino/BazzDoorbell #34: Help on Bell B1S 2.1.7

jpleite, 2021-06-02:
@guino Olá Guino. First thank you for the great job and spent hours on reverse engineering tuba like cams. I went through your repos and got a lot insights. I own a video doorbell which I've never seen here or on your other repos. Thing is I would like to be able to access ONVIF or use your mpeg/jpeg hack to get video on home assistant and press button notification (not a fan of rasp since is usually slow). The problem is I have no idea how to start since it is a Bell S1 which is not described anywhere. I was able to get device info with http://WeEyE:&$ChuTian_91@9@ip/devices/deviceinfo but is pretty much the only command I can use. Synology surveillance station detects as a Generic ONVIF camera but isn pretty much it. Can you point me out what guide should I follow?

{
	"devname":	"Smart Home Camera",
	"model":	"Bell B1S",
	"serialno":	"057348421",
	"softwareversion":	"2.1.7",
	"hardwareversion":	"BEB1S_V10_915",
	"firmwareversion":	"ppstrong-c2-tuya2_xingtel-2.1.7.20190225",
	"uuid":	"FFFFFFFFFFFFFFFFFFF",
	"authkey":	"gBF9AzezOXuM7jDcJ3mNpl17gR0MR5qB",
	"deviceid":	"pp01f9970f7f518dec96",
	"pid":	"mgi0lw7a8o2mlruj",
	"WiFi MAC":	"20:32:33:19:1e:31",
	"ETH MAC":	"08:88:12:40:5a:64"
}

guino, 2021-06-02:
@jpleite Oi! I have not seen this hardware before (BEB1S_V10_915 / c2). The firmware is older so there's even a chance there's no bootloader protection but really the only way to do anything is if you're willing to open it and either solder some wires to the serial port (requires a TTL-USB/Serial adapter) OR remove the flash chop and read the flash with a programmer (which requires heat gun or good soldering skills to remove it AND a hardware programmer). The TTL-USB adapter and hardware programmer are not that expensive (i.e. $15) but considering the price of the device and that you'll only use the tools once it may not be worth your time/money to mess with it.

If you're willing to open the device, take some pictures I can identify the flash chip and/or serial port connections for you but like I said that's only going to help if you're willing to use a TTL-USB/Serial adapter or hardware programmer. Obviously if I had the device I could help since I have the tools but with your hardware the only thing I can do is guide you (and have done it before with a couple of people). In any case I would NOT recommend you do any soldering work if you're not familiar with it as you could damage the device beyond repair. So either you or someone you know (to help you) should have good soldering skills before you decide to move forward.

jpleite, 2021-06-02:
Obrigado @guino. Actually that is a device only sold in China which I bought a couple of years ago produced by Xingtel. http://www.xingtel.com/ProductDetail/2432318.html. I can open it to check the flash chip but definitely I'm not familiar with soldering neither do I have the equipment. My question is if there's a reversible way of checking if there's no bootloader protection by any patch inserted in the SD Card

guino, 2021-06-02:
@jpleite the only ways to know anything about the bootloader is to either solder the TTL-USB/Serial adapter or read the flash with the hardware programmer.

You can most definitely try one of the existing hacks or #11 but being a different hardware chances are the kernel load address is different and the hack just won’t install. I have a feeling you probably already tried but if not there’s very little chance of it causing problems. 

Lastly if you are in Brazil you may be able to find one of those backyard tv shops that could  solder the TTL-USB adapter or remove the flash chip from you and solder it back later but then again you still need to buy the hardware and pay their service, it it is money you could just use to buy a device with RTSP out of the box.

Subtixx, 2021-06-18:
I've disassembled the door bell so here are some images:

Battery "module":
[image]

Main "module":
[image]

Main chip? + uart?
[image]

guino, 2021-06-18:
@Subtixx your UART is likely the for 'golden' round pads on the left of the last picture. The flash chip is the one on the top-right labeled with XMC.

If you can get a serial log during boot we may be able to adjust one of the files from #11 so we can get a copy of the flash without a programmer. That would be the the simplest way to get into the device unless the bootloader isn't protected at all in which case the UART may give you full root access (even if requires minor bootloader tweaks).

guino, 2021-06-18:
@Subtixx please remember the UART is TTL-3.3v level, you need the right type of adapter the connect to it and the board is very fragile so I recommend being very careful if you're going to solder wires on it as any movement of the wires can pull the pads right off the board.

Subtixx, 2021-06-18:
I guess the USB-ISP thing I have from flashing my 3d printers doesn't work right? Is there anything I have here that I could use? Like an arduino or a raspberry pi? Don't feel like buying one :/

guino, 2021-06-18:
@Subtixx if you have any flasher/programmer, send the details I can tell you if it would work or not. A raspberry pi can be used for UART access as it should be 3.3v TTL as well, just have to wire the RX, TX and GND (no 3.3v) then use a serial terminal application in the raspberry pi -- This was done before by another user (on similar camera) and we got it figured out that way.

Subtixx, 2021-06-18:
I have all the tools I need I think.
- Soldering iron to solder wires to the pads
- Multimeter
- Raspberry Pi
- Wires

You just need to tell me what I should do? I dunno what pads are what. I suspect the outer pad is GND. Since it has continuity with the screw here:
[image]

Dunno how to figure out what the other are tho. Nor how I connect it to the pi

guino, 2021-06-18:
@Subtixx that's a good start - yes, the top one should be ground based on your check.

You should probably just solder some wires to the pads and either connect other wires to the pi OR have a way to switch them around on the pi so you can test it. You will NOT damage the ports of the pi or device by switching them around as long as you're only using the RX, TX and GND pins on the pi. Here are the pins on the PI (you may want to double check on your specific pi model):

[image]

Basically you will use pins 6, 8 and 10 above (6 will go to the ground of the camera), 8 and 10 you'll basically have to connect to RX/TX on the pi and if it doesn't show anything during boot you will just switch them around which is why I recommended soldering other wires and just twisting them together with the ones you soldered for the device/bell.

guino, 2021-06-18:
@Subtixx  You will also need to disable the serial console on the p (so it's available to talk with the bell): 

1.     Start raspi-config: sudo raspi-config.
2.     Select option 3 - Interface Options.
3.     Select option P6 - Serial Port.
4.     At the prompt Would you like a login shell to be accessible over serial? answer 'No'
5.     At the prompt Would you like the serial port hardware to be enabled? answer 'Yes'
6.     Exit raspi-config and reboot the Pi for changes to take effect.

Subtixx, 2021-06-18:
:( Doesn't matter anymore.. As much as I have tried to be careful in the end when 2 of 3 were soldered the last solder ripped all pads out... No idea how or why since I was EXTREMLY cautious

guino, 2021-06-18:
@Subtixx I know how it goes man, I have found the hard way how fragile these boards are. In any case, you can still try the hardware flasher/programmer route if you have any, though my experience is usually you can't read/write the flash withour removing the chip (but you could try -- maybe your hardware would allow it).

I don't know how badly you damaged the UART pins but for GND you could just solder anywhere (ie the screw you pointed out), and for the RX/TX -- if you look close at the picture you sent you will see the pads lead to the resistors right next to them, so you could try to solder wires into the resistor pins -- that's how I managed to 'save' UART access on one of my devices. 

You may chose to stop while you haven't damaged your device beyond normal function so I can understand it if you do.

guino, 2021-06-18:
@Subtixx here's what I am mean regarding the resistors:

[image]

Subtixx, 2021-06-19:
Thanks. I'll try this
Found this url while trying to sniff firmware updates:
https://apis-eu-frankfurt.arenti.net/

Subtixx, 2021-07-02:
Finally got the time to fix the serial connection. I'm now trying to do this.

Ierlandfan, 2021-08-09:
[image]
This the same camera as I have, it's stopped responding to WiFi, only red light on but it still saves something to the sdcard as I tried extracting the firmware from it (2.9.6) only got uboot on the card, maybe wrong adres for the kernel and filesystem parts. Anyway, this one has labels on the pcb and one of them says uart (right bottom)

guino, 2021-08-09:
@Ierlandfan if you read the uboot using #11 then it should have read the whole thing unless there's some problem with the flash itself. If the address is correct (and the bootloader supports the commands) it should read all of it. If the address is wrong or the bootloader doesn't support some commands) it should read nothing at all. The only exception to this would be if the uboot was read at the **end** of the flash.bin file (which you should be able to tell from the offset shown by binwalk) -- in that case it may be possible to adjust the address to read the flash correctly.

Ierlandfan, 2021-08-22:
Hi Guino,

i have a programmer and a device that runs 5.0.5 firmware and 2.9.7. (I want to dump the firmware) What do you use for programming software?  AsProgrammer?

guino, 2021-08-22:
@Ierlandfan I just use the standard linux ‘flashrom’ tool, but you can use whatever application that came with your programmer. From what I have seen with these devices you need to remove the chip from the board (which is very fragile) or at least disconnect pin 6 to make it read/write correctly. You also should check the voltage of your programmer as these are 3.3v chips and using 5v may damage the chip/board. The ideal really is to use a heat gun to remove the chip then read/modify and use the heat gun again to solder it back. In 2.9.7 you can read the flash using steps from #11 - this may work on 5.0.5 too if the bootloader and address are the same. From what I have heard 5.0.5 may not be running Linux.