guino/BazzDoorbell #35: Meco J5 Plus - Finding the password

realroywalker, 2021-06-02:
Hi all,

I have a Meco J5 doorbell (seems to be another spin on the same ppstrong hardware) - however this one runs using CloudEdge app (not Tuya).
However, I've played around and have dumped the firmware (using #11) and it seems that its running ppsapp etc. the same as the Tuya models.
The doorbell has port 80 disabled by default, so I've used the ppsFactoryTool trick to get port 8090 open, I can connect to http://admin:056565099@<ip>:8090/devices/deviceinfo the connection works, but the login credentials fail (they just keep popping back up).
I tried Firefox and Chrome but same result, so I guess the password may be different for this unit.

Does anyone know if I might be able to find the password within the firmware dump ? - I've run it through binwalk and extracted the contents, but not sure where it may be buried.

guino, 2021-06-02:
@realroywalker did you try http://admin:admin@:8090/devices/deviceinfo ? I know some Mercury 1080P cameras use that instead of admin:056565099.

 If you post the firmware dump we should be able to extract ppsapp from it and use ghidra to look at whatever password it may be expecting, that said -- if #11 worked chances are the hack (root) would also work in which case you could get ppsapp that way and we could just see if it can be patched for RTSP/ONVIF (Whatever available, if any).

realroywalker, 2021-06-02:
Thanks, yes I've tried admin:admin, admin:blank etc. but unfortunately no luck.
I have the ppsapp from my camera - https://drive.google.com/file/d/1adnpJNEVfBl-3mXGUfdW10g2MIgX_rTg/view?usp=sharing
I've opened it in Ghidra but not sure what i'm looking for exactly, so just aimlessly poking around.

My camera actually has Onvif by default on the firmware (4.3x) but I'd like to use MQTT to Home Assistant and maybe disable the cloud elements at some point.

guino, 2021-06-02:
@realroywalker I'm taking a quick look in ghidra -- did you by any chance try: admin:SERIALNUMBER (using the serial number of your camera ? I would also try using the serial number as both user and password. I'm not saying it will work but it's worth a shot. I'll let you know if I find something in ghidra to try.

guino, 2021-06-02:
@realroywalker It looks like http://IP:8090/tvclient and http://IP:8090/search should work without user:password -- the port may be a different number than 8090 (i.e. 80). 

I see a specific check for user+password admin:admin in basic authentication (not digest authentication) but it is to actually FAIL if that combination is used -- I am not sure there's a way to force firefox/chrome to use basic auth, but I believe you can do it with curl (linux) -- all I am saying is that the type of authentication might make a difference.

Alas, you should try these user:password combinations that seem to be hard-coded in ppsapp:
PpStRoNg:#%&wL1@*tU123zv
WeEyE:&$ChuTian_91

realroywalker, 2021-06-03:
@guino Thanks so much - http://WeEyE:&$ChuTian_91@<ip>:8090/devices/deviceinfo works 👍 

Firmware version comes back as ppstrong-c5-s_meco-4.3.0.20200811.
Now that I can confirm access to that page I'll continue the rest of #13 and see if I can get it talking to HA.

realroywalker, 2021-06-03:
Seems the hack works just fine, I now have telnet access to the doorbell :smile:

I started poking around and noted some subtle differences (for example my config is /home/cfg/dev_settings.json) - but the majority look the same.

I am seeing some problems when trying to use the logparser script from https://github.com/guino/BazzDoorbell/issues/4#issuecomment-740644879 though.

Or more accurately, when I try to run ppsapp from the sdcard - looking at custom.sh it seems it checks for a copy of ppsapp at /mnt/mmc01 and kills the running copy if it finds it, then launches from the SD card instead.

So I copied the ppsapp from /mnt/mmc01/home/app into the sdcard root, and fired up the doorbell (without any of the MQTT log parser stuff added) - the doorbell light goes blue to show it connected, then after about 10 seconds it seems to reboot, and just sits in that cycle.

I added the logparser script to the launch command for ppsapp (just to check it was getting as far as launching ppsapp) and it starts ok and talks to my MQTT broker just fine (for around 10 seconds), then the doorbell reboots like without the logparser added.

Seems like there is maybe some check regarding where the ppsapp runs from and causes a reboot just after it starts?

Any tips on how to figure out what's going on?

guino, 2021-06-03:
@realroywalker glad you got it rooted. 

Regarding the reboot issue -- there are some known issues with killing ppsapp and running it again (even if you don't modify ppsapp) that cause the device to reboot on **some** versions and **some** hardware combinations (ie Speed 9X + 2.9.0 firmware). This is likely caused because of poorly written code and drivers which can't be stopped/started correctly like on other hardware/firmware. 

The solution is to adjust the boot scripts so the only ppsapp executed is the one from the SD card (with your parameters to use the log parser). There's some information on #2  regarding firmware 2.9.0 which you should review then you should compare the S90PPStrong boot script in your firmware with the contents of the S90PPStrong-290 he provided to make sure they're similar enough (you may need to add/remove stuff so it works on yours) but I would expect you should be able to make it work since you're using the hack from #13

The bottom line is that as long as you don't have to 'kill' and 're-run' ppsapp I expect it should work just fine.

You obviously would have the option to modify the firmware itself if you have a hardware programmer but that's a much harder way of messing with it.

realroywalker, 2021-06-03:
Ah ok that sounds promising, I was thinking it was some kind of protection in the app rather than just bad coding 😆 

Thanks for the tips, I will check out the init scripts tonight and see if I have better luck!

realroywalker, 2021-06-04:
I am happy to report that using the tweaks you linked @guino I now have ppsapp running from the SD card without any crashing 👍 

I've managed to update the logparser script to work for my bell (log messages seemed to be different for motion) and have updated the custom.sh to setup MQTT discovery of the motion and button events to HomeAssistant.

I did notice that when trawling the logs of ppsapp while looking for the motion events, that it spams with:-
`[11:45:34.997 ERR pps_dp.c:1153]U k ow  sdcard status 7
[11:45:35.078 DEBUG pps_sdcard_fsck.c:330][fsck]Sdcard fs type is FAT32.
[11:45:35.079 DEBUG pps_sdcard_hisili ux.c:227]Got o e partitio : /dev/mmcblk0p1
[11:45:35.079 DEBUG pps_sdcard_hisili ux.c:159]The /dev/mmcblk0p1 has bee  mou ted to /m t/mmc01!
[11:45:35.079 DEBUG pps_sdcard.c:409]Force umou t /m t/mmc01
[11:45:35.079 ERR pps_sdcard_hisili ux.c:78]umou t failed: Device or resource busy
[11:45:35.079 DEBUG pps_fsck_boot.c:213][fsck]fs.cs:16384, b.cs:32,  fats:2
[11:45:35.079 DEBUG pps_fsck_boot.c:227][fsck]total_sectors:61065216, b.ts:61065216
[11:45:35.079 DEBUG pps_fsck_boot.c:232][fsck]fat_le gth:14912, b.fat_le gth:0, b.fat32_le gth:0
[11:45:35.079 DEBUG pps_fsck_boot.c:243][fsck]fat_start:16384, root_start:15286272, root_e tries:0, data_start:15286272
[11:45:35.079 DEBUG pps_fsck_boot.c:246][fsck]data_size:31250104320
[11:45:35.079 DEBUG pps_fsck_boot.c:261][fsck]Tool  ame:mkfs.fat▒, fstype:FAT32
[11:45:35.084 DEBUG pps_sdcard_fsck.c:168]total file size:31571968, used_size:31686656
[11:45:35.109 ERR pps_sdcard_hisili ux.c:66]mou t failed: Device or resource busy
[11:45:35.129 ERR pps_sdcard.c:465]mou t failed!
`
Oddly the character 'n' doesn't print for some reason!

Is it normal that ppsapp spams with this ? - I checked fstab and it doesn't mention the SD card, so I guess ppsapp itself scans for the card and tries to mount it?
Can that be disabled, or can I just symlink /mnt/mmc01 to wherever ppsapp wants to mount it, so that it thinks it's already mounted?

guino, 2021-06-04:
@realroywalker glad you got it working - if you made any changes for your camera it would be cool if you posted a zip with the files you changed (so future users could get it). Even if you just updated log_parser it would be nice to post it.

Regarding the SD card mount, yes ppsapp monitors and mounts it as it sees fit. I don’t know how “normal” it is for it to spam the log with messages about the SD card mount state but if it’s not affecting anything I would just ignore it. I do know for a fact it expects /mnt/mmc01 to be a directory where the SD card is mounted so if that’s not what you have you could tweak your boot scripts so that directory exists and so the SD card is mounted there (obviously make a backup of your work before making any changes since it works right now).

Did you want to use snap/mjpeg? I don’t know if your camera has support for it already and I did not look for it in ppsapp but I thought I would check anyway.

realroywalker, 2021-06-04:
@guino Yes, absolutely - I'm thinking of cleaning up the modifications to make it reusable and documenting exactly the steps for this specific bell, since it seems to vary slightly from the Tuya models.

I see no obvious side effects from the log spam apart from SD capacity cannot report in the CloudEdge app (not sure if the event recording to SD can work, as I don't have that turned on at the moment).
I wondered if it could affect performance / heat of the bell, since it's spamming that to the logs in a loop every few hundred ms when I checked.
The SD is currently mounted to /mnt/mmc01 so I guess it makes no check for that and just tried to control the mount itself.

Snap/mjpeg wasn't something I looked at, as the bell has onvif out of the box so I can get the feed to HomeAssistant and I'm happy with that :smile:

realroywalker, 2021-06-15:
I have uploaded a copy of the files used here (to be merged with the files from github as per the instructions).

[Meco_J5.zip](https://github.com/guino/BazzDoorbell/files/6654580/Meco_J5.zip)

As a summary for hacking and use with Home Assistant, the Meco J5 uses (at least on firmware 4.3.0 and 4.3.3) **WeEyE** as the username and **&$ChuTian_91** as the password, however http is disabled by default, so requires booting with the attached ppsFactory file from the 'Wifi' folder on your SD, this will enable HTTP on port 8090 on the doorbell.

The firmware can successfully be dumped using https://github.com/guino/BazzDoorbell/issues/11

To root the device follow the instructions on https://github.com/guino/BazzDoorbell/issues/13 however for step 2 you should include additional parts to the 'env' as described by https://github.com/DanTLehman/orion_sc008ha#first-success (you can also reference the one included in the attachment here) but essentially you need to add:
cp_/mnt/mmc01/S90PPStrong-290_/etc/init.d/S90PPStrong;
The other files from within the attached 'Extras' folder can be copied to your SD once the rest of the hack from https://github.com/guino/BazzDoorbell/issues/13 is performed (don't copy my env though, just the other 3 files) - open log_parser.sh and edit the variables for your broker..

You should also enable a password as per the hack guidance and set it in passwd on the SD.
Now you should be able to boot the doorbell with the SD inserted and be able to telnet into it using your credentials.

If that works, you can run;
cp /mnt/mmc01/home/app/ppsapp /mnt/mmc01/ppsapp-custom
reboot

Now wait, hopefully the doorbell boots ok (if not remove the SD and just delete ppsapp-custom) 
It should now connect to Home Assistant and send MQTT messages when motion / button press is detected.

guino, 2021-06-15:
@realroywalker Thank you for sharing the files and details. I'll be sure to mention it to others that have the same device.

kurtqwerty, 2021-06-23:
Hi! 
Thank you so much for uploading this hack! it looks great.
Unfortunately i have also the issue that i'm unable to login.
When i run :8090/search it tells me the following:

{
	"sn":	"064886065",
	"factory_code":	0,
	"factory_code_str":	"",
	"model":	"Bell 8S",
	"p2p_uuid":	"",
	"ip":	"192.168.1.224",
	"mask":	"255.255.255.0",
	"gw":	"192.168.1.254",
	"mac":	"b4:fb:e3:df:34:3c",
	"interface":	"wlan0",
	"version":	"5.0.5"
}

i'm pretty new in this. Can you please help me so i can send the images to my nas?

Thanks so much!

guino, 2021-06-23:
@kurtqwerty if you don't get a response from /devices/deviceinfo or /proc/cmdline I suspect your camera would be too different to work with the existing hacks. If that's the case the only way to move forward would be to open the device to either connect a UART-TTL adapter OR plug in a hardware programmer to read the flash. What app do you use with this camera anyway (just curious) ?

kurtqwerty, 2021-06-23:
> @kurtqwerty if you don't get a response from /devices/deviceinfo or /proc/cmdline I suspect your camera would be too different to work with the existing hacks. If that's the case the only way to [...]

I use the LSC app, its a doorbell from a dutch shop called the Action. I do get a prompt to log in, but all the previous users and password (and variations on that to try :) ) didn't work. The app tells me i'm running V5.0.5

guino, 2021-06-23:
@kurtqwerty the newest versions of firmware I have seen are 4.x so 5.0.5 is not something I've ever seen. There are plenty of people here that have patched LSC 'Bell 8S' devices but they were all on 2.9.x-2.10.x firmware as far as I know. My doorbell is also labeled as 'Bell 8S' and it runs 2.9.6 firmware. 

If you haven't yet, please try: https://github.com/guino/BazzDoorbell/issues/11 to see if you can read your flash -- if you are able to do that I can review the firmware to see what we can do. If #11 doesn't work the only way to go forward is to open the device and hook up a programmer or UART-TTL adapter.

guino, 2021-07-01:
@dexternl when #11 doesn't work (with either set of files)l it usually means you have a different bootloader or load address -- the only way to find out is by opening and connecting a UART-TTL adapter or downloading the flash with a hardware programmer.

bertbijnens, 2021-08-18:
@kurtqwerty 

> I use the LSC app, its a doorbell from a dutch shop called the Action. I do get a prompt to log in, but all the previous users and password (and variations on that to try :) ) didn't work. The app tells me i'm running V5.0.5

Did you try admin:admin? I've got a V5.0.5 LCS doorbell and admin:admin worked for me.

I did not try the next steps though but I'm sceptical because my /proc/cmdline looks like this:
`console=/dev/null LX_MEM=0x3fe0000 mma_heap=mma_heap_name0,miu=0,sz=0x1d00000 pcbversion=BELL5S_S1_V10 sensor=gc2063mipi`
Did not see anything like this before with anyone else

guino, 2021-08-18:
@bertbijnens I haven't seen a cmdline like that either -- all 5.x firmware I have seen runs RTOS (LiteOS) instead of linux and none of them (till now) returned anything /proc/cmdline as far as I heard. You can try #13 but even if the address is correct and it updates the bootloader I expect it just won't work. If you want to try something that doesn't modify the device in any way I would try #11 to see if you can read the flash and that would tell us if it's running linux or something else.

trizmark, 2021-11-29:
I have applied the hack using #2 on my ieGeek camera, but realised that I would need to apply #13 as killing the ppsapp restarts the camera. What's the process to follow? Do I need to revert #2 then reapply #13 ?
/devices/deviceinfo for reference

```
{
  "devname": "Smart Home Camera",
  "model": "Bell 5T",
  "serialno": "061884642",
  "softwareversion": "4.3.3",
  "hardwareversion": "BE5T_H1_V10_433",
  "firmwareversion": "ppstrong-c5-s_joys-4.3.3.20210122",
  "identity": "MR2008150100100883",
  "uuid": "",
  "licence_id": "ppsc4e6ea324a98c4027",
  "WiFi MAC": "7c:25:da:5c:a4:3c"
}
```

guino, 2021-11-29:
@trizmark you should be able to apply #13 directly as long as you use the **original** /proc/cmdline (URL response) information in Step 1 (don't use the information from after applying #2). You can reverse #2 first if you want/prefer, then aaply #13 -- it will have the same effect. If the hack is working with #2 the only reason to switch to #13 would be if you wanted the device to work without the SD card, everything else is basically the same (there's no difference in behavior with ppsapp).

Killing ppsapp causes most devices to reboot but it usually takes a few seconds for it to happen,-- normally there's enough time to start another ppsapp to prevent a reboot. The only way to prevent the device from rebooting when ppsapp is not running is to disable the watchdog or **feed** the watch dog -- I remember someone made a watchdog feeder application in the #4  thread awhile back but that's only useful for development/testing since there's hardly any reason to run the device without ppsapp running (since that's what controls most functions of the device). I don't think we have any way to 'disable' the watchdog as of writing this (and I probably wouldn't recommend it either).

pioneershahid, 2021-12-09:
Hi guys, can anyone help me with accessing device webpage please. 

I am trying to use ieGeek J5 Video Doorbell Camera(Wired) which looks exactly same as Meco J5 doorbell with device version 4.3.3.20210122 using CloudEdge. I bought it from Amazon [https://www.amazon.co.uk/ieGeek-Upgraded-Doorbell-Detection-Waterproof-Grey/dp/B093WHLHW8/ref=asc_df_B093WHLHW8/?tag=googshopuk-21&linkCode=df0&hvadid=499306924014&hvpos=&hvnetw=g&hvrand=2472949296402291497&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1007895&hvtargid=pla-1261682709005&psc=1](url) 

I successfully dumped the firmware (using #11) but I am unable to access device webpage. here is a link of bin file saved in outlook. 

`https://1drv.ms/u/s!ArGK6BaouTlKhvgv9IsUzS0YZ39e_A?e=VPtCkW`

 I saved ppsFactoryTool  in the root directory of the SD card and changed the wifi credential still unable to access the device webpage. 

I read here `https://community.home-assistant.io/t/2k-rtsp-video-doorbell-on-offer-at-amazon-uk/312505/32` that people have successfully applied the hack with no issue. 

thank you in advance for looking into it..

guino, 2021-12-09:
@pioneershahid post a zip of your SD Card files (without the SDT folder) and I can take a look to see if there's anything wrong -- I will still take a look at the firmware too.

pioneershahid, 2021-12-10:
Hi @guino thank you for the reply. 

I think i started from a wrong place at first. I followed #11 first and loaded ppsFactoryTool  into the root directory. i think this is why i was not able to access the webpage.  

However, I started over again format the sd card via cloudege app. I placed ppsFactoryTool into the root directory first.  To my surprise i was able to access webpage. 

`{"devname":"Smart Home Camera","model":"Bell 5T","serialno":"063047963","softwareversion":"4.3.3","hardwareversion":"BE5T_H1_V10_433","firmwareversion":"ppstrong-c5-s_joys-4.3.3.20210122","identity":"","uuid":"","licence_id":"ppsca33fc3b4f6fb46d6","WiFi MAC":"7c:25:da:aa:45:55"}`

Instead of using #11 I directly used #13  and completed steps 1 to 7 with result done at the end. no issue.  see result of the script using `http://WeEyE:&$ChuTian_91@192.168.1.XX:8090/proc/cmdline` 

`mem=36M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd - ip=${T//_/$'\\x20'}:::::;T=\"sleep_5;mkdir_-p_/mnt/mmc01;mount_-t_vfat_/dev/mmcblk0p1_/mnt/mmc01;/mnt/mmc01/initrun.sh&\";eval mtdparts=hi_sfc:384k(bld)ro,64k(env),64k(enc)ro,64k(sysflg),3584k(sys),6656k(app),1536k(cfg),1m(recove),2880k(user),128k(oeminfo) ppsAppParts=5 ppsWatchInitEnd`

Next step would be to access telenet and link it with thome assistant. however, when i open the app to view if everything is working, I cannot access doorbell live stream via cloud edge app. it is not loading the stream and says "establishing encrypted channel" - see snip below and content of sd card 

[image]

SD Card content onedrive like
https://1drv.ms/u/s!ArGK6BaouTlKhvgwlTlnTHEqjvoXZg?e=fJyIwM

Any advice please where i went wrong.. thank you in advance. appreciate all your help.

pioneershahid, 2021-12-10:
@guino  it is mentioned in the original home assistant forum post to delete ppsFactoryTool file after sucessfully install the hack. i did it and it resolve the cloud edge streaming issue. i have the stream now.. next step is to create sensors to report to home assistant when somene press the bell :) .. thank you for the great hack ..

pioneershahid, 2021-12-10:
> I have uploaded a copy of the files used here (to be merged with the files from github as per the instructions).
> 
> [Meco_J5.zip](https://github.com/guino/BazzDoorbell/files/6654580/Meco_J5.zip)
> 
> As a summary for hacking and use with Home Assistant, the Meco J5 uses (at least on firmware 4.3.0 and 4.3.3) **WeEyE** as the username and **&$ChuTian_91** as the password, however http is disabled by default, so requires booting with the attached ppsFactory file from the 'Wifi' folder on your SD, this will enable HTTP on port 8090 on the doorbell.
> 
> The firmware can successfully be dumped using #11
> 
> To root the device follow the instructions on #13 however for step 2 you should include additional parts to the 'env' as described by https://github.com/DanTLehman/orion_sc008ha#first-success (you can also reference the one included in the attachment here) but essentially you need to add: cp_/mnt/mmc01/S90PPStrong-290_/etc/init.d/S90PPStrong; The other files from within the attached 'Extras' folder can be copied to your SD once the rest of the hack from #13 is performed (don't copy my env though, just the other 3 files) - open log_parser.sh and edit the variables for your broker..
> 
> You should also enable a password as per the hack guidance and set it in passwd on the SD. Now you should be able to boot the doorbell with the SD inserted and be able to telnet into it using your credentials.
> 
> If that works, you can run; cp /mnt/mmc01/home/app/ppsapp /mnt/mmc01/ppsapp-custom mv /mnt/mmc01/DISABLED-S90PPStrong-290 /mnt/mmc01/S90PPStrong-290 (wait a few seconds) reboot
> 
> Now wait, hopefully the doorbell boots ok (if not remove the SD and just delete ppsapp-custom) It should now connect to Home Assistant and send MQTT messages when motion / button press is detected.

@realroywalker @guino  can anyone please advise how to send MQTT msges when button pressed. i followed these instructions and reboot the device. I changed the mqtt broker, user name and password but i am not getting any messages in home assistant.. any advise please?

guino, 2021-12-11:
@pioneershahid 
If you followed the steps from https://github.com/guino/BazzDoorbell/issues/35#issuecomment-991285130 and it is working right now, please post an updated /proc/cmdline output and the updated SD card contents I can help review.

If you still have the SD card contents like you posted on onedrive then read below:

Based on the SD card contents (onedrive) I see a few issues below:
1-there's no ppsapp in the root of the SD card -- this is required in order to 'catch' the notifications of button press and/or motion events so the log_parser.sh can trigger them.
2-The custom.sh file needs to be adjusted to run the log parser -- just edit your custom.sh and include `/mnt/mmc01/ppsapp 2>&1 | /mnt/mmc01/log_parser.sh &` (replacing `/mnt/mmc01/ppsapp &`) after `kill $PPSID`
3-your log_parser has some mosquitto_pub commands on top of the script which should not be there. I realize you may have placed them there just to 'see' if they would get executed (and they didn't because of items 1 and 2 above) -- please remember to remove those.

pioneershahid, 2021-12-11:
@guino 

still struggling to get the doorbell to communicate to Home assistant.  I placed ppsapp in the root of the SD card after following instruction from #35 (comment) i.e.

`cp /mnt/mmc01/home/app/ppsapp /mnt/mmc01/ppsapp-custom 
mv /mnt/mmc01/DISABLED-S90PPStrong-290 /mnt/mmc01/S90PPStrong-290 `

after waiting for few seconds, I reboot it. There is no connection to connection to Home Assistant or MQTT messages when motion / button press is detected - not sure where i am going wrong.

I do see `/mnt/mmc01/ppsapp 2>&1 | /mnt/mmc01/log_parser.sh` in the S90PPStrong-290 file per #35 (comment) 

here is the updated /proc/cmdline output 

`mem=36M console=ttyAMA0,115200n8 mtdparts=hi_sfc:384k(bld)ro,64k(env),64k(enc)ro,64k(sysflg),3584k(sys),6656k(app),1536k(cfg),1m(recove),2880k(user),128k(oeminfo) ppsAppParts=5 ppsWatchInitEnd - ip=${T//_/$'\\x20'}:::::;T=\"sleep_5;mkdir_-p_/mnt/mmc01;mount_-t_vfat_/dev/mmcblk0p1_/mnt/mmc01;cp_/mnt/mmc01/S90PPStrong-290_/etc/init.d/S90PPStrong;/mnt/mmc01/initrun.sh&\";eval mtdparts=hi_sfc:384k(bld)ro,64k(env),64k(enc)ro,64k(sysflg),3584k(sys),6656k(app),1536k(cfg),1m(recove),2880k(user),128k(oeminfo) ppsAppParts=5 ppsWatchInitEnd`

Here is the updated SD card content saved in outlook.
`https://1drv.ms/u/s!ArGK6BaouTlKhvgyZomwg1vYFV5fJA?e=5WBmN5`

Thank you once again for looking into it and appreciate your time and effort.

guino, 2021-12-11:
@pioneershahid Please telnet into the device and execute the mosquitto_pub commands by hand to see if it works. Everything else you did appears to be correct (the device is rooted and extracted the ppsapp and executed all scripts) -- there's a good chance that everything is working except for something between mosquitto_pub and home assistant. Executing the command manually in telnet should tell you if/what may be wrong: 

If you execute the command by hand and it shows no errors but home assistant doesn't receive the notification then likely something is wrong home assistant side. 

If you execute the command by hand and home assistant receives the notification correctly then there may be something different in the log output or something wrong in log_parser.sh -- you could modify the top of the log parser to have this instead:
```
DEBUG_FILE=/mnt/mmc01/output.log
#DEBUG_FILE=$1
```
Then check/post the output.log file in the SD card after booting up and trying the doorbell.

pioneershahid, 2021-12-12:
@guino  Thanks once again for taking time and looking into it. 

I tried following commands in telenet with error `bin/sh: mosquitto_pub: not found` and looks like none of the mqtt commands works. 
`mosquitto_pub`
`mosquitto_pub -h localhost -t homeassistant/#`
`mosquitto_pub -t 'test/topic' -m 'helloWorld'`

I unintall mqtt broker and reinstalled it along with restarting home assistant. I also modified log_parse.sh file and reboot the doorbell (strangely as soon doorbell connected to wifi it powercycle automatically then connected statys connected to wifi -  not sure why). When I tried mqtt commands via telenet again I got the same error as above. 

As the doorbell connected to wifi, I started seeing messages in mqtt as the motion and doorbell pressed.
 
[image]

 Iguess my telenet mosquitto_pub commands are incorrect which is why i was probably seeing the error.  I also think reinstalling the mqtt broker probably started showing the mqtt messages. 

attached output.log for the reference not sure if you like to see it. 

thanks once again.

[output.log](https://github.com/guino/BazzDoorbell/files/7697949/output.log)

guino, 2021-12-12:
@pioneershahid in telnet (and in the log larser script) you have to execute the mosquito_pub command with the path to it:
/mnt/mmc01/mosquito_pub

Try it with the path above to see if it works in telnet.

pioneershahid, 2021-12-12:
sorry I misunderstand it previously.. I can report that it is successfully working via telenet and normally when button and motion pressed.   
`/mnt/mmc01/mosquitto_pub -h 192.168.x.x -p 1883 -u mqtt -P mqtt -t "homeassistant/binary_sensor/doorbellMotion/state" -m ON`

Thank you once again for your guidence and patience. really appreciate it..

pioneershahid, 2021-12-15:
@guino i was helping my friend to install the hack. he had same bell (IeGeek J5) ver 4.3.1. I then followed #4 advice and installed older Cloudedge v3.1.4 .apk, which shows an available firmware update to 4.3.3 (the play store / latest version of cloudedge did not offer this update). When applied, everything works great. 

just for others, log_parser.sh is probably only accept limited char as mqtt username and password. it only works for me when i use mqtt for both fields. when i tried 6 alphabet as username "shahid" and "11Jan2021" as password for example, it did not worked for me. I had to change them to mqtt to be able to communicate to HA.

guino, 2021-12-15:
@pioneershahid that is an interesting issue with the user/password limitation, I'll make a note of it in the wiki, thanks for reporting.

trizmark, 2022-01-04:
I returned to this project after I haven't been able to get it fully working. As I mentioned in my previous [comment](https://github.com/guino/BazzDoorbell/issues/35#issuecomment-981504667), I got as far as I'm able to telnet to the camera.

The strange thing is that my 'cgi-bin' directory keeps disappearing. By adding a couple of debug lines to `initrun.sh`, it looks like it's disappearing from the SD card during the run of `custom.sh`. Any idea why this would happen?

Also, if I copy the ppsapp to the root of the SD card and reboot the camera (which would cause `custom.sh` to kill the currently running ppsapp and run it from the card), I lose the ability to telnet to the camera + remote access via the cloudedge app. Am I doing something wrong here?

guino, 2022-01-04:
@trizmark can you post a zip of your SD card contents ? It sounds like the ppsapp from your firmware may be erasing the SD card contents because it thinks it's full. It is possible there's something specific to your firmware that prevents ppsapp from being killed and executed again (we have seen this in a number of firmware versions and/or hardware types). 

You could try backing up your SD card files then reformatting the SD card on a computer (or with the phone app) and copying the files back to see if the same happens (cgi-bin gets deleted). You could also try a different SD card altogether to see if cgi-bin gets deleted.

I would most definitely focus on one problem at a time -- ie either try to get cgi-bin to stay OR try to get ppsapp working but not both at once. Your copy of ppsapp could just be corrupted (thus why I'm asking you to post zip of SD card contents). You can try to get a 'fresh' copy of ppsapp by deleting the 'home' directory from the SD card and booting the device (and wait at least 5 minutes) before removing the SD card.

trizmark, 2022-01-04:
[sdcard.zip](https://github.com/guino/BazzDoorbell/files/7810442/sdcard.zip)
Here's a zip file of the SD card contents. I have tried to reformat the card and put the contents back, but cgi-bin keeps disappearing. It's a 32G card with more than 31G free space on it. (df usage shows 0% used)

guino, 2022-01-05:
@trizmark at a quick look I don't see anything that would explain the cgi-bin directory disappearing. 
Can you try this:
1-create a `www` directory in the SD card
2-create `cgi-bin` inside `www` with the correct files
3-move `index.html` and `upload.html` inside `www`
4-modify the httpd line in custom .sh to have `-h /mnt/mmc01/www` 
5-boot up and see if the www/cgi-bin still exists, and if so: try accessing http://user:password@ip:8080

If the above works you still may need to tweak a few scripts/files to work in the www/cgi-bin directory but at least it won't disappear. 

For your firmware play.cgi address is: 0x0483b94 and mjpeg/snap.cgi address is: 0x04802b0 -- I would expect you can enable onvif without patching by editing tuya_config.json as described here: https://github.com/guino/Merkury1080P/issues/9#issuecomment-926414826

trizmark, 2022-01-06:
Thank you @guino, that did work. The www & cgi-bin directories are not removed now.

Onvif can be enabled by editing /home/cfg/dev_settings.json (plus I realised that this setting is exposed in the CloudEdge app as well).

I am so close, yet so far right now. Without any modifications I've got the video feed integrated into HA via motionEye (as direct Onvif to HA had a massive lag). But I have no way of capturing the button press event.
If I kill the running ppsapp and start it from the SD card, then I can get the button press event into HA, but for some reason the device does not expose the RTSP / Onvif ports anymore. The doorbell doesn't play the 'ding-dong' sound when the button is pressed either. (CloudEdge is not able to connect to the restarted ppsapp, but that's not a huge issue as I'm not planning to use the app.)

Anybody found a way to capture the button press?

trizmark, 2022-01-06:
I think I cracked this one! The issue seems to be that when psapp is killed, not all resources are released, so the newly started psapp is unable to bind to the ports that were already in use by the previous instance.
I simply added a `sleep 5` between the `kill $PPSID` and the ppsapp start from /mnt/mmc01 and everything is working fine! (Sure, there's a 50/50 chance that the doorbell restarts due to the watchdog, but that's a minor issue.)

guino, 2022-01-06:
@trizmark That's great news! I was going to suggest you to use Dan's changes to start ppsapp right away (without ever needing to kill it). 

There's a watchdog feeder in the off-line cloud issue which you can try: https://github.com/guino/BazzDoorbell/issues/4#issuecomment-742434771 , just unzip it onto the SD card and add: `/mnt/mmc01/test_wdt &` before the line `kill $PPSID` and the watchdog should not reboot the device.

The only thing with this watchdog feeder is that it was compiled for armv7 and if your device has older hardware (armv5) then it won't run -- you can test to see if it works on telnet: just run `/mnt/mmc01/test_wdt` and it should say 'feeding' every few seconds. I did not write the tool so I can't compile it for armv5 as I don't have the sources code.

Some firmware versions also have a 'watchdog_feeder' tool in /sbin -- which would be worth a shot if it's available.

trizmark, 2022-01-11:
@guino I didn't try the WDT feeder as I can accept the possible restart once the device boots. It's not really an issue. The doorbell has been working fine for the last 2 days.
I've been trying to improve the HA integration (as the MQTT provisioned entities need re-provisioning once HA restarts), but I would need to get my hand on a mosquitto_sub that would run on the doorbell. I tried to compile everything with buildroot, but so far only managed to compile binaries that segfault. 🤷 Got any pointers (or a binary that you can share)?
Also, beer-money is on the way! 🍺

guino, 2022-01-11:
@trizmark you would need to compile the tool with the right toolchain (armv5 uclibc) so it runs correctly. Definitely try a hello world first then try something bigger. 

Here’s the link to the toolchain I used to compile for armv5:
—editing—

The [Merkury720 repository](https://github.com/guino/Merkury720) has the source code and binary file for jpegarm available to compare/test - just run it without parameters and it should show a usage message if it works.

Thanks for the beers! 
Cheers!

guino, 2022-01-28:
@trizmark I have posted an armv5 mqtt publisher tool in the repos recently -- just thought I'd let you know: https://github.com/guino/BazzDoorbell/issues/59#issuecomment-1023517550

trizmark, 2022-02-15:
Thank you @guino ! With the toolchain I managed to compile the MQTT subscriber, which I needed to properly integrate the doorbell with Home Assistant. (Auto-configured entities need reconfiguring if HA is restarted.)
I'm happy to post my script here if anybody is interested.

Oh, and I also resolved the disappearing cgi-bin directory issue. Lessons learnt: always, always, *always* throw away the SD card that comes with the device and replace it with one from a reliable/reputable brand. Simple as that.

alexbgreat, 2022-02-20:
In case anyone finds this thread and is bummed out they have to go through all the trouble of cross compiling mosquitto for Armv5 because they need a username and password for publishing to work right in their environment... fear not. 
I've attached static builds of mosquitto_pub and mosquitto_sub. I've tested mosquitto_pub to be working on Armv5. Compiled without TLS support, sorry.
[mosquitto_armv5.zip](https://github.com/guino/BazzDoorbell/files/8105080/mosquitto_armv5.zip)

trizmark, 2022-04-04:
I'm back. 😞 So my doorbell has been working fine for a few weeks. Got nicely integrated with HA, push notifications to Telegram/Pushover, got ssh running on it with sftp. Came home from my morning walk and it's flashing red. Took it off, can't see anything wrong on the card. Put ppsFactoryTool.txt back and it boots fine. Connects to WiFi, starts telnet/ssh. Remove ppsFactoryTool and nothing. Just flashing red.
Any ideas as to how to troubleshoot this issue?

guino, 2022-04-04:
@trizmark Did you try booting without the SD card at all ? If you get red blinking light booting without the SD card it could only be because of internet connection issues (to tuya servers) or hardware issues (i.e. flash memory corruption, issues with the video capture sensor, etc). Since you said it boots in factory mode (ppsFactoryTool.txt present) the underlying hardware may be ok but it may be failing when trying to start other hardware features not initialized while in factory mode (i.e. the audio+video capture). I also recommend total power off, waiting a few minutes then power on (not just a reboot) -- one of my cheap indoor cameras freezes every now and then and if I just reboot it doesn't completely start again, but if I power off, wait a few minutes and power back on it usually boots normally (sometimes takes a few tries).

I have heard of at least 1 user who had a flash memory corruption due to powering the device on/off numerous times -- flash memory is susceptible to corruption if the device is writing at the moment power is cut off, these devices don't write frequently to the flash but it's possible (and only way to know for sure is with a hardware programmer).

trizmark, 2022-04-04:
@guino The camera doesn't start without the SD card as the hack I applied the hack from #2 . I added some debug messages to initrun.sh and custom.sh and everything seems to be executed, but the camera does not connect to wifi. I even added a netstat to custom.sh and I can see ssh/telnet/www all listening. Not sure where to go from here.
Should I try to revert the mod and see if the camera works that way and start over.
Oh, one more thing: I *think* the startup sound/chime of the camera has changed. It's a different tune compared to what it was before. (Hope it's not my memory failing me 🤔)

guino, 2022-04-04:
@trizmark if the startup sound changed, the camera ma y have gotten an update (without your permission) -- try removing the ppsapp file and the 'home' folder from the SD card and let it boot to see if it works. Regardless of working or not check if the ppsapp in the home folder (will be recreated) is the same as the one you had before.

There are 'troubleshooting' steps to 'remove' the hack on #2 which you can use to allow the camera to start normally (without SD card), you try that and re-apply the hack later if you wish to try but I'd check the above first -- chances are it got an update and the old ppsapp (from SD card) may not run correctly with the new firmware and you just need to use the new one.

trizmark, 2022-04-23:
@guino The changed startup tone is a red herring - it's probably my memory failing me. I removed the home folder and booted the camera, but the two ppsapp binaries have the same MD5, so there was no update.
Unfortunately, the camera was only working with the ppsFactoryTool in the root, which meant no video. That is until today. I noticed that the camera is off the network. Tried rebooting it, but it no longer boots. Just get a flashing red light. Tried removing the hack, but no luck. No boot, just red flashing light. How long do I have to hold the bell button when using the ppsMmcTool? The guide says 5 seconds from power on, but I don't see it doing anything different when trying to revert the hack. Got a feeling that at this point it's ready for the bin.

guino, 2022-04-24:
@trizmark if the camera booted and created the home directory but didn’t start ppsapp completely (red light) it could be something wrong with the video hardware. 

If you have restored the original settings (troubleshooting steps on #2) and it is not booting  then either the reversal didn’t work or there’s something wrong with the hardware. 

Can you try to get a current /proc/cmdline from the device using ppsFactoryTool.txt? That would tell us the current state. Alternatively you can simply apply #13 and see if you can enable telnet (this may work even though ppsapp isn’t starting completely).

It is ppsapp that controls the led state, so if it doesn’t turn blue it means ppsapp may not be finishing the startup process but it doesn’t mean it isn’t running (`ps -w` Command on telnet would tell us).

The reset button on power on simply tells the device to read ppsMmcTool.txt which sets the boot command line using the env file. I think it may blink blue briefly when it is reading ppsMmcTool.txt but it may be different on some devices. When not using reset button it would probably stay red until later in the boot process. The 5 seconds is just enough time to make sure the device started to boot after power on. Anyway, the only thing to compare is boot without holding reset vs boot holding reset to see if there’s any LED status difference during the first few seconds.

You could also try using ppsFactoryTool.txt and holding reset button after it has been powered on for a few minutes to see if it can reset it back to factory state.

you are welcome to post a zip of the files you’re using to restore original boot settings and/or apply #13 and I will be happy to review.

Lastly if you changed SD cards after you originally applied the hack, there’s a chance the new SD card may not be working (during boot) to change the boot settings — most SD cards work after boot of the device but many don’t work with the boot loader (first few seconds after power on).

trizmark, 2022-04-25:
@guino So, after everything was working, the first thing that went was the video stream. One morning the red light was flashing on the doorbell and it would simply not boot up properly.
Managed to get it on the network using the ppsFactoryTool.txt. This meant the the camera was on the network, I was getting notifications via MQTT, but no video. Also, the RF chime was working.
Day before yesterday the doorbell fell of WiFi and I have been unable to get it on the network. Tried removing the hack using the attached files on the SD card (which has not been changed and has been working fine). With the ppsMmcTool in place all I get is a solid red light. Nothing else. It sounds that the doorbell might be rebooting every 30 or so seconds, but it's still only solid red light.
With ppsFactoryTool in place, the doorbell boots, but never joins WiFi. First the red light blinks slowly (boot started), then flashing frequency speeds up (WiFi connection attempt). Got a UniFi controller, but cannot see any trace of it actually trying to connect.
At this point I'm not sure what else to try.
[doobell_restore.zip](https://github.com/guino/BazzDoorbell/files/8553454/doobell_restore.zip)

guino, 2022-04-25:
@trizmark your files look fine. If the device isn't connecting to the wifi with ppsFactoryTool.txt the only thing you can do is try different SD cards to do the restore (using the files you just posted) -- if one SD card works the device should boot completely (at least to the point of hearing the startup sound). It would not hurt to keep the initrun.sh from #2 in the SD card just in case the hack from #2 is still installed (and hasn't been removed by your attempts). If you exhaust your SD card options I'd repartition+reformat the SD cards available and try again just to be sure. If your device is still under the warranty period I'd try to get a replacement under warranty (as the changes you have performed should have not damaged the hardware in any way -- but this is sounding like a hardware issue).

trizmark, 2022-04-26:
@guino ok, I am totally embarrassed. I really should read the manual before posting stuff. I've been pressing the doorbell button instead of reset. Guess what? As soon as you start doing the right things, you end up with a working doorbell. 😭
Now that I'm back to where I started, I can try to mod the bell again!

guino, 2022-04-26:
@trizmark I am just glad your device is working, hopefully you can set it up like you wanted again.

trizmark, 2022-04-27:
@guino It only took 20 mins to get it back to a state I wanted. Everything is working well again.
One last thing: is there a way to get a hi-res snapshot on the camera? I am using the ONVIF integration with HA and when I take a screenshot it takes 4-5 seconds. If I use the snap.cgi upon button press, the image is instantaneous, but I'm only getting a low-res pic.

guino, 2022-04-27:
@trizmark  Glad you got it all working. The resolution of the jpeg images is fixed by the device (hardware encoder) and are always low resolution. You can get a high-resolution snapshot using something like this https://github.com/guino/rtsp2jpeg but it will add considerable delays unless you're constantly taking snapshots. I have tried to change the code to see if I could make high-res JPG snapshots but was not successful.

pioneershahid, 2024-02-03:
folks, i am back with another query and hope if anyone can help me.. 
I have the hack successfully deployed and working. I can see i have onvif enabled by default. i was hoping if i could get iegeek doorbell motion and button pressed into Home assistant. i followed the advise and the motion and button trigger working intermittently. 

Problem: when I connect doorbell to power and it turns on fine. it detects motions and sends trigger to Ha. However, within few seconds, it restarts and stays connected. when i try to press the button, nothing registers in doorbell.. if i restarts doorbell via telenet or powercycle it. same things happens again. 

[image]

i know there is somewhere minor issue with the code and cannot seem to comprehend what needs to be fixed .. here is the content of my sd card please thank you.

[6 more replies]