guino/BazzDoorbell #36: Orion Grid Connect Smart Wireless Video Doorbell / Bell 7S

NickStallman, 2021-06-14:
This is partially to look for the correct solution for this device, and partially to get this seemingly newer device to show up in Google (is there a hardware list page btw?).

I got a "[Orion Grid Connect Smart Wireless Video Doorbell](https://www.bunnings.com.au/orion-grid-connect-smart-wireless-video-doorbell_p0178707)" which is the same as a "[Meari Bell 7S](https://www.meari.com/bell-7s-product/)"

http://admin:056565099@192.168.1.x:8090/devices/deviceinfo does successfully pull the device info.
```json
{
        "devname":      "Smart Home Bell",
        "model":        "Bell 7S",
        "serialno":     "062879218",
        "softwareversion":      "3.2.1",
        "hardwareversion":      "",
        "firmwareversion":      "ppstrong-b5-tuya2_arlec-3.2.1.20200612",
        "licence_id":   "pp01cfxxxxxxxxxx",
        "licence_key":  "pAlMjeHtMKDDzxxxxxxxxxxxxxxxxxxxxxxxx",
        "identity":     "MR20080300000000000",
        "authkey":      "pAlMjeHtMKDDzwBy5xxxxxxxxxxxxxxxxx",
        "deviceid":     "pp01cf0ce903exxxxxxx",
        "pid":  "vFHpaEFwuxxxxxxxxxx",
        "WiFi MAC":     "7c:25:da:8c:xx:xx",
        "ETH MAC":      "08:88:12:94:xx:xx",
        "mcuversion":   "4.2.3.20200528"
}
```

However /proc/cmdline unfortunately hangs, no error, no HTTP response at all until the HTTP connection times out.

I've also attempted to dump the flash via the ppsMmcTool.txt method however unfortunately that doesn't seem to be doing anything right now.

I'll come back to this soon and do some more tinkering and post my results. :)

allencar99, 2021-09-01:
Hi Nick. Have you had any more luck with getting into the Bell7S (arlec) firmware? Just bought one today, pulled it apart and thinking it may be a valid candidate for a video hack - 1080p with built in PIR, Speaker and Mic. The Hi3518 SoC has an RS232 interface so thinking it may be worth while pocking around there first, any thoughts?  I'm also looking for any valid reason why there is a big sticker above the external power lugs saying "WARNING NOT TO BE USED WITH AC POWER SUPPLY" when the Hangzhou Meari BELL7S manual clearly states 12~24VAC for those same power lugs? Thinking Arlec doesn't trust DIY people enough to connect the correct power adapter - and they are probably right 😏

NickStallman, 2021-09-01:
Nothing yet, I've taken it apart but I managed to misplace my serial adapter so I haven't hooked in to the serial port yet. :)

I had no luck with the SD card exploits so far.

fermentedmilk, 2022-01-10:
I have a rebadged version of the bell 7s.  The ppsFactoryTool.txt worked. I was able to do log upload and save logs onto the SD card.

Tried Merkury 1080/720, BazzDoorBell hacks and no luck loading /proc/cmdline. It seemed as if initrun.sh did not run as i tried to create a new file on the SD card during boot and it not happen. checked the file permissions and it's set to executable.

can anyone kindly point me in the right direction on what else i should try without going down the path of access via UART?
Thank you 

{
	"devname":	"Smart Home Bell",
	"model":	"Bell J1 2.0",
	"serialno":	"065000000",
	"softwareversion":	"2.1.0",
	"hardwareversion":	"",
	"firmwareversion":	"ppstrong-b8-neutral_std-2.1.0.20210609",
	"licence_id":	"ppslxxxxxxxxxxxx",
	"licence_key":	"xxxxxxxxxxxxxxxx",
	"identity":	"M4R00xxxxxxxxxxxxxxx",
	"WiFi MAC":	"c0:e7:bf:ff:ff:ff",
	"ETH MAC":	"08:88:12:ff:ff:ff",
	"mcuversion":	"neutral-2.1.0.20210604"
}

guino, 2022-01-10:
@fermentedmilk so you got /devices/deviceinfo to work (above posted) but did not get /proc/cmdline to work ? When /proc/cmdline doesn't work it is usually a sign that the device is not running linux firmware (i.e. RTOS) and without linux none of the hacks/scripts/tools we have here will work.

The 'log upload' you mentioned is something on your phone app ? 

Were you able to check this url:
http://admin:056565099@IP/proc/self/root/etc/init.d/S90PPStrong

Can you also check this one:
/proc/self/root/etc/init.d/S80network

I don't believe we have confirmation of any 2.1.x version being rooted (I certainly have not seen this firmware version), but if the above URLs work you may have a chance of getting it rooted.

fermentedmilk, 2022-01-11:
@guino Thank you for your response. 

Unfortunately, it doesn't load anything file from the /proc dir. /proc/cmdline, /proc/self/root/etc/init.d/S90PPStrong and /proc/self/root/etc/init.d/S80network were not able to be loaded. HTTP ERROR 404 was returned.

http://admin:056565099@IP:8090/log/upload can be executed via a web browser 

do you happen to have a copy of a linux based firmware that i can flash the device with? (i will not hold you accountable if the device get bricked)

i was able to run most of the commands including the flash upgrade commands from this web site after applying  ppsFactoryTool.txt which then opens port 8090. other ports such as 80 and 53 are closed.
https://research.nccgroup.com/2020/12/18/domestic-iot-nightmares-smart-doorbells/ e.g. /flash/upgrade/ppstrong. 

here is a photo of the board
[image]

guino, 2022-01-11:
@fermentedmilk I don't have any firmware for Bell 7S, even if you find one you want to be sure the hardware ID matches entirely or it would likely not work PLUS you would likely not be able to use it online at the same time as the device from which the firmware was obtained.

The fact that there's no 'Bell 7S' patch ever listed in https://github.com/guino/ppsapp-rtsp/issues/1 is probably a good indication that there's probably no linux style firmware for that hardware (ever).

fermentedmilk, 2022-01-12:
@guino Thanks again for your response. You have done some excellent work in share your knowledge with beginners like me. i have learnt a lot from this forum. 

I will try and hook the camera up via UART and share the logs (if successful) when I have some more spare time. 

Alternatively, i will buy a new door bell from the ppsapp-rtsp list for the purposes of gaining more control over the device.

BurrPro, 2022-05-01:
@fermentedmilk did you have any success? I purchased one earlier today to play around with, and I've had no luck even finding the UART pins

BurrPro, 2022-05-01:
> Hi Nick. Have you had any more luck with getting into the Bell7S (arlec) firmware? Just bought one today, pulled it apart and thinking it may be a valid candidate for a video hack - 1080p with built [...]

I noticed that according the the fcc filings for the bell 7s, theres some chips missing on the arlec version. This will be why we're not allowed to use AC 👎

roabyd, 2022-06-16:
Has anyone had any further luck with this? I'm hoping this Orion dorrbell has not been a wasted purchace... The length of time the video takes to render in the smart life/Grid connect app makes it essentially useless as a doorbell, so If there is a way to have to stream locally to home assitant I would love to know.

NickStallman, 2022-06-16:
Nah I ended up going with a Eufy doorbell instead. No alternate firmware for it but it has been throughly reverse engineered and has good Home Assistant support. Not perfect, but close.