guino/BazzDoorbell #37: unbrick/recover led stays red.

spacecabbie, 2021-07-01:
Hello folks,

Just wondering it happened 2 times now first time I just exchanged it. 
Basically the sd-card fills up I forget to test the remove of recording scripts.
The the device will get stuck at a red light. If I restore the sd-card from backup and use the 5 sec reset it will flicker blue short while and immediately back to red. 

I did managed to get a dump of the device but no clue how to open or read it.

Any suggestions what's going on ? or how to procceed with the dump and how to mount and where to look for indication of the problem.

guino, 2021-07-01:
@spacecabbie Once the hack is installed you should never **need** to re-do the process of holding reset for 5 seconds during boot -- the hack will stay in the device even if the SD card is erased, so all you need to do is to replace the SD card files and reboot and it should work (with the hack) as it did before.

The instructions are clear that if the SD card gets filled (i.e. with recordings) the files will be erased and the hack will not work until you replace the files (thus why I recommend a backup once everything is working). The only way to prevent the SD card from filling up is to disable recording or configure it so it gets cleaned daily while using motion-recording.

If you used #2:
The device can boot without the hack as long as you have initrun.sh (and no other files) on the SD card -- please be sure to try different SD cards and make sure the SD card is formatted as FAT32.

You CAN switch from #2 to #13 by just following the instructions using the original /proc/cmdline from the device and just following the instructions.

If you used #13:
If it doesn't boot without the SD card then your device is simply damaged and won't work (with or without the hack).
If the device boots up FINE (and works) without the SD card, there's nothing wrong with the device (good).

With #13 installed you should be able to boot without SD card and 'check' that the hack is installed by opening the URL under /proc/cmdline -- it should show the the same as after the hack was installed.

If the device boots up fine without SD card but doesn't boot up with the SD card you should:
1-Verify the version of the firmware (in the phone app) **without** the SD card to make sure it is the same version from when you installed the hack. Some devices update firmware without telling you and then you end up trying to run an old version of the ppsapp with a newer firmware and that can freeze up the device.
2-You should try a different SD card -- sometimes the SD card appears fine (and may be for the most part) but bad sectors develop over time and can cause the scripts/files in the SD card to not work correctly in the device (even if they seem fine on the PC).
3-Check your files, make sure you have the right scripts, the right ppsapp, filenames, sizes, etc.

If you checked all the above feel free to post the output of /proc/cmdline (i.e. from now or when you installed the hack) and a zip of your SD card files (except for SDT folder) so I can review.

spacecabbie, 2021-07-09:
> @spacecabbie Once the hack is installed you should never **need** to re-do the process of holding reset for 5 seconds during boot -- the hack will stay in the device even if the SD card is erased, [...]
Prob what i did wrong in the first place, formatted card and copied backup to it then reset.

> The instructions are clear that if the SD card gets filled (i.e. with recordings) the files will be erased and the hack will not work until you replace the files (thus why I recommend a backup once [...]

Yup learned from the first time had made a nice backup but i managed to fubar it anyway.

> If you used #2:
> The device can boot without the hack as long as you have initrun.sh (and no other files) on the SD card -- please be sure to try different SD cards and make sure the SD card is formatted as FAT32.
> 
I tried with 3 diff cards 1 64gb original but due to fat32 and wanted to make absolutely sure also got 2 32gb cards.
Ok  did some trials
Booting with clean sd card 32gb with only init.sh from here on github the device turns on red led no action/sounds whatsoever if i look at the sd card again it looks like this: https://i.imgur.com/b0tSadP.png

Now when i boot again with a clean sd card 32gb but use my init.sh from my backup again just red led nothing but after a minute or so you hear the little chrip that it reboots.(if I leave it on it continues to do this non stop. If it then look on the card it has rebuild some of the directory structure on the card: https://i.imgur.com/l6f6RN1.png
[sd-card.zip](https://github.com/guino/BazzDoorbell/files/6791564/sd-card.zip)

I tried then to overwrite the files with my backup but this makes no diferance it just loops.

It never requests a ip or starts a wifi lan so i cannot talk to it.

> You CAN switch from #2 to #13 by just following the instructions using the original /proc/cmdline from the device and just following the instructions.
> 
> If you used #13:
> If it doesn't boot without the SD card then your device is simply damaged and won't work (with or without the hack).

Didn't use 13 

> If you checked all the above feel free to post the output of /proc/cmdline (i.e. from now or when you installed the hack) and a zip of your SD card files (except for SDT folder) so I can review.

I am sure I fubared it so quite thankful for your help so far if you have any other insights happy to hear them

guino, 2021-07-09:
@spacecabbie the fact that your SD card is getting corrupted (as shown in your picture) is not a good sign -- does this happen with EVERY SD card you tried ?

If you know your original /proc/cmdline you should try the Troubleshoot / Restore steps in #2 which should restore the device to factory settings (your original /proc/cmdline is in your env file if you made a backup).

Alternatively you could just apply #13 using your original /proc/cmdline -- it alllows the device to boot normally without a SD card but also allows for a custom ppsapp to be used (when using SD card).

Regardless of what if every SD card you insert is getting corrupted your device may have some sort of hardware malfunction in regards to the SD card slot.

spacecabbie, 2021-09-03:
> @spacecabbie the fact that your SD card is getting corrupted (as shown in your picture) is not a good sign -- does this happen with EVERY SD card you tried ?
>
> If you know your original /proc/cmdline [...]

This resolved it seems it couldn't handle the 64gb card took me a while to get a smaller sd private life stuff etc.

closed 2021-09-03 by spacecabbie