guino/BazzDoorbell #38: Lidl LSC Smart Wireless Video Doorbell / Bell 8S

SargonofAssyria, 2021-07-11:
I have a Bell 8S with
[image]
I have 2 rtsp streams on
rtsp://admin:admin@192.168.xxx.xxx:8554/Streaming/Channels/101
rtsp://admin:admin@192.168.xxx.xxx:8554/Streaming/Channels/102
They work fine.
The trick with ppsFactoryTool.txt works too, and opens port 8090 as HTTP but username/password don't work.
When port 8090 is open the rtsp port 8554 is closed.
In user management there are 2 users WeEyE and admin both as role administrator.
I would like to discover if there is a MJPEG and/or snapshot url.
I know I could snapshot from the rtsp stream, but I would to do it directly for performance reasons.
Who can help, or has a good hint?

SargonofAssyria, 2021-07-12:
Found out that the username/password of WeEyE/&$ChuTian_91 works on the url http://192.168.2.181:8090/search
Output:
// 20210712161647
// http://192.168.2.181:8090/search

{
  "sn": "064876126",
  "factory_code": 0,
  "factory_code_str": "",
  "model": "Bell 8S",
  "p2p_uuid": "",
  "ip": "192.168.2.181",
  "mask": "255.255.255.0",
  "gw": "192.168.2.1",
  "mac": "b4:fb:e3:da:27:ad",
  "interface": "wlan0",
  "version": "5.0.5"
}
Could not find other urls that work.

SargonofAssyria, 2021-07-13:
Sorry, the above username/password was NOT needed for the url http://192.168.2.181:8090/search
Only for other url's there is a password prompt. I tried all kind of combination of username/password but none worked.
Any hints?

guino, 2021-07-13:
@SargonofAssyria I am sorry to say that if your device doesn't return anything under /proc/cmdline it is likely NOT running linux (runs RTOS) and there's no mod available to make MJPEG/SNAP work in RTOS. RTOS seems to be common for 5.x firmware right now.

SargonofAssyria, 2021-07-14:
Ok, Thanks for the response. Have to do it with the workaround by processing the rtsp stream for getting a snapshot.
The stream works fine with my Synology Surveillance Station.

closed 2021-07-14 by SargonofAssyria