guino/BazzDoorbell #39: SSH or SSHD possible?

WeterPeter, 2021-07-15:
Hi,

is it possible to get ssh or even a sshd running on this doorbell?
any tips on a ‘how to’ ?

regards,
Peter

guino, 2021-07-15:
There's not a lot of memory in these devices so I would stick with telnet/http and tftp but if you want to go the SSH route your best bet is Dropbear which you'll have to cross compile from the sources. This is your best bet to start with: https://topic.alibabacloud.com/a/dropbear-enable-ssh-for-arm-embedded-linux-cross-compilation-based-on-busybox-root-file-system_1_16_30174395.html 

I had tried it before and got the executable to run but ran into a missing symbol error when trying to use it (it would find a symbol which usually means the uClibc which usually means the uclib toolchain I used doesn't match the one on the device.

I didn't want to spend a lot more time on it so I never tried to find a better matching toolchain.

If you do get it working, please post the binaries so others can use it in the future.

guino, 2021-07-15:
@WeterPeter actually still had the stuff on my machine and managed to compile dropbear in static mode (about 3 times the size it can be) but it worked fine (for server and client including scp):
[dropbearmulti.zip](https://github.com/guino/BazzDoorbell/files/6824617/dropbearmulti.zip)

/mnt/mmc01/dropbearmulti ssh user@host
/mnt/mmc01/dropbearmulti scp user@host:/path/file /path/file

(server mode)
mkdir /etc/dropbear
/mnt/mmc01/dropbearmulti dropbearkey -t rsa -f /etc/dropbear/dropbear_rsa_host_key
/mnt/mmc01/dropbearmulti dropbear

You can put the server commands in custom.sh if you want, I just can't vouch for the memory usage.

WeterPeter, 2021-07-15:
Thanks, I got sshd running using a stipped version of [https://github.com/bol-van/bins](https://github.com/bol-van/bins). 
In the end I used only curl for a call to Home Assistant to get my old doorbell ringing again.

My doorbell.sh script on caching the press on the button and call HA :

while true
do
        busybox dmesg -c 2>/dev/null | grep door
        RET=$?
        if [ $RET = 0 ]
        then
                /mnt/mmc01/curl -X POST -H 'Authorization: Bearer  <your Long-Lived Access Token herr>' -H 'Content-Type: application/json'  -d '{"entity_id": "switch.<NAME>"}'  http://<IP>:8123/api/services/switch/turn_on
                sleep 1
                /mnt/mmc01/curl -X POST -H 'Authorization: Bearer <your Long-Lived Access Token here>' -H 'Content-Type: application/json'  -d '{"entity_id": "switch.<NAME>"}'  http://<IP>:8123/api/services/switch/turn_off
        else
                sleep 1
        fi
done    

Add the following to initrun.sh to get it running :

if [ -e /mnt/mmc01/doorbell.sh ]; then
  /mnt/mmc01/doorbell.sh &
fi

Thanks for your work, I got me started on a great solution !!

Nigel1992, 2021-07-27:
> @WeterPeter actually still had the stuff on my machine and managed to compile dropbear in static mode (about 3 times the size it can be) but it worked fine (for server and client including scp):
> [dropbearmulti.zip](https://github.com/guino/BazzDoorbell/files/6824617/dropbearmulti.zip)
> 
> /mnt/mmc01/dropbearmulti ssh user@host
> /mnt/mmc01/dropbearmulti scp user@host:/path/file /path/file
> 
> (server mode)
> mkdir /etc/dropbear
> /mnt/mmc01/dropbearmulti dropbearkey -t rsa -f /etc/dropbear/dropbear_rsa_host_key
> /mnt/mmc01/dropbearmulti dropbear
> 
> You can put the server commands in custom.sh if you want, I just can't vouch for the memory usage. If you run as server you should probably just create the key file once and just copy it to /etc/dropbear instead of generating the key every boot (which would take time/memory/cpu).

Works, but I cant transfer files.

""Cannot execute SCP to start transfer. Please make sure that SCP is installed on the server and path to it is included in PATH. You may also try SFTP instead of SCP.
Command failed with return code 127.
"""

guino, 2021-07-27:
@Nigel1992 I only tried to login and copy files from the device to a computer, it seems to scp from a computer you need to have the scp command in the device. I have updated the server commands to create a link to provide the scp command so that should work -- check the 3rd post again for the 'ln' command added.

Nigel1992, 2021-07-28:
@guino Works perfectly, thanks!