guino/BazzDoorbell #40: I got a new version of the LSC Doorbell from Action firmware V5.05 - My attempt to hack failed

WeterPeter, 2021-07-17:
Hi,

I have followed the hack steps for the version 4.0.x firmware but I fail at the step to get a response form poort 8090.
Port 8090 opens up after I reset 5 sec. with the hack files in the sd-card but no response on http://admin:056565099@<ip>:8090/devices/deviceinfo
nor 
http://admin:056565099@<ip>:8090/proc/self/root/mnt/mmc01/hack

Firmware: 
  Main module: c5.05
  MCU Module: V5.05

Any tips on how to proceed?

Peter

guino, 2021-07-17:
All 5.x firmware I have seen runs RTOS (LiteOS) which is NOT linux, meaning none of the scripts/applications we put together will work on it. 

If your device takes ppsFactoryTool.txt (to connect to wifi and open port) but doesn’t return anything for /proc/cmdline then you like have a non-Linux device.

The only possible way to make any changes to a non-Linux device is to download the flash, modify it and write it back - even then you will be very limited in what you can change.

WeterPeter, 2021-07-17:
Wow, that means it is a no go for me.
Thanks for your quick answer!

regards,
Peter

Ierlandfan, 2021-08-12:
I have the same camera (As well as a soft bricked 2.9.6 one) and an Onvif scan shows port 8000 open on fw 5.0.5. Now trying to find the password...

guino, 2021-08-12:
@Ierlandfan check this post: https://github.com/guino/BazzDoorbell/issues/35#issuecomment-853457003 it neither works I would expect that if you use the right app (tuya or LSC) it should allow you to set the password if onvif is actually available. Other than that the only way to find out would be if we got a copy of the firmware which would require opening and using UART or a hardware programmer.

bertbijnens, 2021-08-18:
@guino 

> All 5.x firmware I have seen runs RTOS (LiteOS) which is NOT linux, meaning none of the scripts/applications we put together will work on it.
>
> If your device takes ppsFactoryTool.txt (to connect to [...]

I've got a 5.0.5 LCS device but these paths do work!
`http://admin:056565099@ip/proc/self/root/etc/init.d/S90PPStrong` doesn't work though

guino, 2021-08-18:
@bertbijnens if S90PPStrong doesn't work then you're most likely not running linux -- unless they moved into a new linux version with a different layout of OS files that I haven't seen yet.

jilleb, 2021-08-30:
While I feared I had bricked my doorbell, I got an additional one, which turned out to be this V5.05 version.
I took it apart, and took pictures of the board:

[image]

Comparing it with the previous, hackable version,  it's quite different, have a look:
[image]

The UART connection is different(or absent, but I higly doubt that), and the  model number is Bell 5T instead of Bell 5S.
Besides the position of some obvious things and mountings, about 75% of the board is different.

almirus, 2021-08-30:
[image]
possible UART

jilleb, 2021-08-30:
that's the old board

guino, 2021-08-30:
I would expect the UART to be one of the two sets of 4 pads on the top left of the picture on the new board - likely the one that is straight (not the square shaped one).

jilleb, 2021-08-31:
Yes, I thought so too. Unfortunately, my soldering skills (and iron) aren't as precise to be able to solder into such a small spot... I wonder if there's anything useful  on the USB connector.

AxelBaas, 2021-08-31:
Already triend USB, nothing there. Seems te be just for powering the doorbell.

guino, 2021-09-06:
@jilleb I just stick 3 of these headers in the holes soldered to the USB-TTL adapter: 
[image]

You may need to 'shave off' a little bit of the header to make it fit but that's how I've always done it. Still on 5.x I would not have much of a hope to do anything on the serial port other than (maybe) be able to download the firmware using the bootloader.

jilleb, 2021-09-06:
Yeah, that's what I normally use as well, but on the new board, there are no holes, just pads. I need to get myself a smaller soldering tip and a magnifying glass 😉

AchilleTalon-Lefuneste, 2021-09-11:
@bertbijnens @jilleb 

I'm interested: close setup, and mine answers to http://x.x.x.x:8090/devices/deviceinfo
It returns :

<html><body>

devname | "Smart Home Camera"
model | "Bell 8S"
serialno | "xxxxxxx"
softwareversion | "5.0.5"
hardwareversion | "BE8S_S1_V10_433"
firmwareversion | "ppstrong-c71-tuya2_lsc-5.0.5.20210301"
identity | "xxxxx"
authkey | "xxxxx"
deviceid | "xxxxx"
pid | "xxxxx"
WiFi MAC | "AA:BB:CC:DD:EE:FF"
ETH MAC | "AA:BB:CC:DD:EE:FF"

</body>
</html>

Firmware is 5.0.5. So we may have the same piece of technology. Let us know if you have any success.

edekeijzer, 2021-09-14:
I have the same device (exactly the PCB as shown in the top image from jilleb) with firmware version 5.0.6, did an nmap scan on it but it only has ports 6668, 8000 and 8554 (after enabling ONVIF) opened.

guino, 2021-09-14:
@edekeijzer you are lucky thtat your device has factory ONVIF (which should include RTSP as part of it). Most likely your device is not running linux so the features you have is all you'll ever have. That said if I could get a hold of the firmware of an ONVIF device (like yours) it is likely I could find if the non-ONVIF devices have similar code and hopefully be able to enable it on those. Still features like native mjpeg/snapshot/motion detection/mqtt will likely never be possible on these non-linux devices -- the best you can do is use the existing features (ie ONVIF) connected a server somewhere which does the motion detection and mjpeg/snapshot/mqtt integration. Personally if I got a device like that I would probably return it (if I had the option).

edekeijzer, 2021-09-15:
@guino 8554 is the RTSP stream indeed. Do you have any clue on how to extract the firmware? I'm not afraid to void warranties, so if I have to solder a UART header somewhere and hook it up to a USB-serial converter I'd surely try that for you.
I'll use it until I find a proper self-hosted solution which allows for two way communication.

guino, 2021-09-15:
@edekeijzer You may be able to use the UART+Bootloader commands to download the firmware from the device onto the SD card -- this however would depend on the version and commands available on the bootloader, so I have no way of knowing the exact commands that would work for you. You could try #11 first to see if it works -- maybe the address is the same. Using the UART+bootloader commands would probably require you to prepare the SD card as described in #11 anyway.

The 'for sure' way of getting the firmware would be using a hardware programmer but it would likely require removing the flash chip with a heat gun for it to work, so I can definitely understand if you don't have the equipment/time to do it.

jilleb, 2021-10-05:
Allright all, a small success here! :-D

Putting  ppsFactoryTool.txt from https://github.com/guino/Merkury720 on it opened up port 8090, and I was able to reach 
http://IP:8090/proc/cmdline
Output:

`console=/dev/null LX_MEM=0x3fe0000 mma_heap=mma_heap_name0,miu=0,sz=0x1d00000 pcbversion=BELL5S_S1_V10 sensor=sc3335mipi`

Now I need to take some more steps...

jilleb, 2021-10-05:
Dumping useful info here, updating as I go along

http://IP:8090/proc/mounts

`rootfs / rootfs rw,size=17068k,nr_inodes=4267 0 0
proc /proc proc rw,relatime 0 0
sysfs /sys sysfs rw,relatime 0 0
tmpfs /tmp tmpfs rw,relatime 0 0
/dev/mtdblock4 /home/cfg jffs2 rw,relatime 0 0
/dev/mtdblock3 /home/tmp jffs2 rw,relatime 0 0
/dev/mmcblk0p1 /mnt/mmc01 vfat rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,usefree,errors=continue 0 0`

http://IP:8090/proc/self/root/home/cfg/tuya_config.json
`{
	"version":	1,
	"sleep_mode":	0,
	"alarm_fun_onoff":	1,
	"alarm_fun_sensitivity":	1,
	"alarm_fun_mode_switch":	0,
	"alarm_fun_time_start":	0,
	"alarm_fun_time_end":	0,
	"flip_onoff":	0,
	"light_onoff":	1,
	"night_mode":	0,
	"sound_detect_onoff":	0,
	"sound_detect_sensitivity":	0,
	"watermark_onoff":	0,
	"event_record_time":	60,
	"enable_event_record":	2,
	"record_enable":	1,
	"motion_trace":	1,
	"motion_area_switch":	0,
	"motion_area":	"",
	"motion_tracking":	0,
	"cry_detection_switch":	0,
	"humanoid_filter":	0,
	"loudspeaker_vol_pct":	80,
	"jingle_mode":	0,
	"jingle_sound":	1,
	"jingle_volume":	100,
	"jingle_exist":	0,
	"onvif_enable":	1,
	"onvif_pwd":	"admin"
}`

guino, 2021-10-05:
@jilleb so looks like you could enable some 'console' if we modified the cmdline to console=ttyS0 -- but I doubt it will be an actual 'linux' console. If your device's hardware was 'similar enough' (i.e same hardware version and chip model) you could try just using a hardware programmer to backup the current flash and flash a copy of another linux firmware over, but I have never seen a c71 board or BE8T_S1_SC3_433 before. Chances are that a linux firmware for c71 + *_S1_SC3_* should be enough match to make it work but I haven't seen anything like that either. Your json file seems to show that your device has onvif and it is enabled but does it work ?

jilleb, 2021-10-05:
Yes, ONVIF is working :-)
And I've checked most of the file paths I know from my other doorbell by using the http://IP:8090/proc/self/root/ method, and was able to confirm most of the internal filesystem has the same setup, although it's pretty hard to guess the right filenames for the init.d scripts 🤣 

I wonder if there's a way to get a reverse shell by exploiting something. But... it's probably easier to dump the current flash and have a look inside what useful stuff is there.

guino, 2021-10-05:
@jilleb I thought someone with 5.x had connected the UART and said it had a 'terminal" in in -- but it was just a RTOS debug terminal not a linux shell.

jilleb, 2021-10-05:
Allright. Too bad..
I just had an idea.. what if we spoof Tuya by using the 'old' Doorbell, but with modified hardware ID and software version, so the Tuya app will want to force a software update? I guess we could trick the server into sending the 5.0.5. firmware that way....

guino, 2021-10-05:
@jilleb I know I have done the opposite, meaning: I modified the firmware version on my ppsapp so it wouls 'think' it's already updated and not try to update it, so there's a good chance that doing the opposite can work (saying the firmware is older to spoof a link to the firmware update file). This may not work unless there's an older firmware (and we say we have that specific older firmware) and this is firmware is a newer replacement for it (otherwise they may not have this version as an update for anything).

That said, if you do get a link, it will be to an update file which is not exactly the firmware dump. You will likely get the main image may have "files" and such but it's all a single image if it's RTOS so I'm not sure how easy that can be modified.

RoelVideler, 2021-10-14:
This is what I get from proc/mounts. Not sure how to execute commands, list dirs or upload data from the HTTP interface. Running 5.0.6. 

rootfs / rootfs rw,size=17068k,nr_inodes=4267 0 0
proc /proc proc rw,relatime 0 0
sysfs /sys sysfs rw,relatime 0 0
tmpfs /tmp tmpfs rw,relatime 0 0
/dev/mtdblock4 /home/cfg jffs2 rw,relatime 0 0
/dev/mtdblock3 /home/tmp jffs2 rw,relatime 0 0

Ierlandfan, 2021-12-10:
I have uploaded a firmware dump from 5.0.5. See #62

spacecabbie, 2022-03-19:
> All 5.x firmware I have seen runs RTOS (LiteOS) which is NOT linux, meaning none of the scripts/applications we put together will work on it.
>
> If your device takes ppsFactoryTool.txt (to connect to [...]

Maybe a silly question but i am still learning as I go... But can't you take the flash chip wipe it and just drop the older version on it since its the same hardware ?

guino, 2022-03-19:
@spacecabbie in theory if you know the hardware is exactly the same you could do that but you would not be able to have both devices online at the same time as they have different key/license information, but you could use it offline as some people use it. 

There’re usually some hardware differences in similar devices so it would be something to try if you have access to a suitable firmware file.

There are some 5.x linux based firmware we found recently and the existing 1080p hack works with it.

Someone was also able to patch a 5.x firmware directly (hardware programmer): https://github.com/guino/BazzDoorbell/issues/62#issuecomment-992367086

bilogics, 2022-09-16:
In the Tuya Snart App ONVIF is standaar available. At least in my latest firmware update per 20220910. You can log in with usr admin psq admin. In the Tuya App you can change it, but for some unknown reason an another password doesn’t seem to work anymore.

krzakx, 2022-11-14:
How, actually do you check frimware version? 
I do have following hardware version:

[image]
[image]

Is this hackable or no way? 
In general I do miss Pushbutton status in HomeAssistance (currently using Tuya integration) so would be appreciate for any walkaround.

WeterPeter, 2022-11-14:
> How, actually do you check frimware version? I do have following hardware version:
[image]

Check in your Tuya app for the firmware version. 
If you have version 2 and you have hacked it and you can telnet into it, I can give you a shell script to send the button press to Home Assistant.

samezrp, 2022-11-21:
I checked on mine, port 8090 does not work, the only ones opened are:

Starting Nmap 7.80 ( https://nmap.org ) at 2022-11-21 13:32 CET
Nmap scan report for 192.168.x.x
Host is up (0.00062s latency).
Not shown: 1998 closed ports
PORT     STATE         SERVICE
**80/tcp   open          http
5353/udp open|filtered zeroconf**
MAC Address: 00:1E:xx:xx:xx:xx (COmputime)
Device type: general purpose
Running: Linux 2.6.X
OS CPE: cpe:/o:linux:linux_kernel:2.6
OS details: Linux 2.6.19 - 2.6.36
Network Distance: 1 hop

so it looks like linux but with all functionality closed.
Web page on port 80 looks like this:

[image]

guino, 2022-11-24:
@samezrp did you check if URLs like http://admin:056565099@192.168.x.x/devices/deviceinfo work ? that's port 80, so if that works then one of the hacks may work. Chances are if your device is battery operated it won't work.

jilleb, 2022-11-25:
I'll hook mine up again and do some research.

WeterPeter, 2022-11-28:
If this is de Linux variant doorbel: I made a sctipt to trigger the REST API from Home Assistant. This can probably be modified to work with MQTT. For the script to work, you have to put the curl command for Android on it to work.

while true
do
        busybox dmesg -c 2>/dev/null | grep "door bell  press down"
        RET=$?
        if [ $RET = 0 ]
        then
                /mnt/mmc01/curl -X POST -H 'Authorization: Bearer YOUR_AUTH_BEARER' -H 'Content-Type: application/json'  -d '{"entity_id": "input_boolean.YOUR_ID"}'  http://YOUR_IP:8123/api/services/input_boolean/turn_on
                sleep 1
                /mnt/mmc01/curl -X POST -H 'Authorization: Bearer YOUR_AUTH_BEARER' -H 'Content-Type: application/json'  -d '{"entity_id": "input_boolean.YOUR_ID"}'  http://YOUR_IP:8123/api/services/input_boolean/turn_off
                date >> $LOG
        else
                sleep 1
        fi
done    

This is how I let ring my doorbel through a switch when de doorbel is pressed.

samezrp, 2022-11-28:
@WeterPeter Thank you, you were faster answering than me editing and deleting MQTT part in my prev message, as I found info about parsing log file.
I will try your solution.

krzakx, 2022-12-28:
Hi back again! 
Update from my side:
My `Frimaware version:
V4.0.7
`

Under http://admin:056565099@192.168.x.x/devices/deviceinfo I got:
`Sorry, the requested file does not exist on this server.`

What is interesting - Video door bell is conneting into router from **two WiFi interfaces**
[image]

where additionally I can navigate into:
[image]

guino, 2022-12-28:
@krzakx I'm not sure of what you're trying describe -- is it getting two ip addresses ? or maybe reporting 2 different mac addresses ? If you have multiple wifi routers/access points with the same ssid I would expect to see some devices hopping from one wifi to another. On the 4.0.7 firmware I have seen there's no text like the one you showed ("Sorry, the requested file does not exist on this server.") so either you got some different 4.0.7 firmware or you may be connecting to the wrong IP.

krzakx, 2022-12-29:
oh... **my bad.**.. I tought that there were two IP on Videodoorbell however that was kind of old cache.... I do not now, anyway the end story is that I do have Videodoorbell with 4.0.7 firmware with nmap scanning results:

```
Starting Nmap 7.93 ( https://nmap.org ) at 2022-12-29 12:54 Środkowoeuropejski czas stand.
NSOCK ERROR [0.3540s] ssl_init_helper(): OpenSSL legacy provider failed to load.
NSE: Loaded 155 scripts for scanning.
NSE: Script Pre-scanning.
Initiating NSE at 12:54
Completed NSE at 12:54, 0.00s elapsed
Initiating NSE at 12:54
Completed NSE at 12:54, 0.00s elapsed
Initiating NSE at 12:54
Completed NSE at 12:54, 0.00s elapsed
Initiating ARP Ping Scan at 12:54
Scanning 192.168.xx.xx [1 port]
Completed ARP Ping Scan at 12:54, 0.15s elapsed (1 total hosts)
Initiating Parallel DNS resolution of 1 host. at 12:54
Completed Parallel DNS resolution of 1 host. at 12:54, 0.01s elapsed
Initiating SYN Stealth Scan at 12:54
Scanning 192.168.xx.xx [65535 ports]

Discovered open port 6668/tcp on 192.168.xx.xx
Discovered open port 8000/tcp on 192.168.xx.xx
Discovered open port 8554/tcp on 192.168.xx.xx
Completed SYN Stealth Scan at 12:54, 34.97s elapsed (65535 total ports)
Initiating Service scan at 12:54
Scanning 3 services on 192.168.xx.xx
Service scan Timing: About 33.33% done; ETC: 12:56 (0:01:24 remaining)
Service scan Timing: About 66.67% done; ETC: 12:57 (0:01:02 remaining)
Completed Service scan at 12:57, 166.33s elapsed (3 services on 1 host)
Initiating OS detection (try #1) against 192.168.xx.xx
NSE: Script scanning 192.168.xx.xx.
Initiating NSE at 12:57
Completed NSE at 12:58, 64.65s elapsed

Initiating NSE at 12:58
Completed NSE at 12:58, 7.05s elapsed
Initiating NSE at 12:58
Completed NSE at 12:58, 0.01s elapsed
Nmap scan report for 192.168.xx.xx
Host is up (0.0044s latency).
Not shown: 65532 closed tcp ports (reset)
PORT     STATE SERVICE   VERSION

6668/tcp open  irc?
|_irc-info: Unable to open connection
8000/tcp open  http-alt?
8554/tcp open  rtsp-alt?
MAC Address: 84:7A:B6:04:7C:41 (AltoBeam (China))
Device type: general purpose
Running: Linux 2.6.X|3.X
OS CPE: cpe:/o:linux:linux_kernel:2.6 cpe:/o:linux:linux_kernel:3
OS details: Linux 2.6.32 - 3.10
Uptime guess: 0.007 days (since Thu Dec 29 12:48:53 2022)
Network Distance: 1 hop
TCP Sequence Prediction: Difficulty=256 (Good luck!)
IP ID Sequence Generation: All zeros
```

guino, 2022-12-29:
@krzakx have you tried this process: https://github.com/guino/Merkury1080P -- many people have had success with 4.0.7 firmware with that process (just double checking as I understand you have a doorbell but that should not matter).

I would also try RTSP with VLC directly since it seems your device has port 8554 open (rtsp://IP@8554 and rtsp://IP:8554//Streaming/Channels/101) 

Let me know.

krzakx, 2022-12-30:
@guino mine main goal is to get _bell button pressed_ action ;) 
As to video throught RTSP I have access.
Regards!

guino, 2022-12-30:
@krzakx you should only need the information on this wiki to get doorbell notifications: https://github.com/guino/BazzDoorbell/wiki/%5BHow-to%5D-Use-notifications-MQTT-log_parser-without-a-patched-ppsapp-%3F -- the log parser posted has support for doorbell notifications (and motion notifications).