guino/BazzDoorbell #41: Stuck at step #6

WeterPeter, 2021-07-20:
HI,

Sorry, but I cannot past step #6, the cmdline stays tha same after I reset/powerof the doorbell. I need help. I tried this over and over.

curl http://admin:056565099@<myip>/devices/deviceinfo :
{"devname":"Smart Home Camera","model":"Bell 8S","serialno":"061206207","softwareversion":"2.9.7","hardwareversion":"BE8S_H1_V10_433","firmwareversion":"ppstrong-c51-tuya2_lcs-2.9.7.20201020","authkey":"x28HrZFvldk5l2XTlHnQsNx1AgMqV5FZ","deviceid":"pp018347a5b13d41040f","identity":"MR2005212301200374","pid":"aaa","WiFi MAC":"7c:25:da:1b:e4:23"}

curl http://admin:056565099@<myip>/proc/cmdline :
mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd

env:
bootargs=mem=36 console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=0 ppsWatchInitEnd ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,ThankYouGuino,T(with 0A00 at the end).

ppsMmcTool.txt :
style=upgrade,,writeAddr=0,,password=nothing,,writeLen=0,,fileName=env;env import 42000000;saveenv,,

Files on SD card:
-rwxrwxrwx  1 peterdejong  staff  1152216 Jul 13 19:57 busybox
drwxrwxrwx  1 peterdejong  staff   131072 Jul 20 12:50 cgi-bin
-rwxrwxrwx  1 peterdejong  staff      621 Jul 15 09:23 custom.sh
-rwxrwxrwx@ 1 peterdejong  staff      926 Jul 20 15:22 env
-rwxrwxrwx  1 peterdejong  staff       16 Jul 13 18:00 httpd.conf
-rwxrwxrwx  1 peterdejong  staff     1327 Jul 13 18:00 index.html
-rwxrwxrwx  1 peterdejong  staff      425 Jan 10  2021 initrun.sh
-rwxrwxrwx  1 peterdejong  staff     7956 Jul 13 18:00 jpeg-arm
-rwxrwxrwx  1 peterdejong  staff       37 Jul 13 18:00 passwd
-rwxrwxrwx  1 peterdejong  staff      102 Dec 22  2020 ppsMmcTool.txt
-rwxrwxrwx  1 peterdejong  staff      263 Jul 13 18:00 set
-rwxrwxrwx  1 peterdejong  staff      161 Jul 13 18:00 upload.html

Can you give me a hint?

WeterPeter, 2021-07-20:
UPDATE: When I looked in the LSC app it asked to format the card, apparantly it was not formatted correcly.  I let the app format the card, put the files on it again and now my cmdline output is:  

mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd - ip=${T//_/$'\\x20'}:::::;T=\"sleep_5;mkdir_-p_/mnt/mmc01;mount_-t_vfat_/dev/mmcblk0p1_/mnt/mmc01;/mnt/mmc01/initrun.sh&\";eval mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd

No telnet login working for now.

WeterPeter, 2021-07-20:
Well, putting the right busy box on it solved my next step.

I am able to log in via telnet.

Thanks for your help! ;-D

guino, 2021-07-20:
Sorry I have been busy with work and had not had a chance to reply earlier. I am glad you got it working. These devices can be picky about the format and type of SD card. I was going to suggest you try #13 which you seem to have done but it looks like your issue was only the format of the partition and the incorrect busybox file.

WeterPeter, 2021-07-20:
Yes, the format of the SD was the issue.

Thanks!

johan-van-marion, 2023-03-01:
> Well, putting the right busy box on it solved my next step.

@WeterPeter which busybox did you end up using? 
I did try some but still getting the HTTP 500 Error 
Formatting sevral sd-card through the app but no good result

WeterPeter, 2023-03-01:
I used the one of the mmc link on this page: https://github.com/guino/BazzDoorbell/issues/2

Is working for quite some time now.

johan-van-marion, 2023-03-01:
@WeterPeter good the heart it's still working. Do you by accident still have the SD content/files you used? I think i'm still missing a piece of the "puzzle" cause i'm getting the returns from other url's but when using the hack url i'm still getting the HTTP 500 error. Maybe by comparing content i can solve this.

WeterPeter, 2023-03-03:
Are you sure you have a doorbell running major version 2? Because if you have version 3, you have a completely different OS on is which is not Linux based. You can find out the version in the Tuya/LCE app.

johan-van-marion, 2023-03-03:
@WeterPeter i'm not sure i follow you?
The Tuya app shows:
Main module 5.2.2
MCU 5.2.2
It's a Nedis wificdp10gy video Doorbell.
I tried the ja k but getting some were, but no luck applying the hack it self

WeterPeter, 2023-03-03:
[image]
Sorry but your doorbell is too new, there is no Linux running on it. You need to find a doorbell with version 2 firmware to be able to hack it.