guino/BazzDoorbell #42: How to access files on SD after successful flash remotely?

Nigel1992, 2021-07-23:
Thanks a ton for your amazing work on this, you really did something awesome!
I want to be able to remotely access the files on my SD card to replace files, edit files etc.. [in example https://github.com/guino/BazzDoorbell/issues/7]

Is this possible to do using WinSCP, FTP or something else ?
And if so, how?

Regards,
Nigel

guino, 2021-07-23:
@Nigel1992 you can download files using http and upload files too (using /upload.html) — those are what I use and recommend since httpd is already running for snap/mjpeg and other features. If you have snap/mjpeg setup you just have to use the same user/pass/URL without the /cgi-bin/snap.cgi part and you should be able to browse and download any files. To upload use the same stuff just with /upload.html at the end (files will go to root of sd card).

Can you install/run ssh (for winscp) - yes but I don’t recommend due to low memory availability. There’s a github issue with a working ‘dropbear’ tool for that purpose - just search for it if you like.

For ftp you would have to compile/run a server for arm yourself (and you should mind the memory usage too).

Other than the above busybox has tftp support but it’s client side only (no server).

Nigel1992, 2021-07-23:
@guino Thanks for your quick response !
Apart from uploading a new audio file to replace the doorbell sound, is there any point of doing it now that I think of it ? haha

guino, 2021-07-24:
@Nigel1992 upload will replace existing files so be careful with it. you can use /devices/reboot to reboot (same user/pass as /proc/cmdline) -- works on mine at least.

Only certain firmware versions allow changing the password for onvif access -- if that's supported it will be in /home/cfg/tuya_config.json -- if it's not there there's no way to change it. You can make firewall rules on your network and/or make a script to rename the patched ppsapp to enable/disable it at will but there's no easy way to change the password.

Nigel1992, 2021-07-24:
I just did the reboot and now none of the hacked functions are working :(

guino, 2021-07-25:
@Nigel1992 the device will format the SD card if it gets full - you need to make sure that doesn’t happen by either disable recording 24/7 or using motion only recording and configuring cleanup cgi.

I always tell people to backup their files once things are working. 

If your SD card was not full then you may just have a faulty SD card - they will go bad after awhile if you’re recording daily to them.

Nigel1992, 2021-07-25:
@guino To what type does it format when it's full ? FAT32 ? Cause mine was RAW

guino, 2021-07-25:
It should format it as fat32 but if you remove the SD while it is formatting who knows how it may be left. An issue with the SD card could also have unknown outcomes.

You still have to configure the cleanup script for it to work (I think you need to put in the serial number in the script).

You can run play.cgi from a script in the device or with a http request. There’s nothing ready to make a sound play when motion is detected. You would have to use the log_parser script to either run the play.cgi script OR to notify some outside server which in turn can make the http request to play audio.