guino/BazzDoorbell #43: Changing WiFi

Nigel1992, 2021-07-28:
If I reset the camera to change WiFi, do I need to reflash?
If yes, is there a way to change WiFi settings inside some file on the device ?

guino, 2021-07-28:
You can probably change/set the wifi ID+password in ppsFactoryTool.txt and just leave it at that.

If you reset the device you do NOT need to re-apply the hack, you should be able to do this:
-Boot without SD card
-Reset device to factory settings
-Configure device (so it uses the WIFI+settings you want)
-Boot with SD card (it should boot with the hack here)

If you have a ppsFactoryTool.txt in the SD card you will need to adjust the WIFI ID + Password in it too or it will connect to whatever was left in the file before you reset the device.

Nigel1992, 2021-07-28:
Its working after changing the WiFi, but i had to fix my /home/cfg/tuya_config.json to get rtsp to work again.

closed 2021-07-28 by Nigel1992