guino/BazzDoorbell #44: Support for V30 doorbell

mshuflin, 2021-08-04:
I am looking at tuya-based video doorbells available in my region and this one seems quite common. 
https://shopee.com.my/product/361966449/8762138469?smtt=0.208419637-1628041873.9

Is there a possibility that these doorbells can be rooted?
Thanks!

guino, 2021-08-04:
@mshuflin hard to say without seeing the url responses. I would stay away from battery powered devices because they are more likely to be running RTOS than linux and they’re possibly not going to be always on.

I am not sure if you have specific reasons to root the device but if you’re still shopping I always recommend buying a device that has all features you need instead of trying to mod it. I understand there are factors such as price and availability but I would only go thru the trouble of rooting if I had to.

mshuflin, 2021-08-04:
Thanks @guino for your quick reply. I am looking for a fully cloudless option that can integrate with Home Assistant, node red, and/or MQTT.

guino, 2021-08-05:
@mshuflin the only cameras that would allow for that without rooting would be the ones that have the 'remote event log' or 'event server' feature which allow you to enter the ip address of a server to receive camera events (including motion detection) which you could create in the same computer/server that runs home assistant.

Obviously many people have done what you want with a range of tuya cameras (using with the posted information) but it can be a gamble finding a device with a rootable version. If you're going to buy without knowing details I would recommend buying from place where you can easily return the camera if it isn't what you need.

mshuflin, 2021-08-08:
Thanks for the advice, I will order (and return if it does not work) and report back here

mshuflin, 2021-09-06:
Here are my results: I could not get any response from any port other than 6668. 

from http://192.168.0.170:6668/devices/deviceinfo I received:

��Uª����������û����3.3������PÖ���Æ¥¿LˆÂéÎ¬<[•Çl¬MBxps*ÀdW:4ågAb¢%ªäËCžuHÈÅ¤“iÐ.Þë}´iñç?|šl*Þ¾Ø²TƒÃ”ËÿcŠï'owp†äý3#.ž²TS­F!¾úŒ†û~ïP¾ÙÚ©Ä(SñA€x4R“}[@/pQçD‹+Ÿ®ŸÑöaåÕEÇÆZøÕßÙ¯ËÙç‚73vÕØ|ŸRß¾I]bd¬x’26ŽIÓª¼ê£+ž|1¢¦:¹Øq¶’±ÁÇTËþ‹€Fò¥}¬¥ÜÕ%wïÏPôì¨†��ªU

and similarly garbled text from other urls.

guino, 2021-09-07:
@mshuflin did you add in ppsFactoryTool.txt to the SD card to see if any other ports open (80 or 8090) ?

The output you got for port 6668 id the standard tuya api port which isn't useful for anything else.

mshuflin, 2021-09-07:
It turns out this model does not have an SD card. The back has a USB port.

guino, 2021-09-07:
@mshuflin it most likely has a SD card slot inside (if you open it) - I have no way to tell how easy it is to access it.

mshuflin, 2021-09-07:
I was able to open it fairly easily, here is what it looks like:
[image]
[image]
[image]

guino, 2021-09-08:
@mshuflin  Your UART is likely the top left 4 holes on the 2nd picture posted (and flash chip looks like the one next to the holes --I couldn't read the numbers on it from the picture). If there's any SD card slot it would likely be behind the top board (where the camera lens are mounted). Without a SD card slot the only hope of making any mods would be using a UART/hardware programmer to read the firmware, modify it and flash it back. Depending on the bootloader it may only be possible to do it with the hardware programmer and having no SD card storage would limit you to AT MOST enabling RTSP/ONVIF (if included in the code at all).

If you have a heat-gun and programmer you can read the flash and I can review it to tell you what is possible to do. If you don't have the tools or familiarity with it I would stop now before risking breaking the device -- you could sell/return it and get another if you really require extra features.

paulloft, 2022-01-09:
I have the same doorbell. All ports on the device are closed, I tried all your options with ppsFactoryTool.txt and none worked.
I connected to the UART pin, but it seems to be a dead end, since I cannot send commands, and the output to the console looks like this:
[image]

I managed to connect with clamps to the XM25QH64AHIG chip and download the firmware from it (https://drive.google.com/file/d/1I0fIbhhVVQDAoBq6kRFSbFJtjyMji2IE/view?usp=sharing)
Next, I ran `binwalk -e -M` like in your example
<details> 
  <summary>its output under the spoiler:</summary>

```
pi@raspberrypi:~ $ binwalk -e -M ./doorbell_v30.bin 

Scan Time:     2022-01-05 22:41:48
Target File:   /home/pi/doorbell_v30.bin
MD5 Checksum:  904f0a83916fa3477f51d17ed13e676e
Signatures:    411

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
18160         0x46F0          gzip compressed data, has original file name: "u-boot.bin", from Unix, last modified: 2020-07-07 03:29:15
327696        0x50010         gzip compressed data, has original file name: "EasyCam.bin", from Unix, last modified: 2021-04-01 07:14:22
5308419       0x510003        MySQL ISAM compressed data file Version 5
5395045       0x525265        MySQL MISAM index file Version 10
5418378       0x52AD8A        MySQL ISAM index file Version 2
5444208       0x531270        MySQL MISAM compressed data file Version 4
5444412       0x53133C        MySQL MISAM index file Version 5
5514744       0x5425F8        MySQL MISAM index file Version 10
5535950       0x5478CE        MySQL MISAM index file Version 5
5547885       0x54A76D        MySQL MISAM index file Version 6
5593564       0x5559DC        MySQL ISAM index file Version 7
5602202       0x557B9A        MySQL ISAM compressed data file Version 8
5641461       0x5614F5        MySQL MISAM compressed data file Version 7
5644284       0x561FFC        MySQL MISAM index file Version 1
5706578       0x571352        MySQL ISAM index file Version 2
5711276       0x5725AC        MySQL MISAM compressed data file Version 4
5721427       0x574D53        MySQL MISAM compressed data file Version 4
5739140       0x579284        MySQL ISAM compressed data file Version 6
5741912       0x579D58        MySQL ISAM compressed data file Version 5
5748403       0x57B6B3        MySQL MISAM compressed data file Version 10
5753885       0x57CC1D        MySQL MISAM index file Version 7
5757997       0x57DC2D        MySQL ISAM compressed data file Version 6
5763292       0x57F0DC        MySQL ISAM index file Version 2
5764060       0x57F3DC        MySQL ISAM index file Version 8
5769029       0x580745        MySQL MISAM index file Version 2
5771205       0x580FC5        MySQL MISAM index file Version 4
5793662       0x58677E        MySQL MISAM compressed data file Version 10
5798838       0x587BB6        MySQL MISAM index file Version 5
5806306       0x5898E2        MySQL MISAM index file Version 6
5810993       0x58AB31        MySQL MISAM index file Version 7
5822940       0x58D9DC        MySQL MISAM index file Version 1
5825339       0x58E33B        MySQL MISAM compressed data file Version 2
5827436       0x58EB6C        MySQL MISAM index file Version 8
5827991       0x58ED97        MySQL MISAM index file Version 1
5830225       0x58F651        MySQL ISAM index file Version 3
5831459       0x58FB23        MySQL ISAM index file Version 3
5837622       0x591336        MySQL MISAM index file Version 3
6007679       0x5BAB7F        MySQL ISAM index file Version 11
6088525       0x5CE74D        MySQL MISAM index file Version 6
6342707       0x60C833        MySQL ISAM compressed data file Version 5
6429333       0x621A95        MySQL MISAM index file Version 10
6452666       0x6275BA        MySQL ISAM index file Version 2
6478496       0x62DAA0        MySQL MISAM compressed data file Version 4
6478700       0x62DB6C        MySQL MISAM index file Version 5
6549032       0x63EE28        MySQL MISAM index file Version 10
6570238       0x6440FE        MySQL MISAM index file Version 5
6582173       0x646F9D        MySQL MISAM index file Version 6
6627852       0x65220C        MySQL ISAM index file Version 7
6636490       0x6543CA        MySQL ISAM compressed data file Version 8
6675749       0x65DD25        MySQL MISAM compressed data file Version 7
6678572       0x65E82C        MySQL MISAM index file Version 1
[... 132 lines cut ...]
7281332       0x6F1AB4        Unix path: /var/lib/jenkins/workspace/Release_IPC_SDK/src/tuya_middleware/tls/mbedtls/ssl_srv.c
7290308       0x6F3DC4        Unix path: /var/lib/jenkins/workspace/Release_IPC_SDK/src/tuya_middleware/tls/mbedtls/ssl_tls.c
7291332       0x6F41C4        PEM certificate
7291528       0x6F4288        PEM certificate
7302488       0x6F6D58        Unix path: /var/lib/jenkins/workspace/Release_IPC_SDK/src/tuya_middleware/tls/mbedtls/ssl_cli.c
7521948       0x72C69C        CRC32 polynomial table, little endian
7565712       0x737190        CRC32 polynomial table, little endian
7570620       0x7384BC        CRC32 polynomial table, little endian
7640972       0x74978C        Base64 standard index table
7695060       0x756AD4        AES S-Box
7696916       0x757214        AES Inverse S-Box
7827092       0x776E94        CRC32 polynomial table, little endian
```

</details>

When unpacking it, I received only two files _u-boot.bin_ and _EasyCam.bin_, which are not further unpacked.
It seems their structure is different from what you had. Unfortunately, I have no experience in decompiling and changing firmware, please tell me what steps can I try next?

<details> 
  <summary>Just in case, I attach a photo of the main module from both sides.</summary>

[image]
[image]
</details>

guino, 2022-01-10:
Taking a quick look at the binwalk output it seems to confirm this device runs RTOS which greatly limits the options for making any changes (only hardware programmer would work for minimal changes). Did you try typing anything on the UART to see if there was any way to interact with it? Usually the RTOS application has some sort of pseudo terminal which allows a few basic commands but I would not expect anything useful from it.

rossrosh007, 2024-06-19:
This maybe crazy. But I remember the old 720p doorbell camera had a sticker on the circuit modules with "admin" and "password". The sticker was behind the board, unscrew the board and check behind it. I can't recall if that was for connecting with a NVR or to login to a config. 

The login info on that sticker was not the same as the sticker under the battery cover, which had the wifi  AP login info