guino/BazzDoorbell #45: @guino Now i am stuck at step 7...  mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd - ip=${T//_/$'\\x20'}:::::;T=\"sleep_5;mkdir_-p_/mnt/mmc01;mount_-t_vfat_/dev/mmcblk0p1_/mnt/mmc01;/mnt/mmc01/initrun.sh&\";eval mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd says the cmdline but the hack cant be installed

guino, 2021-08-11:
@guino Now i am stuck at step 7...  mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd - ip=${T//_/$'\\x20'}:::::;T=\"sleep_5;mkdir_-p_/mnt/mmc01;mount_-t_vfat_/dev/mmcblk0p1_/mnt/mmc01;/mnt/mmc01/initrun.sh&\";eval mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd says the cmdline but the hack cant be installed 
no connection HTTP Error 500

_Originally posted by @Beer-mann in https://github.com/guino/BazzDoorbell/issues/2#issuecomment-897111515_

guino, 2021-08-11:
@Beer-mann if your /proc/cmdline shows up like you posted above it's ok if you don't get a response from the hack URL, just continue with the next steps and it will likely work

Beer-mann, 2021-08-11:
Thanks I will try tomorrow! Man if this work i will definitely buy you a beer! @guino

Beer-mann, 2021-08-12:
After removing the sd card there is no home, lib, or bin folder... so I cannot patch the right ppsapp?

guino, 2021-08-12:
@Beer-mann Can you post the **original** response for the /proc/cmdline URL and the **current** response ? I also would like you to post a zip of your current SD card files EXCLUDING the SDT folder. Hopefully it's just something missing/incorrect on your SD card.

Does the device currently work normally WITHOUT the SD card (using the phone app) ?

Beer-mann, 2021-08-12:
Original response of /proc/cmdline URL : mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd
Current response: mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd - ip=${T//_/$'\\x20'}:::::;T=\"sleep_5;mkdir_-p_/mnt/mmc01;mount_-t_vfat_/dev/mmcblk0p1_/mnt/mmc01;/mnt/mmc01/initrun.sh&\";eval mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd

Camera is working normally without SD Card
[Card.zip](https://github.com/guino/BazzDoorbell/files/6975887/Card.zip)

guino, 2021-08-12:
@Beer-mann this is the first case I see where the /proc/cmdline is changed but the firmware files are not copied to the SD card.
Can you boot with the SD card inserted and post the response for the following please:

http://admin:056565099@ip/devices/deviceinfo
http://admin:056565099@ip/proc/mounts
http://admin:056565099@ip/proc/self/root/etc/init.d/S90PPStrong

Beer-mann, 2021-08-12:
http://admin:056565099@ip/devices/deviceinfo: 
{"devname":"Smart Home Camera","model":"Bell 8S","serialno":"060906730","softwareversion":"2.9.7","hardwareversion":"BE8S_H1_V10_433","firmwareversion":"ppstrong-c51-tuya2_lcs-2.9.7.20201020","authkey":"w9uQmHRXzuHiHpcm92ywzGbgbNA14Azq","deviceid":"pp017029e2a1d167ed1e","identity":"MR2007120100106415","pid":"aaa","WiFi MAC":"d4:d2:d6:b1:cb:e7"}

http://admin:056565099@192.168.0.176/proc/mounts: 
rootfs / rootfs rw,size=15856k,nr_inodes=3964 0 0
proc /proc proc rw,relatime 0 0
sysfs /sys sysfs rw,relatime 0 0
tmpfs /dev tmpfs rw,relatime 0 0
devpts /dev/pts devpts rw,relatime,mode=600,ptmxmode=000 0 0
/dev/mtdblock6 /home/cfg jffs2 rw,relatime 0 0
/dev/mmc01 /mnt/mmc01 vfat rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro 0 0

http://admin:056565099@ip/proc/self/root/etc/init.d/S90PPStrong: 
#!/bin/sh

export PATH=/usr/bin:/sbin/:/usr/sbin:/bin

RED="[1;31m"
NORMAL="[0;39m"

echo "${GREEN} 2015 PPStrong Tech Cop.Ltd.${NORMAL}"

mkdir -p /opt/pps
MTDNUM=`cat /proc/cmdline | sed 's/.*ppsAppParts=\([0-9]\).*/\1/'`

# debug
#MTDNUM=5

echo "------------->mtdnum:${MTDNUM}"

case $MTDNUM in
         5)
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
			break
			;;
		 7|8)
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
		 	break
		 	;;
		 0)     
			sleep 10
			mount -t vfat /dev/mmcblk0p1 /opt/pps
			break
			;;
		 *)
			MTDNUM=5
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
		 	;;
esac

echo "/opt/pps/" > /tmp/PPStrong.runpath
[ -e /opt/pps/initrun.sh ] && cp /opt/pps/initrun.sh /tmp/PPStart && chmod +x /tmp/PPStart && /tmp/PPStart

----------------------------------------------------------------------------------------------------------------------------------------------

I hope we will figure this out...
Do I really do not need the old version of the hack because of the #debug #MTDNUM=5 ?

guino, 2021-08-12:
@Beer-mann #13 is the best approach all around. You have a known firmware, the SD card appears to be mounted and the files appear to be ok. I am going to take a closer look at the files and let you know but right now I am not sure why it isn’t working.

guino, 2021-08-12:
@Beer-mann all your files appear correct based on what I know. The only thing I could suggest is change the slee_5 in the env file to sleep_10 and re-apply the hack (boot holding reset for 5 seconds). Thay may give it enough time for the SD card drivers to load so the commands can work. Still I have never seen this issue so it's kind of unknown territory. You CAN still try #2 if you like but if you do you have to keep in mind that you may brick your camera based on the fact it doesn't seem to be executing the SD card scripts and #2 requires the SD card in order to boot (it won't work without a SD card).

Other than that I can only offer a few things to keep in mind.
-For the hack to work you must power up the device with the SD card inserted (and the files in it -- you can't insert the SD card after the device is booted up.
-You should only hold the reset button when installing the hack (to modify the cmdline), otherwise just power up normally without using the reset button
-With the hack installed as you have right now, you should try a different SD card to see if it works at all -- just copy initrun.sh to the FAT32 sd card and boot it up to see if it creates the files.

Hope you get it sorted out.

Beer-mann, 2021-08-13:
@guino 
I will buy a new SD Card today. It seems that none of the two sd cards laying around in my house work for this.

Beer-mann, 2021-08-13:
I tried 3 SD Cards, it did not work with one of these. Well I am giving up right now...

Beer-mann, 2021-08-13:
There must be something wrong with the initrun.sh file, shouldnt it? This file installes the needed folders right?

Beer-mann, 2021-08-13:
Hey, http://admin:056565099@192.168.0.176/proc/self/root/home/cfg/tuya_config.json gives me: 
{
	"version":	1,
	"sleep_mode":	0,
	"alarm_fun_onoff":	0,
	"alarm_fun_sensitivity":	1,
	"alarm_fun_mode_switch":	0,
	"alarm_fun_time_start":	0,
	"alarm_fun_time_end":	0,
	"flip_onoff":	0,
	"light_onoff":	1,
	"night_mode":	0,
	"sound_detect_onoff":	0,
	"sound_detect_sensitivity":	0,
	"watermark_onoff":	1,
	"event_record_time":	60,
	"enable_event_record":	2,
	"record_enable":	0,
	"motion_trace":	1,
	"motion_area_switch":	0,
	"motion_area":	"",
	"motion_tracking":	0,
	"cry_detection_switch":	0,
	"humanoid_filter":	1,
	"loudspeaker_vol_pct":	100,
	"jingle_mode":	0,
	"jingle_sound":	1,
	"jingle_volume":	100,
	"jingle_exist":	0,
	"flight_bright_mode":	0,
	"flight_light_brightness":	100,
	"flight_pir_set":	0,
	"flight_pir_one":	0,
	"flight_pir_two":	0,
	"flight_pir_three":	0,
	"flight_pir_sensitivity":	0,
	"flight_alarm_fun_onoff":	0,
	"flight_on_off":	0,
	"flight_pir_light_on_time":	30,
	"flight_warn_switch":	0,
	"flight_dualbrite":	0,
	"flight_ontime":	0,
	"flight_highbrightess":	10,
	"flight_lowbrightess":	10,
	"flight_mode":	0,
	"flight_motion_sens":	0,
	"onvif_enable":	0,
	"onvif_pwd":	"admin"
}
couldnt i just edit this file and get rtsp and mqtt working? 
if this works how could I do it? @guino

guino, 2021-08-13:
@Beer-mann the only way to edit a file or run anything in the device requires that initrun.sh to work. Did you try changing your env file switching sleep_5 to sleep_10 and reapplying the hack to see if initrun.sh executes? If you did, please post the updated response from /proc/cmdline so I can double check it.

Beer-mann, 2021-08-13:
@guino I did: mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd - ip=${T//_/$'\\x20'}:::::;T=\"sleep_10;mkdir_-p_/mnt/mmc01;mount_-t_vfat_/dev/mmcblk0p1_/mnt/mmc01;/mnt/mmc01/initrun.sh&\";eval mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd

Also tried 15 secounds... didnt work either... Maybe SD Reader of device broken?

guino, 2021-08-13:
@Beer-mann if it records video files to the SD card (when configured by the phone app) then it isn’t broken. If it can’t record anything to the SD card then it would be a problem.

Beer-mann, 2021-08-13:
@guino How should I test this? i got the folder but there is no video file...

guino, 2021-08-13:
@Beer-mann if you configure the device on the phone app you should have an option to enable recording to the SD card - it should create video files in the SDT directory at least when motion is detected. If files are being written then the SD card is working. If nothing is written then something must be wrong with the device. There’s usually an option to format the SD card in the app too, you could try that then power off, copy initrun.sh back into it and power up with SD card again to see if it helps but it is a long shot.

Beer-mann, 2021-08-13:
@guino There are just data files in the directory no video files...

guino, 2021-08-13:
@Beer-mann the ‘data files’ are the video files. You can play them with no audio on vlc by setting the demuxer to h264 demuxer. But the point is the SD card is working so I have no idea why the script isn’t running.

can you post the response for:
/proc/self/root/etc/init.d/S80network

I doubt this is the problem but I am running out of ideas.

Beer-mann, 2021-08-13:
#!/bin/sh

ipaddr=
bootp=
gateway=
netmask=
hostname=
netdev=
autoconf=

for ipinfo in `cat /proc/cmdline`
do
	case "$ipinfo" in
	ip=*)
		for var in  ipaddr bootp gateway netmask hostname netdev autoconf
		do
			eval read $var
		done << EOF
		`echo "$ipinfo" | sed "s/:/\n/g" | sed "s/^[ 	]*$/-/g"`
EOF
		ipaddr=`echo "$ipaddr" | cut -d = -f 2`
		[ x$ipaddr == x ] && ipaddr=x
		;;
	esac
done

[ -z "$ipaddr" ] && exit 0

echo "      IP: $ipaddr"
echo "   BOOTP: $bootp"
echo " GATEWAY: $gateway"
echo " NETMASK: $netmask"
echo "HOSTNAME: $hostname"
echo "  NETDEV: $netdev"
echo "AUTOCONF: $autoconf"

if [ x$ipaddr == x- ] ; then
	# use DHCP
	:
else
	cmd="ifconfig $netdev $ipaddr"
	[ x$netmask != x- ] && cmd="$cmd netmask $netmask"
	eval $cmd
	[ x$gateway != x- ] && route add default gw $gateway
fi

ifconfig lo 127.0.0.1

Beer-mann, 2021-08-13:
@guino Maybe turn of DHCP? I have no idea either...

guino, 2021-08-13:
@Beer-mann The output is fine, the settings look fine, the SD card is working so I am honestly not sure why this isn't working. The only thing I can suggest you try is change the sleep_5 in env file to something crazy high like sleep_60 -- then re-install the hack (hold reset for 5s during power up), then let it boot up and wait at least 2 minutes after booting up before removing the SD card -- this is because the sleep_60 will make it wait 1 minute after boot before running the initrun.sh script. Then check the SD card to see if the home directory shows up or at least if the 'hack' file shows up the SD card.

Beer-mann, 2021-08-13:
I will try... this is so strange wtf...

Beer-mann, 2021-08-13:
sleep_60 wont let the device boot at all. Just a red light. sleep_30 let the device boot but does not change anything on the sd card. I also checked the cmdline. I am desperate. I try something between 30 and 60 now @guino

Beer-mann, 2021-08-13:
mem=37M console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd - ip=${T//_/$'\\x20'}:::::;T=\"sleep_30;mkdir_-p_/mnt/mmc01;mount_-t_vfat_/dev/mmcblk0p1_/mnt/mmc01;/mnt/mmc01/initrun.sh&\";eval mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd

This is the final result... My device do not start with sleep more than 30. So I think maybe my device is just one thats broken or I am missing out something. @guino thanks for your help

guino, 2021-08-13:
@Beer-mann you can try #2 again and if it fails just use the troubleshoot information in there to restore your original cmdline so the device can boot again (or you can use the files from #13 to make it boot again without a SD card). Based on what we're seeing here I expect the device to not boot at all with #2, but if it does, please post the /proc/cmdline output from it and I can check if it installed ok.

Beer-mann, 2021-08-13:
@guino you are right it does not boot at all with the other method... maybe I try another device or build my own doorbell with an pi

guino, 2021-08-13:
@Beer-mann if you can’t return this device and really want it patched you could use #11 and then #12 to modify the firmware directly. I would only do this if you don’t have other options. I can help you with patching the firmware bin file if you like.

Beer-mann, 2021-08-13:
@guino  I got a new device, same issue... maybe the german fabric of the device is not hackable?

guino, 2021-08-13:
@Beer-mann it is possible they changed something which would prevent the hack from working. Are you able to use #11 to extract the firmware for review?

Beer-mann, 2021-08-13:
@guino I will try to extract the firmware... I am not sure if I can make it thru. I will try now. Never worked with binwalk

Beer-mann, 2021-08-14:
This is the binwalk of the sdb 
[image]
But the binwalk of the partiotion seems to be empty 
[image]

Beer-mann, 2021-08-14:
@guino After a long night i will give it a trie with the new device with older firmware and other sd card. Maybe the used sd card which I tried in the new device was not working correct.

guino, 2021-08-14:
@Beer-mann your binwalk output suggests the address used with #11 is incorrect or it did not save the flash correctly to the SD card (ie reset button didn’t work). It seems to have just random trash/old data from before reading the firmware. To be clear it should be /dev/sdb (not sdb1) as that’s where the data is supposed to be recorded (but it didn’t in your case).

Beer-mann, 2021-08-14:
@guino I got step 7 done now!!! The problem was the sd card format. When formatted from the device (phone app) the sd card gets currupted. Formatting it with linux solved the problem. Now I will provide you the pps data, so you can adjust it and the device does not get stuck. I also would need some help with the mqtt if you could help me out there as well. Thanks a lot so far!!!

Beer-mann, 2021-08-14:
@guino UPDATE: Everytime I put the sd card in the device it gets "broken", windows wants to repair the card then. I will try to put all the files on the card with linux, with the updated ppsapp. Hopefully my device is not bricked afterwards.

guino, 2021-08-14:
@Beer-mann you may want to look at what the app is doing to the sd card and partition/format it in linux as close as possible to it so hopefully it will leave it alone.

If the SD card doesn’t have at least 2Gb of free space the app may erase/format it so keep that in mind.

Beer-mann, 2021-08-14:
How do I enable onvif with the hack now installed? 2.9.7 Version should work without patchin ppsapp right?

guino, 2021-08-14:
@Beer-mann  Onvif should work without patching on your version if you edit the tuya_config.json file to set onvif_enable to 1: 

Modify initrun.sh:
Add above the while loop

if [ ! -e /mnt/mmc01/tuya_config_original.json ]; then cp /home/cfg/tuya_config.json /mnt/mmc01/tuya_config_original.json; fi

Add inside the while loop below sleep 30:

if [ -e /mnt/mmc01/tuya_config.json ]; then
 cp /mnt/mmc01/tuya_config.json /home/cfg/tuya_config.json;
fi

Boot the doorbell once, it will create the tuya_config_original.json file, copy it and rename it to tuya_config.json, open it and set onvif_enable to 1.

Beer-mann, 2021-08-14:
@guino What about mqtt for the button press? How am I able to manage this?

guino, 2021-08-15:
@Beer-mann for motion detection you need to:
-copy ppsapp from home/app/ppsapp to the root of SD card (even without patching)
-download and adjust mosquitto_pub and log_parser.sh from this post: https://github.com/guino/BazzDoorbell/issues/4#issuecomment-740644879
-Adjust custom.sh to use log parser as described in the same post above

When booting the device with the above changes it will kill ppsapp and start a new one using the log_parser which can den trigger your MQTT messages for motion/ring.