guino/BazzDoorbell #46: LCS action doorbell update V2.10.5

gielie, 2021-08-24:
Hi,
I have a Action LCS doorbell on firmware V2.9.7 and it works perfect with the patch explained here. Now he’s prompting for an update to V2.10.5 has anyone done this already? Does it work, do I have to repatch or will the patch stay in place? If not, is there a way to remove the prompting of the update because every time I open the app it asks for an update, I can klick it away but the wife also uses this app, no need to explain further. 

Tnks for this great work.

guino, 2021-08-24:
@gielie if you update you won't have to install the hack again but you will likely need to:
1-remove the patched ppsapp from SD card and reboot before updating (so it doesn't run the old version after update)
2-remove the 'home' directory from SD card and reboot after updating so you get a copy of the new ppsapp under home/app on the SD card
3-Add ppsFactoryTool.txt to enable the http interface for /proc/cmdline /devices/deviceinfo etc (which will likely be on port 8090 on 2.10.x version)
4-Patch the new ppsapp (2.10.5) and copy it to the root of SD card.

I do not know of any way to disable the update popup at this time but I am considering patching the version information on the ppsapp file and rebooting to see if that's enough to 'fool' the server that it is already updated -- I am currently running 2.9.6 and have a popup to 2.10.1 myself but I don't use the tuya app to view the video often so it hasn't bugged me enough to try it. From what I can tell I have enough space in the ppsapp to modify the 2.9.6 to 2.10.1 and give it a try. I may have time to try it later today -- I do NOT encourage anyone to try it without first making a backup of the firmware with #11 because you may risk some changes being downloaded from the server which can't work with the old version (bricking the device).

guino, 2021-08-24:
@gielie the change worked -- I edited the '2.9.6' occurrences on my ppsapp to '2.10.0'' (there were only 2 places) and now the app no longer pops up the update question and the camera continues to work normally. No side effects from what I can tell -- the app and /devices/deviceinfo report 2.10.0 as expected, so this is definitely an option if you don't want to update.

gielie, 2021-08-24:
Great work, I will think what i will do.

closed 2021-08-24 by gielie