guino/BazzDoorbell #47: Attempting to hack Bullet 4S [B4S_V10_S1_GC1]

Nigel1992, 2021-08-29:
I had to follow the steps from: Special note for 4.0.x firmware

Here's the device info: 
`{"devname":"Smart Home Camera","model":"Bullet 4S","serialno":"100192327","softwareversion":"5.0.5","hardwareversion":"B4S_V10_S1_GC1","firmwareversion":"ppstrong-c71-tuya2_lsc-5.0.5.20210301","identity":"M1M001AA3202018818","authkey":"TIiZAi5qoAhO5uM8emvS94EmhMsXPXOm","deviceid":"pp012b51c61aef6473f3","pid":"aaa","WiFi MAC":"b4:fb:e3:fd:2f:47","ETH MAC":"b4:fb:e3:fd:2f:47"}`

Proc/cmdline returns:
`console=/dev/null LX_MEM=0x3fe0000 mma_heap=mma_heap_name0,miu=0,sz=0x1d00000 pcbversion=B3S_S1_V10 sensor=gc2063mipi`

How do I Proceed from step 1?

guino, 2021-08-30:
@Nigel1992 5.x firmware does not run linux so none of the mods work on it.

Nigel1992, 2021-08-30:
@guino So there's no way I can get rstp to work ?

guino, 2021-08-30:
@Nigel1992 the only way to make any changes to this camera is to use a hardware programmer and even then we have no confirmed way to modify the firmware to enable rtsp. So right now we the answer is no. I would not expect a hack for this camera to be made available (from me at least).

Nigel1992, 2021-08-30:
@guino how about downgrading using this

https://developer.tuya.com/en/docs/iot/firmware-upgrade-operation-guide?id=K93ixsft1w3to

Also...
I saw a guy here in this repo who has my camera but runs an older firmware.

https://github.com/guino/ppsapp-rtsp/issues/2

guino, 2021-08-30:
I have no familiarity on the tuya sdk but I believe most of their sdk is for non-camera devices but if you figure something out with that feel free to share your results.

while I would not be surprised if the same camera/model would work on an older firmware it is possible/likely that some cameras have the same information on /devices/deviceinfo but run on different boards which may or may not be similar enough to allow using different firmware. The only real way of knowing would be to swap the flash chip between camera with old and new firmware and see if both or either would work. If the old firmware works on the new firmware device you could potentially flash it with the old firmware and keep it offline as I doubt it would work on the tuya cloud with the old firmware due to the different settings/files in the device.

guino, 2021-12-22:
@Nigel1992 Your device seems similar to #62 where the only way currently to root the device is with a hardware programmer, and there's currently no support for play/mpeg/snap (only onvif/rtsp). If we make any progress with that we'll be sure to post our findings publicly.

Nigel1992, 2021-12-22:
@guino Thanks for the update!
How much are these hardware programmers, and where to buy it ?
Aliexpress?

guino, 2021-12-22:
@Nigel1992 you can get flash programmers on amazon, aliexpress, etc. But before you spend the money on this you should think about the following:
* Many hardware programmers around say they handle 3.3v and 5v but end up sending 5v to 3.3v devices which can damage them. You can modify them to properly feed 3.3v to all pins but it requires soldering work (soldering iron, etc)
* It is likely that the flash chip will need to be removed from the board to be flashed. This means opening the device and removing the board, then using a Heat-Gun to remove and solder it back after flashing. I do not recommend using a soldering iron for this as even people with soldering experience may damage the board which is very fragile (I myself have damaged mine and almost lost it).
* You will need a linux machine to prepare the flash changes.
* You will not get play/mjpeg/snap support from the device as I mentioned above.

So really you have to think of the big-picture: how much time/money are you going to spend on this which may not even have a positive outcome (if you do something wrong) compared to possibly just getting something that does what you need out of the box. If you already have the equipment, sure it's worth a shot but I would not justify getting the equipment to make changes to 1 device.

Nigel1992, 2022-07-04:
@guino Any progress on this ?
Can it be modded just like the doorbell?

Nigel1992, 2022-07-06:
> @Nigel1992 <https://github.com/Nigel1992> without the device for research
> and testing we'll not have a SD card method anytime soon as it will require
> someone with the device and the knowledge to see what can be done (if at
> all possible).

Alright thanks for letting me know.