guino/BazzDoorbell #48: Attempting to hack Mini 9S

almirus, 2021-08-29:
hi! I have new camera with firmware 4.0.2
<pre>{
  "devname": "Smart Home Camera",
  "model": "Mini 9S",
  "serialno": "062713209",
  "softwareversion": "4.0.2",
  "hardwareversion": "M9S_A2_V10_F37",
  "firmwareversion": "ppstrong-a3-tuya2_laxi-4.0.2.20201008",
  "identity": "MR2008240200901278",
  "authkey": "███████████",
  "deviceid": "pp01c██████1a4f9",
  "pid": "aaa",
  "WiFi MAC": "dc:29:19:94:57:cc",
  "ETH MAC": "00:00:00:00:00:00"
}</pre>

<pre>setenv bootargs mem=64M console=ttySAK0,115200n8 loglevel=10   mtdparts=spi0.0:256k(bld),64k(env),64k(enc),64k(sysflg),3m(sys),4032k(app),640k(cfg) ppsAppParts=5 ip=192.168.1.99:::255.255.255.0 eth=00:55:7b:b5:7d:f7</pre>

<pre>#!/bin/sh

export PATH=/usr/bin:/sbin/:/usr/sbin:/bin

RED="[1;31m"
NORMAL="[0;39m"

echo "${GREEN} 2015 PPStrong Tech Cop.Ltd.${NORMAL}"

mkdir -p /opt/pps
MTDNUM=`cat /proc/cmdline | sed 's/.*ppsAppParts=\([0-9]\).*/\1/'`

# debug
MTDNUM=5

case $MTDNUM in
         5)
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
			break
			;;
		 7)
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
		 	break
		 	;;
		 0)     
			sleep 10
			mount -t vfat /dev/mmcblk0p1 /opt/pps
			break
			;;
		 *)
			MTDNUM=5
			mount -t cramfs /dev/mtdblock$MTDNUM /opt/pps
		 	;;
esac

echo "/opt/pps/" > /tmp/PPStrong.runpath
[ -e /opt/pps/initrun.sh ] && cp /opt/pps/initrun.sh /tmp/PPStart && chmod +x /tmp/PPStart && /tmp/PPStart
</pre>
tried this https://github.com/guino/BazzDoorbell/issues/13 and this https://github.com/guino/Merkury720 - unsuccessfully
any idea? my goal is ONVIF support

guino, 2021-08-30:
#13 should work on this camera, just make sure to follow the steps correctly and make sure to try formatting/partitioning the SD card in windows/linux/app and possibly try different SD cards.

almirus, 2021-08-30:
i think this address for my camera is different (but what?)
my\ppsMmcTool.txt                                                                                                                                                                               upgrade,,writeAddr=0,,password=nothing,,writeLen=0,,fileName=env;env import **42000000**;saveenv,,

guino, 2021-08-30:
There’s always a chance of different addresses, but the only way to know would be to dump the firmware or using UART. Did you try #11? You could try the hack for Merkury1080 to see - it will not hurt if the address is wrong.

almirus, 2021-08-31:
my dump contains garbage 
[flash.bin.zip](https://github.com/guino/BazzDoorbell/files/7081742/flash.bin.zip)
and my UART log:
<pre>
SYS_Init++
off513,temp24
p2p0,ap0,active0
p2p0,ap0,active0
StartUp_Indication()
normalScan
directScan
ch 7,type0,2442M.

ch 7,type0,2442M.

ch 7,type0,2442M.

off513,temp29
ch 1,type0,2412M.

agcrevert
ScanWdt,gain40
ch1,noise-79.
ch 2,type0,2417M.

nodBm-80.
ScanWdt,gain40
ch2,noise-80.
ch 3,type0,2422M.

ScanWdt,gain40
ch3,noise-80.
ch 4,type0,2427M.

ScanWdt,gain40
ch4,noise-71.
ch 5,type0,2432M.

ScanWdt,gain40
ch5,noise-81.
ch 6,type0,2437M.

ScanWdt,gain40
ch6,noise-81.
ch 7,type0,2442M.

ScanWdt,gain40
ch7,noise-81.
ch 8,type0,2447M.

ScanWdt,gain40
ch8,noise-81.
ch 9,type0,2452M.

ScanWdt,gain40
ch9,noise-81.
ch 10,type0,2457M.

TPC1
nodBm-81.
Iter0,i(0),q(0),g(-2),p(138)
power982085,60(dB),agc idx-2,up0.
Iter1,i(-24),q(-60),g(-2),p(126)
Iter2,i(-23),q(-59),g(-2),p(126)
Iter2,i:-23,q:-59,g:-2,p126
ch 10,type0,2457M.
</pre

guino, 2021-09-01:
@almirus do you get a prompt/countdown right after power on? Did you try pressing enter during the countdown? If you get a password prompt send me an email.

guino, 2021-09-06:
@almirus you may need to press 'reset' while powering on to get the prompt to show up -- if you get to the prompt send me an email and I can send you a few things to try.

almirus, 2021-09-06:
@guino with or without sd?

almirus, 2021-09-06:
without SD and pressed reset button:
<pre>
L876:Bus suspend
CPU: 0%
CPU: 0%
CPU: 0%
CPU: 0%
CPU: 0%
CPU: 0%
CPU: 0%
CPU: 0%
CPU: 0%
CPU: 0%
CPU: 0%
</pre>

camera not load with red light :(

guino, 2021-09-07:
@almirus If a countdown is going to show it will show regardless of SD card being present (with/without). If it doesn't show it suggests the bootloader may be different or you may not be doing it correctly (less likely).

almirus, 2021-09-25:
@guino 
finally i downloaded the firmware via directly connect on FM25Q64 flash chip

[flash.bin.zip](https://github.com/guino/BazzDoorbell/files/7229808/flash.bin.zip)

guino, 2021-09-25:
@almirus from the flash bin - the mod that should work with this camera is https://github.com/guino/Merkury1080P#conclusion -- as the load address appears to be 81C08000. I had suggested you tried it before (https://github.com/guino/BazzDoorbell/issues/48#issuecomment-908772557) but I don't know if you did, can you confirm you tried it ?

guino, 2021-09-25:
@almirus extracting your ppsapp I get:
mjpeg/snap.cgi address is: 037e634
play.cgi address is: 037eda4

You may want to try https://github.com/guino/Merkury1080P/issues/9#issuecomment-926414826 to see if you can enable onvif by just editing tuya_config.json (assuming you can use the Merkury1080P instructions to access the device).

almirus, 2021-09-26:
@guino for this https://github.com/guino/BazzDoorbell/issues/48#issuecomment-927185571

> fdisk -l /dev/sda
> Disk /dev/sda: 29.95 GiB, 32161923072 bytes, 62816256 sectors
> Disk model: STORAGE DEVICE
> Units: sectors of 1 * 512 = 512 bytes
> Sector size (logical/physical): 512 bytes / 512 bytes
> I/O size (minimum/optimal): 512 bytes / 512 bytes
> Disklabel type: dos
> Disk identifier: 0x2827a5e5
> 
> Device     Boot Start      End  Sectors  Size Id Type
> /dev/sda1       32769 62816255 62783487 29.9G 83 Linux

and this file ppsMmcTool.txt (2.9.x firmware) nothing to happen, I've garbage flash.bin

> root@vivobook:/mnt/d/Distrib/camera/from sd# binwalk -e -M flash.bin
> 
> Scan Time:     2021-09-26 14:03:07
> Target File:   /mnt/d/Distrib/camera/from sd/flash.bin
> MD5 Checksum:  164147da90c2733784fa42a239c4311f
> Signatures:    391
> 
> DECIMAL       HEXADECIMAL     DESCRIPTION
> --------------------------------------------------------------------------------
> 
> root@vivobook:/mnt/d/Distrib/camera/from sd#

this file mini7c.zip nothing to happen, I've garbage flash.bin 

> root@vivobook:/mnt/d/Distrib/camera/from sd# binwalk -e -M flash.bin
> 
> Scan Time:     2021-09-26 14:32:47
> Target File:   /mnt/d/Distrib/camera/from sd/flash2.bin
> MD5 Checksum:  19f34d13536314642d8f52ac88aa00dc
> Signatures:    391
> 
> DECIMAL       HEXADECIMAL     DESCRIPTION
> --------------------------------------------------------------------------------
> 
> root@vivobook:/mnt/d/Distrib/camera/from sd#

almirus, 2021-09-26:
> You may want to try [guino/Merkury1080P#9 (comment)](https://github.com/guino/Merkury1080P/issues/9#issuecomment-926414826)
I haven't telnet access

almirus, 2021-09-26:
I tried https://github.com/guino/BazzDoorbell/issues/13
The port 8090 is default closed in ,y firmware, so i have to use the file **ppsFactoryTool.txt**

but my default proc/cmdline like this "setenv bootargs mem=64M console=ttySAK0,115200n8 loglevel=10   mtdparts=spi0.0:256k(bld),64k(env),64k(enc),64k(sysflg),3m(sys),4032k(app),640k(cfg) ppsAppParts=5 ip=192.168.1.99:::255.255.255.0 eth=00:55:7b:b5:7d:f7" and MTDNUM=5 (without #)
unlike your file "mem=36 console=ttyAMA0,115200n8 mtdparts=hi_sfc:192k(bld)ro,64k(env)ro,64k(enc)ro,64k(sysflg)ro,3136k(sys),4352k(app),320k(cfg) ppsAppParts=5 ppsWatchInitEnd
"
I tried this env file "bootargs=mem=64M console=ttySAK0,115200n8 loglevel=10   mtdparts=spi0.0:256k(bld),64k(env),64k(enc),64k(sysflg),3m(sys),4032k(app),640k(cfg) ppsAppParts=5 ppsWatchInitEnd - ip=\\${T//_/\\$\\'"\\\\x20"\\'}:::::";T=\\"sleep_5;mkdir_-p_/mnt/mmc01;mount_-t_vfat_/dev/mmcblk0p1_/mnt/mmc01;/mnt/mmc01/initrun.sh&\\";eval"

If i copy 3 files, i can't access via web, if i copy 4 files, i get old file "setenv bootargs mem=64M console=ttySAK0,115200n8 loglevel=10   mtdparts=spi0.0:256k(bld),64k(env),64k(enc),64k(sysflg),3m(sys),4032k(app),640k(cfg) ppsAppParts=5 ip=192.168.1.99:::255.255.255.0 eth=00:55:7b:b5:7d:f7"

guino, 2021-09-27:
@almirus can you make a zip of your SD card contents when trying https://github.com/guino/Merkury1080P#conclusion ? The instructions are clear that you need to copy 3 specific files OVER the files from https://github.com/guino/Merkury720/tree/main/mmc - which is a lot more than 4 (four) files. If I double check your files and it still doesn't work then I would suggest doing exactly what I did in https://github.com/guino/BazzDoorbell which is to modify the initrun.sh file so it runs custom.sh from the SD card and flash it to the chip directly (and I can help you with the changes).

almirus, 2021-09-27:
@guino
Yippee! This is https://github.com/guino/Merkury1080P#conclusion work!

guino, 2021-09-27:
@almirus glad it worked! Thanks for the coffee!

Please try https://github.com/guino/Merkury1080P/issues/9#issuecomment-926414826 to see if you can get RTSP/ONVIF working, if not let me know if you need help patching ppsapp.

almirus, 2021-09-27:
success !!! 🎉
[image]

almirus, 2021-09-27:
@guino
Is quality equal via ONVIF  and  RTSP ?
now ONVIF (both Stream) much worse than Smart Live app

guino, 2021-09-27:
@almirus 
There should be two channels on onvif: one low resolution and one high resolution. I am pretty sure the direct RTSP URLs for them should be the same as in previous versions: rtsp://IP:8554//Streaming/Channels/101 and rtsp://IP:8554//Streaming/Channels/102

whatever user/password you used with ONVIF should work with the above.

almirus, 2021-09-27:
@guino 
this is true. thanks

do you have any idea how to make the motion detector work?
and how to get motion events?

almirus, 2021-09-27:
Yes. I want to get events motion) for home assistant (without video analyze on raspberry)

closed 2022-12-31 by almirus