guino/BazzDoorbell #49: Useful URLs

jilleb, 2021-08-30:
Not an issue, merely some research which should end up in the wiki when it's done, I guess.
I took a dive into the ppsapp binary, and found a lot of endpoints the doorbell is publishing on its internal webserver, as well as some additional variables for tuya_config.json which could be useful.

Here you go:

/telnetd/switch
`Telnetd service started ok`

/devices/deviceinfo (device info):

```json
{"devname":"Smart Home Camera","model":"Bell 8S","serialno":"xxxxxxxxxx","softwareversion":"2.9.7","hardwareversion":"BE8S_H1_V10_433","firmwareversion":"ppstrong-c51-tuya2_lcs-2.9.7.20201020","authkey":"xxxxxxxxxx","deviceid":"xxxxxxxxxx","identity":"xxxxxxxxxx","pid":"aaa","WiFi MAC":"xxxxxxxxxx"}
```

/devices/storageformat
No visible output

/devices/formatpercent/
```json
{"formatpercent":100}
```

/flash/upgrade/release_package
`Upgrade from ppstrong release package.`

/search/
```json
{
	"deviceName":	"xxxxxxxxxx",
	"serialno":	"xxxxxxxxxx",
	"sn":	"xxxxxxxxxx",
	"licenseUsed":	1,
	"licenseId":	"xxxxxxxxxx",
	"p2p_uuid":	"v2-xxxxxxxxxx",
	"factory_code":	0,
	"factory_code_str":	"",
	"model":	"Bell 8S",
	"ip":	"xxxxxxxxxx",
	"mask":	"255.255.255.0",
	"gw":	"192.168.1.1",
	"mac":	"xxxxxxxxxx",
	"interface":	"wlan0",
	"version":	"2.9.7"
}
```

/devices/upgradepercent/
```json
{"percent":-1,"upgrade_percent":-1,"download_percent":-1}
```

/devices/reboot/
Reboots the device

More to be added.

guino, 2021-08-30:
The telnet url says it works but never does because there’s no telnet binary in the firmware (had tried it way back). The others can probably go in the wiki.

jilleb, 2021-08-30:
👍🏼 
I  have some more to add, I'll put them in the first post.
Currently recovering my device, after formatting it by accident (one of the URLs does that 😉 )
Maybe I can cook up a simple portal that holds all URLs, with a description so users can access them easily.

almirus, 2021-09-01:
/proc/self/root/home/cfg/tuya_config.json

<pre>

  "version": 1,
  "sleep_mode": 0,
  "alarm_fun_onoff": 0,
  "alarm_fun_sensitivity": 1,
  "alarm_fun_mode_switch": 0,
  "alarm_fun_time_start": 0,
  "alarm_fun_time_end": 0,
  "flip_onoff": 0,
  "light_onoff": 1,
  "night_mode": 1,
  "sound_detect_onoff": 0,
  "sound_detect_sensitivity": 1,
  "watermark_onoff": 1,
  "event_record_time": 60,
  "enable_event_record": 2,
  "record_enable": 1,
  "motion_trace": 1,
  "motion_area_switch": 0,
  "motion_area": "",
  "motion_tracking": 0,
  "cry_detection_switch": 0,
  "humanoid_filter": 0,
  "loudspeaker_vol_pct": 100,
  "onvif_enable": 0,
  "onvif_pwd": "admin"
}
</pre>

almirus, 2021-09-01:
/proc/mounts
<pre>
rootfs / rootfs rw 0 0
proc /proc proc rw,relatime 0 0
sysfs /sys sysfs rw,relatime 0 0
tmpfs /dev tmpfs rw,relatime 0 0
devpts /dev/pts devpts rw,relatime,mode=600,ptmxmode=000 0 0
/dev/mtdblock6 /home/cfg jffs2 rw,relatime 0 0
/dev/mmcblk0p1 /mnt/mmc01 vfat rw,relatime,fmask=0022,dmask=0022,codepage=cp437,iocharset=iso8859-1,shortname=mixed,usefree,errors=continue 0 0
</pre>

youribonnaffe, 2022-01-02:
Some URLs from the CloudEdge application

```
127.0.0.1/devices/charm/pair
127.0.0.1/devices/charm/unpair
127.0.0.1/devices/firmware_upgrade
127.0.0.1/devices/led/all
127.0.0.1/devices/music/id/%s/start
127.0.0.1/devices/music/play/
127.0.0.1/devices/music/play/next
127.0.0.1/devices/music/play/pause
127.0.0.1/devices/music/play/prev
127.0.0.1/devices/music/play/state
127.0.0.1/devices/network
127.0.0.1/devices/reboot
127.0.0.1/devices/record/search_by_day
127.0.0.1/devices/sd_event
127.0.0.1/devices/settings
127.0.0.1/devices/storage
127.0.0.1/devices/temp_humidity/value
127.0.0.1/devices/upgradeprecent
127.0.0.1/devices/voicemail
127.0.0.1/media/audio/output/volume
127.0.0.1/media/audio/output/volume/
```
And the code from the app to see how they are called: https://github.com/youribonnaffe/iegeek-security-camera/blob/main/MeariDeviceController.java

I have not checked if they all work.

youribonnaffe, 2022-01-09:
Even more URLs extracted from https://github.com/Mearitek/MeariSdk/tree/master/Android/library/x86_64 (libsppsdk):
```
http://%s/devices/ntp
http://%s/devices/timezone
http://%s/devices/network
http://%s/devices/reboot
http://%s/devices/setpasswd
http://%s/devices/voicetalk
http://%s/devices/record/stop
http://%s/devices/playback
http://%s/devices/setptzmove
http://%s/devices/setptzstop
http://%s/devices/upgrade
http://%s/devices/lte/online
http://%s/devices/wifi
http://%s/devices/deviceinfo
http://%s/devices/devicecaps
http://%s/devices/storage
http://%s/devices/netlink
http://%s/devices/lte/status
http://%s/devices/energy
http://%s/devices/lte/info
http://%s/devices/snapshot
http://%s/devices/daytonightlevel?videoid=%d
http://%s/devices/videosourceconfig?videoid=%d
http://%s/devices/audiosourceconfig?audioid=%d
http://%s/devices/factoryreset
http://%s/devices/record/search_by_day
http://%s/devices/record/search_by_month
http://%s/devices/record/start
http://%s/devices/record/schedule
http://%s/devices/playback/control
http://%s/devices/firmware_upgrade
http://%s/devices/storageformat
http://%s/devices/mirror?videoid=%d&mirror=%d
http://%s/devices/ftpcfg?videoid=%d
http://%s/devices/alarm/open?videoid=%d&alarmtype=%d
http://%s/devices/alarm/close?videoid=%d
http://%s/devices/lte/offline
http://%s/devices/videosourcecaps?videoid=%d
http://%s/devices/audiosourcecaps?audioid=%d
http://%s/devices/upgradeprecent
http://%s/devices/formatpercent
http://%s/devices/alarm/get?videoid=%d
http://%s/devices/mirror?videoid=%d
```