guino/BazzDoorbell #50: PPSAPP Patch for LSC BEL 8S 2.10.5

rneurink, 2021-09-01:
I didn't find a matching hash so I thought I would post this.

{"devname":"Smart Home Camera","model":"Bell 8S","serialno":"060662226","softwareversion":"2.10.5","hardwareversion":"BE8S_H1_V10_433","firmwareversion":"ppstrong-c51-tuya2_lcs-2.10.5.20210806","authkey":"Legsa820dIilcfwUv7Ml0hId0QYlXLAW","deviceid":"pp01c88015b61d8734ea","identity":"MR2005212400902061","pid":"aaa","WiFi MAC":"d4:d2:d6:d0:0b:89"}

[ppsapp.zip](https://github.com/guino/BazzDoorbell/files/7094263/ppsapp.zip)

WeterPeter, 2021-09-05:
This is needed for the latest firmware update?

peter

rneurink, 2021-09-05:
Yeah this is currently the latest update. I got everything else working already but I would like to get the RTSP working as well.

guino, 2021-09-07:
@rneurink the 2.10.x version should have RTSP available without patching at rtsp://IP:8554//Streaming/Channels/101 and rtsp://IP:8554//Streaming/Channels/102 -- see: https://github.com/guino/BazzDoorbell/issues/2#issuecomment-745618597

If the OLD RTSP code is in the firmware I can make a patch to enable it on port 8555. Sorry I was traveling but should be able to do it soon.

guino, 2021-09-07:
@rneurink I just posted a RTSP patch for this version in https://github.com/guino/ppsapp-rtsp/issues/1

rneurink, 2021-09-07:
Am i missing something? This is the Nmap of the device with the patched app:

http://admin:056565099@192.168.x.x/proc/self/root/tmp/hack shows the done page.

This is what http://user:password@ip:8080/cgi-bin/snap.cgi shows
[image]

Followed the [way to patch the ppsapp](https://github.com/guino/ppsapp-rtsp/issues/1)

PORT     STATE SERVICE VERSION
23/tcp   open  telnet  security DVR telnetd (many brands)
80/tcp   open  http
| fingerprint-strings: 
|   FourOhFourRequest: 
|     HTTP/1.1 401 Unauthorized
|     WWW-Authenticate: Digest realm="www-user@ppstrong",qop="auth",nonce="1631039765",opaque="927932679"
|   GetRequest, HTTPOptions, RTSPRequest: 
|     HTTP/1.1 404 Not Found
|     Content-Length: 0
|   SIPOptions: 
|     HTTP/1.1 401 Unauthorized
|_    WWW-Authenticate: Digest realm="www-user@ppstrong",qop="auth",nonce="1631039773",opaque="927932687"
|_http-title: Site doesn't have a title.
6668/tcp open  irc?
|_irc-info: Unable to open connection
8080/tcp open  http    BusyBox httpd
| http-methods: 
|_  Supported Methods: HEAD

netstat -tulpn through the telnet connection
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name
tcp        0      0 0.0.0.0:6668            0.0.0.0:*               LISTEN      906/ppsapp
tcp        0      0 0.0.0.0:8080            0.0.0.0:*               LISTEN      1066/busybox
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      906/ppsapp
tcp        0      0 0.0.0.0:23              0.0.0.0:*               LISTEN      1062/busybox
udp        0      0 192.168.0.111:57777     0.0.0.0:*                           906/ppsapp
udp        0      0 0.0.0.0:46196           0.0.0.0:*                           906/ppsapp
udp        0      0 0.0.0.0:3702            0.0.0.0:*                           899/ppsdsry
udp        0      0 0.0.0.0:3703            0.0.0.0:*                           899/ppsdsry
udp        0      0 127.0.0.1:52864         0.0.0.0:*                           906/ppsapp

guino, 2021-09-07:
@rneurink Are you using the pached ppsapp ? or the onvif_enable setting in tuya_config.json ? the patched ppsapp must be copied to the root of the SD card to work (and you have to reboot the device with it in there). The tuya_config.json method requires changes to the file and in some cases the change may be 'reset' by the app which may require copying the file every boot. 
The snap/mjpeg.cgi requires setting the address in the snap/mjpeg.cgi file (the address is published with the patch) -- you should be able to use it even without a patched ppsapp.

guino, 2021-09-07:
@rneurink feel free to post a zip of your SD card contents without the SDT folder in it and I can review it.

rneurink, 2021-09-08:
Seems like my SD card was dying. Maybe i messed it up somewhere in the process but i redid the process on another SD card and now it works. 
I got the snap and mjpeg as well as the RTSP stream working. 
Onvif is working on http://IP:8000/onvif/device_service wth admin:admin as user and pass
Thanks

closed 2021-09-08 by rneurink