guino/BazzDoorbell #51: Rotate camera 90 degree?

mippen, 2021-09-11:
In the SmartLife application I can "flip" the camera feed 180 degrees.
Is there a way to flip it 90?
I have mounted the unit on my Roborock vacuum and can't have it standing up.
I do not mind if the lens etc will give me a faulty view.. better than tilting my head when watching the feed .

guino, 2021-09-11:
@mippen A flip (horizontal/vertical) is fairly easy to implement but a rotate function is pretty demanding so that is likely the reason it doesn’t exist within a device with very limited resources. 

If you use the right player however you can rotate the feed when viewing it: https://uca.edu/tlc/files/2012/10/Rotate_a_Video_90_degrees_with_VLC.pdf

I have verified this works fine with my doorbell camera RTSP feed, so it should work with any.

mippen, 2021-09-11:
Thanks for the feedback. I do need to do it at the "source" as I have not found a way to rotate the camera in synology surveillance station

mippen, 2021-09-11:
An alternative would be to rotate the camera module in the device.. But without opening the unit I'm afraid that the sensor is mounted on the PCB itself.
Would be great if someone have a picture of the internal unit

guino, 2021-09-11:
@mippen I can tell you that none of the cameras I have seen have a separate sensor module from the actual board (which is basically in the format of the camera). Some of the 'bullet' cameras have a more squared board format but I doubt you'd be able to rotate them (and those should be easier to rotate+mount physically too).

If you have a server/machine always on you run ffmpeg/vlc/etc to connect to the camera, rotate the video and 'stream' the rotated output so synology surveillance station would connect to it and received the rotated feed. Other than having the machine running always on and added lag to the live feed there's not much of a draw-back for this.