guino/BazzDoorbell #52: ppsapp 2.10.6 request

nakcam, 2021-09-14:
[ppsapp.zip](https://github.com/guino/BazzDoorbell/files/7163798/ppsapp.zip)

cant find a patched fitting ppsapp, can someone please help? This is for the "cleverio" branded doorbell sold by swedish company kjell.com if that helps more people. 

after updating to 2.10.6 (that might have been stupid?) it now need the "pMerkury720/raw/main/ppsFactoryTool.txt" to open port 80 on the sd card all the time, is this possible to resolve? 

All i want from this hack is to enable rtsp to use it with frigate on home assistant. Does anyone know the exact resolution and fps  of the rtsp stream? (frigate is picky).

Thanks!!    

{"devname":"Smart Home Camera","model":"Bell 5S","serialno":"061051002","softwareversion":"2.10.6","hardwareversion":"BE5S_H1_V10_433","firmwareversion":"ppstrong-c51-tuya2_kjell-2.10.6.20210818","authkey":"ijYxNlWAnGuFW5e5sOyozMgiHbAEKbOF","deviceid":"pp018717cc72466de877","identity":"MR2006180401201155","pid":"aaa","WiFi MAC":"d4:d2:d6:63:70:4f"}

guino, 2021-09-14:
@nakcam I have just now posted a patch for this firmware on here: https://github.com/guino/ppsapp-rtsp/issues/1 I don't expect any problems.

nakcam, 2021-09-14:
THANK YOU! Now I can open a rtsp stream at 8555. But it is really unstable and slow (artifacts and several seconds delay). Do you have any tips for getting a more stable connection? is it possible to turn something off to unload it?

guino, 2021-09-14:
@nakcam just make sure you’re using RTSP over TCP (check #2 information on vlc settings for instance). Anything outside of that would have to be either wifi signal poor/congestion or hardware issues. I sometimes have high congestion on my 2.4ghz band and get delays of several seconds and bad skipping. You should be able to compare it with the stream from the phone app.

nakcam, 2021-09-14:
ok, I did try rtsp over tcp and it is the same. The stream in tuya app is very good with about half the delay (about 3sec against rtsp 6sec) and no artifacts.

guino, 2021-09-14:
Did you try the RTSP stream from onvif on port 8554 as pointed out by the patch? Did you try different RTSP players? My RTSP feed works as good as the phone app so I don’t know what else it could be.

nakcam, 2021-09-14:
I cant get the stream to work on port 8554.. I must have missed something. I have tried vlc and motion eye, in vlc the stream will freeze after a few seconds but on motion eye it runs.

nakcam, 2021-09-14:
I cant get  http://admin:056565099@192.168.x.x/devices/deviceinfo if i dont boot on the "https://github.com/guino/Merkury720/raw/main/ppsFactoryTool.txt " is this a problem?

guino, 2021-09-14:
@nakcam it is normal for firmware 2.10.x to only enable the URLs when using ppsFactoryTool.txt - and usually they work on port 8090.

nakcam, 2021-09-14:
when I use telnet and "top" command I can see that cpu load is <2% until I open a rtsp stream then ppsapp jumps up to 99% usage.

nakcam, 2021-09-14:
when viewing the stream on tuya app the load is <10%

guino, 2021-09-14:
@nakcam high cpu is normal when using RTSP over UDP. Over TCP the cpu load should be lower.

nakcam, 2021-09-14:
well it was sort of my bad. Some how both vlc and motion eye used udp (with setting sett to tcp!). I tried frigate now and it works fine with 3% cpu load (low res)! I still have a delay (>3sec) but i guess its the best this camera can do. Thanks for speedy response!!

guino, 2021-09-14:
@nakcam glad you got it working - these are cheap cameras so we just make the best we can.