guino/BazzDoorbell #53: Victure PC660 compatible

Zjokkeh, 2021-09-22:
Has somebody an idea that the victure pc660 is also hackable? Its a great camera but the same like the rest. It needs the Tuya app and is not reachable by IP. No onvif or rtsp... I can't reach it with the provided urls...

guino, 2021-09-22:
@Zjokkeh did you try adding ppsFactoryTool.txt to the SD card ? if that doesn't allow you to use any URLs then the camera must not be running the same type of OS/system as other tuya cameras we have seen.

Zjokkeh, 2021-09-23:
@guino Thank you for your fast reply. When I place the ppsFactorytool on the SD card, the camera deletes the file and creates a DCIM folder. So I guess it will not work?

Zjokkeh, 2021-09-23:
@guino I just tried that but without succes. I do see the IP adress when the cam is online.

guino, 2021-09-24:
@Zjokkeh if you have telnet port open - have you tried to telnet in? there's a github full of passwords previously found to try: https://gist.github.com/gabonator/74cdd6ab4f733ff047356198c781f27d that would be your way to create a hack but I doubt it would be very similar to the stuff we have modded (but I would not mind taking a look at it if you do get in).

Zjokkeh, 2021-09-27:
So I have to manually typ all those passwords? 

Another remarque I forgot that the camera did an update once. I am now on V5.4.6
Would it be possible to downgrade firmware?

guino, 2021-09-27:
@Zjokkeh you're welcome to try using any tools for telnet brute force but most people just try the passwords by hand to make sure nothing goes wrong. It may seem like a lot of work but honestly trying a lot of passwords is very little effort comparing to all the time we spend studying firmware and coming up with hacks/patches. 

Unless you have a copy of your previous firmware, there's no way to downgrade that I know of.

Zjokkeh, 2021-09-27:
I used the brute force tool but without succes.

guino, 2021-09-27:
Unfortunately, without having the device (or similar) I have no way of figuring out the password or hack/mod for it.

Zjokkeh, 2021-09-27:
Yeah, i know. I could try and make a flash dump?

guino, 2021-09-27:
@Zjokkeh to make a flash dump you'll need to open the device and use a hardware programmer to read it. I'd be glad to take a look if you upload a flash dump.

Zjokkeh, 2021-09-28:
I would be more than happy to make a flash dump. Any tips or how to's are welcome to get me started. I know my way around electronics and arduino etc... Never did a flash dump though...

guino, 2021-09-28:
@Zjokkeh that looks like the flash chip, this board looks different than the tuya camers I've seen but I'll help on what I can. You will probably need to remove the chip from the board to read/write it which may require some delicate work (these boards are fragile) -- a heat gun is highly recommended for that. Then it's just a matter of hooking up the programmer and reading the flash using whatever software comes with it (or you can download many free ones for whatever OS you use). Writing is just as simple: hook it up, select the file and tell it to write back. You CAN try reading the flash without disconnecting it from the board but it usually doesn't work -- I have had success by disconnecting pin 6 only but again it's delicate work and you may damage the board so if you have access to a heat gun that's preferred.

If you're familiar with Arduino, etc you may want to try the UART route but that requires a TTL-Serial/USB adapter -- there's no guarantee that you'll be able to do much with the UART. A lot of these boards have a telnet terminal on the UART and in most cases you can access uboot (bootloader) and pass in a parameter to try and login as root.

Zjokkeh, 2021-09-29:
I don't see an UART connection. I shall try the flash dump route first?

guino, 2021-09-30:
@Zjokkeh the flash route usually provides more options. If you decide to try the UART anyway post a few pics of the board and I can try to help you find the UART connections.

Zjokkeh, 2021-10-01:
Hi. I did it. I dumped the flash. I got 2 files, one I made in windows and one in linux. Maybe there's a difference....
[PC660.zip](https://github.com/guino/BazzDoorbell/files/7267722/PC660.zip)

guino, 2021-10-01:
@Zjokkeh both images are the same and it looks correct. I was able to extract the data and it is very different than the other cameras I have seen -- you said it uses the tuya app too or does it use some other app ? I ask cause I don't see the same files used to control the device.

Zjokkeh, 2021-10-01:
It does indeed work with the tuya app. And only with the tuya app. I tried different apps but without succes...

guino, 2021-10-01:
@Zjokkeh your device doesn't even use an ARM processor like the other cameras, it uses a MIPS processor. At a quick look we should be able to modify the flash to set the root password and enable telnet but I don't know about what else we may be able to do at this time.

Zjokkeh, 2021-10-01:
Telnet ls already enabled, I just can't find the login and password...

guino, 2021-10-01:
@Zjokkeh we can set the root password in the flash and you can write it back so you can access it. I have already confirmed that your application 'dgiot' does NOT have any RTSP or ONVIF code in it, so that's not something you'll get by making changes. At the most we may be able to do mjpeg if we can find the jpeg address (and it encodes jpeg in real time like other tuya cameras).

Zjokkeh, 2021-10-03:
All we can do is try :)

guino, 2021-10-03:
@Zjokkeh here's a flash you can write to enable telnet with root access:
[PC660_new.bin.zip](https://github.com/guino/BazzDoorbell/files/7274227/PC660_new.bin.zip)

From looking at the code it seems the device does NOT encode the video as JPG unless a request comes in to either produce a snapshot or to view a MJPEG stream (Which I assume is what the phone app uses). It looks like the motion detection in this device is done using an IR/PIR sensor instead of detecting video changes (thus why the software doesn't do the motion detection itself).

If you get telnet access we can try a few things to find the jpeg buffer and see if it can be streamed or not (likely not) so I am expecting that at-most you can get some sort of notification of motion or doorbell push to whatever you may want.

On another note I saw this video: https://www.youtube.com/watch?v=zu6xpDt0BnE which seems to indicate there's a way to get it to enable camera stream to echoshow/alexa -- which is usually RTSP. The firmware version may be different in the video than on your device -- which could indicate you need/get a firmware update ? It doesn't seem like the Tuya app was used in the video, so maybe same camera just different firmware (tuya vs non-tuya?). The version on the video seemed like 7.1.x.

Let me know if you get telnet working and/or try a different app/firmware (sometimes all it takes is emailing the manufacturer and asking if there's a new firmware and they'll send you an update).

Zjokkeh, 2021-10-05:
So... I flashed the new bin and reinstalled the memory chip, but I think I fried the camera... Nothing happens after power on, just a red light... I think I might have fryed something by trying to do in circuit programming... I don't know...

guino, 2021-10-05:
@Zjokkeh did you try loading the original flash back to see if it works again ? If you loaded the original firmware and it didn't work on the board then chances are something fried (or you could have a solder issue). There are lots of 'programmers' out there that apply 5v where it should be 3.3v and may need to be modified to be safe for 3.3v - I know cause I had one of those. Other than measuring all pins there's no way to tell for sure. I have posted about the programmer voltage and the fragility of these boards many times before, but you seem like you know your way around this stuff so I did not want to sound like a broken record. Worst case scenario I hope it is still under warranty and maybe you can return it ?

Lucasark, 2022-02-23:
@Zjokkeh I am so happy to see this thread, I hope none of you lost hope in this device, I will make my 2cents share this video where you can see clear that is possible access the camera via PC.
https://www.youtube.com/watch?v=rjM74yWugEo&ab_channel=GaraGulp%21

applicma, 2023-11-08:
@Zjokkeh, did you find a solution ?
@Lucasark , have you test his solution ? products and vendor are differents, the way to connect this camera is not the same.