guino/BazzDoorbell #54: Hey! Smart Doorbell

dev-danny, 2021-10-02:
Hey guys, 

I came stumbling along this project when I was trying to get my tuya based doorbell to show its stream on my smart TV when someone rang the bell.
All I needed was the rstp stream and alexa to open the app on my TV. 

I started by following the guides on here but as my firmware was 1.3.25, I followed [THIS](https://github.com/guino/Merkury720) guide. 

I noticed once I copied all contents to my SD card my doorbell would boot up then hang and not be accessible from the tuya app. Once I removed the card it booted up normally so surely something is going on. 

I also tried to access the http://admin:05656 links I get no response. 

I then followed the new ad card option and used the ppFactory.txt file to open port 80/8090 but again non of the http://admin:05656 links worked. I've tried looking for the UART ports on the boards and have provided pictures but I'm at a loss. 

Is there any way forward with this model of camera? 

[image]
[image]
[image]
[image]

dev-danny, 2021-10-04:
> @dev-danny the few 1.x firmware I have seen do not run linux and run RTOS. If none of the URLs worked you're likely not running linux as even the RTOS version should at least ask for user/password and maybe even give some responses like /devices/deviceinfo. There's no hack/mod for RTOS devices.
> 
> Your flash chip is the one labeled 'winbond' on the last picture you posted (in case you wanted to read it with a flash/hardware programmer). The UART is likely the 3 pads to the left of the word MIC on the last picture (possibly along with the pad above the 3 others. Another possibility for the UART is the 4 solder points under the word DSP on the last picture (less likely I'd say). If you have RTOS there will not be much on the UART -- it's just an RTOS terminal which lets you do a few specific commands/queries but there won't be a telnet/linux shell or way to enable one.
> 
> These boards are likely capable of running linux but they were just not developed using it so I be there is no alternative firmware or drivers that would allow it to work. Somoene has posted a dump of the firmware for one of those and I have not made any progress locating anything that could enable RTSP/ONVIF on it (and even that would require a flash programmer to read/write the chip directly). Some of the newer doorbells with version 5.x are the same way unfortunately.

Thank you for your speedy reply and insight. Its a real shame as the doorbell was relatively cheap £60 and is also wireless and adding rtsp to this would have been ideal. 

Does anyone know of a hackable doorbell that is also wireless?

guino, 2021-10-04:
@dev-danny all tuya doorbells are wireless in regards to networking. If you’re looking for wireless as-in ‘battery powered’ doorbells then I don’t believe there are any that will run linux, and if anything they would not support the features we added because they would drain the batteries too fast with constant video processing and communication with the servers. Most battery powered devices stay in low power mode and wake up thru IR motion sensor and such to save power.